zigpy/zha-device-handlers #2704: [BUG] Loading  Next step for ZHA

rudgej, 2023-11-03:
### Bug description

Trying to add a new Sonoff 3.0 Zigbee Dongle Plus

I have two HA system - one on a Pi and one on an Intel Nuc.

On both of them the system finds the dongle. 

I Select Erase Network Settings and Create a New Network 

I then get  " Loading Next Step For ZHA" with a spinning wheel for about 30 seconds followed by "Error" 

The only Zigbee entries I can find in the logs are

Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/zigpy_znp/zigbee/application.py", line 689, in _watchdog_loop
    await self._znp.request(c.SYS.Ping.Req())
  File "/usr/local/lib/python3.11/site-packages/zigpy_znp/api.py", line 1024, in request
    async with async_timeout.timeout(
  File "/usr/local/lib/python3.11/site-packages/async_timeout/__init__.py", line 141, in __aexit__
    self._do_exit(exc_type)
  File "/usr/local/lib/python3.11/site-packages/async_timeout/__init__.py", line 228, in _do_exit
    raise asyncio.TimeoutError

  File "/usr/src/homeassistant/homeassistant/components/zha/config_flow.py", line 387, in async_step_form_new_network
    await self._radio_mgr.async_form_network()
  File "/usr/src/homeassistant/homeassistant/components/zha/radio_manager.py", line 260, in async_form_network
    await app.form_network()
  File "/usr/local/lib/python3.11/site-packages/zigpy/application.py", line 390, in form_network
    await self.backups.restore_backup(
  File "/usr/local/lib/python3.11/site-packages/zigpy/backups.py", line 151, in restore_backup
    await self.app.write_network_info(
  File "/usr/local/lib/python3.11/site-packages/zigpy_znp/zigbee/application.py", line 195, in write_network_info
    await self._znp.write_network_info(
  File "/usr/local/lib/python3.11/site-packages/zigpy_znp/api.py", line 403, in write_network_info
    nib = await self.nvram.osal_read(OsalNvIds.NIB, item_type=t.NIB)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/zigpy_znp/nvram.py", line 168, in osal_read
    length = (await self.znp.request(c.SYS.OSALNVLength.Req(Id=nv_id))).ItemLen
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/zigpy_znp/api.py", line 1021, in request
    self._uart.send(frame)

If anyone can help get this going I would appreciate it.

Thanks 

### Steps to reproduce

Try to configure 

### Expected behavior

It adds the dongle 

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

MattWestb, 2023-11-03:
Putting one Zigbee USB-Stick in one USB3 port is **never** going to working and if you have reading little you have seen way.
(USB3 is using the same frequencies as WiFi 4.2 and Zigbee / BT so you is getting the radio jammed fro the USB 3 connection RF leek)

rudgej, 2023-11-04:
Interesting. That seems a rather large flaw that will kill zigbee and usb3 is the current standard !

MattWestb, 2023-11-04:
Have you trying using one microwave oven and talking in your BT headset or streaming one WiFi 4.2 webcam then its running ?
Its 4.2 Ghz is one crap band and Zigbee is low power and end devices is extreme low power devices so its always is being hitted first of RF interference.

any1, 2023-11-29:
I have the same problem on an Pi 4 running archlinux; here are debug logs: https://gist.github.com/any1/9d9e5a82b670f4552c8b452407c440d8

closed 2024-06-04 by github-actions[bot]

sdrieghe, 2025-01-28:
I'm getting this issue as well.... Was there ever a fix?

VikingGer, 2025-02-09:
> I'm getting this issue as well.... Was there ever a fix?

Same here. Unfortunately there's also this zigbee2mqtt issue not appearing als addon anymore...