zigpy/zha-device-handlers #2706: [Device Support Request] TS0601 _TZE204_pcdmj88b TRV not showing any entities

T0ytoy, 2023-11-04:
### Problem description

I bought some Zigbee TRV, they show up in home assistant as TS0601_TZE204_pcdmj88b but although they are pairing, no entity for control or sensor reading is showing up.

Model link for reference: https://fr.aliexpress.com/item/1005006191259938.html?spm=a2g0o.productlist.main.3.2de8kzSokzSoTw&algo_pvid=fc119493-da4b-462c-86bd-2d78585444c8&algo_exp_id=fc119493-da4b-462c-86bd-2d78585444c8-1&pdp_npi=4%40dis%21EUR%2132.60%2114.67%21%21%2132.60%21%21%402103834816991401988737073e38b3%2112000036203052461%21sea%21FR%21769762047%21&curPageLogUid=sWvhl7kLhrPV

I tried some custom quirks I found (for Moes or Zonnsmart TRVs) but obviously nothing good came out of it. 

### Solution description

I never used or debbuged custom quirks before, but I'm willing to provide help if someone needs more information to create a custom quirks for this model. Thanks a lot!

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE204_pcdmj88b",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

T0ytoy, 2023-11-09:
For information, a converter was made for this valve for zigbee2mqtt: [](https://github.com/Koenkk/zigbee2mqtt/issues/19462)
Maybe the technical details will make it easier to make a quirk for ZHA too :)

jim-fx, 2023-11-15:
Here is the relevant zigbee-herdsmann-converter:

```ts
 {
        fingerprint: tuya.fingerprint('TS0601', [
            '_TZE204_pcdmj88b',
        ]),
        model: 'TS0601_thermostat_4',
        vendor: 'TuYa',
        description: 'Thermostatic radiator valve',
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tuya.tz.datapoints],
        onEvent: tuya.onEventSetLocalTime,
        configure: tuya.configureMagicPacket,
        exposes: [
            e.child_lock(),
            e.battery(),
            e.battery_low(),
            e.climate()
                .withSetpoint('current_heating_setpoint', 5, 35, 0.5, ea.STATE_SET)
                .withLocalTemperature(ea.STATE)
                .withPreset(['schedule', 'holiday', 'manual', 'comfort', 'eco'])
                .withSystemMode(['off', 'heat'], ea.STATE)
                .withLocalTemperatureCalibration(-3, 3, 1, ea.STATE_SET),
            ...tuya.exposes.scheduleAllDays(ea.STATE_SET, 'HH:MM/C HH:MM/C HH:MM/C HH:MM/C HH:MM/C HH:MM/C'),
            e.holiday_temperature().withValueMin(5).withValueMax(30),
            e.comfort_temperature().withValueMin(5).withValueMax(30),
            e.eco_temperature().withValueMin(5).withValueMax(30),
            e.binary('scale_protection', ea.STATE_SET, 'ON', 'OFF').withDescription('If the heat sink is not fully opened within ' +
                'two weeks or is not used for a long time, the valve will be blocked due to silting up and the heat sink will not be ' +
                'able to be used. To ensure normal use of the heat sink, the controller will automatically open the valve fully every ' +
                'two weeks. It will run for 30 seconds per time with the screen displaying "Ad", then return to its normal working state ' +
                'again.'),
            e.binary('frost_protection', ea.STATE_SET, 'ON', 'OFF').withDescription('When the room temperature is lower than ' +
                '5 °C, the valve opens; when the temperature rises to 8 °C, the valve closes.'),
            e.numeric('error', ea.STATE).withDescription('If NTC is damaged, "Er" will be on the TRV display.'),
            e.binary('boost_heating', ea.STATE_SET, 'ON', 'OFF')
                .withDescription('Boost Heating: the device will enter the boost heating mode.'),
        ],
        meta: {
            tuyaDatapoints: [
                [2, 'preset', tuya.valueConverterBasic.lookup(
                    {'schedule': tuya.enum(0), 'holiday': tuya.enum(1), 'manual': tuya.enum(2), 'comfort': tuya.enum(3), 'eco': tuya.enum(4)})],
                [4, 'current_heating_setpoint', tuya.valueConverter.divideBy10],
                [5, 'local_temperature', tuya.valueConverter.divideBy10],
                [6, 'battery', tuya.valueConverter.raw],
                [7, 'child_lock', tuya.valueConverter.lockUnlock],
                [21, 'holiday_temperature', tuya.valueConverter.divideBy10],
                [24, 'comfort_temperature', tuya.valueConverter.divideBy10],
                [25, 'eco_temperature', tuya.valueConverter.divideBy10],
                [28, 'schedule_monday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(1)],
                [29, 'schedule_tuesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(2)],
                [30, 'schedule_wednesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(3)],
                [31, 'schedule_thursday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(4)],
                [32, 'schedule_friday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(5)],
                [33, 'schedule_saturday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(6)],
                [34, 'schedule_sunday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(7)],
                [35, 'fault_alarm', tuya.valueConverter.errorOrBatteryLow],
                [36, 'frost_protection', tuya.valueConverter.onOff],
                [37, 'boost_heating', tuya.valueConverter.onOff],
                [39, 'scale_protection', tuya.valueConverter.onOff],
                [47, 'local_temperature_calibration', tuya.valueConverter.localTempCalibration2],
                [49, 'system_mode', tuya.valueConverterBasic.lookup({'off': tuya.enum(0), 'heat': tuya.enum(1)})],
            ],
        },
    },
```

elf0heart, 2023-11-19:
@T0ytoy Did you get any success integrating @jim-fx 's answer in ZHA through a custom quirk ? I am in the same situation, bought it but can not get the sensors using ZHA. Cheers !

T0ytoy, 2023-11-19:
@elf0heart I tried tonight to convert from @jim-fx  zigbee2mqtt converter to a zha quirk but it seems that is a bit above my skill limit. My quirk load but no entity appears. As of now the only solution seem to be using zigbee2mqtt :(

elf0heart, 2023-11-20:
Thanks for the try : ) I try to gather some information I found through my other research. I was in the same situation with another thermostat valve, topic discussed here : [Link](https://community.home-assistant.io/t/moes-trvs-not-showing-most-of-the-atttributes/604384). After putting the custom quirk "beca", detailed by @Rofo , got the 20 entities showing up in ZHA. Not sure if anyone can help more on this topic ? @Rofo, @R1DEN . I understand it takes a substantial amount of time developing such quirks, but could anyone share the method to get there ? Thx !

GigaDive, 2023-11-24:
@T0ytoy @elf0heart I also used the already known tuya quirks but they do not work so far. 
In this [repository](https://github.com/zigpy/zha-device-handlers/tree/002037ab4f7ec02ad482aaf8a806eb07c4e6a652/zhaquirks/tuya): [ts0601_trv.py](https://github.com/zigpy/zha-device-handlers/blob/002037ab4f7ec02ad482aaf8a806eb07c4e6a652/zhaquirks/tuya/ts0601_trv.py) and [ts0601_trv_sas.py](https://github.com/zigpy/zha-device-handlers/blob/002037ab4f7ec02ad482aaf8a806eb07c4e6a652/zhaquirks/tuya/ts0601_trv_sas.py) do load when importing them as a custom quirk. But the this [ts0601_valve.py](https://github.com/zigpy/zha-device-handlers/blob/002037ab4f7ec02ad482aaf8a806eb07c4e6a652/zhaquirks/tuya/ts0601_valve.py) does not work and results in an error.
I bought this [TRV ](https://de.aliexpress.com/item/1005006166009483.html?spm=a2g0o.order_list.order_list_main.23.4d605c5fUnBtfA&gatewayAdapt=glo2deu)from AliExpress and it seems to be the same. Any method to solve this issue? Currently, I lack the knowledge of the ZHA / zigpy codebase.

T0ytoy, 2023-11-24:
@GigaDive these quirks wouldn't work as they do not define MODELS_INFO for the TVR this thread is about (TS0601_TZE204_pcdmj88b). 

For reference this class loads (because I set the signature to match) :

```
class newTVR(TuyaThermostat):
    
    signature = {
        MODELS_INFO: [
            ("_TZE204_pcdmj88b", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufClusterAttributes.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.THERMOSTAT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    ManufCluster,
                    # SiterwellThermostat,
                    # SiterwellUserInterface,
                    # TuyaPowerConfigurationCluster2AA,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        }
    }
```

The difficult part is to define which replacement clusters to use, and write respective classes that interpret "tuya format" and convert it to zha-understandable format. That remains a bit out of reach for me, sadly.

Teka101, 2023-11-24:
Hello,

Very experimental patch (and very ugly), just add these lines at the end of file ts0601_trv.py:
```
from typing import Tuple
from zhaquirks.const import (
    SKIP_CONFIGURATION,
)

PCDM_PRESET = 1026 #OK
PCDM_TARGET_TEMP_ATTR = 516 #OK
PCDM_TEMPERATURE_ATTR = 517 #OK
PCDM_BATTERY_ATTR = 518 #OK
PCDM_CHILD_LOCK_ATTR = 1073 #nop?
PCDM_SYSTEM_MODE_ATTR = 293 #nop?
#1315 ?window_mode?

class PcdmManufTrvCluster(TuyaManufClusterAttributes):
    """Manufacturer Specific Cluster of some thermostatic valves."""

    class Preset(t.enum8):
        """Working modes of the thermostat."""

        Schedule = 0x00
        Away = 0x01
        Manual = 0x02
        Comfort = 0x03
        Eco = 0x04

    set_time_offset = 1970

    attributes = TuyaManufClusterAttributes.attributes.copy()
    attributes.update(
        {
            PCDM_PRESET: ("operation_preset", Preset, True),
            PCDM_TARGET_TEMP_ATTR: ("target_temperature", t.uint32_t, True),
            PCDM_TEMPERATURE_ATTR: ("temperature", t.uint32_t, True),
            PCDM_BATTERY_ATTR: ("battery", t.uint32_t, True),
            PCDM_CHILD_LOCK_ATTR: ("child_lock", t.uint8_t, True),
            PCDM_SYSTEM_MODE_ATTR: ("system_mode", t.uint8_t, True),
        }
    )

    TEMPERATURE_ATTRS = {
        PCDM_TARGET_TEMP_ATTR: "occupied_heating_setpoint",
        PCDM_TEMPERATURE_ATTR: "local_temperature",
    }
    
    def handle_cluster_request(
        self,
        hdr: foundation.ZCLHeader,
        args: Tuple,
        *,
        dst_addressing: Optional[
            Union[t.Addressing.Group, t.Addressing.IEEE, t.Addressing.NWK]
        ] = None,
    ) -> None:
        _LOGGER.debug(
            "handle_cluster_request: [0x%04x:%s:0x%04x] Received value (command 0x%04x)",
            self.endpoint.device.nwk,
            self.endpoint.endpoint_id,
            self.cluster_id,
            hdr.command_id,
        )
        _LOGGER.debug('%d # %s', len(args), str(args))
        return super().handle_cluster_request(hdr, args, dst_addressing=dst_addressing)
    
    async def write_attributes(self, attributes, manufacturer=None):
        _LOGGER.debug('write_attributes %s', str(attributes))
        return await super().write_attributes(attributes, manufacturer)
[... 98 lines cut ...]
        },
    }

    replacement = {
#        SKIP_CONFIGURATION: True,
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.THERMOSTAT,
#                DEVICE_TYPE: zha.DeviceType.TEMPERATURE_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    PcdmManufTrvCluster,
                    PcdmThermostat,
                    PcdmUserInterface,
                    #TODO MoesWindowDetection
                    TuyaPowerConfigurationCluster2AA,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        }
    }
```

elf0heart, 2023-11-25:
Thanks for looking at the issue @Teka101 . I incorporated the few lines of code in the quirk, it's getting better, got the battery as a sensor, as well as the thermostat, see printscreens below. However, seems there is still a lot of entities missing. It seems that the thermostat card can only 'reads' what the physical TRV is showing. At the moment, I can not set the temperature for instance or get the % opening of the TRV. THanks anyways, this is good progress !

[image]
[image]

Teka101, 2023-11-25:
@elf0heart yes it's only work for reading... i'm working on it 💪

medivel, 2023-11-29:
Thank you for your efforts so far! It's a pity that it is so hard to get it work. Looking forward that you'll get it done!

elf0heart, 2023-12-01:
Thanks for trying @Teka101 , indeed every time the temperature is changed on the thermostat card, the heating setpoint comes back to what it was previously. Can't help much, sorry...

Teka101, 2023-12-03:
@elf0heart  ok i can change temperature and preset now. Can you try if it's okay on your installation ?

elf0heart, 2023-12-04:
@Teka101 , just tried. I confirm that the heating setpoint can now be set through home assistant, works in writing now ! Thanks for that. However the 'switch', whenever turned on in home assistant, comes back to 'off' a few seconds later. FOr my usage I do not mind, since I am using each TRV with the scheduler custom card (i always use it in 'heating mode', by just changing the heating setpoints from 16 to 20 °C for instaance). 
When pairing the TRV this time though, didn't get the usual screen with "3 entities" have been found. But when going to ZHA, devices, the TRV appeared well. The main function is therefore working. (i would still be interested in having a few more entities, like "% opening" of the TRV). Cheers !
[image]

Teka101, 2023-12-04:
@elf0heart switch is for "boost mode" but it doesn't work yet :(
And if i'm right the TRV doesn't report valve state (it's only open or close...)

T0ytoy, 2023-12-04:
@Teka101 well, that is outstanding, thank you for your work :)
I added your quirk to my hass instance and am testing it for a few days. So far it seems to work pretty well. Preset don't seem to be selectable from home assistant (only through the physical button on the TVR)

I have another of those TRV working with zigbee2mqtt at the same time, the implementation has a few more entities available as shown below: 

[image]

I would say the most important ones are `local temperature calibration` (to tweak internal temperature sensor value) and maybe preset target temperature (only default values are available right now).

Again, thank you very much for your work, and please let me know if you're interested in any details regarding those TVR using Z2M.

T0ytoy, 2023-12-04:
It seems preset selection and their temperature is easily accessible through cluster 0xef00 👍

[image]

I wasn't however able to change the value of the local_temperature_calibration attribute (0x0010) in cluster 0x0201: no effect on reported local temperature (0x0000)

Teka101, 2023-12-05:
@elf0heart yes there is more to do :)

@T0ytoy preset can be change with group `PcdmThermostat`

At this time, you can only change:
- target temperature
- child mode lock (no visual integration in HASS but working with service call)
- preset mode : eco / manual / away / schedule

And in read only, we have :
- battery level
- valve is open or closed

I'm still working on it

PS: je vois des français partout ^_^

T0ytoy, 2023-12-11:
@Teka101 how did you find out attributes identifiers? Ex:

```
PCDM_PRESET = 1026 #                010000 000010 2
PCDM_TARGET_TEMP_ATTR = 516 #       001000 000100 4
PCDM_TEMPERATURE_ATTR = 517 #       001000 000101 5
PCDM_BATTERY_ATTR = 518 #           001000 000110 6
```
I'm trying to add temperature_calibration, this is the value I guessed:

`PCDM_TEMPERATURE_CORRECTION_ATTR = 559  # 001000 101111 47`

I added it to attributes.update in `PcdmManufTrvCluster` and added:

```
elif attrid == PCDM_TEMPERATURE_CORRECTION_ATTR:
            _LOGGER.debug("update attribute calibration")
            self.endpoint.device.thermostat_bus.listener_event("temperature_correction_change", value)
```

at the end of the `_update_attribute()` method but it doesn't seem to od anything. 

If you have any clue, I would love to hear about it :)

Teka101, 2023-12-11:
Hello @T0ytoy 

You need to add entry in attributes (variable `attributes`):
```
PCDM_TEMPERATURE_CORRECTION_ATTR: ("temperature_calibration", t.int32s, True),
```
Maybe you need to divide by 10 temperature before send it, i don't know...

If still doesn't work maybe the prefix 512 apply on PCDM_TEMPERATURE_CORRECTION_ATTR is not the good one...
At this time, i don't know when apply prefix 256,512 or 1024...

T0ytoy, 2023-12-11:
I think i did already:

```
class PcdmManufTrvCluster(TuyaManufClusterAttributes):
    """Manufacturer Specific Cluster of some thermostatic valves."""

    def __init__(self, *args, **kwargs):
        """Init."""
        super().__init__(*args, **kwargs)
        global PcdmManuClusterSelf
        PcdmManuClusterSelf = self

    set_time_offset = 1970

    attributes = TuyaManufClusterAttributes.attributes.copy()
    attributes.update(
        {
            PCDM_PRESET: ("operation_preset", t.uint8_t, True),
            PCDM_BATTERY_ATTR: ("battery", t.uint32_t, True),
            PCDM_BATTERY_LOW_ATTR: ("battery_low", t.uint8_t, True),
            PCDM_BOOST_MODE: ("boost_duration_seconds", t.uint32_t, True),
            PCDM_CHILD_LOCK_ATTR: ("child_lock", t.uint8_t, True),
            PCDM_SYSTEM_MODE_ATTR: ("system_mode", t.uint8_t, True),
            PCDM_TARGET_TEMP_ATTR: ("target_temperature", t.uint32_t, True),
            PCDM_TEMPERATURE_ATTR: ("temperature", t.uint32_t, True),
            PCDM_TEMPERATURE_CORRECTION_ATTR: ("temperature_correction", t.int8s),

            PCDM_TARGET_MANUAL_ATTR: ("occupied_heating_setpoint", t.uint32_t, True),
            PCDM_TARGET_CONFORT_ATTR: ("comfort_heating_setpoint", t.uint32_t, True),
            PCDM_TARGET_ECO_ATTR: ("eco_heating_setpoint", t.uint32_t, True),
        }
    )

I'll try different prefixes, thx.
```

Teka101, 2023-12-11:
@T0ytoy when i look at other quirks, `temperature_calibration` is type `t.int32s`

T0ytoy, 2023-12-11:
@Teka101 well it seems it did the trick, I have access to the calibration value now! it doesn't appear as an entity yet though, it only works through zha interface "write attribute", but it works! 

Here is the converter I'm testing:

```
import logging
from typing import Optional, Tuple, Union

from zigpy.profiles import zha
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import (
    AnalogOutput,
    Basic,
    BinaryInput,
    Groups,
    OnOff,
    Ota,
    Scenes,
    Time,
)

from zhaquirks import Bus, LocalDataCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaManufClusterAttributes,
    TuyaPowerConfigurationCluster2AA,
    TuyaThermostat,
    TuyaThermostatCluster,
    TuyaUserInterfaceCluster,
)

_LOGGER = logging.getLogger(__name__)

# MQTT
# {
#   "2": "Mode",
#   "4": "Set temperature",
#   "5": "Current temperature",
#   "6": "Battery capacity",
#   "7": "Child lock",
#   "8": "Temperature scale",
#   "9": "Set temperature ceiling",
#   "10": "The lower limit of temperature",
#   "14": "Window check",
#   "16": "Window temp",
#   "17": "Window time",
#   "18": "Backlight brightness",
#   "19": "Factory data reset",
#   "21": "Holiday temperature",
#   "24": "Home temp", || comfort_temperature
#   "25": "Leave temp", || eco_temperature
#   "28": "Week program",
#   "29": "Week program Tuesday",
#   "30": "Week program Wednesday",
#   "31": "Week program Thursday",
#   "32": "Week program Friday",
#   "33": "Week program Saturday",
#   "34": "Week program Sunday",
#   "35": "Fault alarm",
#   "36": "Frost protection",
#   "37": "Rapid warming",
#   "38": "Rapid heating countdown",
[... 369 lines cut ...]
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.THERMOSTAT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    PcdmManufTrvCluster,
                    PcdmBoost,
                    PcdmThermostat,
                    PcdmUserInterface,
                    # PcdmWindowDetection,
                    TuyaPowerConfigurationCluster2AA,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            }
        }
    }

###
```

I'm still wondering how to make it show up as an entity, but there is progress!

Teka101, 2023-12-14:
@T0ytoy ok thank you for your code.

I just publish a new version with: window detection mode and temperature calibration

T0ytoy, 2023-12-16:
@Teka101 I was finally able to test you update, it seems to work thank you! In the home assistant zha UI, boost mode and window detection mode are both switches without a name, so it isn't clear which does what at first glance, but it's not a big problem.

If I have some time this week-end, I'll try to implement the right class so that temprature calibration gets it's own numeric entity in home assistant, so that it can be used easily.

I use the calibration feature (it has an entity on the z2m integration) to correct the device internal temperature to the temperature of an external zigbee thermometer: that way when the radiator heats up, the TVR temperature does not increase just because it's too close to the radiator. It is done home assistant side, that is why I'm deseperately trying to get that entity :D

I'll let you know if I'm getting anything done. Thank you!

royduin, 2023-12-16:
Nice work @Teka101! Curious; why didn't you create a PR or is it not fully ready yet? For now I'm using the custom quirk

T0ytoy, 2023-12-17:
Good news, I was able to make the temperature offset (or calibration) have it's own entity in Home Assistant. It seems to work well enough, and was the feature I wanted the most :)

Here is the code :

<details>
  <summary>code</summary>

  ```
  import logging
  from typing import Optional, Tuple, Union
  
  from zigpy.profiles import zha
  import zigpy.types as t
  from zigpy.zcl import foundation
  from zigpy.zcl.clusters.general import (
      AnalogOutput,
      Basic,
      BinaryInput,
      Groups,
      OnOff,
      Ota,
      Scenes,
      Time,
  )
  
  from zhaquirks import Bus, LocalDataCluster
  from zhaquirks.const import (
      DEVICE_TYPE,
      ENDPOINTS,
      INPUT_CLUSTERS,
      MODELS_INFO,
      OUTPUT_CLUSTERS,
      PROFILE_ID,
  )
  from zhaquirks.tuya import (
      TuyaManufClusterAttributes,
      TuyaPowerConfigurationCluster2AA,
      TuyaThermostat,
      TuyaThermostatCluster,
      TuyaUserInterfaceCluster,
      TUYA_DP_TYPE_BOOL,
      TUYA_DP_TYPE_VALUE,
      TUYA_DP_TYPE_ENUM,
      TUYA_DP_TYPE_FAULT
  )
  
  _LOGGER = logging.getLogger(__name__)
  
  # Features                                  | Implemented
  # ------------------------------------------|------------
  # - preset:                                 |
  #     * manual mode                         | yes
  #     * holiday mode                        | yes
  #     * eco mode                            | yes
  #     * programming mode                    | partialy => calendar cannot be change
  #     * comfort mode                        | yes
  # - child lock                              | yes
  # - AF antifreeze mode                      | no
  #   -> set to 8°C the temperature           |
  # - BS: quick heat                          | no
  #   -> TRV full open for 5 min              |
  # - CC : offset temperature                 |
  #   -> to adjust internal sensor temp.      | yes
  # - EE: blind spot                          | no
  #   -> adjust heating point by offseting    |
  # - DP: open window detection               | yes
  # - HS : thermal stop -> fully closed       | no
  
  # MQTT
  # {
[... 463 lines cut ...]
                      PcdmThermostat,
                      PcdmUserInterface,
                      TuyaPowerConfigurationCluster2AA,
                  ],
                  OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
              },
              2: {
                  PROFILE_ID: zha.PROFILE_ID,
                  DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                  INPUT_CLUSTERS: [
                      PcdmWindowDetection,
                      PcdmTemperatureOffset,
                  ],
                  OUTPUT_CLUSTERS: [],
              },
          }
      }
  
  ###
  ````
</details>

The important part is the class 'PcdmTemperatureOffset', and be careful to add 'self.temperature_calibration_bus = Bus()' to the init method of class 'PcdmTrv'.

Last feature I'd like to see is entities for presets (switch to enable/disable them, and a number entitiy for each preset to set target temperature), let me know if you start working on it, otherwise I may or may not try it myself later (I need a break right now 😄 )

Teka101, 2023-12-17:
@royduin i'm trying to check if all features implemented works and after i will submit a PR :)

@T0yto great ! i'll check code asap (i only work 1 or 2 hours by week on this project, so sorry for delay in response)
In term of Home-Assistant integration the new version 2023.12 try new things in order to improve integration of ZIGBEE devices... maybe in the futur, we don't have to bring patch to have full features on HA.

Really good job for your blueprint !

ed-wright, 2023-12-18:
@T0ytoy thanks for the hard work, i am testing your latest changes and it does indeed show the offset as an entity, it does however appear to have no effect when changed, at least on my thermostats (i have 4 of these running). I may be doing something incorrect so if that is the case please correct me but if not i am happy to test

T0ytoy, 2023-12-18:
@ed-wright it takes about ~35 seconds for it to applied to the current temperature. After moving the calibration slider, can you chek the 'PcdmManufTrvCluster' cluster calibration attribute (using ZHA) and make sure the read value is equal to the slider value?

ed-wright, 2023-12-18:
@T0ytoy i get the following

>  Failed to call service number/set_value. Failed to send request: Request failed after 5 attempts: <Status.MAC_NO_ACK: 233>

ed-wright, 2023-12-18:
[image]
[image]

So i set the number to 12 and -12 and neither changed the value

T0ytoy, 2023-12-18:
@ed-wright I just checked, the quirk I have running on my HA and the one I linked above are exactly the same, and it's working fine for me, so I don't really know what to think. Could you maybe have a different version of the TRV? Mine came in a blue box, the user manuel front page says "Model: BAB-1413Pro-E".

I guess since the quirk is loaded and would only do so with `"_TZE204_pcdmj88b", "TS0601"`, the signature is right.
Does it work if you change the calibration value in the "manage Zigbee device" menu?

ed-wright, 2023-12-18:
@T0ytoy i have confirmed the quirk it running, mine did come in a blue box anoyingly i dont have the box to hand.

[image]
when i set it manually it works and on the hardware unit itself the temperature is reflected and it shows on the thermostat entity too

ed-wright, 2023-12-18:
@T0ytoy can i ask a really dim question, is the version you have in git the same as the one in your post above?

Again, thanks for the hard work and helping debug!

T0ytoy, 2023-12-18:
@ed-wright I never see `set_value` nor `get_value`, that is why I put comments on those methods. No worry 😄

What do you mean by "the git" ? I only shared my version on the post above, under the spoiler tag "code". Could you clarify?

ed-wright, 2023-12-18:
perfect clarification, just wanted to check that the version you had and the version in this thread was the same, thanks!

ed-wright, 2023-12-18:
@T0ytoy i have also tried changing the str(int(float(value))) and removed the str to see if it works

T0ytoy, 2023-12-18:
What version of home assistant are you running? I am running 2023.12.3,maybe there was a change to how _number_ is handled, I don't know?

ed-wright, 2023-12-18:
I was on 2023.12.1, i am updating to 2023.12.3

ed-wright, 2023-12-19:
@T0ytoy i have found the box and they are the same Model: BAB-1413Pro-E

ed-wright, 2023-12-19:
@T0ytoy Ah! I got it working, I deleted the unit in HA entirely and readded it and it has now magically started working, i have no idea why as I did tell HA to reconfigure the device. Thanks for the troubleshoting work and hard work creating the quirk! Many Thanks

T0ytoy, 2023-12-19:
@ed-wright Good news! It kinda make sense: the issue was probably on the zigbee association side, I'm glad there is no intricated technical issue with the python code, as I'm really not comfortable debugging advanced issues in this context 😄 

To give credit to where it's due: 890% of the work was done by @Teka101, I mostly just worked on the calibration feature. Many thanks to him :)

As a side note, I'm currently experimenting with using an average value over ~5-10 minutes instead of the raw "0.5 °C resolution shit data" locale temperature data from the TVR to feed the blueprint I made, I think it might be working a bit better since the raw temperature is jumping +-1°C all the time and the automation blueprint reacts a lot to try to compensate. I'm hoping it will generate less spikes above and below target temperature.

ed-wright, 2023-12-19:
@T0ytoy looks like I may have spoken too soon I can see in ZHA that the number is being set but it still looks like the actual offset is not, I will debug

Teka101, 2023-12-26:
Hello,

I've made an update of my code (@T0ytoy now slider for temperature calibration is working, thank for your code).

Maybe someone can check if boost works ? Because i think, it's broken on this feature :/

Teka101, 2023-12-30:
PR: https://github.com/zigpy/zha-device-handlers/pull/2873

synchronierer, 2024-04-15:
Hello all,
first things first: thx so much for your work, that's so helpful for the community.

May a noob ask, where I can find the latest version of the code?

Thx in advance!

Teka101, 2024-04-16:
Hello @synchronierer,

Last version of code is in pull request : https://github.com/zigpy/zha-device-handlers/pull/2873

synchronierer, 2024-04-18:
@ Teka101:
Thx so much, HA wouldn't be the same without people like you.
I hope the support for the device will soon be integrated ofiicially.

I use the lastest code for the custom quirk and to them to be controllable. I even set up a heating program with Scheduler und Scheduler Card from HACS, but it's still not very intuitive.
Is there a way to change temperature manually and go back to the schedule temperature after a certain time/event?

closed 2024-10-22 by github-actions[bot]