zigpy/zha-device-handlers #2707: Bosch RBSH-RTH0-ZB-EU Thermostat II switch between heating and cooling

metallmanu, 2023-11-05:
### Problem description

The Bosch Smart Home Thermostat II Supports heating and cooling in the Bosch Smart Home App. Unfortunately this switch between heating and cooling is not possible through ZHA in Homeassistant. Over climate.set_hvac_mode it is only possible to switch between off and heating

### Solution description

The hvac_modes cooling, heating and off are possible to choose

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

jclsn, 2023-11-06:
I am planning to write a quirk to expose all of the endpoints to Home Assistant.

For now you can set the available modes by 

-> going to the ZHA device page
-> pressing the dot menu
-> selecting "manage zigbee device"
-> thermostat endpoint
-> ctrl_sequence_of_oper (id: 0x001b)
-> read current value (and post it here) :)

I have found the following so far

ControlSequenceOfOperation.Heating_Only
ControlSequenceOfOperation.Cooling_Only

But there is probably an easier way to query all possible values from the device.

metallmanu, 2023-11-07:
> I am planning to write a quirk to expose all of the endpoints to Home Assistant.
> 
> For now you can set the available modes by
> 
> -> going to the ZHA device page -> pressing the dot menu -> selecting "manage zigbee device" -> thermostat endpoint -> ctrl_sequence_of_oper (id: 0x001b) -> read current value (and post it here) :)
> 
> I have found the following so far
> 
> ControlSequenceOfOperation.Heating_Only ControlSequenceOfOperation.Cooling_Only
> 
> But there is probably an easier way to query all possible values from the device.
> 
> Ah, here they are: https://docs.bosch-iot-suite.com/gateway-software/API/modules/com.prosyst.mbs.zigbee.driver.api/com/prosyst/mbs/services/zigbee/zcl/hvac/ThermostatCluster.html

Hey wow thats great.
When I checked the cluster for me there are also Only the values
ControlSequenceOfOperation.Heating_Only ControlSequenceOfOperation.Cooling_Only

Would be very nice to see a quirk for this 👏

metallmanu, 2023-11-07:
If it is possible you can also add the Installation mode for the heating: 230V, 24V or Boiler and if there are NO or NC valves

jclsn, 2023-11-07:
@metallmanu There are also

```
ControlSequenceOfOperation.Heating_With_Reheat
ControlSequenceOfOperation.Cooling_With_Reheat
```

The others with the 4_Pipes from the specification do not work. No idea what they are for. Anyway, I think the issue is that the only SystemMode settings are 

```
SystemMode.Heat
SystemMode.Cool
```

There is not SystemMode.Off, that HA would expect. I guess I would need to map it to Off. 

You can find all working Endpoints I could identify yesterday here

https://github.com/zigpy/zha-device-handlers/issues/2476#issuecomment-1796390877

Feel free to test it out and maybe extend the list. My thermostat seems to have connection issues and I may have missed some.

jclsn, 2023-11-07:
> If it is possible you can also add the Installation mode for the heating: 230V, 24V or Boiler and if there are NO or NC valves

Oh, we may not have the same thermostat. I have the TRV for radiators. Not sure if this supports boilers etc. I could maybe support it as well though.

metallmanu, 2023-11-07:
Oh yes it is another thermostat. I added the signature and diagnostic of mine. But it seems to be very similar to yours in states.
For me the SystemMode.Off is working. 
[image]

jclsn, 2023-11-07:
That is odd. I do have this one

[image]

Isn't yours a radiator thermostat as well? There may be many different version then.

metallmanu, 2023-11-07:
I have this one:

[image]

It is for underfloor heating with cable system 230V.
Therefore the cooling part is neccessary when you have a heat pump which can also cool the underfloor and also the valves control would be nice, because there are valves which are normaly open and some which are normaly closed.
Hopefully your quirk will also work for this

jclsn, 2023-11-07:
I see. I just don't see how my radiator valve would be cooling anything. Radiators only heat usually. Really weird!

metallmanu, 2023-11-07:
When you have a Heating pump instead of Oil, Gas etc. you can let cold water Flow through your heating cycle. Therefore you have to put the thermostat also in cooling Mode, so that the valve is Open when you want a cooler temperature and cold water can flow through your Radiator

jclsn, 2023-11-07:
Interesting. That makes sense. Guess it is a quite sophisticated thermostat. It is a pity that I am losing the connection frequently. I have to take out the batteries to be able to access it again. I put back the Tado now that I want to get rid of and will buy a radiator valve tomorrow, so the thermostat can calibrate itself to something. I will then try to implement the quirk.

I think the quirk for your device will be quite similar. There may be some endpoints working that don't work on mine and vice versa. Would be good if you could test those endpoints out, see what they return, compile a similar chart and post it here. On the Bosch documentation page you will find some examples. Some of them might be named a little differently though.

metallmanu, 2023-11-07:
Yes the diagnostic is in my first post. Do you need more?

jclsn, 2023-11-07:
Well, a bit of patience I guess. I have only edited a quirk so far. A have just tried zigbee2mqtt and everything is supported there. Even endpoints that do not work with ZHA. I wonder why that is.

Pity that @dmulcahey is not responding atm. He is usually a great help!

markoberholzer-es, 2024-01-07:
Hi @metallmanu, @jclsn - I stubled accross this thread as I am facing the same issue with my Bosch Thermostat II 230V. Thx to your hints, I created a small quirk which enhances the Thermostat with the possibility to switch it into cooling mode.

However the widget now shows an additional button (the one on the left), which seems to be the auto mode, which is not supported by the affected thermostat. 
[image]

Maybe you guys know how to remove that auto mode ?

Here is the simple quirk that I have written:
[rsbh_rth0_zb_eu.py.zip](https://github.com/zigpy/zha-device-handlers/files/13853835/rsbh_rth0_zb_eu.py.zip)

jclsn, 2024-01-07:
@markoberholzer-es and all the rest.

You all seem to not understand what a quirk does. A quirk only adds missing attributes that don't comply with the Zigbee specification. It does NOT at all determine what you see in Home Assistant. 

The ZHA code in the core is implements custom entities for different brands. There is for example something you could call a renderer for Tuya devices that creates entities for Tuya custom attributes. None of this exists for Bosch and therefore ZHA tries to create entities according to the Zigbee specification. 

So you will not be able to fix it from the quirk side. These renderers have to be added to the HA core, where ZHA lives and I don't have the time or expertise yet to implement that.

markoberholzer-es, 2024-01-07:
Appologies @jclsn, if my comment upset you - I just wanted to share my findings after hours of digging through the repository.

I read through the Zigbee Cluster Library Specification and the problem is that the `ctrl_sequence_of_oper` attribute of ther thermostat is set to `Heating Only` (0x02). The quirk that I posted **fixes that** and it is now **possible** to change the modes through the GUI - the only issue is, as I mentioned, that the thermostat does not support the hvac mode `heat_cool` - so possibly this has to be fixed in HA core.   

[image]

jclsn, 2024-01-07:
@markoberholzer-es No reason to be sorry. I just confused this with the other thread where a few people have asked the same thing over and over. I can also be quite impulsive sometimes. I edited the post to make it easier to understand and sound a bit nicer hopefully :)

Yeah, there are different attributes. The **ctrl_sequence_of_oper** exists for almost every thermostat afaik and is never exposed in HA. If you want to have it, it would probably be best to add it as select entity.

https://github.com/home-assistant/core/blob/dev/homeassistant/components/zha/select.py

Then there is also the "heat" and "cool" and then "pause", "manual" and "auto". Those should probably be added to the climate entity. I suspect that it is not so easy, but I also could be wrong.

docloy, 2024-01-30:
@markoberholzer-es I did try to install your quirk.
1. At first I enabled zha custom quirks and die save your quirk in this folder. 
2. restart ha
3. removed Bosch Thermostat II 230V
4. added it by qr code. 

But there is still no entity to switch to cooling mode. Hopefully you have a hint, why it is not working. Thanks in advance!

jclsn, 2024-01-30:
Yes, because you need cluster handlers for this, which need to be implemented in the HA core. I have already started doing that, but it is not easy and the ZHA devs are very busy atm.

docloy, 2024-02-13:
@jclsn and @markoberholzer-es thanks for your reply and your input. In my configuration.yaml I have those 3 lines. But I don't know how to acitvate use the quirk.
But through the zigbee device settings --> Manage Zigbee Device --> Clusters:  BoschThermostatCluster (Endpoint id: 1, Id: 0x0201, Type: in)
I can set manually the attribute and the value for cooling:
```
Attributes of the selected cluster: ctrl_sequence_of_oper (id: 0x001b)
Value: 0x00
```
And for heating:
```
Attributes of the selected cluster: ctrl_sequence_of_oper (id: 0x001b)
Value: 0x02
```

At the moment I'm fine with this solution because I have only to switch twice a year. 
Maybe it'll be implemented in core. Anyway I have no idea how to use quirks (would have been interesting as well).

metallmanu, 2024-08-11:
Poorly the issue is still open with the new updates of 2024.8

metallmanu, 2024-10-04:
Logger: zigpy.quirks.v2
Quelle: custom_zha_quirks/RBSH_RTH0_ZB_EU.py:26
Erstmals aufgetreten: 23:29:54 (1 Vorkommnisse)
Zuletzt protokolliert: 23:29:54

add_to_registry_v2 is deprecated and will be removed in a future release. Please QuirkBuilder() instead and ensure you call add_to_registry().

atsju, 2024-12-07:
Same problem with Schneider thermostat. It doesn't have a physical button to switch to cooling mode but it definitely support cooling mode (I have an hydronic floor and will need it this summer). This issue is probably common to many thermostats on the market.

Both ControlSequenceOfOperation and SystemMode are read/Write attributes. 
[image]
[image]
[image]
In my thermostat by default they are set to `0x02 Heating Only` and `0x04 Heat` respectively. This does not mean that they cannot be set to cooling. And this is wrongly interpreted by ZHA ? I haven't looked at related code (yet).

Linking #3587 for cooling to be implemented if someone does the custom quirk.

Asky82, 2026-02-09:
Hi all, I have 10 Radiator Thermostat from Bosch connected directly via a dongle Sonoff added to HA.With the last update of this month of HA the climate show only HVAC mode OFF.  :( What can I do?? Helppp

Asky82, 2026-02-10:
Thanks, in the mean time I did a reset and re-pair them.
The climate modes then are back (off, heat) however entities behaviors are not correct. False temperatures, entities non updated..

metallmanu, 2026-09-03:
yesterday I get an update to Firmware 0x03096a90 for my bosch.rbsh_rth0_zb_eu. There is a new switch which I don´t know what it is.
Bosch Firmware support page says:
Raum-Thermostat II 230V (0.3.9): Fehlerbehebung: Der Entkalkungsschutz ist jetzt auch im Modus Heizpause/Aus aktiv.

metallmanu, 2026-10-05:
There is a new Info at zigbee2mqtt for this switch in firmware 0.3.9:
https://www.zigbee2mqtt.io/devices/BTH-RM230Z.html
[Humidity alarm led (binary)](https://www.zigbee2mqtt.io/devices/BTH-RM230Z.html#humidity-alarm-led-binary)
Enables/disables the LED warning when measured humidity is outside the comfortable range (below 30% or above 70%). Only the two observed raw values (0x07 on / 0x06 off) are supported; the meaning of the other bits of this attribute is not confirmed. The attribute is present on firmware 0x03086a90 (0.3.8) as well (confirmed by reading it on an unupgraded device); seemd the Bosch app only added a UI toggle for it in 0x03096a90 (0.3.9).. Value can be found in the published state on the humidity_alarm_led property. To read (/get) the value publish a message to topic zigbee2mqtt/FRIENDLY_NAME/get with payload {"humidity_alarm_led": ""}. To write (/set) a value publish a message to topic zigbee2mqtt/FRIENDLY_NAME/set with payload {"humidity_alarm_led": NEW_VALUE}. If value equals ON humidity alarm led is ON, if OFF OFF.

Can you please add this also to ZHA?