zigpy/zha-device-handlers #2708: [Device Support Request] Silvercrest SAPZ A2  (TS011F par _TZ3000_ynmowqk2 )

jacme31, 2023-11-06:
### Problem description

I try to pair a LIDL Device (Silvercrest SAPZ A2 : a new version sold in France with power consumption measuremenst of the SAPZ A1 which already works well with ZHA) .
During the pairing ZHA loads the quirk "zhaquirks.tuya.ts011f_plug.Plug" and everything seems to works, but the device stays in pairing mode,  it disconnect/ reconnects multiple times before disconnecting defintevly.

### Solution description

A well connected  ZHA device  ... 

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[
[image]
]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x010a",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04",
        "0xe000",
        "0xe001"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_ynmowqk2",
  "model": "TS011F",
  "class": "zhaquirks.tuya.ts011f_plug.Plug"
}]

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.11.1",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.6",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Paris",
[... 677 lines cut ...]
2023-11-06 17:41:08.942 INFO (MainThread) [zigpy.application] Device 0x9327 (a4:c1:38:8a:59:c8:0e:a6) left the network
2023-11-06 17:41:08.959 INFO (MainThread) [zigpy.application] Device 0x9327 (a4:c1:38:8a:59:c8:0e:a6) left the network
2023-11-06 17:41:08.968 INFO (MainThread) [zigpy.application] Device 0x9327 (a4:c1:38:8a:59:c8:0e:a6) left the network
2023-11-06 17:41:08.985 INFO (MainThread) [zigpy.application] Device 0x9327 (a4:c1:38:8a:59:c8:0e:a6) left the network
2023-11-06 17:41:08.992 INFO (MainThread) [zigpy.application] Device 0x9327 (a4:c1:38:8a:59:c8:0e:a6) left the network
2023-11-06 17:41:08.999 INFO (MainThread) [zigpy.application] Device 0x9327 (a4:c1:38:8a:59:c8:0e:a6) left the network
2023-11-06 17:41:09.006 INFO (MainThread) [zigpy.application] Device 0x9327 (a4:c1:38:8a:59:c8:0e:a6) left the network
2023-11-06 17:41:09.019 INFO (MainThread) [zigpy.application] Device 0x9327 (a4:c1:38:8a:59:c8:0e:a6) left the network
2023-11-06 17:41:09.026 INFO (MainThread) [zigpy.application] Device 0x9327 (a4:c1:38:8a:59:c8:0e:a6) left the network
2023-11-06 17:41:09.034 INFO (MainThread) [zigpy.application] Device 0x9327 (a4:c1:38:8a:59:c8:0e:a6) left the network

]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

mdeweerd, 2023-11-11:
I just got several of these plugs too - same behavior.

For reference: [HA forum discussion](https://community.home-assistant.io/t/tuya-zigbee-ts011f-issue/507221/12) .

It indicates that this works with Z2M.

I tested the Tuya Magic Spell - not working better.

mdeweerd, 2023-11-11:
Here are some decodings of relevant packets from a sniffer log.

[_TZ3000_ynmowqk2.zip](https://github.com/zigpy/zha-device-handlers/files/13326145/_TZ3000_ynmowqk2.zip)

It's the device that is requesting to leave the network.

MattWestb, 2023-11-11:
I have only fast looking and the device is getting the network key OK and then the coordinator is requesting and getting the Node Descriptor Request, OK (R22 device) and the device is requesting the same and is getting R22 so shall being OK.
Then the device **shall** requesting updating the TC-Link key (R19 and higher is supporting it) but its not doing it so its policy is it must updating the TC-Link key => Leaving.

Must looking deeper of the flags and comparing with other device sniffs that is working OK.

MattWestb, 2023-11-11:
By the way is the Ti coordinator running on updated firmware (2023) ?

mdeweerd, 2023-11-11:
It's not a TI coordinator for me - it's an EZSP.

I opened a topic to see if anybody has the way to update the EZSP:
https://community.home-assistant.io/t/how-bellows-zigbee-repairs-nv3-interface-not-available-in-this-firmware-please-upgrade/635319 .

I've seen and add-on that seems to help with that, but not much of a procedure.

MattWestb, 2023-11-11:
Im very sorry but its only 2 frames with key and bot looks the same only network key and the sender is 0x000= and its .. . . 
```
Frame 83: 73 bytes on wire (584 bits), 73 bytes captured (584 bits) on interface unknown, id 0
IEEE 802.15.4 Data, Dst: 0x0d6a, Src: 0x0000, Bad FCS
ZigBee Network Layer Data, Dst: 0x0d6a, Src: 0x0000
    Frame Control Field: 0x0008, Frame Type: Data, Discover Route: Suppress Data
    Destination: 0x0d6a
    Source: 0x0000
    Radius: 30
    Sequence Number: 239
    [Extended Source: TexasIns_00:01:6a:41:0c (00:12:4b:00:01:6a:41:0c)]
    [Origin: 4]
ZigBee Application Support Layer Command
    Frame Control Field: Command (0x21)
    Counter: 206
    ZigBee Security Header
        Security Control Field: 0x30, Key Id: Key-Transport Key, Extended Nonce
        Frame Counter: 552963
        Extended Source: TexasIns_00:01:6a:41:0c (00:12:4b:00:01:6a:41:0c)
        Message Integrity Code: f2f2bfe3
        [Key: 5a6967426565416c6c69616e63653039]
        [Key Label: ZigBeeAlliance09 Key]
    Command Frame: Transport Key

Frame 282: 73 bytes on wire (584 bits), 73 bytes captured (584 bits) on interface unknown, id 0
IEEE 802.15.4 Data, Dst: 0x0d6a, Src: 0x0000, Bad FCS
ZigBee Network Layer Data, Dst: 0x0d6a, Src: 0x0000
    Frame Control Field: 0x0008, Frame Type: Data, Discover Route: Suppress Data
    Destination: 0x0d6a
    Source: 0x0000
    Radius: 30
    Sequence Number: 188
    [Extended Source: TexasIns_00:01:6a:41:0c (00:12:4b:00:01:6a:41:0c)]
    [Origin: 4]
ZigBee Application Support Layer Command
    Frame Control Field: Command (0x21)
    Counter: 18
    ZigBee Security Header
        Security Control Field: 0x30, Key Id: Key-Transport Key, Extended Nonce
        Frame Counter: 552964
        Extended Source: TexasIns_00:01:6a:41:0c (00:12:4b:00:01:6a:41:0c)
        Message Integrity Code: a3e26415
        [Key: 5a6967426565416c6c69616e63653039]
        [Key Label: ZigBeeAlliance09 Key]
    Command Frame: Transport Key
```
Have you burning one new IEEE in the Silabs chip ?

`Extended Source: TexasIns_00:01:6a:41:0c (00:12:4b:00:01:6a:41:0c)`

MattWestb, 2023-11-11:
The latest EZSP and RCP 7.3.x / 4.3.x can loading one software IEEE that is not burned in the chip (stored in the NVM3) but its no problem running older firmware that cant do it only if migrating to one other hardware if not having that support ZHA is burning one new IEEE in the chip and cant being reverted (with SWD flashing its possible then its in user data that cant being OTA flashed normally).

mdeweerd, 2023-11-11:
> Have you burning one new IEEE in the Silabs chip ?

Yes - I migrated from a CC2531.

MattWestb, 2023-11-11:
Shall working OK i have not testing it as addon only before in veritual env.
Try with different EZSP version if some is working OK. The 6.7.8.X is stable production and 6.10.X is candidate. All 6.8 and 6.9 is alpa and is not good for production.
The 7.1.X is little beta and the same with 7.2.X and normally is 7.3.2 the best if having one MG21 device.
Witch device is you using in your system is it any free firmware for it ?

mdeweerd, 2023-11-11:
I have SonOFF ZBDongle-E to enable - it had v6.10.3 .
I just made an update using the [webflasher](https://dialedin.com.au/blog/sonoff-zbdongle-e-rcp-firmware#web-flasher).

The Multiprotocol Add-on did not want to startup (did not find the port), so I have reboots the HAOS system (OS itself).  It still complains about "Failed to connect, secondary seems unresponsive" .

Trying to find out what to do...

mdeweerd, 2023-11-11:
I did not program the correct firmware, it was Zigbee FW with new version.
After flashing the RCP version, the integration started.
However I have lost all related entities.

The insturctions indicate:
> If you had an existing ZHA setup, you will need to remove this entirely. > Make sure to backup your network first!

So I removed it using the UI and then the configuration as well, but after restoration I do not have the entities ..  Kind of furious...

However, I could pair the TS011F

MattWestb, 2023-11-11:
One very interesting question is the RCP using the original IEEE or the burned one from the CC2531 but now you shall also having the possibility changing it without burning the chip IF our **Puddly** have making all right.
Then if its using the original IEEE i think ZHA is thinking its one new instance and do not reusing all old saved devices.

mdeweerd, 2023-11-12:
It's using the TI IEEE address.

To recover entities (different than those I had), I:
- Reflashed the key (as I could not reconfigure ZHA to connect to it after removing it);
- Reconnected to the socket;
- Restored from the stored backup (previously I recovered from a backup I downloaded using the UI).

Luckily I had an export of the critical devices in devices.csv with zha-toolkit which helped map the devices to their original entity names and locations.

MattWestb, 2023-11-12:
If the IEEE was the same in the "new network" and the old backup its likely the frame counter was set to zero and the network was throwing all frames from the coordinator as replay attack.

Great  that you is up and running after some hardware digging (its little different for code digging but its more my thing as code is yours) !!

jacme31, 2023-11-12:
@MattWestb and @mdeweerd  : Thank you very much for your investigation into this issue.
I am not competent enough to understand all your exchanges in detail.
It seems to me that I have generally understood from your analyzes and experiences that the problem came from the zigbee coordinator and that a new radio firmware could solve the problem.
For my part I use an HA Yellow with the on-board radio chip:
```
"ezsp": {
                   "manufacturer": "Nabu Casa",
                   "board": "Yellow v1.2",
                   "version": "6.10.3.0 version 297",
                   "stack_version": 8,
                   "can_burn_userdata_custom_eui64": false,
                   "can_rewrite_custom_eui64": false
               }
```
I have no idea how this network firmware can be updated even though I saw the following links:
  https://github.com/home-assistant/addons/tree/master/silabs_flasher.
https://github.com/NabuCasa/silabs-firmware/wiki/Flash-Silicon-Labs-radio-firmware-manually
Generally speaking, I'm a little fearful, and I'd rather not use my new SAPZ devices than lose my network and have to re-pair all existing devices...
​

mdeweerd, 2023-11-12:
> If the IEEE was the same in the "new network" and the old backup its likely the frame counter was set to zero and the network was throwing all frames from the coordinator as replay attack.

I migrated many months ago already and the migration was flawless!

I think that changing to the RCP FW using the multiprotocol Add-on should also be don by initiating a migration rather than removing "everything" - the only issue being the port - ZHA might grab the port before the add-on does.  So the trick would be to change the ZHA port to a bad port (currently by hacking) or disabling it until the add-on starts up I guess.

> Great that you is up and running after some hardware digging (its little different for code digging but its more my thing as code is yours) !!

My things are HW and SW (I design HW boards and SW from FW to application level - I've also worked on design and verification of baseband ASICs for phones) !   

However you noticed the Link key issue!

> Generally speaking, I'm a little fearful, and I'd rather not use my new SAPZ devices than lose my network and have to re-pair all existing devices...

I'ld suggest you to stick with the Zigbee SW version.

Nabu Casa indicates there are no known issues with the current FW, but this is one!

https://skyconnect.home-assistant.io/firmware-update/

The [add-on silabs-flasher](https://github.com/home-assistant/addons/tree/master/silabs_flasher) seems to be an official method to update the key.

I'ld make sure to have a recent backup, then proceed with the upgrade (implies disabling - not removing - zha) and enable zha again.  The network info is probably maintained and if not you can restore it.
IMHO the only reason for the issues I had was that the documentation for the alternative code said to remove zha... .
I did not have to repair the devices, but I did have to remap the devices to the entities I was using - and one of the side-effets there is that in several cases the entity_name postfix was incremented (_6 to _7 for instance).

Of course - up to you if and when you want to try the upgrade.  And I would report this to Nabu Casa anyway.  A Coordinator FW change fixes it so it can be argued that there is an issue with the coordinator FW.

jacme31, 2023-12-15:
Finally  I configured the Multi Protocol Support of my HA Yellow (even though I'm not using Thread at the moment)
After that, pairing worked well.
We might think that the multiprotocol has a different zigbee stack...
​

MacDada, 2024-02-15:
> During the pairing ZHA loads the quirk "zhaquirks.tuya.ts011f_plug.Plug" and everything seems to works, but the device stays in pairing mode, it disconnect/ reconnects multiple times before disconnecting defintevly.

I have the same exact issue with `Nous A1Z Smart ZigBee Socket` (identified by ZHA as `TS011F by _TZ3000_2putqrmw`,         `Firmware: 0x0000004d`).

`HA OS 2024.2.1`
`Sonoff Dongle–P` (identified by ZHA as `CC2652 by Texas Instruments`, `Firmware: Z-Stack 20230507`)

1. ZHA -> Add device
2. Hold the button on the socket for a few seconds -> it starts flashing its led
3. ZHA (re)discovers it
4. It works! I can control the socket from HA (it is still flashing its led! even though the pairing should have completed [?])
5. Finally, the led stops flashing -> the socket is not longer controllable from HA (and the values don't get updated)
6. Go to 1 :-P

aalwash, 2024-06-11:
Any update on this? I have the same issue as @MacDada 
Can I provide any logs, if so, which logs and how can I collect them?

013k, 2025-01-04:
I have exactly the same problem

mdeweerd, 2025-01-04:
Regarding the problem exposed by @MacDada, is the same as what is exposed in the initial post and through my packet snifffs but in my cas it was a  this looks to be the same as what I exposed with the SonOFF ZBDongle-E.

It is possible to set the log level of the zigpy.zcl layer to debug but I think that will not help a log as the issue seems to be at a lower level handled by the zigbee processor itself.  That is where the TC-Link key is handled.
So for more details one would have to sniff the packets directly.

```yaml
service: logger.set_level
data:
  zigpy.zcl: debug
```

I can see on forums that a new FW is available for the Sonoff Dongle-P: 20240710 .  It may help to fix this issue.

013k, 2025-01-04:
I'm using Nabu Casa SkyConnect stick as a Zigbee Controller. I have just flashed the stick with the latest firmware and Silvercrest SAPZ A2 started to work normally!!

To update the firmware i have used the Home Assistant Add-on: Silicon Labs Flasher. 
The process is very simple:

1.First you install the add-on
2.Disable any integrations that are using the Controller - in my case I had to disable ZHA.
3. Firmware flash using Silicon Labs Flasher - process was very fast and fully automatic. It flashed the firmware from version 7.1.1.0 build 273 7.4.4.0 build 0

Good luck with updating your sticks :-)

closed 2025-08-15 by github-actions[bot]