zigpy/zha-device-handlers #2722: [BUG] IKEA TRADFRI SHORTCUT Button - Battery Level wrong state

mouz124, 2023-11-07:
### Bug description

Since update to Firmware 0x24040006, the battery status has been reported incorrectly. In my case this only happens with one of the two buttons. I have already deleted and reset the device, reconnected it, power on/off, replaced the battery but nothing helped.

Another user reports the same problem [here](https://community.home-assistant.io/t/ikea-tradfri-shortcut-button-battery-shows-255-percent-intead-of-100-percent/632023)

### Steps to reproduce

Firmware update OTA was carried out

### Expected behavior

Actually the battery status should be at 100%, the measured voltage (2.9V) is correct but the scaling doesn't seem to fit.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0820",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0009",
        "0x0020",
        "0x1000"
      ],
      "output_clusters": [
        "0x0003",
        "0x0004",
        "0x0006",
        "0x0008",
        "0x0019",
        "0x1000"
      ]
    }
  },
  "manufacturer": "IKEA of Sweden",
  "model": "TRADFRI SHORTCUT Button",
  "class": "zhaquirks.ikea.shortcutbtn.IkeaTradfriShortcutBtn2"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.11.1",
[... 355 lines cut ...]
            "unsupported_attributes": {}
          },
          "0x1000": {
            "endpoint_attribute": "lightlink",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

Mayhem-SWE, 2023-11-07:
Also happens with the Trådfri two button remote/dimmer that has the same form factor as the shortcut button.

kitkat270, 2023-11-10:
Same issue, not impacting all buttons though even though they are identical.
[image]

stevegroom, 2023-11-10:
Also affects Tredansen blinds. 

[image]

MattWestb, 2023-11-11:
You need deleting the blinds and waiting one minute or restart  HA and adding it new and the as new so the undated signature i OK and the old quirk is not loading then all is being OK.

soonic6, 2023-12-18:
> You need deleting the blinds and waiting one minute or restart HA and adding it new and the as new so the undated signature i OK and the old quirk is not loading then all is being OK.

that worked für my Stybar, but not for TRADFRI on/off switch.

any news about this?

mhl66, 2023-12-18:
Did not work for STYRBAR, and worse: no more group bindings with STYRBAR!

mhl66, 2023-12-18:
> You need deleting the blinds and waiting one minute or restart HA and adding it new and the as new so the undated signature i OK and the old quirk is not loading then all is being OK.

"restart HA" did not improve anything at all in my year-long experience. ZHA did never work satisfactorily (the green check mark over the "BIND" button did not mean anything at all).

TheJulianJES, 2023-12-20:
Related/same issues:
- https://github.com/zigpy/zha-device-handlers/issues/2203
- https://github.com/zigpy/zha-device-handlers/issues/2418
- https://github.com/zigpy/zha-device-handlers/issues/2440
- https://github.com/zigpy/zha-device-handlers/pull/2773

Comment on why this is still an issue:
- https://github.com/zigpy/zha-device-handlers/pull/2773#issuecomment-1826892883

To summarize, older devices require the percentage duplication, but newer ones don't. We can't match per firmware version though.
An idea/workaround would be to re-scale the battery voltage to a percentage, as it's consistent across versions.
We should ideally "emulate" the device behavior perfectly, so we'd need to find out if it's a linear function that's used to map from voltage to percentage and if it's the same for all/most devices (if we want to use this approach).

TheJulianJES, 2024-02-03:
This issue should be fixed in HA Core 2024.2.0 (currently in beta).
Relevant PR:
- https://github.com/zigpy/zha-device-handlers/pull/2948

If you still see new "double battery percentage" reports, wait a day.
You can also try to press a button on the device or manually read the `sw_build_id` attriubte. Re-pairing/reconfiguring should also work.

If it's not fixed for you, please create a new issue and provide a new diagnostics file.

closed 2024-02-03 by TheJulianJES