zigpy/zha-device-handlers #2724: Loratap SC400ZB-EU-V5S (TS130F curtain switch) percentage position not working

jopeyyyy, 2023-11-07:
### Problem description

Switch is calibrated and working correctly up/down/stop and even reporting correct state and position but go_to_lift_percentage function does not work

### Solution description

get the go_to_lift_percentage function working or a tool to send commands up/down and stop calculated based on the calibration time.

### Screenshots/Video

[image] #2707 

### Device signature

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0203",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0102",
        "0xe001"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_qa8s8vca",
  "model": "TS130F",
  "class": "ts130f.TuyaTS130ESTC"
}

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.10.5",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.5",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Paris",
[... 1030 lines cut ...]
                    TuyaWithBacklightOnOffCluster,
                    TuyaCoveringCluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

### Additional information

I'm new to home assistant and managed it to got the local quirk working but I would say that's it.
I spent a lot of hours ready and searching for solutions but I didn't find the right thing.

What about "time based cover synced", can that help me in my case? I tried to install it (adding to the configuration.yaml) but I don't know how to use it...

Hope someone can help me as it's the only zigbee roller shutter switch which has physical buttons which are superposed and not always backlight on and to a good price. The light from other switches disturbs me in the night and to find the button without light with a glas touch panel is not so easy...

thanks

jopeyyyy, 2023-11-09:
Hi, nobody who can help me out there?

gignonje, 2023-11-13:
Hi,

I have same issue with Loratap SC400ZB-EU-V2 (with glass panel), identified as _TZ3000_qa8s8vca TS130F. Up and down work, position don't.

Hope someone finds a fix.

jopeyyyy, 2023-11-13:
Great! Or not great but nice that I'm not alone... I'm sure  there are more out there with this issue...

It would be great to have something were you can put the time for up and down from each roller so you have exact positioning. Of course the manual operation would not be registered or used but that is like never the time (for me).
Even if working based on the feedback time which the switch gives would be nice, so if positioning would not be 100% exact due to different up and down time, but then even the physical operation would be known by home assistant.

@gignonje: does your switch gives feedback of the actual position in id: 0x0008 too?

gignonje, 2023-11-13:
> Great! Or not great but nice that I'm not alone... I'm sure there are more out there with this issue...
>
> It would be great to have something were you can put the time for up and down from each [...]

Yes it is !
[image]

jopeyyyy, 2023-11-13:
Perfect! I thinks that's something the specialists can work with...

I would like too but I have not enough knowledge in that domain 😢

jopeyyyy, 2023-11-29:
finaly got the cover_rf_time_based integration working and for me it's ok.

I also Tried to by the double switch version:
[image]

This one, the percentage works out of the box, but trying to send the commands got_to_lift_percentage directly also ends with the same error. 

However, I now ordered several of the simple version for me.

trudetta, 2024-01-22:
> finaly got the cover_rf_time_based integration working and for me it's ok.
>
> I also Tried to by the double switch version: [...]

How did you manage, can you explain to us? Thanks in advance.

jopeyyyy, 2024-01-22:
After setting up the shutter switch in HA and calibrating it or entering the calibration time in the zigbee modul itself
You need to install the integration via hacs and add to the configuration yaml an entry for each shutter like this with the appropriate times for up and down and the name of the new created entity(name:...):

[image]

Then you need to restart HA.

After that you can add in the dashboard a manual type: 
[image]

And that's it!

Note: now, if you use the hardware buttons on the switch, the dashboard will not reflect the actual position.

**But as the switch itself return the actual shutter position (due to the defined calibration_ time) there must be a way to use this information to update the position in the dashboard (for the new created entity). There is this option for this in the integration but I didn't get that working for now. Is there anybody who could help here?? Then everything would be perfect as even hardware button would react onto the dashboard....**

jopeyyyy, 2024-02-01:
I used this one
https://github.com/barmazu/home-assistant-custom-components-cover-rf-time-based

jopeyyyy, 2024-02-01:
YEAH!

Just got it working like I wanted, that's the automation:

`alias: Rollladen Büro Position
description: ""
trigger:
  - entity_id:
      - cover.rollladen_buero_o
    platform: state
    attribute: current_position
condition: []
action:
  - data:
      confident: true
      position: **"{{ state_attr('cover.rollladen_buero_o', 'current_position') }}"**
      position_type: current
      entity_id: cover.rollladen_buro
    service: cover_rf_time_based.set_known_position
mode: single
`

Each time you  the shutter stops, the entities from the rf cover time based are updated in Dashboard! PERFECT!

closed 2024-02-03 by jopeyyyy

albertomercurio, 2025-01-03:
Hello, I have the same problem. I will for sure try this workaround of using a custom integration, but can we define this issue as fixed? I think that this issue is still in zigpy.