zigpy/zha-device-handlers #2725: [Device Support Request] ZTH01: outdoor temp/humidy sensor detected but not working

MarkL1969, 2023-11-07:
### Problem description

Hi,

just tried to connect a new outdoor sensor, model ZTH01.

When adding it using ZHA with Sonoff Zigbee 3.0 Plus dongle it is detected as

```
"manufacturer": "_TZE200_yjjdcqsq",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_sensor.TuyaTempHumiditySensorVar03"
```

But Humidity, Temperature and Battery keep on been "unknown". No entries in device protocol.

Because of an aqara H1 switch I have `enable_quirks: True` in my configuration.
In device information it is shown as "zhaquirks.tuya.ts0601_sensor.TuyaTempHumiditySensorVar03"
When I try to connect the device with `enable_quirks: False` it doesn't get any entities at all (unknow device).

According to https://zigbee.blakadder.com/Tuya_ZTH01.html the device should work with ZHA.

Any help highly appreciated.

### Solution description

Working sensors.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

aquatix, 2023-11-12:
I have the same issue, using ZHA with a Conbee II

struppis77, 2023-11-13:
Ich habe das gleiche Problem mit einem Indoor Sensor. Habe ZHA mit Soniff Zigbee Stick. Wird gar nicht erkannt

TheJulianJES, 2023-12-24:
The issue template isn't filled-out. Diagnostic information and/or device signature are missing.

TheJulianJES, 2023-12-24:
> Because of an aqara H1 switch I have enable_quirks: True in my configuration.

`enable_quirks` is `true` by default. You should never set it to `false`. There's no reason to include it in your `configuration.yaml` at all.

mp68, 2024-01-10:
Same issue using the ConBee II. All states remain unknown.
<details><summary>Device signature</summary>

```
IEEE: *redacted*
Nwk: 0xae3b
Device Type: EndDevice
LQI: 255
RSSI: -49
Last ssen: 2024-01-10T18:23:54
Energy source: Battery or Unknown
Type: zhaquirks.tuya.ts0601_sensor.TuyaTempHumiditySensorVar04
``` 
</details>

<details><summary>Diagnostic information</summary>

```
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.1.2",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.6",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "6.1.59",
    "supervisor": "2023.12.0",
    "host_os": "Home Assistant OS 11.1",
    "docker_version": "24.0.6",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "zwift": {
      "version": "3.3.3",
      "requirements": [
        "zwift-client==0.2.0"
      ]
    },
    "dwd_weather": {
      "version": "v2.1.0",
      "requirements": [
        "simple_dwd_weatherforecast==2.0.29",
        "markdownify==0.6.5",
        "suntimes==1.1.2"
      ]
    },
    "xiaomi_cloud_map_extractor": {
      "version": "v2.2.0",
      "requirements": [
        "pillow",
        "pybase64",
        "python-miio",
        "requests",
        "pycryptodome"
      ]
    },
    "hacs": {
      "version": "1.33.0",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    },
    "xiaomi_miot": {
      "version": "0.7.15",
[... 538 lines cut ...]
2024-01-10 18:25:20.355 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x17e50026001f00220200000102a8c90104010504080008460a000021c60f00afffa2c61400d0
2024-01-10 18:25:20.357 DEBUG (MainThread) [zigpy_deconz.api] Received command CommandId.aps_data_indication{'status': <Status.SUCCESS: 0>, 'frame_length': 38, 'payload_length': 31, 'device_state': DeviceState(network_state=<NetworkState2.CONNECTED: 2>, device_state=<DeviceStateFlags.APSDE_DATA_REQUEST_FREE_SLOTS_AVAILABLE: 8>), 'dst_addr': DeconzAddress(address_mode=<AddressMode.NWK: 2>, address=0x0000), 'dst_ep': 1, 'src_addr': DeconzAddress(address_mode=<AddressMode.NWK: 2>, address=0xC9A8), 'src_ep': 1, 'profile_id': 260, 'cluster_id': 1029, 'asdu': b'\x08F\n\x00\x00!\xc6\x0f', 'reserved1': 0, 'reserved2': 175, 'lqi': 255, 'reserved3': 162, 'reserved4': 198, 'reserved5': 20, 'reserved6': 0, 'rssi': -48} (seq 229)
2024-01-10 18:25:20.359 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(timestamp=datetime.datetime(2024, 1, 10, 17, 25, 20, 359461, tzinfo=datetime.timezone.utc), src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xC9A8), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=None, profile_id=260, cluster_id=1029, data=Serialized[b'\x08F\n\x00\x00!\xc6\x0f'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=255, rssi=-48)
2024-01-10 18:25:20.360 DEBUG (MainThread) [zigpy.zcl] [0xC9A8:1:0x0405] Received ZCL frame: b'\x08F\n\x00\x00!\xc6\x0f'
2024-01-10 18:25:20.361 DEBUG (MainThread) [zigpy.zcl] [0xC9A8:1:0x0405] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=70, command_id=10, *direction=<Direction.Client_to_Server: 1>)
2024-01-10 18:25:20.362 DEBUG (MainThread) [zigpy.zcl] [0xC9A8:1:0x0405] Decoded ZCL frame: RelativeHumidity:Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=uint16_t, value=4038))])
2024-01-10 18:25:20.362 DEBUG (MainThread) [zigpy.zcl] [0xC9A8:1:0x0405] Received command 0x0A (TSN 70): Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=uint16_t, value=4038))])
2024-01-10 18:25:20.363 DEBUG (MainThread) [zigpy.zcl] [0xC9A8:1:0x0405] Attribute report received: measured_value=4038
2024-01-10 18:25:20.373 DEBUG (MainThread) [zigpy.zcl] [0xC9A8:1:0x0405] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=70, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>)
2024-01-10 18:25:20.374 DEBUG (MainThread) [zigpy.zcl] [0xC9A8:1:0x0405] Sending reply: Default_Response(command_id=10, status=<Status.SUCCESS: 0>)
2024-01-10 18:25:20.374 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending packet: ZigbeePacket(timestamp=datetime.datetime(2024, 1, 10, 17, 25, 20, 374934, tzinfo=datetime.timezone.utc), src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xC9A8), dst_ep=1, source_route=None, extended_timeout=False, tsn=70, profile_id=260, cluster_id=1029, data=Serialized[b'\x18F\x0b\n\x00'], tx_options=<TransmitOptions.ACK: 1>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2024-01-10 18:25:20.376 DEBUG (MainThread) [zigpy_deconz.api] Sending CommandId.aps_data_request{'request_id': 60, 'flags': <DeconzSendDataFlags.NONE: 0>, 'dst': DeconzAddressEndpoint(address_mode=<AddressMode.NWK: 2>, address=0xC9A8, endpoint=1), 'profile_id': 260, 'cluster_id': 1029, [... cut ...]
2024-01-10 18:25:20.463 DEBUG (MainThread) [zigpy_deconz.api] Received command CommandId.aps_data_confirm{'status': <Status.SUCCESS: 0>, 'frame_length': 19, 'payload_length': 12, 'device_state': DeviceState(network_state=<NetworkState2.CONNECTED: 2>, device_state=<DeviceStateFlags.APSDE_DATA_REQUEST_FREE_SLOTS_AVAILABLE: 8>), 'request_id': 60, 'dst_addr': DeconzAddressEndpoint(address_mode=<AddressMode.NWK: 2>, address=0xC9A8, endpoint=1), 'src_ep': 1, 'confirm_status': <TXStatus.SUCCESS: 0>, 'reserved1': 0, 'reserved2': 0, 'reserved3': 0, 'reserved4': 0} (seq 231)
``` 
</details> 

Thank you for looking into this! :)

johnklox, 2024-07-11:
same for me:

ZHA with ConBee_II and TS0601
from _TZE200_yjjdcqsq

all sensor data unknow

IEEE: a4:c1:38:83:be:41:48:d3
Nwk: 0xbf75
Device Type: EndDevice
LQI: 255
RSSI: -55
Zuletzt gesehen: 2024-07-11T22:09:11
Energiequelle: Battery or Unknown
Eigenart: zhaquirks.tuya.ts0601_sensor.TuyaTempHumiditySensorVar04

PR #3817 (2025-01-31): Fix Tuya temp/humidity sensor quirks (`_TZE200_upagmta9`, ...)
## Proposed change

Add Tuya `tze200 cirvgep4` and combine quirks to match z2m.

See: https://github.com/Koenkk/zigbee-herdsman-converters/blob/09cfce5153d3931c7cc009aa3ae3107d9b394f5c/src/devices/tuya.ts#L1073

## Additional information

Closes: #2577
Closes: #2854 
Closes: #2304 
Closes: #3260 
Closes: #2725 
Closes: #2929 

## Checklist

- [ ] The changes are tested and work correctly
- [x] `pre-commit` checks pass / the code has been formatted using Black
- [x] Tests have been added to verify that the new code works
closed 2025-01-31 by TheJulianJES