zigpy/zha-device-handlers #2728: [Device Support Request] Popp Thermostat Radiator Valve 701721

deechte, 2023-11-08:
### Problem description

This Popp TRV is a great product, hardwire-wise: https://popp.eu/popp-smart-thermostat-701721/
However ZHA only exposes a thermostat card. Much more could be exposed to Home Assistant if I look at what Zigbee2mqtt offers: https://www.zigbee2mqtt.io/devices/701721.html.

This product has identical hardware to the Danfoss Aly TRV.

### Solution description

Could what Zigbee2mqtt exposes please be implemented in ZHA as well?
I'm especially looking for:
- reading the state of the valve opening percentage;
- reading and setting the window open feature;
-  doing OTA updates to the TRV.

Thank you!

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4678, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)", 
  "endpoints": { 
    "1": { 
      "profile_id": "0x0104", 
      "device_type": "0x0301", 
      "input_clusters": [ 
        "0x0000", 
        "0x0001", 
        "0x0003", 
        "0x000a", 
        "0x0020", 
        "0x0201", 
        "0x0204", 
        "0x0b05" 
      ], 
      "output_clusters": [ 
        "0x0000", 
        "0x0019" 
      ] 
    } 
  }, 
  "manufacturer": "D5X84YU", 
  "model": "eT093WRG", 
  "class": "zigpy.device.Device" 
} 
]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.11.1",
    "dev": false,
[... 455 lines cut ...]
</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

### Additional information

Don't know if it's relevant, but I'm using the HA SkyConnect USB stick.

MattWestb, 2023-11-08:
I think its the POP version of the updated Aly and need being added here in the Danfoss quirk:
https://github.com/zigpy/zha-device-handlers/blob/cbd0f037fbc3c5b696e499bdf13bf935bc8746ff/zhaquirks/danfoss/thermostat.py#L130

deechte, 2023-11-08:
Thanks for checking it out Matt!!
Who could add that to the quirk?

MattWestb, 2023-11-08:
Some must testing doing it with local quirk and if its working making one PR and its being added to next ZHA release.

deechte, 2023-11-08:
Thanks. I'd be happy to test a quirk. I'm familiar with using local quirks, but not to making them. I suppose the Danfoss quirk needs changing to work for the Popp?

MattWestb, 2023-11-08:
Put the danfoss quirk in your local quirk forlder and then adding the new device `MODELS_INFO:` one one new line after the last one i was showing and save and restart HA and look if its loading the quirk.

Caius-Bonus, 2023-11-08:
There is already a PR for this in ZHA and here: https://github.com/home-assistant/core/pull/86907 and https://github.com/zigpy/zha-device-handlers/pull/2150/. It just needs a few other bits into place before it can get merged. In the meantime you can run custom components and custom quirks.

Caius-Bonus, 2023-11-08:
And OTA updates are already possible using https://www.danfoss.com/nl-nl/products/dhs/smart-heating/smart-heating/danfoss-ally/danfoss-ally-support/#tab-software and the information here: https://www.home-assistant.io/integrations/zha under OTA firmware updates . Haven't confirmed this yet personally, but it should work. Note that the new devices (G prepended serial) have no firmware files available yet.

Caius-Bonus, 2023-11-08:
Also duplicate: https://github.com/zigpy/zha-device-handlers/issues/1726

deechte, 2023-11-08:
> Put the danfoss quirk in your local quirk folder and then adding the new device `MODELS_INFO:` one one new line after the last one (131) i was showing and save and restart HA and look if its loading the quirk.

Ok, so I've done that. Added a line
 (D5X84YU, "eT093WRG"),
which is the model number of my Popp 701721.
Removed the device from ZHA and factory reset the Popp itself, rebooted HA and repaired the Popp. I can now clearly see it is using the quirk in the Popp's device zigbee info. Unfortunately, no new entities or sensors have appeared. It's exactly the same as before without the quirk.

deechte, 2023-11-08:
> Haven't confirmed this yet personally, but it should work.

It appears to me you have this device as well, right? But do you have the Popp 701721 or the Danfoss Aly? Cause the custom quirk isn't working, like I wrote one post higher.

MattWestb, 2023-11-08:
Try restarting HA its some times happening ll handles is not being loaded then the device is being added but the device is OK.

deechte, 2023-11-08:
> Try restarting HA its some times happening ll handles is not being loaded then the device is being added but the device is OK.

Like I wrote, I did that already in the process :-( . In fact, I did it multiple times already since installing the custom quirk.

Caius-Bonus, 2023-11-08:
The updated quirk will work for D5X84YU eT093WRG. Adding this to the old quirk will not change anything. If you want to have entities in Home Assistant, you need to use the code from both my PRs. If you only use the custom quirk, it will not do anything in home assistant.

deechte, 2023-11-08:
OK, thanks, but that's beyond my level of knowledge. How do I do that?

Caius-Bonus, 2023-11-08:
If it is, I would advice waiting for my PRs to be merged. It would require you to redo this every large release and it requires editing specific files in specific ways that might break things if not done with knowledge of the intention. I hope to get it merged before the December release, but I can't promise anything. It works, but it requires other PRs to be in place first.

deechte, 2023-11-09:
> If it is, I would advice waiting for my PRs to be merged. It would require you to redo this every large release and it requires editing specific files in specific ways that might break things if not [...]

You're right, that'll be over my head. Fingers crossed your changes come live soon
BTW, you have the Popp or the Danfoss trv?

AllardKatan, 2023-12-06:
> If it is, I would advice waiting for my PRs to be merged. It would require you to redo this every large release and it requires editing specific files in specific ways that might break things if not [...]

It looks like your PR's are almost ready? **Great work, thanks a lot**. It comes at a good time for me, I am getting a completely new heating system (boiler, heat pump, valves, only the radiators stay the same) next week and I was thinking about getting POPP Zigbee TRV's for all my 10 radiators. I have one of them already (but no matching radiator valve to connect it to), and it connects just fine, but I can only read temperature and write setpoint through ZHA now.

If you need anyone to run tests, just let me know. I haven't used custom quirks and stuff like that before but I'm tech-savvy enough to find out how that works. I run my HA in a Virtualbox, so I can simply clone it and run that one for tests without the danger of messing everything up.

Caius-Bonus, 2023-12-28:
@deechte I am doing my testing with Popp TRVs. @AllardKatan If you can figure it out yourself, I would say: go for it. It is fully functional and stable on my setup and more testing is better before merging. However mind that it would require certain hacks in python to get it working and I don't have the time to assist in this.

AllardKatan, 2023-12-28:
I'll see how far I get. I have all 10 valves installed now and connected to ZHA, they are all Popp but one of them is the older (O) type, the rest is the newer (G). So that should be a decent testing ground.
For my understanding, would I need to set up a full dev environment in VScode or would a clone of my VM with a couple of files replaced with the ones from your PRs be enough? I think I can make time for the latter but I think the former is more of an investment than I can currently afford. I was already considering diving in to Ha core development but it's going to take me quite some time to get familiar with it so that will not be soon.

deechte, 2024-06-25:
@github bot. This issue isn't solved.

closed 2024-12-29 by github-actions[bot]