zigpy/zha-device-handlers #2730: [Device Support Request] Adaprox Fingerbot Plus (Zigbee)

bst27, 2023-11-08:
### Problem description

The Adaprox Fingerbot Plus is connected to my Home Assistant but it is only shown as a switch. I am not able to change how long the button is pressed or in which mode the fingerbot is working.

### Solution description

- I would like to set the duration to wait between push and release of a button with the Fingerbot
- I would like to change the working mode of the Fingerbot so that it toggles between push/release.

### Screenshots/Video

It is the following device: https://www.adaprox.io/products/single-kit?variant=41614756184269

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0000",
        "0x0006",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3210_j4pdtz9v",
  "model": "TS0001",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2023.2.3",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "armv7l",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "5.10.103-v7l+",
    "run_as_root": true
  },
  "custom_components": {},
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "config_flow": true,
    "documentation": "https://www.home-assistant.io/integrations/zha",
    "requirements": [
[... 269 lines cut ...]
</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

### Additional information

I am not really sure, but maybe this is somehow related: https://github.com/Koenkk/zigbee2mqtt/issues/17841

bst27, 2023-11-08:
I just found the following related issue:
https://github.com/home-assistant/core/issues/98811

The issue mentions the following zigpy issue:
https://github.com/zigpy/zha-device-handlers/pull/2506

Though for me it looks like the manufacturer mentioned in the device signature is different to my device signature. Could it be that there is only a small change needed to also detect my device signature and then it should be fine?

bst27, 2023-11-09:
I upgraded my home assistant docker installation to the latest home assistant release 2023.11.1 and readded the device. Now I get the following info:

<details>
<summary>Device Signature:</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0006",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3210_j4pdtz9v",
  "model": "TS0001",
  "class": "zhaquirks.tuya.ts0001_fingerbot.TuyaFingerbot"
}
```
</details>

<details>
<summary>Diagnostic Information:</summary>
```json
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2023.11.1",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.11.6",
    "docker": true,
    "arch": "armv7l",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "5.10.103-v7l+",
    "run_as_root": true
  },
  "custom_components": {},
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "after_dependencies": [
      "onboarding",
      "usb"
    ],
    "codeowners": [
      "@dmulcahey",
      "@adminiuga",
      "@puddly"
    ],
    "config_flow": true,
    "dependencies": [
      "file_upload"
    ],
    "documentation": "https://www.home-assistant.io/integrations/zha",
    "iot_class": "local_polling",
    "loggers": [
      "aiosqlite",
[... 271 lines cut ...]
          },
          "0xef00": {
            "endpoint_attribute": "tuya_manufacturer",
            "attributes": {},
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```
</details>

N3orun, 2024-01-09:
An news on this?

georaspi, 2024-02-08:
Same issue here, running 2024.1 HA.

FalcoonM, 2024-03-28:
HA 2024.03.03 - changed cluster mode to SWITCH, now it works properly
[image]

closed 2024-10-01 by github-actions[bot]