zigpy/zha-device-handlers #2732: [Device Support Request] Tuya Temperature&Humidity Sensor with backlight and Battery report TZE200_vvmbj46n

WLCD, 2023-11-08:
### Problem description

This Unit is TS601 based sensor, I tried to integrate it by copying the code from Variation 04 with some success : I can get the Temp and Humidity readings, but the battery doesn't seem to report correctly, plus I don't know what to do to update the Date & time on the device itself, but I'm pretty sure the gateway is supposed to do that.

Here is the link to the device itself : https://fr.aliexpress.com/item/1005005897159196.html?spm=a2g0o.order_list.order_list_main.11.7a875e5bryD8IU&gatewayAdapt=glo2fra

### Solution description

I'm using the custom quirk for the Tuya air quality sensor which is awesome, I am no coder, but I'm willing to provide help if someone needs more information to create a custom quirks for this model. Thanks a bunch !

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0302",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0004",
        "0x0005",
        "0x0402",
        "0x0405",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_vvmbj46n",
  "model": "TS0601",
  "class": "ts0601_sensor.TuyaTempHumiditySensorVar05"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.11.1",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.6",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Paris",
    "os_name": "Linux",
    "os_version": "6.1.59",
[... 373 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

WLCD, 2023-12-02:
Just added my customized code, the battery status won't show up, unfortunately.

joji90, 2024-02-17:
anyone found a working quirk for this?

paulsijben, 2024-02-29:
I have a quirk that now gives me values but the humidity is off by a factor of 10x . Does anyone have an idea to fix this?

This is the snippet of the quirk I have on the basis of the bigger quirk for these devices 

class TuyaTempHumiditySensorVar03(CustomDevice):
    """Tuya temp and humidity sensor (variation 03)."""
    RH_MULTIPLIER = 10
    signature = {
        # "profile_id": 260,
        # "device_type": "0x0051",
        # "in_clusters": ["0x0000","0x0004","0x0005","0xef00"],
        # "out_clusters": ["0x000a","0x0019"]
        MODELS_INFO: [
            ("_TZE200_yjjdcqsq", "TS0601"),
            ("_TZE200_9yapgbuv", "TS0601"),
            ("_TZE200_qyflbnbj", "TS0601"),
            ("_TZE200_utkemkbs", "TS0601"),
            ("_TZE200_vvmbj46n", "TS0601")

        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TemperatureHumidityManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            }
        },
    }

    replacement = {
        SKIP_CONFIGURATION: True,
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.TEMPERATURE_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TemperatureHumidityManufCluster,
                    TuyaTemperatureMeasurement,
                    TuyaRelativeHumidity,
                    TuyaPowerConfigurationCluster2AAA,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            }
        },
    }

richardcc, 2024-02-29:
Hi

I had a similar problem and from file ts0601_sensor.py I modified
RH_MULTIPLIER, but I don't remember which one or where. The version I
attached is working correctly for me.

BR
Ricardo

paulsijben, 2024-02-29:
Hi @richardcc , looks like the attachment did not make it. Could you please upload that one separately?

richardcc, 2024-02-29:
Let me know if you can get it from here. Otherwise I'll try to upload the
file.

"""Tuya temp and humidity sensors."""

from typing import Any, Dict

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time
from zigpy.zcl.clusters.measurement import (
    RelativeHumidity,
    SoilMoisture,
    TemperatureMeasurement,
)

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    SKIP_CONFIGURATION,
)
from zhaquirks.tuya import TuyaLocalCluster, TuyaPowerConfigurationCluster2AAA
from zhaquirks.tuya.mcu import DPToAttributeMapping, TuyaMCUCluster

class TuyaTemperatureMeasurement(TemperatureMeasurement, TuyaLocalCluster):
    """Tuya local TemperatureMeasurement cluster."""

class TuyaSoilMoisture(SoilMoisture, TuyaLocalCluster):
    """Tuya local SoilMoisture cluster with a device RH_MULTIPLIER
factor if required."""

class TuyaRelativeHumidity(RelativeHumidity, TuyaLocalCluster):
    """Tuya local RelativeHumidity cluster with a device RH_MULTIPLIER
factor."""

    def update_attribute(self, attr_name: str, value: Any) -> None:
        """Apply a correction factor to value."""

        if attr_name == "measured_value":
            value = value * (
                self.endpoint.device.RH_MULTIPLIER
                if hasattr(self.endpoint.device, "RH_MULTIPLIER")
                else 100
            )
        return super().update_attribute(attr_name, value)

class TemperatureHumidityManufCluster(TuyaMCUCluster):
    """Tuya Manufacturer Cluster with Temperature and Humidity data points."""

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        1: DPToAttributeMapping(
            TuyaTemperatureMeasurement.ep_attribute,
            "measured_value",
            converter=lambda x: x * 10,  # decidegree to centidegree
        ),
        2: DPToAttributeMapping(
            TuyaRelativeHumidity.ep_attribute,
            "measured_value",
            # converter=lambda x: x * 10,  --> move conversion to
TuyaRelativeHumidity cluster
        ),
        4: DPToAttributeMapping(
            TuyaPowerConfigurationCluster2AAA.ep_attribute,
            "battery_percentage_remaining",
            converter=lambda x: x * 2,  # double reported percentage
        ),
[... 267 lines cut ...]
                    TemperatureHumidityManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            }
        },
    }

    replacement = {
        SKIP_CONFIGURATION: True,
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.TEMPERATURE_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    SoilManufCluster,
                    TuyaTemperatureMeasurement,
                    TuyaSoilMoisture,
                    TuyaPowerConfigurationCluster2AAA,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            }
        },
    }

richardcc, 2024-02-29:
By the way my sensor is identified as: TS0601 /  _TZE200_vvmbj46n

BR
Ricardo

paulsijben, 2024-02-29:
@richardcc yes that solved it. Thank you!

Now for my education, I an going to try and find the differences between my (stolen and mutilated ;-) file and yours.

edgarveersel, 2024-04-29:
Very happy with this custom quirk for my TS0601 /  _TZE200_vvmbj46n. It's the first time I heard of with quirks, and just tried it, and it works!

Would there be a way to set the time and the date on the device?

iTigere, 2024-06-02:
How should it be inserted? ?

gillenoarbe, 2024-06-05:
Hi. Thanks for the quirk.

I own the same Tuya device from Aliexpress. Once adding the custom quirk, although both Temp and Humidity and battery show up as entities, the status remains as Unknown and no data is flowing. Is there any special step of config tweak you performed? Is weird that exactly the same device with same config doesn't work...

Thanks.

gillenoarbe, 2024-06-05:
In my case, inserted the code by WLCD in one of the github available TS0601.py files at the end and worked after full restart of the docker container wih HA.

> Please, can you tell me where I must insert this custom quirk?

> How should it be inserted? ?

cecinitmortem, 2024-08-25:
When I use the quirk the time does not get written to the device properly. It just starts counting from the moment the device is powered on. The date is also not set. I have tried to restart HA and rejoin the device multiple times but that does not help either. 
Did you also have the problem with date and time @richardcc @paulsijben  ?

marvelloard, 2025-02-11:
Date and time don't work for me either. Would be great if this was fully supported.

prairiesnpr, 2025-02-17:
This was added in #3776, please remove any custom quirks and test.

closed 2025-02-17 by prairiesnpr