zigpy/zha-device-handlers #2734: [Device Support Request] Add support of TZ3000_6zvw8ham

Thibaut-tech, 2023-11-09:
### Problem description

This Unit is TS0203 based sensor.
I have add the sensor to my dashboard but the the opening and closing entity is not common and does not allow automation.

Link to the product : 

https://fr.aliexpress.com/item/1005005066429576.html?spm=a2g0o.order_list.order_list_main.4.2c4a5e5b6dz0rY&gatewayAdapt=glo2fra

### Solution description

Connect the two entities with a switch when the sensor is open or closed

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

kimme1024, 2023-12-26:
Following.
After adding the device shows 2 (?) contact sensors. However none of them react to changes in HA. The led on the sensor itself reacts just fine on changes.

Maxetto, 2024-03-29:
Following, same issue, same device ID.

giulianosrebuli, 2024-08-07:
> Following, same issue, same device ID.
> 
> EDIT: Nevermind, after many pairing attempts, it finally worked and paired up and working fine since 3 months.

Hi, did you manage it to work only doing many pairing attempts?
The reason I'm asking is because i use a Sonoff ZBBridge Pro flashed with Tasmota to act as a Zigbee Gateway via TCP, and i bought a _TZ3000_6zvw8ham contact sensor and it didn't work. After many retries, i decided to disable the ZHA integration and try my gateway with zigbee2mqtt, which i was also unsuccessful.
After also giving up on z2m, i went back to ZHA and it showed my devices already connected, and the _TZ3000_6zvw8ham showed up, connected and working. The device also had two entities, just like @kimme1024 said, but one of them worked (and is still working).
Thinking that it was all well, i tried another _TZ3000_6zvw8ham that i had here, only to it not work again.
I don't want to keep uninstalling and reinstalling ZHA each time i add a new device.

In ZHA, these are my logs:

```
[0x5C6A:1:0x0500] Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/zigpy/device.py", line 339, in request
    await send_request()
  File "/usr/local/lib/python3.12/site-packages/zigpy/application.py", line 841, in request
    await self.send_packet(
  File "/usr/local/lib/python3.12/site-packages/zigpy_znp/zigbee/application.py", line 1100, in send_packet
    raise DeliveryError(
zigpy.exceptions.DeliveryError: Request failed after 5 attempts: <Status.NWK_NO_ROUTE: 205>

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/zha/core/cluster_handlers/__init__.py", line 81, in wrapper
    with wrap_zigpy_exceptions():
  File "/usr/local/lib/python3.12/contextlib.py", line 158, in __exit__
    self.gen.throw(value)
  File "/usr/src/homeassistant/homeassistant/components/zha/core/cluster_handlers/__init__.py", line 73, in wrap_zigpy_exceptions
    raise HomeAssistantError(message) from exc
homeassistant.exceptions.HomeAssistantError: Failed to send request: Request failed after 5 attempts: <Status.NWK_NO_ROUTE: 205>

```

I have no idea on what to do now

giulianosrebuli, 2024-08-07:
Update:

After thinking about what could replicate what the uninstall-reinstall process did to ZHA, i decided to just restart Home Assistant, as it would also restart ZHA. After that, my device showed simmilarly as before, with 2 contact sensors, but only one working. The difference is that this time the battery value is 'Unknown', but after resetting/waking-up the device, it all went ok.
I have one more _TZ3000_6zvw8ham sensor to replicate the procedure and if it works, i will tell you guys

giulianosrebuli, 2024-08-07:
Well, i couldn't replicate the procedure, as the regular pairing process succeeded at my first try. I bought two more of these sensors, so if i have any problem on pairing them, i can try it again.

dJani97, 2024-09-05:
First time pairing, the device paired with no sensors. Second time pairing, it showed up with two contact sensors, out of which only one works, obviously. But since it shows two sensors with the same name, it's easy to select the wrong one by accident when assembling an integration.

Still does not show any battery information though.