zigpy/zha-device-handlers #2735: [New device support]: CK-BL702-SWP-01(7020)

gromgsxr, 2023-11-09:
### Problem description

shows as manufacture eWeLink CK-BL702-SWP-01(7020) available from amazon ebay ect https://www.ebay.co.uk/itm/314855773062

appears to be working in z2mqtt see here for info on the clusters https://github.com/Koenkk/zigbee2mqtt/issues/19334#issuecomment-1784120749

### Solution description

please add this device 

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4742, maximum_buffer_size=127, maximum_incoming_transfer_size=242, server_mask=11264, maximum_outgoing_transfer_size=242, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0009",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x1000",
        "0xfc11",
        "0xfc57"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019",
        "0x0b04"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "eWeLink",
  "model": "CK-BL702-SWP-01(7020)",
  "class": "zigpy.device.Device"
}

```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

### Additional information

https://github.com/Koenkk/zigbee2mqtt/issues/19334#issuecomment-1784120749

gromgsxr, 2023-11-22:
so i managed to cobble together a quirk that seems too work values reported by the voltage, current and wattage seem feasible.

`
import logging
import zigpy.types as t

import copy

from typing import Dict
from zigpy.profiles import zha
from zigpy.quirks import CustomCluster, CustomDevice
from zigpy.zcl.foundation import ZCLAttributeDef, ZCLCommandDef

from zhaquirks import Bus, LocalDataCluster

from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    OnOff,
    Ota,
    Scenes,
    Time,   
    AnalogInput,
    )
 
from zigpy.zcl.clusters.lightlink import LightLink
from zigpy.zcl.clusters.smartenergy import Metering
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    SKIP_CONFIGURATION,
    )

CURRENT_REPORTED = 0x7004
VOLTAGE_REPORTED = 0x7005
POWER_REPORTED = 0x7006

 
class ZBElectricalMeasurement(LocalDataCluster):
    
    cluster_id = 0xFC11	
    name = "measure"
    attributes = {
        0x7004: ("CURRENT_REPORTED",  t.CharacterString, True),
        0x7005: ("VOLTAGE_REPORTED", t.CharacterString, True),
        0x7006: ("POWER_REPORTED", t.CharacterString, True),
        }
    def _update_attribute(self, attrid, value):
        super()._update_attribute(attrid, value)        
        if attrid == CURRENT_REPORTED:
            self.endpoint.electrical_measurement.current_reported(value)
        elif attrid == POWER_REPORTED:
            self.endpoint.electrical_measurement.power_reported(value)
        elif attrid == VOLTAGE_REPORTED:
            self.endpoint.electrical_measurement.voltage_reported(value)
            
class ElectricalMeasurementCluster(LocalDataCluster, ElectricalMeasurement):
    """Custom class for power, voltage and current measurement."""

    POWER_ID = 0x050B
    VOLTAGE_ID = 0x0505
    CURRENT_ID = 0x0508
    
    AC_VOLTAGE_MULTIPLIER = 0x0600
    AC_VOLTAGE_DIVISOR = 0x0601
    AC_CURRENT_MULTIPLIER = 0x0602
    AC_CURRENT_DIVISOR = 0x0603
    AC_POWER_MULTIPLIER = 0x0604
    AC_POWER_DIVISOR = 0x0605

    _CONSTANT_ATTRIBUTES = {
        AC_VOLTAGE_MULTIPLIER: 1,
        AC_VOLTAGE_DIVISOR: 1000,
        AC_CURRENT_MULTIPLIER: 1,
        AC_CURRENT_DIVISOR: 1000,
        AC_POWER_MULTIPLIER: 1,
        AC_POWER_DIVISOR: 1000,
    }
    def voltage_reported(self, value):
        """Voltage reported."""
        self._update_attribute(self.VOLTAGE_ID, value)
    def power_reported(self, value):
        """Power reported."""
        self._update_attribute(self.POWER_ID, value)    
    def current_reported(self, value):
        """Ampers reported."""
        self._update_attribute(self.CURRENT_ID, value)
        
class PlugAC(CustomDevice):
    """test.""" 
    signature = {
        MODELS_INFO: [("eWeLink" ,"CK-BL702-SWP-01(7020)",)],
        ENDPOINTS: {
        # <SimpleDescriptor endpoint=1 profile=260 device_type=9
        # device_version=1
        # input_clusters=[0, 3, 4, 5, 6, 4096, 64529, 64599]
        # output_clusters=[10, 25, 2820]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.MAIN_POWER_OUTLET,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id, 
                    LightLink.cluster_id,
                    ZBElectricalMeasurement.cluster_id,
                    0xfc57,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, ElectricalMeasurement.cluster_id, Time.cluster_id],
            },
            
            242: {
            # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
            # input_clusters=[]
            # output_clusters=[]
            PROFILE_ID: 41440,
            DEVICE_TYPE: 97,
            INPUT_CLUSTERS: [],
            OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
    replacement = {
        SKIP_CONFIGURATION: True,
        ENDPOINTS: {
            1: {
            PROFILE_ID: zha.PROFILE_ID,
            DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
            INPUT_CLUSTERS: [
                Basic.cluster_id,
                Identify.cluster_id,
                Groups.cluster_id,
                Scenes.cluster_id,
                OnOff.cluster_id, 
                LightLink.cluster_id,
                ElectricalMeasurementCluster,
                ZBElectricalMeasurement,
               
            ],
            OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            },
            242: {
            # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
            # input_clusters=[]
            # output_clusters=[33]
            PROFILE_ID: 41440,
            DEVICE_TYPE: 97,
            INPUT_CLUSTERS: [],
            OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
`

Alan5001, 2023-11-28:
Hi,

could you please add the "default behaviour after power Out" (Don't konw if this is called correct ...)
Before using the quirk i could define this (On, Off, Last State), but had no power measurement-
No, with the quirky i can't define this anymore, but because i have the ability for power measurement now this is much better WITH the quirk 😉👍

And because i'm relatively new to HA, I had a real hard time to get the quirk working because of the formatting problems above 😅

Regards & Kudos
Alex

gromgsxr, 2023-11-28:
I will try and see what I can do. It's my first attempt at creating a quirk

Alan5001, 2023-11-28:
That would really be great, i'd appreciate this ! 🤘

Alan5001, 2023-11-28:
As is said, i'm a newbee, but i assume it could be THIS:

[image]

Alan5001, 2023-11-28:
Hmm,

but i can't write an attribute "Last" or "LastState" here (just ON of OFF) ... maybe i'm completely wrong 😁
Before the quirk i had three options: On, Off, LastState

gromgsxr, 2023-11-28:
you can write the attribute with

start_up_on_off (id: 0x4003)
0 = StartUpOnOff.Off
1 = StartUpOnOff.On
2 = StartUpOnOff.Toggle
255 = StartUpOnOff.PreviousValue

Alan5001, 2023-11-29:
Thanks, so i can at least manually set or check it 😉👍

eepyaich, 2023-12-12:
Just wanted to say thanks very much for pulling together this quirk @gromgsxr - it's great to be able to get the power output from these switches now.

Not sure why github insists on messing up the formatting of the quirk, when pasted in here, but for future generations here's a version in a file, with the correct indentation, etc...

[eWelinkQuirk - Copy.txt](https://github.com/zigpy/zha-device-handlers/files/13648681/eWelinkQuirk.-.Copy.txt)

edg2s, 2024-01-03:
I've installed the quirk but all the sensors are reporting as "Unknown". Any ideas?
[image]

gromgsxr, 2024-01-03:
> I've installed the quirk but all the sensors are reporting as "Unknown". Any ideas? ![Screenshot_20240103-225007](https://private-user-images.githubusercontent.com/180672/294046740-5c1084c5-ca92-4816-8b83-23c802133126.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQzMjM1MTgsIm5iZiI6MTcwNDMyMzIxOCwicGF0aCI6Ii8xODA2NzIvMjk0MDQ2NzQwLTVjMTA4NGM1LWNhOTItNDgxNi04YjgzLTIzYzgwMjEzMzEyNi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTAzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDEwM1QyMzA2NThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05OTM1NGI2YjQ5MWRjNDYwNTU4ZGEwMjhlZjM0NWIyMjMzMGYxZWFkZTMzOWIzOGQ4Y2MxOGZmMmU3YjJjODc5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.Vh30xQH69KiY2_lZwEF5vy9OnTS1YkpdYQED1Oou6U0)

What version of ha? 2023.07.03 works for me earlier versions didn't.

gromgsxr, 2024-01-04:
> > > I've installed the quirk but all the sensors are reporting as "Unknown". Any ideas? ![Screenshot_20240103-225007](https://private-user-images.githubusercontent.com/180672/294046740-5c1084c5-ca92-4816-8b83-23c802133126.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQzMjM1MTgsIm5iZiI6MTcwNDMyMzIxOCwicGF0aCI6Ii8xODA2NzIvMjk0MDQ2NzQwLTVjMTA4NGM1LWNhOTItNDgxNi04YjgzLTIzYzgwMjEzMzEyNi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTAzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDEwM1QyMzA2NThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05OTM1NGI2YjQ5MWRjNDYwNTU4ZGEwMjhlZjM0NWIyMjMzMGYxZWFkZTMzOWIzOGQ4Y2MxOGZmMmU3YjJjODc5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.Vh30xQH69KiY2_lZwEF5vy9OnTS1YkpdYQED1Oou6U0)
> > 
> > 
> > What version of ha? 2023.07.03 works for me earlier versions didn't.
> 
> 2024.01

Oh not tried after 2023.07.03. what happens if you read the attributes on active power, rms voltage and RMS current

gromgsxr, 2024-01-04:
i will try and update at some point to try 

AC_VOLTAGE_MULTIPLIER = 0x0600
AC_VOLTAGE_DIVISOR = 0x0601
AC_CURRENT_MULTIPLIER = 0x0602
AC_CURRENT_DIVISOR = 0x0603
AC_POWER_MULTIPLIER = 0x0604
AC_POWER_DIVISOR = 0x0605

if you try reading those attributes i wonder if its not pulling he multiply across

edg2s, 2024-01-04:
All the multipliers are 1 and all the divisors are 1000.

I left a light plugged in for a while and it eventually started reading 4W. It appears the problem is very slow updates (several hours).

gromgsxr, 2024-01-04:
I had a go on the latest and the updates almost instantly

edg2s, 2024-01-05:
Based on [this comment](https://community.home-assistant.io/t/changing-refresh-rate-of-a-device/463112) about when ZHA sets the refresh times, I tried reconfiguring the device and it appears to be updating quickly now.

Many thanks for your help!

closed 2024-07-10 by github-actions[bot]

gromgsxr, 2025-08-11:
What version of homeassistant you using? Think I'm on 2025.5 ISH believe it still works then