zigpy/zha-device-handlers #2740: [Device Support Request]  Smoke Detector TS0601 by _TZE200_rccxox8p

Kri72, 2023-11-11:
### Problem description

Hello, I've purchased this zigbee smoke Detector  (TS0601 by _TZE200_rccxox8p)
https://fr.aliexpress.com/item/1005005855776130.html?spm=a2g0o.order_list.order_list_main.23.15355e5bMWrOw4&gatewayAdapt=glo2fra
I've installed the quirk, that enable me to see new a "Smoke" sensor. However, this one remains clear, even when the detector detect smoke and start ringing.

### Solution description

I'll like to get the smoke sensor working in HA and reflecting the detector state.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0402",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0500",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_rccxox8p",
  "model": "TS0601",
  "class": "ts0601_smoke.TuyaSmokeDetector0601"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.11.2",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.6",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Paris",
    "os_name": "Linux",
    "os_version": "6.1.21-v8",
    "supervisor": "2023.11.0",
    "host_os": "Home Assistant OS 11.1",
[... 384 lines cut ...]
</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

### Additional information

Thanks in advance for your support!

Coennie, 2023-11-23:
I have the same issue, would love it to get visible under HA

systemerror-codenotfound, 2023-12-02:
Same problem here, it's not working over zigbee

systemerror-codenotfound, 2023-12-02:
@Coennie  and @Kri72 i just got it to work following this:

https://community.home-assistant.io/t/tuya-smart-smoke-detector-not-supported-ts0601-by-tze200-uebojraa/476288/4

Just change one of the lines, restart HA and it will detect it

    signature = {
        MODELS_INFO: [
            ("_TZE200_rccxox8p", "TS0601"),
            ("_TZE200_ntcy3xu1", "TS0601"),
            ("_TZE200_vzekyi4c", "TS0601"),

bbriers, 2023-12-09:
@Coennie @Kri72 same problem here, did anyone succeed to get it visible as smoke detector 
i tried the above solution from systemerror but it did not change anything.

Kri72, 2023-12-09:
@bbriers , unfortunately not.. haven't succeeded to get them working ,😭

clutgen, 2023-12-09:
Followed suggestion of @systemerror-codenotfound, paired the device again, but it didn't work as well.

Kri72, 2023-12-15:
Hello, I tried once more.. but it still doesn't want to trigger when they're is smoke..😭

bbriers, 2023-12-19:
even after i tested it with the tuya integration (bought tuya gateway in order to test) it does not trigger when there is smoke. the only thing that reports correctly trough the integration is the battery status. also the smoke density is not updated. 
These smokedetectors do not even report back the smoke density in the tuya app, only smoke alarm and battery level.
Since i bought 8, i will have to use them on tuya app and only monitor battery level in home assistant untill i find some real zigbee devices for a decent price.

InfinityMe, 2024-02-08:
О, у меня такой же

https://github.com/sprut/Hub/issues/3040

MattWestb, 2024-02-08:
> О, у меня такой же
> 
> [sprut/Hub#3040](https://github.com/sprut/Hub/issues/3040)

GT = Oh I have the same one

Kri72, 2024-04-14:
Hello.. I recently switched my home assistant from zha to z2mqtt, and tried to integrate the smoke detectors again.. and it works!!! 😄

mjacoulet, 2024-10-11:
> @Coennie and @Kri72 i just got it to work following this:
>
> https://community.home-assistant.io/t/tuya-smart-smoke-detector-not-supported-ts0601-by-tze200-uebojraa/476288/4
>
> Just change one of the [...]

Thanks. it worked perfectly for me.

closed 2024-12-09 by TheJulianJES