zigpy/zha-device-handlers #2741: [BUG] MOES/Tuya TS0601 _TZE200_la2c2uo9 smart dimmer - slider does not show up

glcos, 2023-11-11:
### Bug description

Smart dimmer is properly detected by ZHA with quirk zhaquirks.tuya.ts0601_dimmer.TuyaSingleSwitchDimmer but only on-off switch is added, the brightness slider does not show up.
However I can get and set the brightness by managing the cluster and related attribute.

mcu_version is 2.0.2

### Steps to reproduce

1. pair device
2. see that only a on-off switch is added, brightness slider is missing

### Expected behavior

Brightness slider should show up after pairing device

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_la2c2uo9",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_dimmer.TuyaSingleSwitchDimmer"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant Core",
    "version": "2023.11.2",
    "dev": false,
    "hassio": false,
    "virtualenv": true,
    "python_version": "3.11.6",
[... 297 lines cut ...]
            "unsupported_attributes": {}
          },
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

vukisz, 2023-11-16:
I have same _TZE200_la2c2uo9 paired like 3 weeks ago and I got brightness slider. Always with latest stable HA version.
Maybe there were some HA zha quirk updates in HA core so now pairing is different?
Actuator: SkyConnect

ZdenekM, 2023-11-23:
I just paired it and got a slider. But it behaves weirdly, when on, I can set whatever level and after some time it always goes to 13% by itself.

glcos, 2023-11-26:
I attemped a pairing on a fresh installation of Home Assistant core 2023.11.3 with no other ZigBee devices but I get the same result, device is pairing but the brightness slider does not show up.

Pairing log:

```
2023-11-26 13:22:34.798 DEBUG (MainThread) [zigpy.appdb] SQLite version for <module 'sqlite3' from '/usr/local/lib/python3.11/sqlite3/__init__.py'>: 3.34.1
2023-11-26 13:22:35.105 INFO (MainThread) [zigpy.serial] Using pyserial-asyncio-fast in place of pyserial-asyncio
2023-11-26 13:22:36.382 DEBUG (MainThread) [zhaquirks] Loading quirks module 'zhaquirks.adeo'
2023-11-26 13:22:36.385 DEBUG (MainThread) [zhaquirks] Loading quirks module 'zhaquirks.adeo.color_controller'
2023-11-26 13:22:36.395 DEBUG (MainThread) [zhaquirks] Loading quirks module 'zhaquirks.aduro'
2023-11-26 13:22:36.397 DEBUG (MainThread) [zhaquirks] Loading quirks module 'zhaquirks.aduro.adurolightncc'
2023-11-26 13:22:36.399 DEBUG (MainThread) [zhaquirks] Loading quirks module 'zhaquirks.aurora'
2023-11-26 13:22:36.401 DEBUG (MainThread) [zhaquirks] Loading quirks module 'zhaquirks.aurora.aurora_dimmer'
2023-11-26 13:22:36.405 DEBUG (MainThread) [zhaquirks] Loading quirks module 'zhaquirks.bitron'
2023-11-26 13:22:36.407 DEBUG (MainThread) [zhaquirks] Loading quirks module 'zhaquirks.bitron.thermostat'
2023-11-26 13:22:36.411 DEBUG (MainThread) [zhaquirks] Loading quirks module 'zhaquirks.bosch'
2023-11-26 13:22:36.413 DEBUG (MainThread) [zhaquirks] Loading quirks module 'zhaquirks.bosch.isw_zdl1_wp11g'
2023-11-26 13:22:36.417 DEBUG (MainThread) [zhaquirks] Loading quirks module 'zhaquirks.bosch.motion'
2023-11-26 13:22:36.421 DEBUG (MainThread) [zhaquirks] Loading quirks module 'zhaquirks.centralite'
2023-11-26 13:22:36.424 DEBUG (MainThread) [zhaquirks] Loading quirks module 'zhaquirks.centralite.cl_3130'
2023-11-26 13:22:36.429 DEBUG (MainThread) [zhaquirks] Loading quirks module 'zhaquirks.centralite.cl_3157100'
2023-11-26 13:22:36.431 DEBUG (MainThread) [zhaquirks] Loading quirks module 'zhaquirks.centralite.cl_3300S'
2023-11-26 13:22:36.432 DEBUG (MainThread) [zhaquirks] Loading quirks module 'zhaquirks.centralite.cl_3305S'
2023-11-26 13:22:36.435 DEBUG (MainThread) [zhaquirks] Loading quirks module 'zhaquirks.centralite.cl_3310S'
2023-11-26 13:22:36.436 DEBUG (MainThread) [zhaquirks] Loading quirks module 'zhaquirks.centralite.cl_3321S'
2023-11-26 13:22:36.439 DEBUG (MainThread) [zhaquirks] Loading quirks module 'zhaquirks.centralite.cl_3460L'
2023-11-26 13:22:36.442 DEBUG (MainThread) [zhaquirks] Loading quirks module 'zhaquirks.centralite.ias'
2023-11-26 13:22:36.444 DEBUG (MainThread) [zhaquirks] Loading quirks module 'zhaquirks.centralite.motion'
2023-11-26 13:22:36.445 DEBUG (MainThread) [zhaquirks] Loading quirks module 'zhaquirks.centralite.motionandtemp'
2023-11-26 13:22:36.449 DEBUG (MainThread) [zhaquirks] Loading quirks module 'zhaquirks.const'
2023-11-26 13:22:36.449 DEBUG (MainThread) [zhaquirks] Loading quirks module 'zhaquirks.danfoss'
2023-11-26 13:22:36.451 DEBUG (MainThread) [zhaquirks] Loading quirks module 'zhaquirks.danfoss.thermostat'
2023-11-26 13:22:36.474 DEBUG (MainThread) [zhaquirks] Loading quirks module 'zhaquirks.develco'
2023-11-26 13:22:36.479 DEBUG (MainThread) [zhaquirks] Loading quirks module 'zhaquirks.develco.air_quality'
2023-11-26 13:22:36.483 DEBUG (MainThread) [zhaquirks] Loading quirks module 'zhaquirks.develco.heat_alarm'
2023-11-26 13:22:36.484 DEBUG (MainThread) [zhaquirks] Loading quirks module 'zhaquirks.develco.motion'
2023-11-26 13:22:36.485 DEBUG (MainThread) [zhaquirks] Loading quirks module 'zhaquirks.develco.open_close'
2023-11-26 13:22:36.488 DEBUG (MainThread) [zhaquirks] Loading quirks module 'zhaquirks.develco.power_plug'
2023-11-26 13:22:36.490 DEBUG (MainThread) [zhaquirks] Loading quirks module 'zhaquirks.develco.smoke_alarm'
2023-11-26 13:22:36.491 DEBUG (MainThread) [zhaquirks] Loading quirks module 'zhaquirks.echostar'
2023-11-26 13:22:36.492 DEBUG (MainThread) [zhaquirks] Loading quirks module 'zhaquirks.echostar.bell'
2023-11-26 13:22:36.494 DEBUG (MainThread) [zhaquirks] Loading quirks module 'zhaquirks.ecolink'
2023-11-26 13:22:36.495 DEBUG (MainThread) [zhaquirks] Loading quirks module 'zhaquirks.ecolink.contact'
2023-11-26 13:22:36.499 DEBUG (MainThread) [zhaquirks] Loading quirks module 'zhaquirks.edpwithus'
2023-11-26 13:22:36.502 DEBUG (MainThread) [zhaquirks] Loading quirks module 'zhaquirks.edpwithus.redy_plug'
2023-11-26 13:22:36.503 DEBUG (MainThread) [zhaquirks] Loading quirks module 'zhaquirks.elko'
2023-11-26 13:22:36.507 DEBUG (MainThread) [zhaquirks] Loading quirks module 'zhaquirks.elko.smart_super_thermostat'
2023-11-26 13:22:36.512 DEBUG (MainThread) [zhaquirks] Loading quirks module 'zhaquirks.eurotronic'
2023-11-26 13:22:36.516 DEBUG (MainThread) [zhaquirks] Loading quirks module 'zhaquirks.eurotronic.spzb0001'
2023-11-26 13:22:36.517 DEBUG (MainThread) [zhaquirks] Loading quirks module 'zhaquirks.feibit'
2023-11-26 13:22:36.518 DEBUG (MainThread) [zhaquirks] Loading quirks module 'zhaquirks.feibit.switch'
2023-11-26 13:22:36.520 DEBUG (MainThread) [zhaquirks] Loading quirks module 'zhaquirks.gledopto'
2023-11-26 13:22:36.521 DEBUG (MainThread) [zhaquirks] Loading quirks module 'zhaquirks.gledopto.glc009'
2023-11-26 13:22:36.522 DEBUG (MainThread) [zhaquirks] Loading quirks module 'zhaquirks.gledopto.glc009p'
2023-11-26 13:22:36.523 DEBUG (MainThread) [zhaquirks] Loading quirks module 'zhaquirks.gledopto.gls007z'
[... cut ...] l_number'}, {'id': 2, 'name': 'stack_version'}, {'id': 16384, 'name': 'sw_build_id'}, {'id': 0, 'name': 'zcl_version'}]
2023-11-26 13:25:46.564 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x8725](TS0601): Device seen - marking the device available and resetting counter
2023-11-26 13:25:46.565 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x8725](TS0601): Update device availability -  device available: True - new availability: True - changed: False
2023-11-26 1

glcos, 2023-12-13:
I reinstalled a fresh instance of HA creating a new ZigBee network and rejoined device. Apparently everything goes well during pairing but the brightness slider entity is not created. I can change brightness from the "Manage Zigbee device" screen.
I'm attaching a fresh log.
Please have a look.

`2023-12-13 14:38:53.002 INFO (MainThread) [zigpy_znp.zigbee.application] Permitting joins for 254 seconds
2023-12-13 14:38:53.003 DEBUG (MainThread) [zigpy_znp.zigbee.application] Sending packet ZigbeePacket(timestamp=datetime.datetime(2023, 12, 13, 13, 38, 53, 3790, tzinfo=datetime.timezone.utc), src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.Broadcast: 15>, address=<BroadcastAddress.ALL_ROUTERS_AND_COORDINATOR: 65532>), dst_ep=0, source_route=None, extended_timeout=False, tsn=7, profile_id=0, cluster_id=<ZDOCmd.Mgmt_Permit_Joining_req: 0x0036>, data=Serialized[b'\x07\xfe\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2023-12-13 14:38:53.005 DEBUG (MainThread) [zigpy_znp.api] Sending request: ZDO.MgmtPermitJoinReq.Req(AddrMode=<AddrMode.Broadcast: 15>, Dst=0xFFFC, Duration=254, TCSignificance=0)
2023-12-13 14:38:53.030 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.MgmtPermitJoinReq.Rsp(Status=<Status.SUCCESS: 0>)
2023-12-13 14:38:53.033 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.MgmtPermitJoinRsp.Callback(Src=0x0000, Status=<Status.SUCCESS: 0>)
2023-12-13 14:38:53.037 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.Broadcast: 15>, address=0xFFFC), DstEndpoint=0, DstPanId=0x0000, SrcEndpoint=0, ClusterId=54, TSN=7, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK: 32>, Radius=0, Data=b'\x07\xFE\x00')
2023-12-13 14:38:53.042 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.MsgCbIncoming.Callback(Src=0x0000, IsBroadcast=<Bool.false: 0>, ClusterId=32822, SecurityUse=0, TSN=3, MacDst=0x0000, Data=b'\x00')
2023-12-13 14:38:53.046 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(timestamp=datetime.datetime(2023, 12, 13, 13, 38, 53, 46619, tzinfo=datetime.timezone.utc), src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=0, source_route=None, extended_timeout=False, tsn=3, profile_id=0, cluster_id=32822, data=Serialized[b'\x03\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2023-12-13 14:38:53.048 DEBUG (MainThread) [zigpy.zdo] [0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_rsp: [<Status.SUCCESS: 0>]
2023-12-13 14:38:53.049 DEBUG (MainThread) [zigpy.zdo] [0x0000:zdo] No handler for ZDO request:ZDOCmd.Mgmt_Permit_Joining_rsp([<Status.SUCCESS: 0>])
2023-12-13 14:38:53.059 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
2023-12-13 14:39:09.412 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.Ping.Req()
2023-12-13 14:39:09.418 DEBUG (MainThread) [zigpy_znp.api] Received command: SYS.Ping.Rsp(Capabilities=<MTCapabilities.SYS|AF|ZDO|UTIL|GP|APP_CNF: 1625>)
2023-12-13 14:39:14.356 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.TCDevInd.Callback(SrcNwk=0x3FDF, SrcIEEE=18:7a:3e:ff:fe:9e:a4:a9, ParentNwk=0x0000)
2023-12-13 14:39:14.359 INFO (MainThread) [zigpy_znp.zigbee.application] TC device join: ZDO.TCDevInd.Callback(SrcNwk=0x3FDF, SrcIEEE=18:7a:3e:ff:fe:9e:a4:a9, ParentNwk=0x0000)
2023-12-13 14:39:14.361 DEBUG (MainThread) [zigpy_znp.api] Sending request: ZDO.ExtRouteDisc.Req(Dst=0x3FDF, Options=<RouteDiscoveryOptions.UNICAST: 0>, Radius=30)
2023-12-13 14:39:14.422 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.ExtRouteDisc.Rsp(Status=<Status.SUCCESS: 0>)
2023-12-13 14:39:14.877 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.EndDeviceAnnceInd.Callback(Src=0x3FDF, NWK=0x3FDF, IEEE=18:7a:3e:ff:fe:9e:a4:a9, Capabilities=<MACCapabilities.Router|MainsPowered|RXWhenIdle|AllocateShortAddrDuringAssocNeeded: 142>)
2023-12-13 14:39:14.880 DEBUG (MainThread) [zigpy_znp.api] Command was not handled
2023-12-13 14:39:14.882 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.MsgCbIncoming.Callback(Src=0x3FDF, IsBroadcast=<Bool.true: 1>, ClusterId=19, SecurityUse=0, TSN=129, MacDst=0xFFFF, Data=b'\xDF\x3F\xA9\xA4\x9E\xFE\xFF\x3E\x7A\x18\x8E')
2023-12-13 14:39:14.883 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(timestamp=datetime.datetime(2023, 12, 13, 13, 39, 14, 883466, tzinfo=datetime.timezone.utc), src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x3FDF), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.Broadcast: 15>, address=<BroadcastAddress.ALL_ROUTERS_AND_COORDINATOR: 65532>), dst_ep=0, source_route=None, extended_timeout=False, tsn=129, profile_id=0, cluster_id=19, data=Serialized[b'\x81\xdf?\xa9\xa4\x9e\xfe\xff>z\x18\x8e'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2023-12-13 14:39:14.885 INFO (MainThread) [zigpy.application] New device 0x3fdf (18:7a:3e:ff:fe:9e:a4:a9) joined the network
2023-12-13 14:39:14.886 DEBUG (MainThread) [zigpy.device] [0x3fdf] Scheduling initialization
[... cut ...]
2023-12-13 14:39:28.119 DEBUG (MainThread) [zigpy.zcl] [0x3FDF:1:0x0000] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=100, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>)
2023-12-13 14:39:28.120 DEBUG (MainThread) [zigpy.zcl] [0x3FDF:1:0x0000] Sending reply: Default_Response(command_id=10, status=<Status.SUCCESS: 0>)
`

closed 2024-06-17 by github-actions[bot]