zigpy/zha-device-handlers #2748: [Device Support Request] Gas detector tuya zigbee TS0601 da _TZE204_chbyv06x

antoweb, 2023-11-13:
### Problem description

Device paired but no entities

### Solution description

Please create a custom quirks

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.11.2",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.6",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Rome",
    "os_name": "Linux",
    "os_version": "6.1.21-v8",
    "supervisor": "2023.11.0",
    "host_os": "Home Assistant OS 11.1",
    "docker_version": "24.0.6",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "iphonedetect": {
      "version": "1.4.0",
      "requirements": []
    },
    "tuya_local": {
      "version": "2023.9.2",
      "requirements": [
        "pycryptodome~=3.18",
        "tinytuya==1.12.11"
      ]
    },
    "hacs": {
      "version": "1.33.0",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    },
    "candy": {
      "version": "1.0.0",
[... 344 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

tidharmor, 2024-01-06:
I also have this device.
I tried using [this](https://github.com/sergey-suloev/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_gas.py) quirk but it didn't work at first.
I added another Endpoint (242) and after some tweaking I got it to use the quirk when connecting the device.
However, while it shows an entity for gas detection, it doesn't react when I test the device.
I think it may be using a different attribute then the linked quirk.

My device diagnostics are attached, if anyone can help to fix the quirk so it works with this device.

[zha-bdd3c06377f38529ecc9fdcb9ca57872-_TZE204_chbyv06x TS0601-b4f66de4a2cdde186916db2ca4189c85.json (2).txt](https://github.com/zigpy/zha-device-handlers/files/13850020/zha-bdd3c06377f38529ecc9fdcb9ca57872-_TZE204_chbyv06x.TS0601-b4f66de4a2cdde186916db2ca4189c85.json.2.txt)

jacadcaps, 2024-01-20:
@tidharmor 
Mind sharing your modifications?

tidharmor, 2024-01-21:
My quirk file is attached.

[ts0601_gas_leak.txt](https://github.com/zigpy/zha-device-handlers/files/14000604/ts0601_gas_leak.txt)

OleksiiSaviuk, 2024-05-31:
@tidharmor  Thank you. I use your quirk with ZHA. But during Alarm I don\t receive Alert in HA. Maybe you have a fix?

GuidoC1, 2024-09-04:
I have the same issue, no event nor state change triggered in HA when gas/CO2 is detected, when using the custom quirk provided above by Tidhamor.

Does anyone have a working custom quirk for it?

guslma, 2024-09-12:
up

RenatoLemons, 2024-09-30:
I worked on @tidharmor's shared modification and got a solution to the issue.
Now the device shows the entity and reacts when I test.

[ts0601_gas__TZE204_chbyv06x.py.txt](https://github.com/user-attachments/files/17168369/ts0601_gas__TZE204_chbyv06x.py.txt)

Thanks guys

GuidoC1, 2024-09-30:
@RenatoLemons Thank you for sharing your update of the custom quirk! I tested it and it works like a charm for me, much appreciated!

@antoweb @guslma @OleksiiSaviuk @jacadcaps @tidharmor --> if you guys are still interested in a fully working version for the gas/CO2 sensor, you might want to try the custom quick file shared by RenatoLemons.

jacadcaps, 2024-10-03:
@GuidoC1 tested, works fine with the new quirk.

josecarre, 2024-10-03:
> I worked on @tidharmor's shared modification and got a solution to the issue. Now the device shows the entity and reacts when I test.
> 
> [ts0601_gas__TZE204_chbyv06x.py.txt](https://github.com/user-attachments/files/17168369/ts0601_gas__TZE204_chbyv06x.py.txt)
> 
> Thanks guys

Many thanks ([I had same issue](https://github.com/zigpy/zha-device-handlers/issues/3389)), i've just loaded it also and it's working perfect :)

albertoacj, 2024-12-01:
Can an update  brick a quirk ? my TZE204_chbyv06x doesn't work with ZHA and quirk.

kisabgit, 2024-12-17:
> Can an update brick a quirk ? my TZE204_chbyv06x doesn't work with ZHA and quirk.

Used this quirk today with current HA build and it works fine.

burbon-pl, 2024-12-24:
> I worked on @tidharmor's shared modification and got a solution to the issue. Now the device shows the entity and reacts when I test.
> 
> [ts0601_gas__TZE204_chbyv06x.py.txt](https://github.com/user-attachments/files/17168369/ts0601_gas__TZE204_chbyv06x.py.txt)
> 

Hi there!
@RenatoLemons quirk works. Thanks! But - I have this device in Tuya wifi version also, and there apart of alarm state it shows also LEL (lower explosive level) in %. It is very sensitive and nicely reacts even on burning gas stove. Do you have any idea how to get to this data?

RenatoLemons, 2025-01-06:
> Hi there! @RenatoLemons quirk works. Thanks! But - I have this device in Tuya wifi version also, and there apart of alarm state it shows also LEL (lower explosive level) in %. It is very sensitive [...]

Hi @burbon-pl, I have no idea, sorry.
Maybe looking at the code of similar devices that has the same sensor.

Pols88, 2025-01-13:
Hi, I  used your quirk and added:

 zha:
     database_path: /config/zigbee.db
     enable_quirks: true
     custom_quirks_path: /config/zha_quirks/  

But the devices has only firmware upgrad. My device is Tuya gas detector TS0601 _TZE204_chbyv06x. TY to all.

[image]

kisabgit, 2025-01-15:
> Hi, I used your quirk and added:
>
> zha: database_path: /config/zigbee.db enable_quirks: true custom_quirks_path: /config/zha_quirks/
>
> But the devices has only firmware upgrad. My device is Tuya gas [...]

I have the same sensor and this worked for me:

add this code to configuration.yaml

```
zha:
  enable_quirks: true
  custom_quirks_path: config/custom_zha_quirks/
```

Create config/custom_zha_quirks folder in the root and copy ts0601_gas__TZE204_chbyv06x.py (watch out for the correct file type, on the first try, i made the mistake of keeping .TXT after .PY :)
[ts0601_gas__TZE204_chbyv06x.py.txt](https://github.com/user-attachments/files/18423764/ts0601_gas__TZE204_chbyv06x.py.txt)

If you recreate this steps to the letter, it should work :)

Pols88, 2025-01-16:
In works! TY!

kisabgit, 2025-02-06:
FYI Just updated HA core and three new entities appeared :) preheating, silent alarm, low explosive threshold.

PR #3734 (2025-01-21): Add additional Tuya gas sensors
## Proposed change

Adds Tuya explosive gas detectors:

- _TZE200_yojqa8xn
- _TZE204_zougpkpy
- _TZE204_chbyv06x
- _TZE204_yojqa8xn
- _TZE200_ggev5fsl
- _TZE200_u319yc66
- _TZE200_kvpwq8z7

Merge: #3635 first

Units are in LEL, not available in zigpy or HA.

## Additional information

Based on: https://github.com/Koenkk/zigbee-herdsman-converters/blob/40e72bc2515deb67ef35dddab04437342b7b3eb9/src/devices/tuya.ts#L1362

Closes: #2324
Closes: #2748

## Checklist

- [ ] The changes are tested and work correctly
- [x] `pre-commit` checks pass / the code has been form [... cut]