zigpy/zha-device-handlers #2750: [Device Support Request] Avatto TRV TRV07   TS0601 _TZE200_bvrlmajk

jkonieczny, 2023-11-14:
### Problem description

the TRV is paring, but not displaying any controls/entities.
I've tried all the quirks I could find from other TRVs (Avatto and others) but none of them worked

### Solution description

zha device handler/quirk  that works for TRV07  _TZE200_bvrlmajk tuya smart trv  that would allow controlling the thermostat temperature setting

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_bvrlmajk",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2023.6.3",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.11.4",
    "docker": true,
    "arch": "armv7l",
    "timezone": "Europe/Warsaw",
    "os_name": "Linux",
    "os_version": "5.15.93-sunxi",
    "run_as_root": true
  },
[... 270 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

kkossev, 2023-11-16:
Here is the needed information for making a quirk for TS0601 _TZE200_bvrlmajk

   "1":"Mode", mode Enum {"range": ["auto", "manual", "off", "on"]}
   "2":"Target temperature", temp_set Integer { "unit": "°C",  "min": 50, "max": 350, "scale": 1, "step": 5}
   "3":"Current temperature", temp_current Integer {"unit": "°C", "min": -100, "max": 500, "scale": 1, "step": 5}
   "6":"Working status", work_state	Enum {"range": ["closed", "opened"]}
   "7":"Window status", window_state Enum {"range": ["closed", "opened"]}
   "8":"Open window", window_check Boolean "{true,false}"
   "12":"Child lock", child_lock Boolean "{true,false}"
   "13":"Battery",battery_percentage Integer {"unit": "%", "min": 0, "max": 100, "scale": 0, "step": 1}
   "14":"Fault alarm", fault Bitmap {"label": ["fault_sensor", "fault_motor","fault_low_batt", "fault_ug_low_batt"], "maxlen": 4}
   "15":"Min. limit temperature", lower_temp Integer {"unit": "°C", "min": 50, "max": 150, "scale": 1, "step": 10}
   "16":"Max. limit temperature", upper_temp Integer {"unit": "°C", "min": 200, "max": 350, "scale": 1, "step": 10}
   "17":"Heating schedule", week_program_13_1 Raw{}
   "18":"Week program Tuesday", week_program_13_2 Raw{}
   "19":"Week program Wednesday", week_program_13_3	Raw	{}
   "20":"Week program Thursday", week_program_13_4 Raw {}
   "21":"Week program Friday", week_program_13_5 Raw {}
   "22":"Week program Saturday", week_program_13_6 Raw {}
   "23":"Week program Sunday", week_program_13_7 Raw	{}
   "101":"Room sensor calibration",	temp_correction Integer {"unit": "°C", "min": -100, "max": 100, "scale": 1, "step": 1 }
   "108":"Valve", valve_open_degree Integer {"unit": "%", "min": 0, "max": 1000, "scale": 1, "step": 10}
   "109":"Model", mfg_model	String {"maxlen": 255}
   "110":"Motor thrust", motor_thrust Enum {"range": ["strong", "middle", "weak"]}
   "111":"Display brightness", brightness Enum {"range": ["high", "mid", "low"]}
   "112":"Software version", soft_version Integer {"unit": "", "min": 0, "max": 65536, "scale": 0, "step": 1}
   "113":"Screen orientation", screen_orientation Enum {"range": ["up", "right", "down", "left"]}
   "114":"System mode", system_mode Enum {"range": ["comfort_mode","Eco_mode"]} 
   "115":"Switch deviation (energy-saving mode only)", switch_wrap Integer {"unit": "°C", "min": 5, "max": 50, "scale": 1,"step": 1}
   "116":"Motor data" motor_data String {  "maxlen": 255 }}

Thanks @hubivlad

Jedrek2021, 2023-12-23:
@kkossev thanks for the info but do you know what the temperature/time schedule format should look like, maybe an example of a schedule?

kkossev, 2024-01-05:
@Jedrek2021 Sorry for the late reply... Unfortunately, I don't know the schedules (Week Programs) format.

HunorRr, 2024-01-06:
I have the same device, TS0601 by _TZE200_bvrlmajk. Did you get it working?

Jedrek2021, 2024-01-08:
Yes, thanks @eteodun (https://github.com/eteodun/Avatto-TRV-TRV07-TS0601-_TZE200_bvrlmajk) I made a small modification, it works in manual mode and is controlled from HA, I use Zigbee2MQTT. This is enough for me for now (does not read schedules in auto mode). I'm including my quirk, maybe it will be useful to someone for further work
[tuyats601.zip](https://github.com/zigpy/zha-device-handlers/files/13859400/tuyats601.zip)

jacekk015, 2024-03-09:
https://github.com/jacekk015/zha_quirks
Quirk name:
ts0601_trv_rtitek2.py

Test it if possible. In case of errors Debug level logs needed from pairing process or specific action.

closed 2024-09-12 by github-actions[bot]