zigpy/zha-device-handlers #2751: [Device Support Request] Garaze opener TS0601 by _TZE204_nklqjk62

BrtnikFM, 2023-11-14:
### Problem description

Hello,
 I bought [garage opener](https://www.aliexpress.com/item/1005005408244662.html?spm=a2g0o.order_list.order_list_main.17.5f821802XHOvjP) and it was not recognized by HA and Sonoff dongle . I found lot of articles about MQTT and HA Addons and how HA Core in docker is bad. That's why I'd like to thank everybody here to this activity.
I'd like to ask you to add garage opener  based on TS0601 from manufacturer _TZE204_nklqjk62".

### Solution description

I've made script based on https://github.com/zigpy/zha-device-handlers/issues/1260, which works for me. I just changed model signature.

I'd like to ask about the signal. Currently it's switch, which means that it holds state ON. For controlling gate is used short signal. E.g. push button. Should it be solved by automation or is possible to change definition of signal to simulate push button?

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x0402",
      "input_clusters": [
        "0x0500"
      ],
      "output_clusters": []
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE204_nklqjk62",
  "model": "TS0601",
  "class": "garage2.TuyaGarageSwitchTO"
}
```

</details>

### Diagnostic information

[... 573 lines cut ...]
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    ContactSwitchCluster
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
```

</details>

### Additional information

Thank you in advance!

xxrb86, 2023-11-18:
Hi there,

I have the same TS0601 by _TZE204_nklqjk62 via Sonoff dongle and ZHA

would be good for others if we could make it work without a Zigbee Quirk

lNOFXl, 2023-12-08:
yup i have 2 of these as well, going through the setup right now, would be better if it was streamlined a little