zigpy/zha-device-handlers #2753: [Device Support Request] TS0601 Cover. Zemismart (Tuya) TZE204_1fuxihti curtain motor

Silicon51, 2023-11-15:
### Problem description

Tuya released new version of curtain motor with a name (manufacturer) TZE204_1fuxihti, device is same as described in [#1245](https://github.com/zigpy/zha-device-handlers/issues/1245) and [#1602](https://github.com/zigpy/zha-device-handlers/issues/1602) and [#1953](https://github.com/zigpy/zha-device-handlers/issues/1953) but none of those solutions worked.
with a blank configuration it show no entities, once i create a quirk with following code there appear 3 buttons (open, close, stop). If I click any button then error message occur "Failed to call service cover/close_cover. Failed to close cover: <Status.UNSUP_MANUF_CLUSTER_COMMAND: 131>"

What is important - integration show if curtain is open or close, it could read state of curtain, just cannot send anything.

### Solution description

Best solution would be upgrade of code which I send to allow remote communication 

### Screenshots/Video

<details><summary>Screenshot</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.11.2",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.6",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Warsaw",
    "os_name": "Linux",
    "os_version": "6.1.59",
    "supervisor": "2023.11.3",
    "host_os": "Home Assistant OS 11.1",
    "docker_version": "24.0.6",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "alarmo": {
      "version": "v1.9.10",
      "requirements": []
    },
    "hacs": {
      "version": "1.33.0",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    },
    "xiaomi_miot": {
      "version": "0.7.13",
      "requirements": [
        "construct>=2.10.56",
        "python-miio>=0.5.6",
        "micloud>=0.3"
      ]
    }
  },
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "after_dependencies": [
      "onboarding",
      "usb"
    ],
[... 426 lines cut ...]

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.WINDOW_COVERING_DEVICE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaManufacturerWindowCover,
                    TuyaWindowCoverControl,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }
```

</details>

AlterSu24, 2023-11-28:
My model is TZE200_xu4a5rhj and my situation is the same as you are.
It showed no entities at first, so I asked for support and add the custom quirk porvided by @javicalle.
Then the 3 buttons appeared, but cannot be controlled at all.
[https://github.com/zigpy/zha-device-handlers/issues/2483#issue-1813225750](url)

I replyed my debug log and waited for a few months until now.
No further tips or instructions.

Seems like the tuya cover motor do have a problem with ZHA.
[https://github.com/zigpy/zha-device-handlers/issues/2766#issue-2004521949](url)
I found the same issue here.

I tried with zigbee2mqtt, it works perfectly.
But I change back to ZHA because my ohter devices works more stable with ZHA.

Hope someone can help this out.

closed 2024-06-02 by github-actions[bot]

maxustm, 2024-06-10:
Also have _TZE204_1fuxihti, but i can't understand how make it works in ZHA.

Prozpect, 2025-01-07:
This post worked for me.  Only had to replace "_TZE204_xu4a5rhj"  with  "_TZE204_1fuxihti" 

[https://github.com/zigpy/zha-device-handlers/issues/3181#issuecomment-2154063608](url)

googled _TZE204 zha quirk curtain and the second link got me there.

gbrest, 2025-06-30:
Hey everyone! 👋

Just wanted to share a bit of my experience with the **MOES Zigbee Curtain Motor** (`_TZE204_1fuxihti`, model `TS0601`) and how I finally got it working in **Home Assistant with ZHA**, after a lot of digging and frustration.

---

📆 **As of mid-2025, the issue is still ongoing**

I bought a couple of these curtain motors in 2025 and paired them to Home Assistant using ZHA. To get them into pairing mode, I:

* Pressed the reset button **three times** quickly
* Then held it down until the LED started flashing

The device paired and showed up in the ZHA device list, but — **no entities appeared.**

---

❗**Why? Because it’s not really Zigbee-native**

This is another case of a Tuya TS0601 device. It communicates using Tuya’s proprietary MCU protocol over the `0xEF00` cluster, which is **not natively understood by ZHA**.

That’s why the device shows up, but doesn’t expose any entities or usable functionality unless you provide a **custom quirk**.

I found an old discussion where user **Prozpect** had posted a working quirk, but the link is gone. I had to do a deep dive across GitHub, Reddit and the Wayback Machine to find pieces of it and put it all back together.

---

🧩 **So here’s the full working solution, in one place, for rookies like me**

This is everything I did to get the curtain working properly with ZHA, including open/close/stop and position slider.

---

 ✅ **Step 1: Create the quirks folder**

Inside your `/config` directory (where `configuration.yaml` lives), create a new folder called:

**quirks**

So the full path is:

/config/quirks/

---

 ✅ **Step 2: Create an empty `__init__.py`**

Inside the `quirks` folder, create a new file called:

__init__.py

It can be completely empty, but it **must exist**.

---

 ✅ **Step 3: Create the custom quirk file**

Still inside the `quirks` folder, create a file called:

moes_curtain.py

And paste the following code into it (this is the full working quirk):

```

"""Tuya MCU based cover and blinds."""

from typing import Dict, Optional, Union
from zigpy.profiles import zha
import zigpy.types as t
[... 205 lines cut ...]
✅ **Step 4: Tell ZHA to load custom quirks**

In your `configuration.yaml`, add or update the `zha:` section like this:

```
zha:
  custom_quirks_path: quirks
```

> **Do not** include `/config/quirks` — just `quirks`

---

 ✅ **Step 5: Restart and pair again**

1. Restart Home Assistant completely (not just ZHA).
2. Remove any previous pairing of the curtain motor.
3. Put the curtain back into pairing mode (3 presses, then long press).
4. Pair it again from **ZHA > Add Device**.

You should now see a new **`cover.xxxxxx` entity**, and you’ll be able to open/close/stop and use the position slider.

---

Let me know if this works for you or if you get stuck — happy to help!