zigpy/zha-device-handlers #2755: [Device Support Request]

DJ-Specter, 2023-11-16:
### Problem description

Hi, 

I'm submitting this request following the conversation on this ticket: 

https://community.home-assistant.io/t/popular-moes-thermostat-not-showing-entities-after-zha-pairing/641041/1

I bought a Moes thermostat (gas boiler + zigbee version), that pairs with Zigbee but doesn't show any proper entities. 

I have a friend with what I thought was an identical model but it seems there are lots of subvariants of this one. Either way, it should definitely work. 

I have also done all the troubleshooting outlined in this, including removing and re-pairing several times: https://community.home-assistant.io/t/guide-for-zigbee-interference-avoidance-and-network-range-coverage-optimization/515752

I'm not sure what other info is required here, but please don't hesitate to ask :) 

Thank you very very much in advance for any help! 

### Solution description

As far as I understand, something to do with quirks is required for this to work properly with Zigbee Home Assistant, and the device needs to be opted into ZHA. I _think_ that's all that's necessary? 

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE204_aoclfnxz",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information
[... 550 lines cut ...]
    }
  }
}```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

glcos, 2023-11-20:
I have the same device and managed to make it working using this quirk:
https://github.com/jacekk015/zha_quirks/blob/main/ts0601_thermostat_avatto.py

however this device has a number of issues not dependent on Home Assistant or the ZigBee protocol itself, see here
https://github.com/jacekk015/zha_quirks/issues/29

darknessblade, 2023-12-04:
@glcos Thank you.

Have this model as well [gotten it with the 11.11 sale, at a steep discount], with this the 3 most important features are working for me

One question: Do you know if its possible to FULLY deactivate the build in programming/scedule

I have a Avatto one as well, of which this is possible. Having it only be In Manual modus.

https://www.aliexpress.com/item/1005002960679997.html?
And want to know if its possible on the MOES one as well

darknessblade, 2023-12-04:
Not sure what caused it, but For some reason the thermostat wants to heat to over 9-90 times the surface temperature of the sun

[image]

darknessblade, 2023-12-05:
@glcos Contacted the seller for the advanced menu instructions, and It does not look like its possible to disable the build in scedule

Now I only needs to find out why HASS thinks I want to heat my room to a temperature hotter than the sun

[image]

darknessblade, 2023-12-05:
See following log [Due large file size and randomness in the spikes, I cut the section of the most recent spike with roughly 3 temp/set checks before the spike]

[home-assistant log [CUT].log](https://github.com/zigpy/zha-device-handlers/files/13560445/home-assistant.log.CUT.log)

DJ-Specter, 2023-12-05:
Thanks for all the time and research guys, I really appreciate it 😄  

I don't think I put it anywhere in the original post so I edited to reflect what I now understand to be the specific model identifier: _TZE204_aoclfnxz

darknessblade, 2023-12-05:
@DJ-Specter Do you have the same issue with those Solar-flare temperature spikes?

DJ-Specter, 2023-12-05:
This also seems like a good opportunity to flag another post that much to my distress hasn't had any traction, but which I'm pretty sure must be related: 

https://community.home-assistant.io/t/call-service-set-target-temperature-only-sets-temporarily/649184

glcos, 2023-12-05:
@DJ-Specter @darknessblade in my installation I don't see these spikes in temperature.
I use ZHA and a ZigBee coordinator based on Texas Instruments CC2652P

darknessblade, 2023-12-05:
@glcos I use the "Sonoff Zigbee 3.0 plus type E" with ZHA.

Currently have changed the build in scedule to have everything set at 00:00 and 15C.
Did this as a test to see if this would have any effect on the spikes.

The weirdest part is that I have not touched or interacted with the Climate card/integration or panel. yet it still has spikes

DJ-Specter, 2023-12-11:
Does anyone know how long these kind of device support requests take to be actioned? 

I'm wondering now about just using the smelly Smart Life app to unlock the full content of these and some TRVs I got, but I really don't want to.

glcos, 2023-12-18:
@darknessblade I received a new Avatto thermostat and tested, now I think I understand what you ment for disabling scheduling mode.
I want my thermostat to be in manual mode, but once in while for some reason it goes into this screen by itself:

[image]

I could not find a way to completely disable scheduling or advanced programming mode and put it into manual.
I cannot understand if it's Home Assistant that is pushing this mode for some reason or it's the thermostat itself.

darknessblade, 2023-12-27:
Not sure waht causes that blank screen.
it might be because you turned the thermostat OFF trough the off button/turn of heating
which turns of the screen.

Also a small update regarding My moes thermostat

after updating the zigbee coordinator, it still gives the same spikes as I had before.

DJ-Specter, 2024-01-09:
Do the spikes only happen when the mode is switched from heat to off, or vice versa?

Or are they when the temperature is changed, maybe? It's hard for me to see from the graphs, but I feel like it's possible to test.

I'm also convinced this is the thing that's causing my thermostat to gradually decrease the temperature it's set to, per the HA community ticket I linked above...

darknessblade, 2024-01-12:
Caught the error in action:
[image]

It is only on HASS its side, as the thermostat itself does not show strange behavior or try to turn on even though hass thinks it is set to 2133C

DJ-Specter, 2024-01-12:
So is this a manufacturer problem then? How can we move forward to get it fixed?

darknessblade, 2024-01-12:
@DJ-Specter 
The only way I can think of is that HASS needs to Ignore ALL inputs/data changes that are above 50C
Like in my Image 2133C. would be cut off to 50C.  

This way the "bug" still is present, but the log data is more readable, as the spikes are cut off at 50C instead of UNCAPPED. which results in rare spikes of 1.3BILLION degrees Celsius.

My reason for 50C, is because nobody would use a set temp above 30C ""under normal use"". [Unless people use it for a Infared sauna, trough the floor sensor with the electric heating panel]. 
---------
As I am not well versed in coding, I am not sure if its possible to do this with the custom quirks.

darknessblade, 2024-01-20:
@DJ-Specter tried contacting the manufacturer/seller, for the tuya cluster information, but they do not want to give this information.

Would it be possible to hardlock the max reported set-temp to 50C, in the custom quirk? where it ignores anything above said value

DJ-Specter, 2024-01-20:
That's very annoying, damn. Nice one for getting in contact with them!

I strongly endorse your suggestion but have no idea how to make it happen
I'm afraid 😅

Hopefully it can be factored into the official ZHA quirk when this ticket
gets actioned?

Or some enterprising dev can swoop in to the rescue with an amendment to
the custom quirks listed above, or something fresh if that's more
constructive?

darknessblade, 2024-01-20:
@DJ-Specter 

True, luckily the thermostat has a roughly 10 second delay before activating the relay. so the 1MS "false positive" update isn't the biggest issue in terms of function, its only annoying for logging purposes, as it KILLS the graph.

In my opinion they should HARD-limit all thermostats recording/messages to a default limit.
This will prevent the same from happening on any future different thermostat.

jacekk015, 2024-03-10:
Fixed quirk at repo:
https://github.com/jacekk015/zha_quirks

All target room temp spikes over 50 degrees are not reported to HA.

DJ-Specter, 2024-03-11:
Whoa, out of nowhere, a new hero swoops in!

Amazing, thank you so much dude! I've just added it and will keep an eye on my temps over the next week and report back.

jacekk015, 2024-03-11:
Those spikes are seen in your previous log, so device send it, unfortunately.
```
2023-12-05 10:14:49.413 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(timestamp=datetime.datetime(2023, 12, 5, 9, 14, 49, 413554, tzinfo=datetime.timezone.utc), src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xEA98), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=25, profile_id=260, cluster_id=61184, data=Serialized[b'\tU\x02t\xb9\x10\x02\x00\x04\x00\xaa\x04\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=152, rssi=-62)
2023-12-05 10:14:49.414 DEBUG (MainThread) [zigpy.zcl] [0xEA98:1:0xef00] Received ZCL frame: b'\tU\x02t\xb9\x10\x02\x00\x04\x00\xaa\x04\x00'
2023-12-05 10:14:49.414 DEBUG (MainThread) [zigpy.zcl] [0xEA98:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False), tsn=85, command_id=2, *direction=<Direction.Client_to_Server: 1>)
2023-12-05 10:14:49.415 DEBUG (MainThread) [zigpy.zcl] [0xEA98:1:0xef00] Decoded ZCL frame: AvattoManufCluster:set_data_response(param=Command(status=116, tsn=185, command_id=528, function=0, data=[4, 0, 170, 4, 0]))
2023-12-05 10:14:49.415 DEBUG (MainThread) [zigpy.zcl] [0xEA98:1:0xef00] Received command 0x02 (TSN 85): set_data_response(param=Command(status=116, tsn=185, command_id=528, function=0, data=[4, 0, 170, 4, 0]))
2023-12-05 10:14:49.415 DEBUG (MainThread) [zhaquirks.tuya] [0xea98:1:0xef00] Received value [0, 170, 4, 0] for attribute 0x0210 (command 0x0002)
2023-12-05 10:14:49.416 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0xEA98:1:0x0201]: Attribute report 'AvattoThermostat'[occupied_heating_setpoint] = 1114214400
2023-12-05 10:14:49.416 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0xEA98:1:0x0201]: Attribute report 'AvattoThermostat'[occupied_heating_setpoint] = 1114214400
2023-12-05 10:14:49.418 DEBUG (MainThread) [zigpy.zcl] [0xEA98:1:0xef00] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=85, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>)
2023-12-05 10:14:49.418 DEBUG (MainThread) [zigpy.zcl] [0xEA98:1:0xef00] Sending reply: Default_Response(command_id=2, status=<Status.SUCCESS: 0>)
2023-12-05 10:14:49.419 DEBUG (MainThread) [bellows.zigbee.application] Sending packet ZigbeePacket(timestamp=datetime.datetime(2023, 12, 5, 9, 14, 49, 419008, tzinfo=datetime.timezone.utc), src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xEA98), dst_ep=1, source_route=None, extended_timeout=False, tsn=85, profile_id=260, cluster_id=61184, data=Serialized[b'\x18U\x0b\x02\x00'], tx_options=<TransmitOptions.ACK: 1>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2023-12-05 10:14:49.419 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0xea98, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_RETRY|APS_OPTION_ENABLE_ROUTE_DISCOVERY: 320>, groupId=0, sequence=85), 64, b'\x18U\x0b\x02\x00')
2023-12-05 10:14:49.419 DEBUG (MainThread) [homeassistant.components.zha.entity] climate.tze204_moes_ts0601_thermostat: Attribute 'occupied_heating_setpoint' = 1114214400 update
2023-12-05 10:14:49.420 DEBUG (MainThread) [homeassistant.components.zha.entity] climate.tze204_moes_ts0601_thermostat: Attribute 'occupied_heating_setpoint' = 1114214400 update
```

darknessblade, 2024-03-12:
Can verify I am only getting **DIPS** that drop the set_temp to 0-8C

Still slightly annoying, but still better than spikes of 100MIL+, especially for logging data

From the logs: {Isolated the section of the **DIP**}

[Log Temp dip.txt](https://github.com/zigpy/zha-device-handlers/files/14572277/Log.Temp.dip.txt)

jacekk015, 2024-03-12:
You know that you have a broken device. There's no cure for that.
Those 3 commands below were sent under half of second.
Your Zigbee mesh is probably totally overloaded with those messages.
I can't make the quirk to omit the 15 degrees setting - it's still in the usable range.
We can play more and count time and omit incoming messages within 1 second, but that's on the HA side.
Zigbee Mesh will still be utilizing a ton of trash messages.

```
2024-03-12 13:04:05.141 DEBUG (MainThread) [zhaquirks.tuya] [0xb993:1:0xef00] Received value [0, 0, 0, 15] for attribute 0x0210 (command 0x0002)
2024-03-12 13:04:05.194 DEBUG (MainThread) [zhaquirks.tuya] [0xb993:1:0xef00] Received value [0, 0, 0, 0] for attribute 0x0210 (command 0x0002)
2024-03-12 13:04:05.242 DEBUG (MainThread) [zhaquirks.tuya] [0xb993:1:0xef00] Received value [0, 0, 0, 15] for attribute 0x0210 (command 0x0002)
```

darknessblade, 2024-03-12:
> You know that you have a broken device. There's no cure for that. Those 3 commands below were sent under half of the second. Your Zigbee mesh is probably totally overloaded with those messages. I can't make the quirk to omit the 15 degrees setting - it's still in the usable range. We can play more and count time and omit incoming messages within 1 second, but that's on the HA side. Zigbee Mesh will still be utilizing a ton of trash messages.
> 
> ```
> 2024-03-12 13:04:05.141 DEBUG (MainThread) [zhaquirks.tuya] [0xb993:1:0xef00] Received value [0, 0, 0, 15] for attribute 0x0210 (command 0x0002)
> 2024-03-12 13:04:05.194 DEBUG (MainThread) [zhaquirks.tuya] [0xb993:1:0xef00] Received value [0, 0, 0, 0] for attribute 0x0210 (command 0x0002)
> 2024-03-12 13:04:05.242 DEBUG (MainThread) [zhaquirks.tuya] [0xb993:1:0xef00] Received value [0, 0, 0, 15] for attribute 0x0210 (command 0x0002)
> ```
> 
> > I use the "Sonoff Zigbee 3.0 plus type E" with ZHA.
> 
> Did you upgrade the firmware of the dongle or are you on stock version? I did upgrade my Sonoff-E with, I think, version 7.3.1 at https://darkxst.github.io/silabs-firmware-builder/

I upgraded the firmware from the stock firmware, I am running "Firmware: 7.3.1.0 build 176"
I am not running it in DUAL modus [Zigbee+threads]
_______________________
In what sense would the device be broken?
And what can I do to resolve this?

_______________________
I have roughly 90 zigbee devices in my network. Of which 12 are switches with Neutral "functioning as repeaters".
According to the ''current'' logs: _**Zigbee channel 20 utilization is 97.98%!**_
The thermostat is the Highlighted one [top right] [Connection: 125/60]

"Wifi_analyzer does not show any overlapping Wifi networks in the range of the zigbee channel"
_______________________
[image]
_______________________

jacekk015, 2024-03-12:
>In what sense would the device be broken?

ZHA is in reader mode. It just hears what is sent to HA.
If your device send's 3 different commands, under half of the second, with temp changes that's the device fault.
ZHA doesn't ask anything itself.

You can't even blame repeater or etc. because those commands are different: 15, 0 and 15 degrees back.

> Zigbee channel 20 utilization is 97.98%!
If that's only on the restart of HA that's normal.

darknessblade, 2024-03-12:
> ZHA is in reader mode. It just hears what is sent to HA. If your device send's 3 different commands, under half of the second, with temp changes that's the device fault. ZHA doesn't ask anything [...]

I can try connecting it to TUYA [have a Zigbee hub, I only use for updates]
Will have to check then what the TUYA IOT platform shows regarding logs. 

Can also check if I can still get a refund on aliexpress {even if it has been 5 months after ordering}

closed 2024-09-16 by github-actions[bot]

MrdUkk, 2026-10-20:
necroposting. if you have skills to compile firmwares - you can burn via simple UART-USB bridge central microcontroller with opensource thermostat code for CA51F MCU family. it will fix this and many more problems with this thermostat.
tuya RF module here only pass packets from central MCU to network (acts as simple network gateway). so spam of network with temperature and heater status will gone if main central MCU was flashed with new firmware.  
https://github.com/slacky1965/tuya_thermostat_full_custom_r02_zrd/

and if you want *original* firmware with only problems fixed - you can always contant me.