zigpy/zha-device-handlers #2757: [Device Support Request] Smart energy monitor TS0601 by _TZE200_ves1ycwx

Scorpio256, 2023-11-17:
### Problem description

No entity found.

Here is some diagnosting data
https://github.com/sprut/Hub/issues/2585

### Solution description

As far as I understand, something to do with quirks is required for this to work properly with Zigbee Home Assistant, and the device needs to be opted into ZHA. I think that's all that's necessary?

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
"data": {
    "ieee": "**REDACTED**",
    "nwk": 48283,
    "manufacturer": "_TZE200_ves1ycwx",
    "model": "TS0601",
    "name": "_TZE200_ves1ycwx TS0601",
    "quirk_applied": false,
    "quirk_class": "zigpy.device.Device",
    "quirk_id": null,
    "manufacturer_code": 4098,
    "power_source": "Mains",
    "lqi": 255,
    "rssi": -44,
    "last_seen": "2023-11-17T11:53:09",
    "available": true,
    "device_type": "Router",
    "signature": {
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
      "endpoints": {
        "1": {
          "profile_id": "0x0104",
          "device_type": "0x0051",
          "input_clusters": [
            "0x0000",
            "0x0004",
            "0x0005",
            "0xef00"
          ],
          "output_clusters": [
            "0x000a",
            "0x0019"
          ]
        }
      },
      "manufacturer": "_TZE200_ves1ycwx",
      "model": "TS0601"
    },
    "active_coordinator": false,
    "entities": [],
    "neighbors": [],
    "routes": [],
    "endpoint_names": [
      {
        "name": "SMART_PLUG"
      }
    ],
    "user_given_name": "ZHA - METER - Main power",
    "device_reg_id": "462e6e24bbed327d795732059be0fde4",
    "area_id": null,
    "cluster_details": {
      "1": {
        "device_type": {
          "name": "SMART_PLUG",
          "id": 81
        },
        "profile_id": 260,
        "in_clusters": {
          "0x0000": {
            "endpoint_attribute": "basic",
            "attributes": {
              "0x0001": {
                "attribute_name": "app_version",
                "value": 68
              },
              "0x0004": {
                "attribute_name": "manufacturer",
                "value": "_TZE200_ves1ycwx"
              },
              "0x0005": {
                "attribute_name": "model",
                "value": "TS0601"
              }
            },
            "unsupported_attributes": {}
          },
          "0x0004": {
            "endpoint_attribute": "groups",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0005": {
            "endpoint_attribute": "scenes",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0xef00": {
            "endpoint_attribute": null,
            "attributes": {},
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
```

</details>

Scorpio256, 2023-11-17:
https://vi.aliexpress.com/item/1005006034226090.html

Scorpio256, 2023-11-17:
https://github.com/zigpy/zha-device-handlers/blob/fb794323fb11014c41843bebc5825bc9004cd23a/zhaquirks/tuya/ts0601_din_power.py#L245

I tried to add a model for my device here and one sensor appeared in Home Assistant: state_increasing

[image]

dkssn, 2023-12-15:
Looking forward to this one too.

It appears that Zigbee2mqtt already supports it:
https://zigbee.blakadder.com/Zemismart_SPM02-D2TZ.html

blka, 2023-12-17:
duplicate of #2463. see this topic about SPM01 and SPM02 (is what you need) https://github.com/zigpy/zha-device-handlers/pull/2603

blka, 2023-12-17:
HowTo for someone like me:
1. create directory for custom quirks, eg `/config/zha_quirks`
2. setup custom quirk directory in configuration.yml
```
zha:
  enable_quirks: true
  custom_quirks_path: /config/zha_quirks
```
4. put this quirk into custom_quirks directory: [ts0601_din_power.py](https://raw.githubusercontent.com/agocsdaniel/zha-device-handlers/dev/zhaquirks/tuya/ts0601_din_power.py)
5. you can restart HA now, some metrics should be available in zha gui under your device (SPM02), eg "sensor summation_delivered" which can be added to your energy card right now.

The next steps are for people, who dont want to wait until zha core changes for 3 phase will be merged into official HA:
7. create directory `/config/custom_components/zha` - yes, after few stepst, you will overwrite your ZHA with custom one:)
8. go into this directory and wget repo branch, where all needed changes was made:
```
cd /config/custom_components/zha
wget https://codeload.github.com/Roeland54/core/zip/refs/heads/feature/3phase-2023-12
mv 3phase-2023-12 3phase-2023-12.zip
unzip 3phase-2023-12.zip
mv core-feature-3phase-2023-12/homeassistant/components/zha/* .
rm -rf 3phase-2023-12.zip core-feature-3phase-2023-12
```
10. in web ui, under dev tools, check configuration. if everything is fine, restart HA
11. after restart, goto your device, new stuff should appear. if ZHA didn't goes up, just reload again (was needed for me)
12. profit

remember to subscribe to PRs, to have info about merges to official one.
in the future you will need to remove this custom zha to have your installation updated.

Scorpio256, 2023-12-18:
> HowTo for someone like me:
>
> 1. create directory for custom quirks, eg `/config/zha_quirks`
> 2. setup custom quirk directory in configuration.yml
>
> ```
> zha:
>   enable_quirks: true
>   custom_quirks_path: [...]

[image]
Don't think it is working correctly.

blka, 2023-12-18:
#SOA1

maiandrei, 2024-01-07:
> https://codeload.github.com/Roeland54/core/zip/refs/heads/feature/3phase-2023-12

Hi! 
The aforementioned link is not valid anymore.
Does it work for you?
Thanks

blka, 2024-01-08:
@maiandrei,
base on author activity, branch name is now with "-4" at the end,
so
`https://codeload.github.com/Roeland54/core/zip/refs/heads/feature/3phase-2023-12`

 see https://github.com/Roeland54/core/branches

tbaumann, 2024-01-20:
I tried. The Meter device has no new entries.

checked out github.com/Roeland54/core branch `feature/3phase-2023-12-4` to `/config/custom_components/zha`

Is there a clever way to make sure the custom component is indeed used?

blka, 2024-01-20:
Check quirk location and configuration.
Works with z2m by default…

tbaumann, 2024-01-29:
> Check quirk location and configuration. Works with z2m by default…

Thanks. I assumed I need the quick OR the branch.

tbaumann, 2024-03-13:
Hope this can be merged soon. Staying on a custom branch for too long gives me the heebie-jeebies.

mathmorier, 2024-04-25:
I Just fellow the steps,
To made working I add in the code file [ts0601_din_power.py](https://raw.githubusercontent.com/agocsdaniel/zha-device-handlers/dev/zhaquirks/tuya/ts0601_din_power.py) the correcte number of my device  _TZE200_v9hkz2yn

[image]

Mow I get this infos :
[image]

mrehanabbasi, 2024-04-25:
> I Just fellow the steps,
> To made working I add in the code file [ts0601_din_power.py](https://raw.githubusercontent.com/agocsdaniel/zha-device-handlers/dev/zhaquirks/tuya/ts0601_din_power.py) the correcte number of my device  _TZE200_v9hkz2yn
> 
> [image]
> 
> I reload the zha configuration and that was good
> 
> Mow I get this infos :
> [image]
> 

Yes, but you only get values for one phase.

mathmorier, 2024-04-25:
You don't thinks is the add or avrage of the 3 phase ?

mrehanabbasi, 2024-04-25:
Nope, it's just one phase.

mathmorier, 2024-04-25:
You have an idea to correct that ?

mrehanabbasi, 2024-04-25:
@mathmorier One possible workaround is to replace zha with a custom one like https://github.com/Roeland54/core

rrataj, 2024-06-19:
Hey There! I created a quirk for `_TZE200_v9hkz2yn` version which supports 3 phase Voltage, Current and Power stats:
https://github.com/rrataj/zha-device-handlers/blob/0a6b19ab591378c53f6a8887e6c65b76ea31084b/zhaquirks/tuya/ts0601_din_power_3.py

Hope that helps someone :)

el3m3l3k, 2024-06-21:
@rrataj - amazing job! All functionalities are unlocked :) I have removed the old quirk, uploaded yours, picked up a 2kW hairdryer, and played a bit in the tester, finding what follows:

1. This is what "Reconfigure" device shows as identified with your quirk: 
[image]

2. The sensors for the first phase are duplicated (maybe I did something wrong?):
[image]

3. Besides being duplicated, the first phase works very accurately for all 3 sensors (Power, Voltage, Current).

4. The second phase shows only current correctly :( Voltage and Power are significantly lower than it should be.

5. I was not able to identify any electrical socket or light switch in my house that would change something in the readings for phase 3 (it looks like frozen). 

6. Not sure how the "Summation delivered" works... it should aggregate value from all 3 phases, right?

rrataj, 2024-06-21:
Hey! Thanks :)
For answer to your question no 2 see here: https://github.com/zigpy/zha-device-handlers/issues/3184#issuecomment-2178663005

Basically 1st entry shows always the same value as 2nd entry (1st entry was suppose to be a total sum for all phases, but it's not working yet, while 2nd entry shows 1st phase)

Regarding 3,4,5 I think you may just connect it in a wrong way. You need to connect all 3 phase wires to different phases (adequate to Z, Y, Z). Please try to change wires for phase 2 with phase 3 and check again (Or phase 1 with 2, hard to tell which one is the X,Y,Z). I don't have access to device right now so can't check it.

Regarding 6 - Yes, I think it's aggregated total for all 3 phases.

vasektomas, 2024-06-21:
I also would like to thank you.  It's perfect!  Off topic.  Does anybody know what command can be used to clear internal values of the summation received and the summation delivered?

chriexpe, 2024-06-22:
> Hey There! I created a quirk for `_TZE200_v9hkz2yn` version which supports 3 phase Voltage, Current and Power stats: https://github.com/rrataj/zha-device-handlers/blob/0a6b19ab591378c53f6a8887e6c65b76ea31084b/zhaquirks/tuya/ts0601_din_power_3.py
> 
> Hope that helps someone :)

Thanks for that quirk, I have one that is almost identical, _TZE204_v9hkz2yn, I tried yours, changed the device name and it unfortunately didn't work, tho if I reconfigure it detect something, where do I extract more info about my device?
[image]

pomianowski, 2024-07-02:
Hey, i see #2726 related. I also bought `TS0601 by _TZE200_x8fp01wi`, adding the name to the https://github.com/rrataj/zha-device-handlers/blob/0a6b19ab591378c53f6a8887e6c65b76ea31084b/zhaquirks/tuya/ts0601_din_power_3.py does not help. I see similar issues for @chriexpe. Can I somehow help diagnose this device to get more information?

flaneur90, 2024-07-29:
> Hey There! I created a quirk for `_TZE200_v9hkz2yn` version which supports 3 phase Voltage, Current and Power stats: https://github.com/rrataj/zha-device-handlers/blob/0a6b19ab591378c53f6a8887e6c65b76ea31084b/zhaquirks/tuya/ts0601_din_power_3.py
> 
> Hope that helps someone :)

Great stuff, plug and play! Thks @rrataj !

ha-primalb, 2024-08-12:
@pomianowski - could you please share complete working file for "_TZE200_x8fp01wi" or create new branch of ts0601_din_power_3.py? thank you in advance.

iorifly, 2024-08-29:
How to reset a value recorded on a ZigBee device.

Dear all, I have this value (energy = 1696.32kWh) recorded on my “Zemismart three-phase ZigBee meter”

[image]

I can’t reset it! Does anyone know what to do? If I reset it on Home Assistant, it comes back… because the information is on the device.
I’ve already reset it and it still doesn’t work.
I’ve already disconnected it from the power supply, and in fact, I received it with these measurements.

Some images that might help.

[image]

[image]

arno-millenaar, 2024-10-18:
Thanks a lot for your effort to make this quirk.  I also had the XYZ (thin voltage signal) wires mixed up and after reading your comment above, the power output numbers now match the power information from the Solar Inverter I am using.  Added the template power sensor so I now also have the total power available.

rimallo, 2024-11-26:
[image]

Thanks for efforts adding this type of devices into zha. Do you have any idea on how to integrate this 3-phase variation?
_TZE200_ny94onlb [[aliexpress](https://www.aliexpress.com/item/1005006801446920.html)]

rainerliebe, 2025-01-05:
I'm also interested. Maybe with the right command in Dev console: Endpoint / Cluster / Command / Payload

markusd1984, 2025-01-07:
@rimallo is the _TZE200_ny94onlb an older version compared to _TZE204_v9hkz2yn or other way around?

Did you make it work adopting the quirk and approach from rimallo?

@rimallo did you too purchase the "Zemismart Zigbee Smart Energy Meter 3 Phase 4 Wire Power Monitor Sensor" ?

thomo66, 2025-01-30:
> <img alt="Screenshot 2024-11-26 um 12 51 15" width="1047" src="https://private-user-images.githubusercontent.com/2337951/389916057-f833f592-c965-4a7f-bdb9-1d5392e69c59.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzgyNDI3NjMsIm5iZiI6MTczODI0MjQ2MywicGF0aCI6Ii8yMzM3OTUxLzM4OTkxNjA1Ny1mODMzZjU5Mi1jOTY1LTRhN2YtYmRiOS0xZDUzOTJlNjljNTkucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDEzMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTAxMzBUMTMwNzQzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NzEyYTA5N2NiZTkwZmQyYjI5Y2U3YmU1OWQ2MjJhZDhjMDU4MDE3MWFmMTcxMDAwMDhjM2QyZTdjMjdjMzViOSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.icoreQfCPQcHy2kwGkYRu7lyCvCBBFJPN15NGXu-EV4">
> Vielen Dank für Ihre Bemühungen, diese Art von Geräten in zha zu integrieren. Haben Sie eine Idee, wie man diese 3-Phasen-Variante integrieren kann? `_TZE200_ny94onlb`[ [aliexpress](https://www.aliexpress.com/item/1005006801446920.html) ]
> 
> Auf dem Screenshot ist das Ergebnis der Verwendung [der aktuellen Eigenart](https://raw.githubusercontent.com/rrataj/zha-device-handlers/0a6b19ab591378c53f6a8887e6c65b76ea31084b/zhaquirks/tuya/ts0601_din_power_3.py) + Hinzufügen einer Zeile mit meinem Gerätetyp zu sehen:
> 
> ```
> MODELS_INFO: [
>             ("_TZE200_byzdayie", "TS0601"),
>             ("_TZE200_ewxhg6o9", "TS0601"),
>             ("_TZE200_ny94onlb", "TS0601"),
>         ],
> ```

Show here: [https://github.com/zigpy/zha-device-handlers/issues/3799](url)

bzumik1, 2025-06-19:
Hi, I have just discovered, that this device is not supported by ZHA, for me it reports as TS0601 from  _TZE200_ves1ycwx. Happy to help anyone with the testing

bzumik1, 2025-06-19:
Ok, I got it working, thanks @blka [your quirk](https://raw.githubusercontent.com/agocsdaniel/zha-device-handlers/dev/zhaquirks/tuya/ts0601_din_power.py) worked for my **TS0601** from _**TZE200_ves1ycwx**. Why is this still not merged to ZHA? Does someone know how to get it there?