zigpy/zha-device-handlers #2763: [BUG] TRADFRI bulb GU10 WW 345lm8

Ichhabekeinenase, 2023-11-18:
### Bug description

I cant Dimm this Light via the Dashboard. When i do it over the OTA Cluster it works. If i do the Turn on Brightness i can dimm it to a lower Brightness but not Higher.

### Steps to reproduce

1. Go to the Light.
2. Try to Dimm the light.
3. Light doesnt Dimm.
4. Try to Dimm over the Turn on Slider. Dimms lower but doesnt go Higher.

### Expected behavior

Dimm the Bulb!

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

Ichhabekeinenase, 2023-11-21:
Is it that new one? Because I cant dimm it over the Dashboard. Mine is just the WW one without Color or Cold White.

MattWestb, 2023-11-21:
Yes was baying it yesterday and its still working after firmware upgrade.
The "old" version is 400lm and is having the old IKEA Zigbee module and the new have the Silabs module.

MattWestb, 2023-11-21:
LED2104R3

MattWestb, 2023-11-21:
[image]

Ichhabekeinenase, 2023-11-21:
Which Dongle are you using? Because like i said I cant dimm them over the Sliders just over the Cluster Commands. On All 6 of them. Im on Home Assistant 2023.11.2.

MattWestb, 2023-11-21:
HA 2023.11.0 and one "IKEA Billy EZSP" (the old IKEA module with latest EZSP 6.10.7).

Ichhabekeinenase, 2023-11-21:
Ok thats weird! Im using an Sonoff ZB-Dongle E with the efr32mg21. I can try 11.0 in an Docker Image with a Bulb. Maybe that Fixes the Issue? Thats the only difference!

TheJulianJES, 2023-11-21:
Try to remove the device from ZHA and then re-pair it. Perhaps the interview didn't complete properly.

TheJulianJES, 2023-11-21:
But before you do that, please download the diagnostic information from HA for that bulb.
It's missing in the issue post.

MattWestb, 2023-11-21:
Also after deleting it wait one minute or restart HA so the device is deleted OK in the database or its coming back with the "old" clusters that we dont like then new added is the only way getting all OK.

Ichhabekeinenase, 2023-11-21:
Now my Test Light Dimms over the Normal Dimmer but only if i turn it off and on again in Home Assistant?
Heres the now Diagnostic Information!

 "data": {
    "ieee": "**REDACTED**",
    "nwk": 34862,
    "manufacturer": "IKEA of Sweden",
    "model": "\u001aTRADFRI bulb GU10 WW 345lm8",
    "name": "IKEA of Sweden \u001aTRADFRI bulb GU10 WW 345lm8",
    "quirk_applied": false,
    "quirk_class": "zigpy.device.Device",
    "quirk_id": null,
    "manufacturer_code": 4476,
    "power_source": "Mains",
    "lqi": 124,
    "rssi": -69,
    "last_seen": "2023-11-21T17:56:33",
    "available": true,
    "device_type": "Router",
    "signature": {
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
      "endpoints": {
        "1": {
          "profile_id": "0x0104",
          "device_type": "0x0101",
          "input_clusters": [
            "0x0000",
            "0x0003",
            "0x0004",
            "0x0005",
            "0x0006",
            "0x0008",
            "0x1000",
            "0xfc7c"
          ],
          "output_clusters": [
            "0x0019"
          ]
        },
        "242": {
          "profile_id": "0xa1e0",
          "device_type": "0x0061",
          "input_clusters": [
            "0x0021"
          ],
          "output_clusters": [
            "0x0021"
          ]
        }
      },
      "manufacturer": "IKEA of Sweden",
      "model": "\u001aTRADFRI bulb GU10 WW 345lm8"
    },
    "active_coordinator": false,
    "entities": [
      {
        "entity_id": "number.test_auf_dem_niveau",
        "name": "IKEA of Sweden \u001aTRADFRI bulb GU10 WW 345lm8"
      },
      {
        "entity_id": "number.test_ein_aus_ubergangszeit",
        "name": "IKEA of Sweden \u001aTRADFRI bulb GU10 WW 345lm8"
      },
      {
        "entity_id": "number.test_einschaltstrompegel",
        "name": "IKEA of Sweden \u001aTRADFRI bulb GU10 WW 345lm8"
      },
      {
        "entity_id": "button.test_identifizieren",
        "name": "IKEA of Sweden \u001aTRADFRI bulb GU10 WW 345lm8"
[... 124 lines cut ...]
      },
      "242": {
        "device_type": {
          "name": "PROXY_BASIC",
          "id": 97
        },
        "profile_id": 41440,
        "in_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}

TheJulianJES, 2023-11-23:
Can you click on the light entity icon or entity title in HA to open the "more info" dialog and send a screenshot of how it looks?

MattWestb, 2023-11-23:
My light is the same batch the its have the same bad data setting of the model string : `"value": "\u001aTRADFRI bulb GU10 WW 345lm8"` and i thing the last 8 is also one miss but its not the first time the factory have making bad data setting for IKEA devices but it shall behavior the same as my.

closed 2024-05-28 by github-actions[bot]