zigpy/zha-device-handlers #2767: TZE204_lzriup1j_TS0601-Avatto-Zigbee-thermostat-panel-Water-heating

schueron, 2023-11-22:
### Problem description

Device/ item link:

https://de.aliexpress.com/item/1005004149361837.html?spm=a2g0o.productlist.main.39.421dl2ZYl2ZYKE&algo_pvid=69cccfc7-969d-4640-97ce-0a0a4d66e966&algo_exp_id=69cccfc7-969d-4640-97ce-0a0a4d66e966-19&pdp_npi=4%40dis%21EUR%2140.77%2123.84%21%21%2143.45%21%21%40211b5e2c17006537203875901e7e28%2112000033940405463%21sea%21DE%210%21AB&curPageLogUid=F3Rj3KWcEr3u

Device is detected by ZHA but does not show any sensor/controls information.

### Solution description

Full climate support for the Thermostat panel. with all functions available under ZHA.

### Screenshots/Video

[image]

### Device signature

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE204_lzriup1j",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

### Diagnostic information

{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.11.2",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.6",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "6.1.21-v8",
    "supervisor": "2023.11.3",
    "host_os": "Home Assistant OS 11.1",
    "docker_version": "24.0.6",
    "chassis": "embedded",
    "run_as_root": true
  },
[... 308 lines cut ...]
          "id": 97
        },
        "profile_id": 41440,
        "in_clusters": {},
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}

### Logs

[home-assistant_2023-11-22T12-03-00.214Z.log](https://github.com/zigpy/zha-device-handlers/files/13438982/home-assistant_2023-11-22T12-03-00.214Z.log)

### Custom quirk

I have already added a _TZE200_2ekuz3dz_TS0601-Avatto-Zigbee-thermostat-panel-Water-heating and it works

https://github.com/jacekk015/zha_quirks/blob/main/ts0601_thermostat_avatto.py

javicalle, 2023-11-23:
Duplicate of #2589

schueron, 2023-11-24:
I have placed the pyhton script below in the directory /config/custom_zha_quirks/

Sensor status is unknown

debug modus is activated

https://github.com/jacekk015/zha_quirks/blob/main/ts0601_thermostat_avatto2.py

[image]

[home-assistant_2023-11-24T12-09-35.064Z.log](https://github.com/zigpy/zha-device-handlers/files/13458901/home-assistant_2023-11-24T12-09-35.064Z.log)

une18, 2024-01-21:
Any updates on this?

closed 2024-01-23 by schueron