zigpy/zha-device-handlers #2770: [Device Support Request] TS0601 _TZE204_wbhaespm Din breaker SMTONOFF

tdoolaege, 2023-11-23:
### Problem description

the device connects to ZHA but no power monitoring is possible or anything else
[image]
[image][

[image]
(https://nl.aliexpress.com/item/1005005143771083.html?spm=a2g0o.order_list.order_list_main.5.3e5b79d28k4Q12&gatewayAdapt=glo2nld)

### Solution description

a working device :) or at least energy/power meter

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE204_wbhaespm",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.11.2",
    "dev": false,
[... 877 lines cut ...]
                'output_clusters': [
                    0x0021,  # Replace with the actual cluster ID
                ],
            },
        },
        'manufacturer': _TZE204_wbhaespm,
        'model': TS0601,
        'class': zigpy.device.Device,
    }

    replacement = {
        # Add any cluster replacements if needed
    }

    device_automation_triggers = {
        # Define automation triggers if needed
    }

```

</details>

### Additional information

i tried creating a costume quirk with the help of openAI... but unfortunatly ... and i lack the knowledge of zigbee standard to create one myself

tdoolaege, 2023-11-23:
[home-assistant_zha_2023-11-23T19-18-56.459Z.zip](https://github.com/zigpy/zha-device-handlers/files/13453310/home-assistant_zha_2023-11-23T19-18-56.459Z.zip)
added full debug log

AlleyneH, 2024-01-21:
I would also like to support this request

closed 2024-07-26 by github-actions[bot]

tdoolaege, 2025-01-03:
> I would also like to support this request as well

if have found that you can use some functionality from following quirck

zhaquirks.tuya.ts0601_powermeter.TuyaPowerMeter_GPP

added the line, restarted ZHA. and good to go

class HikingPowerMeter(TuyaSwitch):
>     """Hiking Power Meter Device - DDS238-2."""
>
>     signature = {
>         # "node_descriptor": "<NodeDescriptor byte1=1 byte2=64
> mac_capability_flags=142 manufacturer_code=4098
>         #                       maximum_buffer_size=82
> maximum_incoming_transfer_size=82 server_mask=11264
>         #                       maximum_outgoing_transfer_size=82
> descriptor_capability_field=0>",
>         # device_version=1
>         # input_clusters=[0x0000, 0x0004, 0x0005, 0xef00]
>         # output_clusters=[0x000a, 0x0019]
>         MODELS_INFO: [
>             ("_TZE200_bkkmqmyo", "TS0601")
>             *("_TZE204_wbhaespm", "TS0601")*
>             ],
>         ENDPOINTS: {
>             # <SimpleDescriptor endpoint=1 profile=260 device_type=51
>             # device_version=1
>             # input_clusters=[0, 4, 5, 61184]
>             # output_clusters=[10, 25]>
>             1: {
>                 PROFILE_ID: zha.PROFILE_ID,
>                 DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
>                 INPUT_CLUSTERS: [
>                     Basic.cluster_id,
>                     Groups.cluster_id,
>                     Scenes.cluster_id,
>                     TuyaManufClusterAttributes.cluster_id,
>                 ],
>                 OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
>             }
>         },
>     }

ststamat, 2025-02-16:
I would also like to support this request