zigpy/zha-device-handlers #2772: [Device Support Request] Please add so far unsupported attributes to Sonoff TRVZB

Cangoo, 2023-11-26:
### Problem description

The new Sonoff TRVZB (https://zigbee.blakadder.com/Sonoff_TRVZB.html) is working in general but missing some attributes that can be exposed and changed. For instance **local_temperature_calibration** can be adjusted sending a Zigbee SET command, but is not showing up in the list of available attributes.

### Solution description

Like in zigbee2mqtt, it would be great if the missing attributes like **local_temperature_calibration** could be added to ZHA as well. See https://github.com/Koenkk/zigbee2mqtt/issues/19269 for details.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4742, maximum_buffer_size=82, maximum_incoming_transfer_size=255, server_mask=11264, maximum_outgoing_transfer_size=255, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0301",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0006",
        "0x0020",
        "0x0201",
        "0xfc11",
        "0xfc57"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "SONOFF",
  "model": "TRVZB",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2023.11.3",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.11.6",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "6.1.21-v8+",
    "run_as_root": true
[... 412 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

onurunaldi, 2023-11-30:
I am seeking ZHA support for extended features since I have four of those. Sonoff is advertising this product as Home Assistant and ZHA compatible but I can use very few of its features, so I support this request.

dmytrohuskov, 2023-11-30:
It will be great to get the valve positioning control, and the auto mode control. So I support this request.

MarcinResearchLab, 2023-12-01:
I support this request as well. I am having 6 of these already and plan to go further with up to 25 in total if only these features are supported.

MarkHessburg, 2023-12-03:
" idle_steps, closing_steps“ so the TRVZB’s firmware actually already can more than just turn on and off?

asdkjasdfla, 2023-12-03:
Please try to prioritise this as it's a quite popular TRV

len5771, 2023-12-03:
Yep, have a couple of these and would be great to get the extra attributes.

dmytrohuskov, 2023-12-04:
> > I think that if you set the device to 'Auto' mode by short-pressing the main button, it will gradually adjust the valve. However, in this mode, you cannot then manually adjust the temperature; it is fixed at either 19 or 16 degrees. Interestingly, when set this way, the device maintains the room temperature at around 21-22 degrees.
> 
> Wow, interesting. Must test that. Shouldn’t the attribute "occupied heating setpoint“ work with that? Set it to auto and set the occupied heating setpoint instead of „temperature“.

I just tried to change the year from 1900 to 2100 in the YAML file and set the state, but it did not work. On the TRV screen, it still showed '19', and when I hit the refresh button, it reverted to 1900.

Kombustor, 2023-12-08:
I'd also love to see the same support zigbee2mqtt offers for this TRV in ZHA, as I am running 6 of these great thermostats.

asdkjasdfla, 2023-12-08:
Considering that z2m has got it would it not be possible to import the parameters that z2m.uses for this device?

OldSmurf56, 2023-12-11:
I would also really appreciate full ZHA support  for the Sonoff TRVZB. I am currently converting all my old Danfoss Connect to Sonoff.

poikjo, 2023-12-12:
I also got few of these and now struggling with manual / automode. On manual mode there's just off and heat modes and on auto mode (heat/cool) it does not cool down so maybe it's calibration issue. Radiator temperature is something like 27-28 °C and TRV in idle all the time... well at least it's warm 😀

lio0909, 2023-12-14:
Support this request as well since I'm using a douzen of them through ZHA
Thanks!

MagicBugsBunny, 2023-12-15:
Support this request as well :-)

MoustacheMonkey, 2023-12-17:
I support this as well

velasub, 2023-12-18:
I support this request

ggdagg, 2023-12-19:
I support this request

defixio, 2023-12-19:
I support this too

KaiDroste, 2023-12-21:
I have started with a quirk to implement the local temperature calibration. Now I am stuck because I don't know how to set it as a float input value. Maybe someone can help over here? 
Furthermre I found out that "local_temperature_calibration" is defined in the zigbee cluster library. Why is it than not supported out of the box? 

```python
"""Quirk for SONOFF TRVZB."""
# import typing

from zigpy.profiles import zha 
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import (
    Basic,
    Identify,
    OnOff,
    Ota,
    PollControl,
    PowerConfiguration,
    Time,
)
from zigpy.zcl.clusters.hvac import (
    Thermostat,
)
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)

# Thermostat_Cluster_ID: typing.Final = 0x0201

class Thermostat(CustomCluster):
    "Cluster for Thomostatic values"

    cluster_id = Thermostat.cluster_id
    TempCalibration = 0x0010

    def __init__(self, *args, **kwargs):
        """Init."""
        super().__init__(*args, **kwargs)
        self.endpoint.device.voltage_bus.add_listener(self)
        self.endpoint.device.consumption_bus.add_listener(self)
        self.endpoint.device.power_bus.add_listener(self)

    def power_reported(self, value):
        """Power reported."""
        self._update_attribute(self.TempCalibration, value)

class SonoffTrvzb(CustomDevice):
    """Sonoff TRVZB custom device implementation."""

    signature = {
        MODELS_INFO: [("SONOFF", "TRVZB")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.THERMOSTAT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    OnOff.cluster_id,
                    PollControl.cluster_id,
                    Thermostat.cluster_id,
                    0xfc11,
                    0xfc57,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.THERMOSTAT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    OnOff.cluster_id,
                    PollControl.cluster_id,
                    Thermostat.cluster_id,
                    0xfc11,
                    0xfc57,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            },
        },
    }
```

Cangoo, 2023-12-22:
Thanks for starting to write a quirk! Unfortunately I have absolute no experience here. I just wanted to add that the local_temperature_calibration seems to be an int. For example "-10" means an offset of -1,0°C.

[image]

hbalzerbss, 2023-12-23:
I support this request

kilianramharter, 2023-12-25:
I support this request

magickaribu, 2023-12-28:
I support this request and having the possibility to use an external sensor (Ex SNZB02) would be a killer feature

t0medic, 2023-12-29:
+1 Support for this request!

Thanks to all Dev's that know how to do that!

Regards,

PacoCT, 2023-12-30:
+1 support the request

JanDotNet, 2023-12-30:
I support this request

igitbits, 2023-12-31:
I lately bought this device and integrated it via ZHA in my Home Assistant installation. I fully support the request to make more of the functionality available via ZHA - especially as a lot of people recently seem to reconvert their installations to Z2M due to lack of functionality in ZHA. I strongly support the further development of ZHA and hope that this popular device will be a showcase of what can be achieved via ZHA. Really appreciate the support of the developers!

fire4you, 2024-01-01:
I support this request. I bought 5 devices!

Balthasar213, 2024-01-03:
I support this request. Got a couple of them too.

rigalp, 2024-01-04:
I support the request. I could set the temp offset but could not set the thermostat temperature in automations. E.g. set 19C at 21:00, set 21C at 7:00

bagnico, 2024-01-04:
I support this request. Got a couple of them too.

r0bb10, 2024-01-10:
i support it also, got two in extra discount for my bathrooms.

rstpeng, 2024-01-12:
I support this request

majopes, 2024-01-13:
I support this request

Le10E, 2024-01-16:
I support this too, I have approx a dozen of these and trying to migrate off Ewelink

snickers2k, 2024-01-16:
+1 
its the best zigbee radiator thermostat out there right now

snickers2k, 2024-01-17:
> Thanks for starting to write a quirk! Unfortunately I have absolute no experience here. I just wanted to add that the local_temperature_calibration seems to be an int. For example "-10" means an [...]

this is weird.. after using this attribute to write the offset, i got a new UI entry (Local Temp Offset) afterwards
[image]

JGrossholtz, 2024-01-21:
It would be nice to be able to schedule the temperature by the time of day and day of week. It would be a great way to optimize power consumption IMO.
Thanks !

mdglawson, 2024-01-22:
I support this issue - would be great to see the functionality that is there exposed when using ZHA.

snickers2k, 2024-01-23:
i only can advise you to stay away from these thermostats... i bought 8 pcs - 6 of them measures temperatures in different rooms far beyond the possible 2.5°C offset. more like 5°C .. it' impossible to work with them

el-ef, 2024-01-26:
> i only can advise you to stay away from these thermostats... i bought 8 pcs - 6 of them measures temperatures in different rooms far beyond the possible 2.5°C offset. more like 5°C .. it' impossible to work with them
> 
> i used tado on the same spots for years without problems. sonoff is simply bad

I own für of theese (TRVZB) and can confirm that the temp is off to about 5°C. Pretty random and practically unusable control the room temp.
I have heard that such problems do more or less affect all thermostats where the sensor is directly integrated, placed near the hot water stream.

Anyways. I hope there will be some fixes for Sonoff soon cause their thermostats can only do full-on and full-off at the moment, using ZHA.
I have tried BetterThermostat with an external sensor and the overheating protection enabled. Doesn't really fix the issue with control itself, though.

sebidude, 2024-01-27:
I also cannot recommend those thermostats. They only know two states, open and close. This means it is hard to maintain a constant temperature in a room. I used them for three days in my workspace and it did not feel comfortable. Here is an example for one working day:

[image]
The blue line is the target temp, the thermostat was open on the yellow areas.

I switched back to the Danfoss Ally, as they learn how to heat up the room over time and maintain the temperature quite well.

If your room is well isolated, this might not be relevant for you and the Sonoff may work well. Every room in my house has two exterior walls which are not well isolated. Therefore the room and radiator temperatures decrease quite quickly, when the Sonoff closes the valve.

This is just my experience with the Sonoff TRVZB.

MagicBugsBunny, 2024-01-28:
This is my workspace with the Sonoff TRV. Have only +- 0,5°C with Better Thermostate.
[image]

MagicBugsBunny, 2024-01-28:
Which Zigbee TRVs can work better as a normal mechanical TRV with HA?

sebidude, 2024-01-28:
> Which Zigbee TRVs can work better as a normal mechanical TRV with HA?
I think, we should not discuss this here, as this issue is about the Sonoff TRVZB. There are many threads about this on home-assistant. IMHO, the perfect does not exist or not all features are implemented, yet.

el-ef, 2024-01-28:
Am I right that the Sonoff is technically/mechanically capable of incrementally opening and closing the valve instead of hard-flipping between 0% and 100% all the time? Is such already working via Zig2MQTT?

Just asking because the overheating that comes with that is really annoying. I hope that can be fixed with a better ZHA integration soon. I also asked Sonoff the other day if there are opportunities to fix such behaviour via firmware in the future. Might be totally up to ZHA, though.

Will give the Danfoss a try until then. Thanks guys!

asdkjasdfla, 2024-01-28:
Yes hope someone can clarify this

MagicBugsBunny, 2024-01-28:
how does the TRVs work with the org. app of ewelink?
also only open/close?

el-ef, 2024-01-29:
Got response from the Sonoff support. They say the fully open/close is a known issue, they will look into it.

But who knows _when_ ...

zipzipro, 2024-02-05:
podporuji žádost, nastavení hodnoty termostatu v automatizaci není funkční (Nastavit hodnotu pro posun místní teploty)

I support the request, setting the thermostat value in automation is not functional. (Set value for local temperature shift)

zipzipro, 2024-02-05:
> jak funguje TRV s org. aplikace ewelink? také pouze otevřít/zavřít?

v eWelink funguje vše správně, jako termostat

in eWelink, everything works correctly, like a thermostat

el-ef, 2024-02-05:
English pls.

MattWestb, 2024-02-05:
@el-ef Hi have written in both original and English so its very OK.

paologroppo, 2024-02-09:
I have bought many of them and was expected to work nice with HA but it's actually totally unusable. I hope that Sonoff will fix this issue ASAP. I support this request.

quettih, 2024-03-10:
I'm going to buy 2 of them because price is really interesting. hoping that ZHA will improve compatibility and generally speaking, will improve development process as zigbee2mqtt does. how can we help ?

Cangoo, 2024-05-22:
> since FW 1.1.5 it seems a lot is now available and supposedly working, would be nice to have proper ZHA support especially the valve opening / closing to adress the current, not ideal, ON/OFF working behaviour.
Is there a way to update the FW via HomeAssistant, without buying the Sonoff-Zigbee-Bridge and using the Sonoff app?

sleveille4, 2024-05-23:
> > since FW 1.1.5 it seems a lot is now available and supposedly working, would be nice to have proper ZHA support especially the valve opening / closing to adress the current, not ideal, ON/OFF working behaviour.
> 
> Thanks for your reply. Is there a way to update the FW via HomeAssistant, without buying the Sonoff-Zigbee-Bridge and using the Sonoff app?

yes, i activated OTA in ZHA by adding some line in my config file, here is my ZHA section, added the OTA section, reboot and updates showed up:

```
zha:
  zigpy_config:
    source_routing: true
    ezsp_config:
        CONFIG_SOURCE_ROUTE_TABLE_SIZE: 41
    ota:
      enabled: true
      broadcast_enabled: true
      broadcast_initial_delay: 14400
      broadcast_interval: 14400
      z2m_remote_index: "https://raw.githubusercontent.com/Koenkk/zigbee-OTA/master/index.json"
      sonoff_provider: true
```

then either you wait and notifications will appear that new FW is available,or you go in system -> update and scan and you should see that new FW is available.

i updated several devices including the TRVZB, except my zlinky, for some reason it doesn't, it goes to 100% then disconnect, reconnecting it restores it's function but it still has the old FW.

note that it takes time to update 1 TRVZB (+-40 min), and that in some case trvzb that were "far" from the conbee II were not able to update (despite properly working for month so i suspect update requires a stronger more stable network)

solved the issue by putting my RPI4 on battery + wifi and moving to the rooms updates were failing.

MagicBugsBunny, 2024-05-24:
take time ...

[image]

sleveille4, 2024-05-24:
> take time ... ~ 60s for 1%
> 
> ![image](https://private-user-images.githubusercontent.com/43914643/333490710-bb9ce292-8901-4a79-ae41-9bffbe22e896.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY1MzYzMTUsIm5iZiI6MTcxNjUzNjAxNSwicGF0aCI6Ii80MzkxNDY0My8zMzM0OTA3MTAtYmI5Y2UyOTItODkwMS00YTc5LWFlNDEtOWJmZmJlMjJlODk2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTI0VDA3MzMzNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTRhODAzOTBkZDAzM2MzNGFmYzg1OWM5NWQ4MGJhYmY5Y2Q3MzY2YWRmYjhiYmY1MmU4Y2ZjMWNjYzNmNmI1ZmQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.L_v2T3zj6meeMVA3yC33FsNjUWN-P7mU7qnoSLF81Tw)

odd, my plugs updated fast, like a minute or so, but my trvzb took around 40min each.

good for you if it is fast then :)

sleveille4, 2024-05-27:
and i thought my 40 min to update one was bad......

sleveille4, 2024-06-08:
update:

i decided to give Z2MQTT a test as i want my heating system working for next fall, and i will not have time to tinker by then:

it works, and i also discovered that on all the other devices there where more entitys available now, mainly one i was missing: OFFSETS! (for humidity / voltage / power / temperature etc...)!!!!

only downside is that the DB is twice the size now, but it seems to report more often values, i always thought values were reported as they changed, seems not.

while easier to setup, i realize ZHA is largely incomplete and missing way more than i thought even on devices that are available for years.

the offsets i mentionned actually did already exist 3 years ago in the devices as there was no FW update for those.

i spent hours trying to make fake sensors to apply offset, while the function actually exist within almost ALL the devices, just that ZHA doesn't expose them

i was even able to update my zlinky FW as a bonus

[10 more replies]

closed 2024-11-25 by TheJulianJES