zigpy/zha-device-handlers #2776: [Device Support Request] Hive SLR2B and SLT3b by Computime

andyb2000, 2023-11-27:
### Problem description

I'm using the Hive devices SLR2B (Boiler relay/controller) and SLT3b (Thermostat/wall unit) by Computime which are the Hive zigbee units.
I have paired them up with information available based on SLR1B (https://zigbee.blakadder.com/Hive_SLR1b.html The difference between 1B and 2B are the channels, 2B is for heating and hot water, 1B is heating only). The two channels work fine, I've just had to note/name the entities to distinguish them.
The issue is the connection to Zigbee, this seems to keep dropping with these devices causing them to go offline (and turn the heating relays off).

After pairing them, in Home Assistant ZHA I go into the device info and can see the current room temperature, current target and what status the relays are in (heating or idle).
However, after a varying amount of time (I have recorded, 6 minutes, 8 minutes, 9 minutes, etc) the unit will turn the red light on (on the boiler relay, this also switches off both relays to my boiler), the thermostat will say "Receiving" or "No signal" and the system will not respond to changes.
It then will start working again, this time with a more regular amount of time (6:01minutes, 6:11, 6:06, 5:57 - so always around the 6 minute mark) and resume the functions it was doing (relays turn back on. Green light comes on boiler relay unit).

### Solution description

The zigbee connection to remain connected. It appears to be some type of heartbeat or signal getting missed which causes them to lose sight of each other. They then appear to re-connect automatically and resume operations on their own.

### Screenshots/Video

The SLR2B in ZHA:

[image]

You can see the cycling, during these periods the heating was supposed to be on constantly but due to the signal/drop it goes idle then re-connects and resumes

[image]

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4153, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=0, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "5": {
      "profile_id": "0x0104",
      "device_type": "0x0301",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0009",
        "0x000a",
        "0x0201",
        "0xfd00"
      ],
      "output_clusters": [
        "0x0000",
        "0x000a",
        "0x0019",
        "0x0402",
        "0xfd00"
      ]
    },
    "6": {
      "profile_id": "0x0104",
      "device_type": "0x0301",
      "input_clusters": [
        "0x0201"
      ],
      "output_clusters": []
    },
    "7": {
      "profile_id": "0x0104",
      "device_type": "0x0301",
      "input_clusters": [
        "0x0402"
      ],
      "output_clusters": []
    },
    "8": {
      "profile_id": "0x0104",
      "device_type": "0x0301",
[... 835 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

alrassia, 2023-11-27:
Could you also post the logs from zha? Let’s make sure something else like weak signal or interference is not an issue as well.

andyb2000, 2023-11-29:
Actually, I think you may be right with connectivity, although not a 'typical' signal drop.
Looking at my visualisation, the SLR3b doesn't show a link at all, yet it is connected and shows correctly in "Last Seen". Not sure if that is because the SLR3b is connected 'through' the SLR2b pairing or not though and that is expected behaviour.

[image]

I also added another Zigbee router 'inbetween' physically the two to ensure it's not a signal loss situation and I can confirm the Zigbee signal to both the SLR2b and SLR3b are good.

AlexC, 2024-01-27:
I'm also experiencing this issue, where the thermostat and receiver disconnect from each other and the red light appears on the receiver. Similar to the above image @andyb2000, when this happens you'll note that the `DeviceType` of the thermostat changes from `Router` to `EndDevice`

AlexC, 2024-01-29:
@andyb2000 are you using the Sonoff ZigBee 3.0 P dongle by any chance? Last night I flashed the firmware to the latest, and things appear to be much more stable. I also no longer get hundreds of ZHA Check-in events for the Hive thermostat.

Looking at the changelog for the firmware, I see the following which may have solved this for me:

> Increase message timeout from 7 to 8 seconds to increase message delivery success rate for devices using a 7.5 seconds poll interval (https://github.com/Koenkk/zigbee2mqtt/issues/13478#issuecomment-1501188485)

See https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.x.0/CHANGELOG.md

mikestir, 2024-06-30:
I've done some work on this after having what sounds like the same issue. It seems to be related to the reporting interval that ZHA sets up in the thermostat, which appears to default to 15 minutes. This is too slow and the receiver thinks the thermostat has gone away.

Using zha_toolkit I was able to execute the following command from within HA in developer tools->services:

```
service: zha_toolkit.execute
data:
  ieee: sensor.hive_thermostat_temperature
  command: conf_report
  endpoint: 9
  cluster: 0x402
  attribute: 0
  min_interval: 5
  max_interval: 120
  reportable_change: 50
  tries: 100
  event_done: zha_done
```

Change the entity name as required.

closed 2025-01-03 by github-actions[bot]