zigpy/zha-device-handlers #2777: [Device Support Request] - SIN-4-FP-21_EQU

APCM-1, 2023-11-27:
### Problem description

I have a Pilot wire heating module (Leroy Merlin version) and I want to control it via ZHA. 

Brand : ADEO
Model : SIN-4-FP-21_EQU

[image]

At the moment we don't have mode handling, only On and Off (Hors Gel) with power measurement.

Some people seems to have handle this module with Z2M but not ZHA. 

[https://www.zigbee2mqtt.io/devices/SIN-4-FP-21_EQU.html](url)

[https://github.com/Koenkk/zigbee2mqtt/issues/19169](url)

### Solution description

We need to have mode handling. There are 6 modes : confort, confort -1, confort -2, eco, Hors Gel, Off.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4727, maximum_buffer_size=82, maximum_incoming_transfer_size=500, server_mask=11264, maximum_outgoing_transfer_size=500, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x1000",
        "0xfc00"
      ],
      "output_clusters": [
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0066",
      "input_clusters": [
        "0x0021"
      ],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "Adeo",
  "model": "SIN-4-FP-21_EQU",
  "class": "zigpy.device.Device"
}
```
[... 562 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

ConteZer0, 2023-11-29:
Hello, i have those devices since a couple of days. I'm using Deconz (Phoscon) to add device to my zigbee network: devices are recognized as Lamp in Phoscon and i can only On and Off but without power measurement. So there is also no Power Measurement in Home Assistant (ZHA).
What have you done to see power measurement?

APCM-1, 2023-11-29:
Hello,

I just paired modules to Home Assistant and they were recognized as a socket and the consumption information came automatically

ConteZer0, 2023-11-30:
Ok thanks. I finally removed DeconZ and i replaced it with ZHA. I have now "Commutateur", "Demande instantanée" and "Summation delivered" but not "Consommation chauffage". Is it usefull?

APCM-1, 2023-11-30:
Oh yes sorry this one is a manual entry to know how much consumption I have each day

romain563, 2023-12-19:
I'm also interested. It will be interesting to check if the device is the same with SIN-4-fp-21 because they are both made by nodon

Ce post m'intéresse aussi. Il serait intéressant de voir si un développement peut permettre de faire fonctionner la référence du fabricant Nodon: la sin-4-fp-21

APCM-1, 2023-12-20:
> Great! Can you explain me how to do that please?

Into parameters, in entries tab, create a new one, you need to choose a utiliy meter (water, electricity) in french its "Compteur de services publics (eau, gaz, électricité…)" and set the same parameters as the screenshot  

[image]

APCM-1, 2023-12-20:
> I'm also interested. It will be interesting to check if the device is the same with SIN-4-fp-21 because they are both made by nodon
> 
> Ce post m'intéresse aussi. Il serait intéressant de voir si un développement peut permettre de faire fonctionner la référence du fabricant Nodon: la sin-4-fp-21

I think it's the same, I see somewhere else on github that people reference the SIN-4-FP-21 as the same device as the SIN-4-FP-21_EQU.

ConteZer0, 2024-01-02:
> > Great! Can you explain me how to do that please?
> 
> Into parameters, in entries tab, create a new one, you need to choose a utiliy meter (water, electricity) in french its "Compteur de services publics (eau, gaz, électricité…)" and set the same parameters as the screenshot
> 
> ![Capture](https://private-user-images.githubusercontent.com/141181269/291830337-ff8281aa-f13d-41f0-b1d2-8bacba04677a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQyMzE3OTQsIm5iZiI6MTcwNDIzMTQ5NCwicGF0aCI6Ii8xNDExODEyNjkvMjkxODMwMzM3LWZmODI4MWFhLWYxM2QtNDFmMC1iMWQyLThiYWNiYTA0Njc3YS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTAyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDEwMlQyMTM4MTRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05YWY4YWYzODQ1MjEzNzhkODRjMmU4MzM0MGQ1N2ZiZjkzZThlMDg2YjIwMDM0ODZhYWJhZjY0ZTY5N2E5NWI4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.hGYRkeGn-sbNKDV35viMlrrTNDrAg429fM5uG6aUAI4)

Thnaks!!

stadros83, 2024-01-03:
Any news on this ?
Is everything working with ZHA ?

APCM-1, 2024-01-04:
> Any news on this ? Is everything working with ZHA ?

Hi,

I asked directly the manufecturer for information. For Z2M the integration will be available in late January but for ZHA it will be available later.

anoril1979, 2024-02-14:
I flag here to follow. I have several of these modules on my heaters and I would like more that consumption and on/off. :)

popof1664, 2024-02-20:
hi,
I am also interested in this integration under ZHA: the SIN-4-FP-21 boxes seem very good, I am waiting to be able to control my radiators with them.

laur3nt95, 2024-03-01:
Same issue here. added yesterday to HA through ZHA, it shows a switch toggle,  and consumption, but no control to the heater. 
I aslo have to mention that the "switch" part of the module has no effect on y heater, which is kind of weird to me.

seb-tinet, 2024-03-01:
Hi i've asked to Nodon's support and they told me the ZHA intégration is on
the pipe and it will be released in the coming weeks.
Seb

Skunnyk, 2024-03-07:
Hi,

I have a `SIN-4-FP-21` (the one from Nodon), and I am able to use it with ZHA, via [zha_toolkit](https://github.com/mdeweerd/zha-toolkit), with the help of https://github.com/Koenkk/zigbee2mqtt/issues/19169#issuecomment-1801181411 . 

You can use `zha_toolkit.scan_device` to scan your zigbee device for all its informations, and look at the "Manufacture specific" cluster_id: 
```
        "0xfc00": {
          "cluster_id": "0xfc00",
          "title": "Manufacturer Specific",
          "name": "manufacturer_specific",
          "attributes": {
            "0x0000": {
              "attribute_id": "0x0000",
              "attribute_name": "0",
              "value_type": [
                "0x20",
                "uint8_t",
                "Analog"
              ],
              "access": "READ|WRITE|REPORT",
              "access_acl": 7,
              "manf_id": 4747,
              "attribute_value": 1
            }
          },
          "commands_received": {},
          "commands_generated": {}
        }
```

Now you can send command to the module:
```
service: zha_toolkit.attr_write
data:
  ieee: xx:xx:xx:xx:xx:xx # your sin-4-fp-21 module id
  manf: 4747 # 0x128b, nodon
  endpoint: 1
  cluster: 64512 # 0xfc00
  attribute: 0 # which attribute we want to write, only 0 / 0x0000 is available in this cluster
  attr_val: "1" # 1: Confort, 2: Eco, 3: Hors Gel, see https://github.com/Koenkk/zigbee2mqtt/issues/19169#issuecomment-1801181411
```

Et voila©®, you can now integrate it in custom configuration in HA.

I hope this can help you, until Nodon/ZHA-device-handlers integrate this natively :)

Mysticronos, 2024-03-15:
They just changed their support page with devices compatibility... Bad news for SIN-4-FP-21, it seems that HA is through Z2M only...

[image]

steenlarsen, 2024-03-20:
Does anybody know if the information above can be used to write a Quirk for ZHA? Btw. Thanks a lot @Skunnyk for finding this information!

oxillo, 2024-05-28:
I've posted a quirk in #3040 that may also support ADEO variant.
Feel free to test.

garc33, 2024-09-18:
Hello I made some adaptations of Nodon quirk in #3040 to make it working with Adeo device.
Here is the code of custom quirk working for me:
```python
"""Adeo SIN-4-FP-21_EQU pilot wire module."""
from typing import Any, Final
from zigpy.quirks import CustomCluster
from zigpy.quirks.v2 import QuirkBuilder, EntityType, EntityPlatform
from zigpy.zcl.foundation import (
    BaseAttributeDefs,
    BaseCommandDefs,
    ZCLAttributeDef,
    ZCLCommandDef,
    WriteAttributesStatusRecord,
    Status
)

from zigpy.zcl import ClusterType

import zigpy.types as t

import logging
_LOGGER = logging.getLogger(__name__)

class NodonPilotWireMode(t.enum16):
    OFF = 0x00
    COMFORT = 0x01
    ECO = 0x02
    FROST_PROTECTION = 0x03
    COMFORT_MINUS_1 = 0x04
    COMFORT_MINUS_2 = 0x05

NODON_PILOT_WIRE_CLUSTER_ID = 0xfc00  #64512

class NodonPilotWireCluster(CustomCluster):
    """NodOn pilot wire manufacturer-specific cluster."""

    cluster_id: Final = NODON_PILOT_WIRE_CLUSTER_ID
    name: Final = "Pilot Wire"
    ep_attribute: Final = "pilot_wire_cluster"
    manufacturer_id_override: Final = 4747
    
    class AttributeDefs(BaseAttributeDefs):
        pilot_wire_mode: Final = ZCLAttributeDef(
            id=0x0000, 
            type=NodonPilotWireMode, 
            access="r"
        )

    class ClientCommandDefs(BaseCommandDefs):
        setMode: Final = ZCLCommandDef(
            id=0x0000,
            schema={"mode": NodonPilotWireMode},
            direction=True,
            is_manufacturer_specific=True
        )
    
    async def write_attributes(
        self, attributes: dict[str | int, Any], manufacturer: int | None = None
    ) -> list:
        """Override writes to the pilot_wire_mode attribute."""
        if "pilot_wire_mode" in attributes:
            await self.client_command(0, attributes.get("pilot_wire_mode"), manufacturer=4747)
            await self.read_attributes(attributes)
            return [[WriteAttributesStatusRecord(Status.SUCCESS)]]
        return await super().write_attributes(attributes, manufacturer=4747)

    async def read_attributes(
        self,
        attributes: list[int | str],
        allow_cache: bool = False,
        only_cache: bool = False,
        manufacturer: int | t.uint16_t | None = None,
    ) -> Any:
        return await super().read_attributes(attributes, allow_cache=False, only_cache=False, manufacturer=4747)

(

    QuirkBuilder("ADEO", "SIN-4-FP-21_EQU")
    .also_applies_to("Adeo","SIN-4-FP-21_EQU")
    .replaces(NodonPilotWireCluster)
    .enum("pilot_wire_mode",NodonPilotWireMode,NODON_PILOT_WIRE_CLUSTER_ID,
        ClusterType.Server,1,EntityType.STANDARD,EntityPlatform.SELECT,False,True)
    .add_to_registry()
)
```

anoril1979, 2024-09-18:
Will test it ASAP! THANKS!

Mookunicorn, 2024-09-19:
> Hello I made some adaptations of Nodon quirk in #3040 to make it working with Adeo device. Here is the code of custom quirk working for me:
>
> ```python
> """Adeo SIN-4-FP-21_EQU pilot wire module.""" [...]

Shouldn't the last part with "QuirkBuilder" be like this?
```python
(

    QuirkBuilder("ADEO", "SIN-4-FP-21_EQU")
    .also_applies_to("NodOn","SIN-4-FP-21")
    .replaces(NodonPilotWireCluster)
    .enum("pilot_wire_mode",NodonPilotWireMode,NODON_PILOT_WIRE_CLUSTER_ID,
        ClusterType.Server,1,EntityType.STANDARD,EntityPlatform.SELECT,False,True)
    .add_to_registry()
)
```

garc33, 2024-09-19:
I’m not sure the manufacturer override works for nodon device. 
Furthermore the name of manufacturer in signature is `Adeo` not `ADEO` on my setup. I don’t it is always the case or not.

anoril1979, 2024-09-29:
Hi! It took me a while but I tested the quirk right now.
Quirk execute with no issue. My warmers are well discovered and use the quirk. I guess that as a lot of attribute names changed.
Alas, I still have only switch, power on demand and total summed power.
Nothing about changing the pilot-wire setup from eco to something else.

I'm not sure the quirk was doing this, either. But I hoped :)

ikruglov, 2024-11-27:
FYI: https://github.com/zigpy/zha-device-handlers/pull/3364 got merged. It supports both NodOn and Adeo. The code should become part of the 2024.12 release. Unfortunately, the change comes with only ZHA attributes (one which changes via "Manage Zigbee Device") and not with enum in the HA interface due to concerns from maintainers.

dprslt, 2024-12-12:
To anyone interested in a temporary workaround based on the recent #3364 

Here is a template Select allowing you to change mode directly within HA.
Unfortunately, I do not found a way to read the attribute value yet. If anyone as a clue about how i can sync it with se template select state, i'll take it !

```
template:
  - select:
        name: "Mode seche serviette"
        icon: mdi:radiator
        unique_id: "select_mode_seche_serviette"
        options: >-
            {{
            [
                "Off",
                "Confort",
                "Eco",
                "Hors Gel",
                "Confort -1",
                "Confort -2"
            ]
            }}
        state: >-
            {% set cluster_value = state_attr('zha.d4:48:67:ff:fe:aa:33:e0', 'cluster_0xfc00_in_0x0000')|default('0') %}
            {% set modes = {
            '0x00': 'Off',
            '0x01': 'Confort',
            '0x02': 'Eco',
            '0x03': 'Hors Gel',
            '0x04': 'Confort -1',
            '0x05': 'Confort -2'
            } %}
            {{ modes[cluster_value|string] }}
        select_option:
            - service: zha.set_zigbee_cluster_attribute
              data:
                cluster_type: "in"
                endpoint_id: 1
                cluster_id: 64512
                ieee: "d4:48:67:ff:fe:aa:33:e0"
                attribute: 0
                value: >-
                    {% set modes = {
                    'Off': 'NodOnPilotWireMode.Off',
                    'Confort': 'NodOnPilotWireMode.Comfort',
                    'Eco': 'NodOnPilotWireMode.Eco',
                    'Hors Gel': 'NodOnPilotWireMode.FrostProtection',
                    'Confort -1': 'NodOnPilotWireMode.ComfortMinus1',
                    'Confort -2': 'NodOnPilotWireMode.ComfortMinus2'
                    } %}
                    {{ modes[option] }}
```

dprslt, 2024-12-12:
To anyone interested in a temporary workaround based on the recent https://github.com/zigpy/zha-device-handlers/pull/3364

Here is a template Select allowing you to change mode directly within HA.
Unfortunately, I do not found a way to read the attribute value yet. If anyone as a clue about how i can sync it with se template select state, i'll take it !

```
template:
  - select:
        name: "Mode seche serviette"
        icon: mdi:radiator
        unique_id: "select_mode_seche_serviette"
        options: >-
            {{
            [
                "Off",
                "Confort",
                "Eco",
                "Hors Gel",
                "Confort -1",
                "Confort -2"
            ]
            }}
        state: ""
        select_option:
            - service: zha.set_zigbee_cluster_attribute
              data:
                cluster_type: "in"
                endpoint_id: 1
                cluster_id: 64512
                ieee: "<device-address>"
                attribute: 0
                value: >-
                    {% set modes = {
                    'Off': 'NodOnPilotWireMode.Off',
                    'Confort': 'NodOnPilotWireMode.Comfort',
                    'Eco': 'NodOnPilotWireMode.Eco',
                    'Hors Gel': 'NodOnPilotWireMode.FrostProtection',
                    'Confort -1': 'NodOnPilotWireMode.ComfortMinus1',
                    'Confort -2': 'NodOnPilotWireMode.ComfortMinus2'
                    } %}
                    {{ modes[option] }}
```

romain563, 2025-08-07:
Works fine for me since few updates. Thanks

closed 2026-02-11 by github-actions[bot]