zigpy/zha-device-handlers #2781: [BUG] INNR SP 240 "summation delivered" kWh value is 100x expected value

rrozema, 2023-11-28:
### Bug description

Upon inclusion of the INNR SP 240 in ZHA, among others a "Summation delivered" sensor gets created. The value returned for this sensor is however 100x larger than expected: I ran a filament dryer for 5 hours of the SP 240 and the device measured 17kWh instead of the expected 0,17 kWh as measured using a Fibaro wall plug V2.

When I requested help from INNR support they responded that ZHA probably has incorrectly interpreted the configuration their device returned. My translation of their response: "An attribute "SummationFormatting" on the Metering cluster has the value 0x2A, which designates: 2 digits to the right of the decimal point, 5 digits to the left of the decimal point, don't suppress leading zeroes". Looks like his system (red: my ZHA integration in HA)) doesn't interpret or incorrectly interprets the value of the "SummationFormatting" attribute." 

The full mail text I got from INNR support (in Dutch, sorry):

> Beste Richard,
> 
>  
> 
> (This message is in Dutch because I assume based on your name and e-mail address that you also speak Dutch.)
> 
> Ik heb de vraag even bij één van onze engineers gelegd, en dit was zijn reactie:
> 
> Wat betreft Watt meting die op 0 decimalen wordt afgerond:
> 
> De waarde wordt opgeslagen en gecommuniceerd in een “int16” type, wat een bereik heeft van -32768 tot 32767. Als we op 1 decimaal zouden afronden zou het maximum slechts 3276.7 W zijn, maar de plug kan 3680 W aan dus de keuze was simpel. De afronding dynamisch maken (meer decimalen bij lagere waarden) is geen optie want dat ondersteunen de meeste systemen niet (voor zover ik weet geen enkel systeem).
> 
>  
> 
> Wat betreft de kWh formatting: het “SummationFormatting” attribuut van het Metering cluster heeft waarde 0x2A, wat betekent: 2 digits to the right of the decimal point, 5 digits to the left of the decimal point, don’t suppress leading zeroes. Lijkt er dus op dat zijn systeem de waarde van het “SummationFormatting” attribuut van het Metering cluster niet of verkeerd interpreteert.
> 
>  
> 
> Hopelijk heb je iets aan deze uitleg.
> 
>  
> 
> Met vriendelijke groet,
> 
>  
> 
>  
> 
> **REDACTED**
> 
> Innr Servicedesk 
> Innr Lighting | Innr.com  
>  
> Innr Lighting BV  |  Catharina van Renneslaan 20  |  1217 CX Hilversum
> 

@TheJulianJES requested me to tag him in here.

### Steps to reproduce

Include the INNR SP 240 into the network using ZHA,
plug in a filament dryer of 145W,
Run the dryer for 5 hours, Watt history graph shows a peak at the beginning, then dropping to roughly 30 Watt.
Multiply the measured Watt value with the number of hours: 5 * roughly 30 Watt = roughly 170 Wh, or 0,17 kWh.
See how the actual sensor in ZHA however shows 17kWh.

### Expected behavior

The summation delivered sensor should have shown a usage of 0,170 kWh

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

[image]

</details>
``
[image]

[... 822 lines cut ...]
        "in_clusters": {},
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

TheJulianJES, 2023-11-28:
This might really be a ZHA issue. If the measurement type is kWh (0), we do not use the `summation_formatting` attribute for some reason. We only use the multiplier and divider attributes in that case: [zha/sensor.py#L522-L531](https://github.com/home-assistant/core/blob/2c196baa7a548291b72283d6acf0d8e0090e545a/homeassistant/components/zha/sensor.py#L522-L531)

@Adminiuga If you're around, do you have any idea why you added [that check](https://github.com/home-assistant/core/blob/2c196baa7a548291b72283d6acf0d8e0090e545a/homeassistant/components/zha/sensor.py#L524-L525) in the [original PR](https://github.com/home-assistant/core/pull/56666)? I'm guessing that it's unexpected the formatter is still needed for kWh? Or did some devices perhaps provide something wrong?

---

This can be worked around with a quirk. I'm not sure if that's a good idea though, as the plug seems to provide `summation_formatting` correctly: `00101010` =>  2 digits to right of decimal point, 5 digits to left of decimal point.
Changing the ZHA code could break some other devices though. Especially the ones we have quirks for which set the multiplier / dividers right now.

.. but apparently we already use quirks for other Innr plugs though: https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/innr/innr_sp120_plug.py
I guess I'll get one ready for this plug too for now.

TheJulianJES, 2023-11-28:
@rrozema Please test [this file](https://raw.githubusercontent.com/zigpy/zha-device-handlers/fba96b883340a5e55926714d474002cf00eee1e9/zhaquirks/innr/innr_sp240_plug.py) as a [custom quirk](https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274).
The comment above is still relevant for the isuse though.

tube0013, 2023-11-29:
I think the US version of this plug the 234 may have the same issue. I have my LumenPNP on one and it's like a 3d printer without heaters and some small vacuum pumps when running.  I don't think it's put out 50 kWh during it's lifetime. 

[image]

Looks like the current quirk is just changing the voltage divisor: https://github.com/zigpy/zha-device-handlers/blob/d30dac2f3ffe99cc303690afb91513cb8836acc4/zhaquirks/innr/innr_sp234_plug.py#L34 I

If I query the Summation_Formatting attribute I get 4 (0x2A) as well.

I can try and add the fix you added for the 240 in the quirk for the 234 to see if the work around works as well.

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4454, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x010a",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04",
        "0x0b05",
        "0x1000",
        "0xfc57",
        "0xfc82"
      ],
      "output_clusters": [
        "0x0003",
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "innr",
  "model": "SP 234",
  "class": "zhaquirks.innr.innr_sp234_plug.SP234"
}
```

TheJulianJES, 2023-11-29:
@tube0013 I've pushed some changes to:
- https://github.com/zigpy/zha-device-handlers/pull/2782

Let me know if all your plug measurements are correct then.

To install the SP234 quirk easily as a custom quirk, use this:
<details>
<summary>SP234 custom quirk test code (click to exapnd)</summary>

```python
"""Innr SP 234 plug."""
from zigpy.profiles import zgp, zha
from zigpy.quirks import CustomCluster, CustomDevice
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    OnOff,
    Ota,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.homeautomation import Diagnostic, ElectricalMeasurement
from zigpy.zcl.clusters.lightlink import LightLink
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)

class ElectricalMeasurementCluster(CustomCluster, ElectricalMeasurement):
    """Fix divisor for AC power."""

    _CONSTANT_ATTRIBUTES = {ElectricalMeasurement.AttributeDefs.ac_power_divisor.id: 1}

class MeteringClusterInnr(CustomCluster, Metering):
    """Provide constant multiplier and divisor.

    The device supplies the summation_formatting attribute correctly, but ZHA doesn't use it for kWh at the moment.
    """

    _CONSTANT_ATTRIBUTES = {
        Metering.AttributeDefs.multiplier.id: 1,
        Metering.AttributeDefs.divisor.id: 100,
    }

class SP234(CustomDevice):
    """Innr SP 234 smart plug."""

    signature = {
        MODELS_INFO: [("innr", "SP 234")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    Metering.cluster_id,
                    ElectricalMeasurement.cluster_id,
                    Diagnostic.cluster_id,
                    LightLink.cluster_id,
                    0xFC57,
                    0xFC82,
                ],
                OUTPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            },
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    MeteringClusterInnr,
                    ElectricalMeasurementCluster,
                    Diagnostic.cluster_id,
                    LightLink.cluster_id,
                    0xFC57,
                    0xFC82,
                ],
                OUTPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            },
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
```

</details>

tube0013, 2023-11-29:
Thanks got it running now.. and reset the summation delivered to get a fresh summation with the updated quirk.

TheJulianJES, 2023-11-29:
Z2M also adds it for your plug ([code here](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/innr.ts)). They are doing some other things we're not doing too.

Can you check if the current also needs to be divided by 1000 for your plug, @tube0013?

tube0013, 2023-11-29:
Amperage seems okay to me:

[image]

ac_current_divisor is 1000:

[zha-e29ceebf7c284fe6a18d61622537f595-innr SP 234-bd4503f7c2ae9e99b0befd345d61ad65.json.txt](https://github.com/zigpy/zha-device-handlers/files/13495963/zha-e29ceebf7c284fe6a18d61622537f595-innr.SP.234-bd4503f7c2ae9e99b0befd345d61ad65.json.txt)

rrozema, 2023-11-29:
[image]
I confirm the quirk fixes my kWh sensor.

tube0013, 2023-11-29:
SP 234 quirk changes have been working well.

Interesting I missed that fact that innr had fw updates posted. so I guess the quick need to be adjusted for that... I'm pulling it down and will update. @TheJulianJES should we move the SP 234 to a new issue sorry fop combining it with the SP 240!

TheJulianJES, 2023-11-29:
Ah, the info with the older firmware versions is interesting. It explains why Z2M still sets those dividers manually.

https://github.com/zigpy/zha-device-handlers/pull/2782 will be included in 2023.12.0b0 and should fix the issue even for older firmware versions of the SP 234, as it always sets the EM current and EM power divisors for the SP 234 now.
The total sum delivered (Metering) fix is also included for all Innr plugs (setting divisor, as ZHA ignores `summation_formatting` at the moment).

So, the behavior should be fixed in all upcoming 2023.12.x versions.
If there any remaining issues in 2023.12.x (beta should be out in about one to two hours or so), please open a new issue for other plugs and tag me.

tube0013, 2023-11-30:
My SP 234 was already on latest fw, so everything seems working with the modified one as I have been testing over last day or so.

thecoop, 2024-02-14:
This is also an issue for the SP 242

chonkusZZ, 2024-03-13:
> This is also an issue for the SP 242

I too am having the same issue with SP 242, FW 1.4.3.

I can't seem to even find this device on Innr's website anymore despite only recently purchasing them, but am exhibiting the same issue with the summation being much higher than it should be, assuming it is the same problem.

dmulcahey, 2024-03-13:
> The new plugs SP 240, SP 242, and SP 244 all use the same firmware code base so the same quirk can be applied. Just for completeness: it is not a bug of the smart plugs; the plugs correctly provide the SummationFormatting attribute 0x0303 with value 0x2A (meaning 2 digits to the right 5 to the left of the decimal point), which is the only mandatory attribute for formatting the CurrentSummationDelivered attribute. The Multiplier and Divisor attributes 0x0301 and 0x0302 are optional.

Hrm we should look at the HA side of this to make sure we are doing things correctly then…

TheJulianJES, 2024-03-13:
> Hrm we should look at the HA side of this to make sure we are doing things correctly then…

Yeah, I think there's an issue with ZHA at the moment, see this comment: https://github.com/zigpy/zha-device-handlers/issues/2781#issuecomment-1830832448 (formatter isn't used for kWh)
Even if we fix ZHA code, we risk breaking multiple devices with existing quirks. Some devices report it correctly and others not, so we don't know which quirks should stay and which should be removed (without knowing the attributes reported by the device).

exalyte, 2024-07-06:
Sorry to add to an old bug but it seems the newer innr 242 has the same bug summation is 100x expected 

Firmware reported by zha is 0x14033682 

Please let me know if this should it's own ticket

EXA

epd5, 2024-09-21:
Loading the SP 240 .py for my SP 242 worked perfectly. Thanks for providing this.

UltimateGG, 2024-11-20:
Having this on SP 244. Can we get the quirk to match those newer models too? I assume it's just not matching because the model is different

pajtaz, 2025-03-22:
This is still an issue with SP 240 with latest firmware 1.9.30. I tried the custom quirk but it is not working.

IWR69, 2025-04-25:
Also still an issue with SP 242 running 1.9.30

josephnfoley, 2025-05-15:
I was struggling with this all day as I have 4 of these plugs. I am using Zigbee2Mqtt which may or may not help but in my case I have found that on the previous firmware 1.7.23, the summaFormatting was set to 42 which is aparently correct however the multiplier and divisor attributes were not configured and when I tried to set their values I got an error saying they were not supported. When I pushed an OTA update to 1.9.3 I found a few things. First the summaFormatting is now set to 51 which shows that they have changed something in the energy formatting, also the multiplier and divisor are now there and set to 1 and 1000 respectively though these are read only. Also the current now reads 1000x what it should be. To resolve the current issue I went to the Settings (Specific) tab and set the Current Calibration to 0.001 which correctly calculates the current draw. I also set the Energy Precision to 2 to give me 2 decimal places. Now everything seems to be resolved for now.

TLDR: Pushed OTA update to plug to Version 1.9.3, set Current Calibration to 0.001 and Energy Precision to 2.

bazcurtis, 2025-09-07:
The Innr "fix" mentioned above made mine even worse. I am not sure how to reverse it. I probably should have tested one of them and not done both at the same time 😀

You can see the before and after in these screenshots.

[image]
[image]
[image]
[image]

puddly, 2025-09-07:
@bazcurtis can you upload diagnostics for the device after the update?

bazcurtis, 2025-09-08:
> [@bazcurtis](https://github.com/bazcurtis) can you upload diagnostics for the device after the update?

I am very happy to do that. Do I need to install the quirk first?

Byh0ki, 2025-09-29:
Hello @TheJulianJES ,

I've setup the diagnostic quirk on 2 devices with different versions, here are the info for both of them:

```
SP-240_1:
  sw_build_id: 1.9.34
  FW: 0x19223685
  current_file_version: 421672581
```
Diagnostic file: [zha-01JYF9S0ZD4W66030GEFDNEADM-innr SP 240-b757a5773128d03ebe54222e624d5516.json](https://github.com/user-attachments/files/22604752/zha-01JYF9S0ZD4W66030GEFDNEADM-innr.SP.240-b757a5773128d03ebe54222e624d5516.json)

```
SP-240_2:
  sw_build_id: 1.9.30
  FW: 0x191e3685
  current_file_version: 421410437
```

Diagnostic file: [zha-01JYF9S0ZD4W66030GEFDNEADM-innr SP 240-dcd8deb59750f28160ffbb010b23e905.json](https://github.com/user-attachments/files/22604762/zha-01JYF9S0ZD4W66030GEFDNEADM-innr.SP.240-dcd8deb59750f28160ffbb010b23e905.json)

From what I can see, the reported values are good for both of the sockets. So it would confirm that the version >= 1.9.30 no longer need the quirks to report the proper values.

AllardKatan, 2025-10-22:
I had noticed for a while that my Innr SP240 plugs were reporting too high energy values, too much to be true, while the power readings made sense with what my total home energy was reporting. I first thought this was due to the integration, and actually set up an integral sensor on the reported power to fix this. By comparing the integral sensor and the reported 'summation delivered' I found out that there is exactly a factor 10 difference by downloading the history of both entities and plotting them together.  My firmware version is 0x191d3685 accordign to the device info page, or 1.9.29 according to sw_build_id cluster attribute . If I check the 'MeteringClusterInnr' and read the 'summation_formatting' attribute, I get the following:
`NumberFormatting<0x33>(num_digits_right_of_decimal=3, num_digits_left_of_decimal=6, suppress_leading_zeros=0`

I also did debug logging of the ZHA events. I found the following lines: 
```2025-10-22 16:58:29.517 DEBUG (Thread-4) [aiosqlite] executing functools.partial(<built-in method execute of sqlite3.Connection object at 0x7fe208d7aa70>, '\n            INSERT INTO attributes_cache_v13\n            VALUES (:ieee, :endpoint_id, :cluster_type, :cluster_id, :attr_id, :value, :timestamp)\n                ON CONFLICT (ieee, endpoint_id, cluster_type, cluster_id, attr_id) DO UPDATE\n                SET value=excluded.value, last_updated=excluded.last_updated\n                WHERE\n                    value != excluded.value\n                    OR :timestamp - last_updated > :min_update_delta\n            ', {'ieee': a4:c1:38:57:0f:ad:1a:f4, 'endpoint_id': 1, 'cluster_type': <ClusterType.Server: 0>, 'cluster_id': 1794, 'attr_id': 0, 'value': 111770, 'timestamp': 1761145109.500436, 'min_update_delta': 30.0}) ```

The 'value'of 111770 is translated to HA as 1,117.700 kWh for this plug, as visible on the device info page. That is not consistent with the number formatting attribute, which states that there should be three digits to the right of the decimal, so .770. The 1,117.7 value is a factor 10 too high, I am sure of that. So whatever is happening to translate the value to kWh, it needs a fix.

GerryInnr, 2025-11-17:
Hi guys,

The Innr plugs firmware history regarding the kWh (CurrentSummationDelivered) reading is as follows:
- FW < 1.9.27: divisor and multiplier attributes of Metering cluster not supported, CurrentSummationFormatting = 0x24 (5 digits left of decimal, 2 right).
- FW >= 1.9.27: divisor and multiplier attributes of Metering cluster supported and set to resp. 1000 and 1, CurrentSummationFormatting = 0x33 (6 digits left of decimal, 3 right).

So starting from firmware version 1.9.27 (hexadecimal 0x191b3685) reporting is in Wh (1/1000th of kWh) and divisor and multiplier are supported (they are optional in Zigbee while CurrentSummationFormatting is mandatory, which is why we didn't support the divisor and multiplier at first).

@tlinnet your plug has version 0x19223685 (1.9.34) but your debug output shows "MeteringClusterInnr: divisor: 100". That must be coming from the quirk, overriding the correct value from the plug which is 1000 (the quirk was there because the divisor was not supported yet). Hence the 10x factor of your readings in the UI. The quirk must be corrected or maybe not used for plugs with FW >= 1.9.27.

Cheers,
Gerry.

GerryInnr, 2025-11-17:
Hi @AllardKatan you summarized it correctly. In all our firmware versions, CurrentSummationFormatting correctly specifies the formatting of CurrentSummationDelivered, and will continue to do so in the future. We don't foresee further changes though, it will remain 6.3 digits for the foreseeable future.
While CurrentSummationFormatting is mandatory and the divisor and multiplier optional, more systems not only HA use the divisor and multiplier, and many or most smart plugs also use divisor and multiplier, probably because those are also used in the Electrical Measurement cluster for voltage, current, and power.

I'm not sure how HA works exactly but assuming that currently:

1. Unless a quirk is active that sets other values,
2. If the device does not provide a divisor and multiplier for CurrentSummationDelivered, they are assumed to be 1.

Then a good change could be:

1. Unless a quirk is active that sets other values,
2. If the device does not provide a divisor and multiplier for CurrentSummationDelivered, check CurrentSummationFormatting and set divisor and multiplier accordingly, and if the device does not provide CurrentSummationFormatting, assume divisor and multiplier 1.

With that change in HA, the quirk for SP 24x and OSP 240 smart plugs could be removed and it would work correctly for all firmware versions. Giving preference to the divisor and multiplier over CurrentSummationFormatting will very likely prevent issues for other smart plugs (which may have correct divisor and multiplier but have the mandatory CurrentSummationFormatting set to some incorrect default value).

AllardKatan, 2025-11-17:
Oh, yes indeed with that table it looks like divisor and multiplication are the sole determinants of the final value. Thanks for the clarification, because the wording is rather confusing.

GerryInnr, 2025-11-18:
Hi @puddly and @AllardKatan that table is also in the Zigbee Cluster Library R8 spec. The start of the section with the table says:
"The following set of attributes provides **_the ratios and formatting_** hints required to transform the received summations, consumptions, temperatures, or demands/ rates into displayable values. **_If the Multiplier and Divisor attribute values are non-zero_**, they are used in conjunction with the SummationFormatting, ConsumptionFormatting, DemandFormatting, and TemperatureFormatting attributes."

That is a strong suggestion that if the Multiplier and Divisor attribute values are zero (or not supported; remember that they are optional while CurrentSummationFormatting is mandatory), the Formatting attributes determine the ratios of the raw values. The table only explains how it works if there are non-zero Divisor and Multiplier.

Anyway, I agree that the spec is unclear on this point (and in many other places).

TheJulianJES, 2026-01-07:
For a status update on this issue for the SP 240, see the latest post in the other issue [here](https://github.com/zigpy/zha-device-handlers/issues/3857#issuecomment-3716999928).
TL;DR: This should be fixed in Home Assistant Core 2026.1.0, releasing later today/tomorrow.