zigpy/zha-device-handlers #2783: [BUG] Ewelink zb-sw01 switch identifies as light

AllardKatan, 2023-11-28:
### Bug description

I added an Ewelink zb-sw01 device, which is aa zigbee connected single relay, to my Zha network. It showed up in Home Assistant as a light. I think a switch would be more appropriate.

### Steps to reproduce

1 go to 'Add device' in ZHA
2 If device is in pairing mode, it pairs and shows up as a light. 

(that is basically all there is to it) 

### Expected behavior

I expect the domain to be a switch

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=0, maximum_buffer_size=80, maximum_incoming_transfer_size=160, server_mask=0, maximum_outgoing_transfer_size=160, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "output_clusters": [
        "0x0000"
      ]
    }
  },
  "manufacturer": "eWeLink",
  "model": "ZB-SW01",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
   "home_assistant": {
      "installation_type": "Home Assistant OS",
      "version": "2023.11.3",
      "dev": false,
      "hassio": true,
      "virtualenv": false,
      "python_version": "3.11.6",
      "docker": true,
      "arch": "x86_64",
      "timezone": "Europe/Amsterdam",
[... 292 lines cut ...]
            },
            "out_clusters": {
               "0x0000": {
                  "endpoint_attribute": "basic",
                  "attributes": {},
                  "unsupported_attributes": {}
               }
            }
         }
      }
   }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

TheJulianJES, 2023-11-29:
See https://www.home-assistant.io/integrations/zha/#modifying-the-device-type for fixing this yourself.

Add the following to your config (replace `IEEE` with the IEEE of the affected Aqara switch):
```yaml
zha:
  device_config:
    IEEE-1:  # format: ieee-endpointId
      type: "switch"  # corrected device type
```

We could also change this with a quirk, but this would break the implementation for everyone using it as a light.
As this happens with a variety of devices, it's best not to add a quirk changing the device type for everything for now.

AllardKatan, 2023-11-29:
The device is a switch, not a light. It's literally a relay with a control voltage that is commanded through a  Zigbee radio and nothing else. You can use it as a light switch of course, but it will still be a switch. Why is it better to not correct this if you know it is wrong? Is it a matter of saving time and energy for more important things, or a deliberate choice to always accept the manufacturer transmitted device type even if it is not correct? I don't quite agree with the logic "people are already used to this being broken, so let's not fix it". 
That's just my opinion as a user of course. I know how to fix it in HA (in fact, I already had done this, I was just reporting this because I thought it would be helpful to the project). So if you decide to keep it like this I will just go on with my life :-)

closed 2024-06-03 by github-actions[bot]