zigpy/zha-device-handlers #2788: [BUG] An other TS011F with the wrong amps offset

kodiakastu, 2023-11-30:
### Bug description

Hi! I have a circuit breaker that reports it's an other TS011F. 

Reading a lot, I've seen many  reports of those devices that are being off by a factor of a 1000. I know that power should be good and voltage seems okay. I do not thrust the total tho. I feel, it should at least be divided by 10.

### Steps to reproduce

Simply add the device like any other.

### Expected behavior

Current seems to be reported in mA from the device. 

Summation delivered should be divided by 10

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x010a",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0402",
        "0x0702",
        "0x0b04",
        "0xe000",
        "0xe001"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_6l1pjfqe",
  "model": "TS011F",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information
[... 560 lines cut ...]
        "in_clusters": {},
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

kodiakastu, 2023-11-30:
Additional infos 2: I do not really understand at an 100% how it work, but reading from the debug file, the device doesn't seem to provide the offsets. Trying to set them in the GUI will always revert to none when trying to write then read the values at 0x0602 and 0x0603.

TheJulianJES, 2023-11-30:
Yeah, a quirk needs to be added. It should be relatively simple, as there are a bunch of quirks for these plugs already.
I hope I can take a look at this soon.

I'll send you a custom quirk to test then.

ogvalt, 2024-05-17:
I encountered the same issue with Tomzn TOB9Z-63M and solved it by adding custom quirk, but as fas as I could tell 
specifying 
```python
MODELS_INFO: [("_TZ3000_qeuvnohg", "TS011F")],
```
is not a good idea. 
If I understand correctly if `zigpy` encounteres several classes with same `MODEL` specified in a signature than it iteratates over them and select one that is most suitable for Descriptors provided by device. 
So in my case speciyfing `MODEL_INFO` actually hurts, since `Plug_CB_Metering` specifies my device signature exactly, but filters class because `MODEL_INFO (_TZ3000_qeuvnohg)` is different. 
Try deleting `MODEL_INFO` altogether

PR #3164 (2024-05-24): Remove manufacturer matching for Tuya plug TS011F variant
## Proposed change
Fixes #2788.
Tested on Tomzn TOB9Z-63M device

## Additional information

## Checklist

- [x] The changes are tested and work correctly
- [ ] `pre-commit` checks pass / the code has been formatted using Black
- [ ] Tests have been added to verify that the new code works
closed 2024-05-24 by TheJulianJES