zigpy/zha-device-handlers #2793: [Device Support Request] Aqara T2 Dual Relay (LLKZMK12LM / acn047) - Interlock support

pilot51, 2023-12-02:
### Problem description

I bought two Aqara T2's to control two roller shades. I thought I'd be able to toggle interlock mode (prevents both switches being on at the same time) through ZHA in Home Assistant, but so far I've been unable to find a way to do it even in the clusters/attributes under Manage Zigbee Device. I see no "interlock" attribute or anything that looks similar.

### Solution description

Interlock mode available for the Aqara T2 through ZHA in Home Assistant.

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4447, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0012",
        "0x0702",
        "0x0b04",
        "0xfcc0"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0012",
        "0xfcc0"
      ],
      "output_clusters": []
    },
    "21": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x000c"
      ],
      "output_clusters": []
    }
  },
  "manufacturer": "Aqara",
  "model": "lumi.switch.acn047",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[... 603 lines cut ...]
        "out_clusters": {
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

### Additional information

This is almost identical to #639 for another Aqara relay.

Interlock for the Aqara T2 is supported by Zigbee2MQTT: https://www.zigbee2mqtt.io/devices/LLKZMK12LM.html

dmulcahey, 2023-12-09:
No clue if this will work but you can try it as a custom quirk and let me know: 

```python
"""Aqara T2 relay device."""
from zigpy import types as t
from zigpy.profiles import zha
from zigpy.zcl.clusters.general import (
    Basic,
    Groups,
    Identify,
    MultistateInput,
    DeviceTemperature,
    OnOff,
    Ota,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.xiaomi import (
    AnalogInputCluster,
    BasicCluster,
    ElectricalMeasurementCluster,
    MeteringCluster,
    XiaomiAqaraE1Cluster,
    XiaomiCustomDevice,
)

class OppleCluster(XiaomiAqaraE1Cluster):
    """Opple cluster."""

    class SwitchType(t.enum8):
        """Switch type."""

        Toggle = 0x01
        Momentary = 0x02
        NoSwitch = 0x03

    class StartupOnOff(t.enum8):
        """Startup mode."""

        On = 0x00
        Previous = 0x01
        Off = 0x02
        Toggle = 0x03

    class DecoupledMode(t.enum8):
        """Decoupled mode."""

        Decoupled = 0x00
        ControlRelay = 0x01

    class Mode(t.enum8):
        """Mode."""

        Power = 0x00
        Pulse = 0x01
        Dry = 0x03

    attributes = {
        0x000A: ("switch_type", SwitchType, True),
        0x0517: ("startup_on_off", StartupOnOff, True),
        0x0200: ("decoupled_mode", DecoupledMode, True),
        0x02D0: ("interlock", t.Bool, True),
        0x0289: ("mode", Mode, True),
        0x00EB: ("pulse_length", t.uint16_t, True),
    }

class AqaraT2Relay(XiaomiCustomDevice):
    """Aqara T2 in-wall relay device."""

    signature = {
        MODELS_INFO: [("Aqara", "lumi.switch.acn047")],
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=0
            # device_version=1
            # input_clusters=[0, 3, 4, 5, 6, 18, 1794, 2820, 64704]
            # output_clusters=[10, 25]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Scenes.cluster_id,
                    Groups.cluster_id,
                    OnOff.cluster_id,
                    MultistateInput.cluster_id,
                    Metering.cluster_id,
                    ElectricalMeasurement.cluster_id,
                    OppleCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            # <SimpleDescriptor endpoint=2 profile=260 device_type=0
            # device_version=1
            # input_clusters=[3, 4, 5, 6, 18, 64704]
            # output_clusters=[]>
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Scenes.cluster_id,
                    Groups.cluster_id,
                    OnOff.cluster_id,
                    MultistateInput.cluster_id,
                    OppleCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
            # <SimpleDescriptor endpoint=21 profile=260 device_type=0
            # device_version=1
            # input_clusters=[12]
            # output_clusters=[]>
            21: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [AnalogInputCluster],
                OUTPUT_CLUSTERS: [],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    BasicCluster,
                    Identify.cluster_id,
                    Scenes.cluster_id,
                    Groups.cluster_id,
                    OnOff.cluster_id,
                    MultistateInput.cluster_id,
                    DeviceTemperature.cluster_id,
                    MeteringCluster,
                    ElectricalMeasurementCluster,
                    OppleCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Scenes.cluster_id,
                    Groups.cluster_id,
                    OnOff.cluster_id,
                    MultistateInput.cluster_id,
                    OppleCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
            21: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [AnalogInputCluster],
                OUTPUT_CLUSTERS: [],
            },
        },
    }
```

closed 2023-12-27 by dmulcahey

bsunderhus, 2024-12-02:
Is this already supported on ZHA? I just got an Aqara T2 that I want to use in interlock mode, is there any documentation on how to set this up? @pilot51  @dmulcahey 

I can see there's a `DecoupledMode` in my ZHA device configuration
[image]

Would that have anything to do with that?

luistelmocosta, 2024-12-16:
Hello. I have the exact same issue and I would like to know if ZHA currently supports it or if the only way is to use Z2M is the only way to set up the interlock mode for Aqara T2 devices.

pilot51, 2024-12-19:
@bsunderhus @luistelmocosta

1. In **Device Info**, open the menu to the right of **Reconfigure** and select **Manage zigbee device**.
2. In the **Clusters** dropdown, select the first **OppleCluster**.
3. In the **Attributes of the selected cluster** dropdown, select **interlock**.
4. Click **Read Attribute** and the **Value** should return `Bool.false`. If it says `None`, you're in the wrong **OppleCluster**, try the other one.
5. Change the value to `Bool.true` and click **Write Attribute**.
6. Test to confirm that attempting to turn on both switches causes the other one to turn off.

luistelmocosta, 2024-12-25:
Thank you. Its working as intended now <3

LunaeSomnia, 2025-02-16:
I'm finding some issues with the steps provided by @pilot51.

[image]

Thought it was a configure issue and I tried the 'Reconfigure' button, but then I found this out:

[image]

What does this mean??