zigpy/zha-device-handlers #2798: [Device Support Request] Schneider Electric Wiser CCT5010-0002W dimmer module

julienhd, 2023-12-03:
### Problem description

By default this dimmer module has a quite low range for dimming light (minimum brightness is quite high and brightness is roughly the same between 50% to 100%).

When paired to the Wiser gateway and the Wiser app, the following 3 configuration settings are proposed:
- Minimum brightness
- Maximum brightness
- Mode: RC LED (default) or RL LED

The 2 first ones allow to adjust the range of brightness to get a better behavior.

When paired with ZHA, these settings are not available, therefore the default values are used that lead to the default behavior described above when used with HA.

### Solution description

Implement the same configuration settings as the Wiser system in order to be able to adjust the range of brightness (and set the kind of LED):
- Minimum brightness
- Maximum brightness
- Mode: RC LED (default) or RL LED

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4190, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "3": {
      "profile_id": "0x0104",
      "device_type": "0x0101",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0301",
        "0x0b05"
      ],
      "output_clusters": [
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "Schneider Electric",
  "model": "PUCK/DIMMER/1",
  "class": "zigpy.device.Device"
}

</details>

### Diagnostic information

[... 383 lines cut ...]
    }
  }
}

</details>

### Logs

<details><summary>Logs</summary>

```logs
[Paste your logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

dflemstr, 2023-12-03:
I think this is the same device, so should be possible to port over from zigbee2mqtt: https://www.zigbee2mqtt.io/devices/WDE002961.html

JoNil, 2024-01-06:
I have a lot of  CCT5010-0002, i would be willing to test this if we find a solution

dflemstr, 2024-01-07:
Just for transparency, I haven't had time to do anything about this.  I'm not familiar with contributing a quirk so don't count on me for actually coding something up :)

julienhd, 2024-01-07:
Based on answers of @dflemstr I installed zha-toolkit via HACS and changed those settings through 
`developer-tools -> service -> 🧰 ZHA Toolkit: Write Attribute`
To summize:

For MinLevel (Minimum Brightness):

- Target Cluster: 769
- Attribute Id: 16
- Attribute Value: 1 to 254

For MaxLevel (Maximum Brightness):

- Target Cluster: 769
- Attribute Id: 17
- Attribute Value: 1 to 254

For Control Mode:

- Manufacturer Id: 4190
- Target Cluster: 769
- Attribute Id: 57344
- Attribute Value: 0 to 3 
-  > Automatic (0x00), RC Mode (0x01), RL Mode (0x02), or RL LED Mode (0x03) 

It's working well and this is sufficient for me, since it has to be done only one time per device.
If someone is willing to create a quirk to support it, it would be welcome (I will also try but never done that), but in the meantime this solution this is working fine.
Thank you @dflemstr !

prilly-dev, 2024-03-30:
I belive this is the same device as ELKO multiwire dimmer. 

Enhet informasjon
NHROTARY/UNIDIM/1
av Schneider Electric

closed 2024-10-03 by github-actions[bot]