zigpy/zha-device-handlers #2800: [BUG] IKEA Fyrtur reports double battery level after new update

vBrolin, 2023-12-05:
### Bug description

After update to version 24.4.13 the blinds reports double percentage.
[image]
[image]

### Steps to reproduce

Charge battery to more than 50%

### Expected behavior

Value need to be divided by 2

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

MattWestb, 2023-12-06:
The battery reporting i change in the updated firmware and its not need one quirk longer.
Deleting the device from the device card wait one minute (or restart HA) and adding it new and its working OK

TheJulianJES, 2024-02-03:
This issue should be fixed in HA Core 2024.2.0 (currently in beta).
Relevant PR:
- https://github.com/zigpy/zha-device-handlers/pull/2948

If you still see new "double battery percentage" reports, wait a day.
You can also try to press a button on the device or manually read the `sw_build_id` attriubte. Re-pairing/reconfiguring should also work.

If it's not fixed for you, please create a new issue and provide a new diagnostics file.

closed 2024-02-03 by TheJulianJES