zigpy/zha-device-handlers #2810: [Device Support Request] Aqara H1 Double Rocker No Neutral Switch (Xiaomi WS-EUK02/lumi.switch.l2aeu1)

bjarnekrottje, 2023-12-07:
### Problem description

Hi all,

Since the merge of #2784 (Thanks for that @TheJulianJES) in release 0.0.107 (which has now been merged into [Home Assistant version 2023.12.0](https://github.com/home-assistant/core/releases/tag/2023.12.0)), a new quirk has been added for the device with id `lumi.switch.l2aeu1`.

Unfortunately, it looks like not all functionality of the device is yet supported. Looking at the Zigbee2MQTT [list of supported functionalities for this device](https://www.zigbee2mqtt.io/devices/WS-EUK02.html) it looks like some of the functionalities appear to not be available in this new quirk, like:
- Power outage memory
- Flip indicator light
- LED disabled night
- Operation mode (for each rocker separate)
- Mode Switch
- Some Actions (`single_left`, `double_left`, `single_right` and `double_right` appear to be working, `single_both` and `double_both` appear to not yet be working. Also the `button_hold` event appears to be the same for both the left and right rocker, shouldn't these be different for each rocker?)

Please add a comment or correct something if I am missing functionalities or have incorrectly labeled them here.

### Solution description

I would like to see the quirk being updated to support the missing functionalities. I have tried looking into this myself, but unfortunately have not been successful yet.
Especially the Operation mode (Decoupled or Coupled/Control_relay) functionality is important to me currently.

It does look like (some of) the functionalities have already been implemented.

**For example:**
The quirk the device currently uses ([`zhaquirks.xiaomi.aqara.switch_h1_double.AqaraH1DoubleRockerSwitchNoNeutralAlt`](https://github.com/zigpy/zha-device-handlers/blob/aeca76da0ea81ecba5b17e0f7314620cd686faae/zhaquirks/xiaomi/aqara/switch_h1_double.py#L150C5-L150C5)) has a base reference to [`XiaomiOpple2ButtonSwitchBase`](https://github.com/zigpy/zha-device-handlers/blob/aeca76da0ea81ecba5b17e0f7314620cd686faae/zhaquirks/xiaomi/aqara/opple_switch.py#L121), which itself has an endpoint of 1 and 2 with the input_cluster [`OppleSwitchCluster`](https://github.com/zigpy/zha-device-handlers/blob/aeca76da0ea81ecba5b17e0f7314620cd686faae/zhaquirks/xiaomi/aqara/opple_switch.py#L88). This input cluster contains a lot of the missing functionalities (like operation mode and switch type).
I am wondering why they aren't showing up in the device when using this quirk then. I have yet to be able to determine the cause for this, maybe someone could clarify it.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>
Current functionality (since Home Assistant version 2023.12.0):
[image]
</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x0000",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0009",
        "0x0012",
        "0x0702",
        "0x0b04",
        "0xfcc0"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
[... 1059 lines cut ...]
2023-12-07 19:52:18.125 DEBUG (MainThread) [zigpy.zcl] [0x0F3A:2:0x0006] Received command 0x0A (TSN 243): Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=Bool, value=<Bool.false: 0>)), Attribute(attrid=0x00F5, value=TypeValue(type=uint32_t, value=50331392))])
2023-12-07 19:52:18.125 DEBUG (MainThread) [zigpy.zcl] [0x0F3A:2:0x0006] Attribute report received: on_off=<Bool.false: 0>, 0x00F5=50331392
2023-12-07 19:52:18.126 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received incomingRouteRecordHandler: [0x0f3a, 54:ef:44:10:00:90:56:5c, 136, -66, [0xd217]]
2023-12-07 19:52:18.126 DEBUG (MainThread) [bellows.zigbee.application] Received incomingRouteRecordHandler frame with [0x0f3a, 54:ef:44:10:00:90:56:5c, 136, -66, [0xd217]]
2023-12-07 19:52:18.126 DEBUG (MainThread) [bellows.zigbee.application] Processing route record request: (0x0f3a, 54:ef:44:10:00:90:56:5c, 136, -66, [0xd217])
2023-12-07 19:52:18.127 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received incomingMessageHandler: [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=6, sourceEndpoint=2, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=27), 124, -69, 0x0f3a, 255, 255, b'\x18\xf3\n\x00\x00\x10\x00\xf5\x00#\x00\xff\xff\x02']
2023-12-07 19:52:18.127 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=6, sourceEndpoint=2, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=27), 124, -69, 0x0f3a, 255, 255, b'\x18\xf3\n\x00\x00\x10\x00\xf5\x00#\x00\xff\xff\x02']
2023-12-07 19:52:18.128 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(timestamp=datetime.datetime(2023, 12, 7, 18, 52, 18, 128427, tzinfo=datetime.timezone.utc), src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0F3A), src_ep=2, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=27, profile_id=260, cluster_id=6, data=Serialized[b'\x18\xf3\n\x00\x00\x10\x00\xf5\x00#\x00\xff\xff\x02'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=124, rssi=-69)
2023-12-07 19:52:18.129 DEBUG (MainThread) [zigpy.zcl] [0x0F3A:2:0x0006] Received ZCL frame: b'\x18\xf3\n\x00\x00\x10\x00\xf5\x00#\x00\xff\xff\x02'
2023-12-07 19:52:18.130 DEBUG (MainThread) [zigpy.zcl] [0x0F3A:2:0x0006] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=243, command_id=10, *direction=<Direction.Client_to_Server: 1>)
2023-12-07 19:52:18.131 DEBUG (MainThread) [zigpy.zcl] [0x0F3A:2:0x0006] Decoded ZCL frame: OnOffCluster:Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=Bool, value=<Bool.false: 0>)), Attribute(attrid=0x00F5, value=TypeValue(type=uint32_t, value=50331392))])
2023-12-07 19:52:18.132 DEBUG (MainThread) [zigpy.zcl] [0x0F3A:2:0x0006] Received command 0x0A (TSN 243): Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=Bool, value=<Bool.false: 0>)), Attribute(attrid=0x00F5, value=TypeValue(type=uint32_t, value=50331392))])
2023-12-07 19:52:18.133 DEBUG (MainThread) [zigpy.zcl] [0x0F3A:2:0x0006] Attribute report received: on_off=<Bool.false: 0>, 0x00F5=50331392
2023-12-07 19:52:18.134 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'37ffa1a92a2a60b87e'
2023-12-07 19:52:18.134 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8430fc7e'
2023-12-07 19:52:18.146 DEBUG (MainThread) [zigpy.appdb] Error handling '_save_attribute' event with (54:ef:44:10:00:90:56:5c, 1, 0, 5, 'lumi.switch.l2aeu1', datetime.datetime(2023, 12, 7, 18, 52, 16, 639847, tzinfo=datetime.timezone.utc)) params: FOREIGN KEY constraint failed
2023-12-07 19:52:18.146 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received setExtendedTimeout: []
```
If you need any more (or different) logs, please let me know. Happy to provide them.
</details>

### Additional information

I hope you can help me out figuring out how we can add these functionalities. I am willing to help create/improve the quirk, although I have been unsuccessful thus far.
Thank you very much in advance for your help.

TheJulianJES, 2023-12-07:
Did you check the "Manage Zigbee device" -> "clusters" dialog? (Three dots on the device page)

TheJulianJES, 2023-12-10:
> Unfortunately you are not able at the moment to change this through the UI

You can change it on that page. Change the upper "value" field to `0` or `1` for example. Leave the lower text field empty. Then, press "write attribute". Afterwards, you can press "read attribute" to see if it correctly wrote that value.

But yes, ZHA config entities should be added for these custom attributes.

bjarnekrottje, 2023-12-10:
> Change the upper "value" field to `0` or `1` for example.
Thank you very much for this tip. I tried entering `OppleOperationMode.Coupled` or `OppleOperationMode.Decoupled` and then pressing "write attribute", but that didn't work. Using `0` and `1` did the trick!

> But yes, ZHA config entities should be added for these custom attributes. (just not implemented yet)
If that's something I could help with, please let me know. Happy to help and contribute to this, just not sure where to start to achieve this at the moment.

Archomeda, 2023-12-13:
Just a quick related question, is the l1aeu1 also supposed to support the same features as the l2aeu1? In HA I only see the changes for my l2aeu1 device, and my l1aeu1 devices have not changed.

Hedda, 2023-12-20:
FYI, there is an indirectly inconsistency issue where the Aqara Wall Switch single and double rocker have different device type:

https://github.com/zigpy/zha-device-handlers/discussions/1957

There also seems to be a related bug that prevents users modifying the device type in ZHA without using a custom quirk:

https://github.com/home-assistant/core/issues/101093

joker11940, 2024-10-07:
This is not solved just yet, still looking for now to see "on power on, use last state" anyone done this?

pokytom, 2024-11-04:
> This is not solved just yet, still looking for now to see "on power on, use last state" anyone done this?

I'm having the same issue. Which cluster can I use to toggle power off memory status?

bjarnekrottje, 2024-11-06:
> > > Change the upper "value" field to `0` or `1` for example.
> > 
> > 
> > Thank you very much for this tip. I tried entering `OppleOperationMode.Coupled` or `OppleOperationMode.Decoupled` and then pressing "write attribute", but that didn't work. Using `0` and `1` did the trick!
> > > But yes, ZHA config entities should be added for these custom attributes. (just not implemented yet)
> > 
> > 
> > If that's something I could help with, please let me know. Happy to help and contribute to this, just not sure where to start to achieve this at the moment.
> 
> I have been testing the double rocker no neutral version about the decoupled mode as that was a deal breaker for my use case
> 
> Did you find any way to set the switch to be on (rather than off) by default?

Hi @joker11940,

Didn't see your reply in my notifications. The way I temporarily 'fixed' it for now is to check on a button press (which throws an event in Home Assistant) if the switch is set to off. If that's the case, it will turn on the switch. This way it makes sure that even after a power outage, the switch gets turned on even though the switch is in decoupled mode. 
Hope this helps you out temporarily at least.

joker11940, 2024-12-13:
> Hi @joker11940,
>
> Didn't see your reply in my notifications. The way I temporarily 'fixed' it for now is to check on a button press (which throws an event in Home Assistant) if the switch is set to [...]

Oh that's not a bad work around, I may use that for some other switches. My main issue going this route is its powering smart bulbs that are Zigbee routers, when they go off the longer they are off the worse my network gets 😅

I do think I have found something that kind of does what I need and that is changing the following option to True.

My understanding is it will remember that it was on and turn turn back on. Any ideas if I am right? Not had a power cut yet 😅

[image]

t-animal, 2025-01-10:
Hey @bjarnekrottje

I'm trying to set decoupled mode. When I run the action you've used in your script, I get "unknown error" and weirdly, the endpoint 1 doesn't even have that cluster id and the cluster  for endpoint 2 doesn't have any attributes. Do you have an idea why? I'm not too familiar with low-level zigbee, unfortunately. [image]

joker11940, 2025-01-15:
> Hey @bjarnekrottje
> 
> I'm trying to set decoupled mode. When I run the action you've used in your script, I get "unknown error" and weirdly, the endpoint 1 doesn't even have that cluster id and the cluster for endpoint 2 doesn't have any attributes. Do you have an idea why? I'm not too familiar with low-level zigbee, unfortunately.
> 
> ![image](https://private-user-images.githubusercontent.com/1775745/401962508-90b3e349-6e96-4732-be8a-040654c4eb14.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY5NDAwMzYsIm5iZiI6MTczNjkzOTczNiwicGF0aCI6Ii8xNzc1NzQ1LzQwMTk2MjUwOC05MGIzZTM0OS02ZTk2LTQ3MzItYmU4YS0wNDA2NTRjNGViMTQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDExNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTAxMTVUMTExNTM2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZWI1YjgxZTlhNmM1NzUzMmJiNzJiODE2ZWZmMGJhODdmOGY5Zjg2NThiZTM4MzQxNzVjOGQ0YmY2ZmFlNjgwMyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.EU7FLW5_JtzjN-eLyQrFivjoR5StCPvrcpPg6ol3Gvw)
> 
> The log says
> 
> ```
> websocket_api script: Error executing script. Unexpected error for call_service at pos 1: Cluster 64704 not found on endpoint 1 while writing attribute 512 with value False
> websocket_api script: Error executing script. Unexpected error for call_service at pos 1: Cluster 64704 not found on endpoint 1 while writing attribute 512 with value 0
> ```
> 
> Is that a hardware or software issue?

Is that a Aqara H1 Double Rocker? Your list does look different to mine

t-animal, 2025-01-15:
Yes, it's a
> lumi.switch.l2aeu1
> von LUMI

joker11940, 2025-02-27:
> Yes, it's a
> 
> > lumi.switch.l2aeu1
> > von LUMI

That is odd. What firmware is it running?

t-animal, 2025-02-27:
I don't know... The home assistant dashboard shows "unknown".

[image]