zigpy/zha-device-handlers #2824: [BUG] XiaomiOpple2ButtonSwitchFace1 is no longer applied to Aquara lumi.switch.b2naus01

dschall, 2023-12-09:
### Bug description

I have a bunch of Aquara double rocker switches (and single rocker too, for that matter, but the single rocker never caused problems).
The double rockers identify as `lumi.switch.b2naus01`, and in the past, the quirk `zhaquirks.xiaomi.aqara.opple_switch.XiaomiOpple2ButtonSwitchFace1` had been applied to them.
After upgrading to HA 2023.12.1, the switches no longer worked and upon repairing them, the quirk is no longer applied.
Log excerpt:
```
Checking quirks for LUMI lumi.switch.b2naus01 (mac redacted)
Considering <class 'zhaquirks.xiaomi.aqara.opple_switch.XiaomiOpple2ButtonSwitchFace1'>
Fail because endpoint list mismatch: {1, 2, 41, 42, 242, 51, 21, 31} {1, 2, 242}
```

That issue had happened prior to 2023.12.1, too, and repairing them once or twice usually got the quirk applied.
But now, even after retrying 10+ times, the quirk is never applied.

### Steps to reproduce

Pair a lumi.switch.b2naus01 with HA.

### Expected behavior

The quirk zhaquirks.xiaomi.aqara.opple_switch.XiaomiOpple2ButtonSwitchFace1 is applied.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

TheJulianJES, 2023-12-09:
Please fill-out the issue template. Attach either the device diagnostics or the device signature.

dschall, 2023-12-09:
Hey Julian, sorry about that. I pasted the log from the device discovery.

TheJulianJES, 2023-12-09:
I see the device diagnostics now. Thanks!
However, it looks like the device signature changed. https://github.com/zigpy/zha-device-handlers/pull/2784 did do some changes, but didn't change `XiaomiOpple2ButtonSwitchFace1` for your device.

There are some possibilities:
- the device "randomly" changes its reported signature during each pairing (each removal + re-pairing might cause a different signature)
- the device changed its reported signature due to a coordinator firmware upgrade (which sends Xiaomi manufacturing codes during pairing now)
- the device had a firmware upgrade which changed its reported signature (after a re-pairing)
- it was not using `XiaomiOpple2ButtonSwitchFace1` before, but a slightly different quirk. Maybe `Face2`?

So, are you 100% sure it was using the `XiaomiOpple2ButtonSwitchFace1` quirk before, or could it have been a different quirk?

dschall, 2023-12-09:
Hey Julian,

it's not just a single switch. I have 5 of them all they all stopped working :/
I still have the device info from a "good" one that successfully paired ONCE after the upgrade.
I can't repro it now, and the other switches never worked.
However, that one time it worked, it was showing `zhaquirks.xiaomi.aqara.opple_switch.XiaomiOpple2ButtonSwitchFace1`, see below.

<details><summary>Good Device</summary>

```python
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.12.1",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.6",
    "docker": true,
    "arch": "x86_64",
    "timezone": "America/Los_Angeles",
    "os_name": "Linux",
    "os_version": "6.1.63-haos",
    "supervisor": "2023.11.6",
    "host_os": "Home Assistant OS 11.2",
    "docker_version": "24.0.7",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "opensprinkler": {
      "version": "1.2.5",
      "requirements": [
        "pyopensprinkler==0.7.11"
      ]
    },
    "browser_mod": {
      "version": "2.3.0",
      "requirements": []
    },
    "hacs": {
      "version": "1.33.0",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    },
    "zha_toolkit": {
      "version": "v1.1.8",
      "requirements": [
        "pytz"
      ]
    }
  },
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "after_dependencies": [
      "onboarding",
      "usb"
    ],
    "codeowners": [
      "@dmulcahey",
      "@adminiuga",
      "@puddly",
      "@TheJulianJES"
    ],
    "config_flow": true,
    "dependencies": [
      "file_upload"
    ],
[... 1088 lines cut ...]
          }
        },
        "out_clusters": {}
      },
      "242": {
        "device_type": {
          "name": "PROXY_BASIC",
          "id": 97
        },
        "profile_id": 41440,
        "in_clusters": {},
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

TheJulianJES, 2023-12-09:
> I still have the device info from a "good" one that successfully paired ONCE after the upgrade.

This is good to know. So, it seems like there's an issue with the device (seen in other Aqara devices too) where it doesn't report its full signature sometimes. I'll take a look at it adding the signature posted initially (without the quirk).

dschall, 2023-12-09:
Just to double check, can I see that quirk info somewhere in the backup?

TheJulianJES, 2023-12-09:
Maybe using `zigbee.db`. Can you download this paste as a .py file and test it as a [custom quirk](https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274): https://dpaste.org/TR6qu/raw

Path could also be `/homeassistant/custom_zha_quirks/` now (instead of `/config`, I'm not sure)

dschall, 2023-12-09:
I added the file (called it lumi.py) to /config/custom_zha_quirks, but it gives me an error:

Unexpected exception importing custom quirk 'lumi'
```python
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/zhaquirks/__init__.py", line 458, in setup
    spec.loader.exec_module(module)
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_zha_quirks/lumi.py", line 58, in <module>
    from .opple_remote import MultistateInputCluster, OppleCluster
ImportError: attempted relative import with no known parent package
```

TheJulianJES, 2023-12-09:
Ah, change that input in line 58 to `from zhaquirks.xiaomi.aqara.opple_remote import MultistateInputCluster, OppleCluster`

dschall, 2023-12-09:
Quirk is recognized, but doesn't work:
```python
Checking quirks for LUMI lumi.switch.b2naus01 (04:cf:8c:df:3c:76:31:1e)
Considering <class 'lumi.XiaomiOpple2ButtonSwitch2'>
Fail because input cluster mismatch on at least one endpoint
Considering <class 'lumi.XiaomiOpple2ButtonSwitch1'>
Fail because endpoint list mismatch: {1, 2, 41, 42, 242, 51, 21, 31} {1, 2, 242}
Considering <class 'zhaquirks.xiaomi.aqara.opple_switch.XiaomiOpple2ButtonSwitchFace1'>
Fail because endpoint list mismatch: {1, 2, 41, 42, 242, 51, 21, 31} {1, 2, 242}
Considering <class 'zhaquirks.xbee.xbee_io.XBeeSensor'>
Fail because endpoint list mismatch: {232, 230} {1, 2, 242}
Considering <class 'zhaquirks.xbee.xbee3_io.XBee3Sensor'>
Fail because endpoint list mismatch: {232, 230} {1, 2, 242}
Considering <class 'zhaquirks.tuya.ts0201.MoesTemperatureHumidtySensorWithScreen'>
Fail because endpoint list mismatch: {1} {1, 2, 242}
Considering <class 'zhaquirks.smartthings.tag_v4.SmartThingsTagV4'>
Fail because endpoint list mismatch: {1} {1, 2, 242}
Considering <class 'zhaquirks.smartthings.multi.SmartthingsMultiPurposeSensor'>
Fail because endpoint list mismatch: {1} {1, 2, 242}
Considering <class 'zhaquirks.netvox.z308e3ed.Z308E3ED'>
Fail because endpoint list mismatch: {1} {1, 2, 242}
Considering <class 'zhaquirks.gledopto.soposhgu10.SoposhGU10'>
Fail because endpoint list mismatch: {11, 13} {1, 2, 242}
```

TheJulianJES, 2023-12-09:
Can you attach the device diagnostics from that device (in a new post) here too? It must differ again from the one you provided in your OP (or I'm really missing something).

dschall, 2023-12-09:
This is the latest:
<summary></summary><details>

```python
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.12.1",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.6",
    "docker": true,
    "arch": "x86_64",
    "timezone": "America/Los_Angeles",
    "os_name": "Linux",
    "os_version": "6.1.63-haos",
    "supervisor": "2023.11.6",
    "host_os": "Home Assistant OS 11.2",
    "docker_version": "24.0.7",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "opensprinkler": {
      "version": "1.2.3",
      "requirements": [
        "pyopensprinkler==0.7.8"
      ]
    },
    "browser_mod": {
      "version": "2.3.0",
      "requirements": []
    },
    "hacs": {
      "version": "1.33.0",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    }
  },
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "after_dependencies": [
      "onboarding",
      "usb"
    ],
    "codeowners": [
      "@dmulcahey",
      "@adminiuga",
      "@puddly",
      "@TheJulianJES"
    ],
    "config_flow": true,
    "dependencies": [
      "file_upload"
    ],
    "documentation": "https://www.home-assistant.io/integrations/zha",
    "iot_class": "local_polling",
    "loggers": [
      "aiosqlite",
      "bellows",
      "crccheck",
      "pure_pcapy3",
      "zhaquirks",
      "zigpy",
      "zigpy_deconz",
      "zigpy_xbee",
      "zigpy_zigate",
[... 671 lines cut ...]
            }
          }
        },
        "out_clusters": {}
      },
      "242": {
        "device_type": {
          "name": "PROXY_BASIC",
          "id": 97
        },
        "profile_id": 41440,
        "in_clusters": {},
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```
</details>

TheJulianJES, 2023-12-09:
Yeah, that's different again. I'll add that too. I wonder how many "signature" combinations there are (and what causes the device to report a different one).

> edit: I may have posted a different switch in the first post, but they are all the same type. Aqara double rocker with neutral (for US market).

Yeah, as long as they work after adding them to a/this quirk, it's good. 👍

dschall, 2023-12-09:
> I wonder how many "signature" combinations there are

I wonder if something in ZHA changed that is exposing this misbehavior more agressively now.
As I said in the original post, even prior 2023.12, the quirk was occasionally not applied to them, but removing and adding the switch usually helped. Never had to do that more than twice for the same switch.

TheJulianJES, 2023-12-09:
Append this to your `.py` file:

```python
class XiaomiOpple2ButtonSwitch3(XiaomiOpple2ButtonSwitchBase):
    """Xiaomi Opple 2 Button Switch. Signature 3."""

    signature = {
        MODELS_INFO: [(LUMI, "lumi.switch.b2naus01")],
        ENDPOINTS: {
            # input_clusters=[0, 2, 3, 4, 5, 6, 9, 1794, 2820], output_clusters=[10, 25]
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    DeviceTemperature.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    Alarms.cluster_id,
                    Metering.cluster_id,
                    ElectricalMeasurement.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            },
            # input_clusters=[0, 3, 4, 5, 6], output_clusters=[]
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [
                    GreenPowerProxy.cluster_id,
                ],
            },
        },
    }
```

dschall, 2023-12-09:
This worked!
[image]

Let me check the other ones.
By the way, it's good to know that the quirks are apparently applied after a HA restart.
So no need to remove and add the Zigbee device to have new custom quirks applied..(?)

dschall, 2023-12-09:
Yep, all 5 switches have the quirk `XiaomiOpple2ButtonSwitch3` now, thank you so much, Julien!

TheJulianJES, 2023-12-09:
> By the way, it's good to know that the quirks are apparently applied after a HA restart.
So no need to remove and add the Zigbee device to have new custom quirks applied..(?)

Yep. Although firmware updates and other stuff might cause a device signature to change. ZHA only reads the device signature when it doesn't know it (so when it's paired as a fresh device).
For your case, restarting is enough.

> nit: One more thing: With the quirk, the switches are shown as "switch", not "light switch", can we change that? (It was even the case before 2023.12, so maybe I should file a separate bug report.)

This is an odd one. Generally, quirks shouldn't change the device type anymore. However, we're doing it for all Aqara devices at the moment, as they are all relays. Changing it back to lights would break other user setups, so we can't really do that.
However, you can change the type from switch to light using the HA UI (it'll create a new light entity and hide the switch one) using the entity settings. ("Switch as X")

Or, you can add this to your `configuration.yaml` with the device IEEE dash the endpoint you want to modify (1 and another entry for 2): https://www.home-assistant.io/integrations/zha/#modifying-the-device-type

> Yep, all 5 switches have the quirk XiaomiOpple2ButtonSwitch3 now
Nice! If you have some time, can you (remove) and re-pair one device a couple of times to see if it picks up an unquirked signature or `Switch1` or `Switch2` (instead of always 3?)
That way, we can make sure this is completely fixed.

dschall, 2023-12-09:
I tried re-pairing all 5, and two came back "wrong".
One did not get any quirk:
[no quirk.txt](https://github.com/zigpy/zha-device-handlers/files/13625730/no.quirk.txt)

And one got quirk `XiaomiOpple2ButtonSwitch2`, which, I guess, works fine, too:
[quirk 2.txt](https://github.com/zigpy/zha-device-handlers/files/13625731/quirk.2.txt)

dschall, 2023-12-09:
> However, you can change the type from switch to light using the HA UI (it'll create a new light entity and hide the switch one) using the entity settings. ("Switch as X")

Thanks, that's what I had been doing.
I was just curious, because the no-quirk version is showing me light switches. Only when a quirk is applied, it changes to generic switches. So I figured the quirks do incorrectly change the type of switch for these devices. But that's a minor issue.

TheJulianJES, 2023-12-09:
Append this (and try to see if you can find more unknown signatures):
```python
class XiaomiOpple2ButtonSwitch4(XiaomiOpple2ButtonSwitchBase):
    """Xiaomi Opple 2 Button Switch. Signature 4."""

    signature = {
        MODELS_INFO: [(LUMI, "lumi.switch.b2naus01")],
        ENDPOINTS: {
            # input_clusters=[0, 2, 3, 4, 5, 6, 18, 64704], output_clusters=[10, 25]
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    DeviceTemperature.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    MultistateInput.cluster_id,
                    OppleSwitchCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            },
            # input_clusters=[0, 3, 4, 5, 6, 18, 64704], output_clusters=[]
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    MultistateInput.cluster_id,
                    OppleSwitchCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [
                    GreenPowerProxy.cluster_id,
                ],
            },
        },
    }
```

dschall, 2023-12-09:
Do I have to restart HA to load the new quirks?

TheJulianJES, 2023-12-09:
I think so. You can also try just reloading the ZHA integration using the integrations page and see if that works.

TheJulianJES, 2023-12-09:
Created https://github.com/zigpy/zha-device-handlers/pull/2825 with the three additional signatures now.
Were you able to find more unquirked signatures or is everything working now (with Switch1, 2, 3, 4)?

dschall, 2023-12-09:
I had tried removing and adding all of them one time, and all work with `XiaomiOpple2ButtonSwitch3` except one, which uses `XiaomiOpple2ButtonSwitch2`.
So looks like we're good for now :)

Thanks again for your help!

TheJulianJES, 2023-12-09:
> I tried re-pairing all 5, and two came back "wrong".
One did not get any quirk:
[no quirk.txt](https://github.com/zigpy/zha-device-handlers/files/13625730/no.quirk.txt)

I've added `Switch4` for that "no quirk.txt". Just to confirm, what do you mean by "two came back 'wrong'.".
They just came back with a different signature, but still a quirked one?

dschall, 2023-12-09:
Oh, yeah, after adding `Switch4` to the quirks, I repaired the switch that didn't have a quirk and it came back as another `Switch3`.. fussy, these Aqara switches, aren't they? ;)

closed 2023-12-10 by TheJulianJES