zigpy/zha-device-handlers #2827: [Device Support Request]  Haozee 24G Presence Sensor (Tuya TS0601 _TZE204_kyhbrfyl)

anthony-spruyt, 2023-12-10:
### Problem description

I've added a new presence sensor (TS0601 by _TZE204_kyhbrfyl). Pairing was successful, but the device doesn't show any entities.

### Solution description

Please add support for [this](https://www.aliexpress.com/item/1005005946786561.html?spm=a2g0o.order_list.order_list_main.27.163018029MLyQm) 24G mmWave presence sensor from Haozee.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE204_kyhbrfyl",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.12.1",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.6",
    "docker": true,
[... 611 lines cut ...]
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.OCCUPANCY_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    HaozeeMmwRadarManufCluster,
                    HaozeeTuyaOccupancySensing,
                    HaozeeTuyaAnalogInput,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id, 
                    Ota.cluster_id
                ],
            }
        }
    }
```

</details>

anoril1979, 2024-01-02:
Interrested too! :)

anthony-spruyt, 2024-01-04:
My custom quirk had been working well in the interim, just haven't been able to change any config such as sensitivity etc.

anoril1979, 2024-02-02:
Thanks Anthony for the quirk. I installed your custom and device is well detected as Occupancy detector.
But for now, it detected nothing. :/

anoril1979, 2024-02-12:
Hey! I updated and now it works! Thanks a lot. (It's a pity we can't yet configure the settings, but it basically works for now).

nublemeduble, 2024-02-24:
I have the same sensor So yes, add me to the list of people who are interested. It works great with your custom quirk. It would be great if you can add the settings for the sensor

nublemeduble, 2024-02-25:
Hello everyone, I've now managed to get the controls to be displayed, but I don't think the sliders have any effect. I use this quirk. I use variant 4  

class TuyaMmwRadarOccupancyVariant4(CustomDevice):
[TS0601_radar.txt](https://github.com/zigpy/zha-device-handlers/files/14397193/TS0601_radar.txt)

The detection itself works

[image]

anoril1979, 2024-03-04:
Hello!
Just to say that after a HA/Pi reboot, I had a new Integration discovered, Hi-Link LD2410 BLE, with 1 device (HLK-LD2410_BC51) and 27 entities. This one is actually the Radar Presence sensor detrected as Bluetooth device! (and for some reason, the ZigBee is no more active...)

nublemeduble, 2024-03-05:
@anoril1979  Hm, that's strange because the sensor that is shown to you is not a Zigbee sensor but a Bluetooth sensor.
https://de.aliexpress.com/item/1005004775187154.html
Have you already found the solution?

anoril1979, 2024-03-05:
Yes, it is actually.

I have no other sensor arround me (to my knowledge) and the only one is the zigbee radio sensor. I suppose it embeds a ble one too.
I managed to get both working in my HA and they behave very similarily. (Presence and Movement, BLE looks faster BTW)
[image]

nublemeduble, 2024-03-05:
@anoril1979  Is it possible that you have this sensor? https://www.szneo.com/en/products/show.php?id=271
Because the normal one only has Zigbee
https://www.szneo.com/en/products/show.php?id=277

nublemeduble, 2024-03-05:
@anoril1979 You're right, I turned off bluetooth because I didn't need it. After I turned it on, it worked for me too. 
[image]

There is even a Bluetooth app in the Google Play Store for settings.
https://play.google.com/store/apps/details?id=com.hlk.hlkradartool

anoril1979, 2024-03-05:
That's weird, isn't it?! I disabled my Bluetooth integration for the same reasons, and I'm surprised the device use both Zigbee and BLE as it is not written in the device specs! (I bought it on Aliexpress: https://fr.aliexpress.com/item/1005005965164376.html is it explicitly sell as a Zigbee device, no mention of BLE).

Anyway, I suppose I will use it that way (as it has better performances).

It is almost the same as the one you shown earlier, but with no little window (PIR) on it. 
Anyway, Using the dedicated App is a great alternative until a working quirk is available! <3 (Note that I installed the app but it did not detected the device... I will try by resetting it)

I think we are anyway going further the initial topic of this thread :') Sorry readers for the divergence.

steflrx, 2024-06-23:
Hello, many thanks for sharing the quirk, it save my lfe !

g1r1, 2024-07-09:
This quirk worked and the device is quite reliable. 

Is there a way to expose other values which are seen in the Bluetooth app like Static Target  / Moving Target or distance from target? Also is there a way of switching off bluetooth radio on this device?

anthony-spruyt, 2025-01-05:
Interesting findings about the Bluetooth support! Still would like to be able to get this supported without a custom quirk.

anthony-spruyt, 2025-01-06:
I replaced my custom quirk with prairiesnpr's and it works great! Finally able to set sensitivity, range and get more detailed detection info. Thank you very much prairiesnpr!

closed 2025-01-16 by TheJulianJES