zigpy/zha-device-handlers #2828: [Device Support Request]

alexeiw123, 2023-12-10:
### Problem description

This switch is detected by ZHA without a custom quirk, but is missing some core settings functionality that is supported using the clipsal/Schneider zigbee hub.

as described [here](https://www.productinfo.schneider-electric.com/clipsal/wiser-iconic-connected-switch-10ax_device-user-guide_clp/English/Wiser%20Iconic%20Connected%20Switch%2010AX_Clipsal_Device%20user%20guide_0001066123.xml/$/LEDIndication_WSE_Iconicdimmer_CLP_PDLCPT_0001077114#LEDIndication-F1B6B586) the switch has a configurable backlit LED behind the button which can operate in one of four modes: Reverse with load, Consistent with load, Always on and Always off.

Additionally, there are also minimum and maximum ballast settings which allow configuration of the minimum and maximum brightness controls, which helps to configure for different lights that have different dimmer responses.

### Solution description

Add UI controls for "LED indicator mode", "ballast minimum" and "ballast maximum".

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4190, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "3": {
      "profile_id": "0x0104",
      "device_type": "0x0101",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0301",
        "0x0b05"
      ],
      "output_clusters": [
        "0x0019"
      ]
    },
    "21": {
      "profile_id": "0x0104",
      "device_type": "0x0104",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0b05",
        "0xff17"
      ],
      "output_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0102"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "Schneider Electric",
  "model": "CH/DIMMER/1",
  "class": "zigpy.device.Device"
}
```

[... 1797 lines cut ...]
</details>

[Z2MQTT support of these features](https://www.zigbee2mqtt.io/devices/41EPBDWCLMZ_354PBDMBTZ.html)
[Github discussion on adding support to Z2MQTT](https://github.com/Koenkk/zigbee-herdsman-converters/issues/5635)

My interpretation of the above is that the cluster "0xff17" as seen in the scan results is related to the LED indicator mode (code snippet below). The ZHA UI for this cluster does not show any attributes or commands.
```json
        "0xff17": {
          "cluster_id": "0xff17",
          "title": "Manufacturer Specific",
          "name": "manufacturer_specific",
          "attributes": {
            "0x0000": {
              "attribute_id": "0x0000",
              "attribute_name": "0",
              "value_type": [
                "0x30",
                "enum8",
                "Discrete"
              ]
```

[image]

I am more than happy to spend time testing and debugging but I must stress that I have next to no experience with ZHA and am totally lost when it comes to json, quirks, etc.

alexeiw123, 2023-12-12:
I have found that calling the following service using ZHA toolkit successfully changes the LED indicator mode:

```
service: zha_toolkit.attr_write
data:
  ieee: light.entityname
  endpoint: 21
  cluster: 0xFF17
  attribute: 0
  attr_val: 0
  manf: 0x105E
```

where `attr_val:` is:
0 = indicator is on when load is on, 1 = Indicator is always on, 2 = indicator is on when load is off, 3 = Indicator is always off

How can we expose these as a UI option for ZHA?

alexeiw123, 2023-12-12:
likewise, the values for min and maxi brightness can be adjusted by calling the following services and adjusting attr_val between 1 and 254:

Min level:
```
service: zha_toolkit.attr_write
data:
  ieee: light.entityname
  endpoint: 3
  cluster: 0x0301
  attribute: 0x0010
  attr_val: 1
```

Max level:
```
service: zha_toolkit.attr_write
data:
  ieee: light.entityname
  endpoint: 3
  cluster: 0x0301
  attribute: 0x0011
  attr_val: 254
```

meremortals70, 2024-06-09:
I think the bot should stay out of it😀

the controls are still needed for this device. Please add.

closed 2024-12-13 by github-actions[bot]