zigpy/zha-device-handlers #2830: [BUG] Tuya LCD temperature sensors' output erroneous  negative values

almostlunatic, 2023-12-12:
### Bug description

I didn't get any help from other forums but @buddly was so kind and guided me here, there's my final hope!

I have a lot of Tuya ZigBee TS0601 Temp/Hum LCD-display sensors installed in every room and several Sonoff sensors without LCD-display outside and inside. Until October, everything worked with a Tuya Smart app and their cloud.

In October I’ve got a Raspberry Pi 4B and installed HAOS onto it with Sonoff’s ZigBee 3.0 USB Dongle Plus. Again, all were ok with the sensors, until the temperatures in some rooms have felt below 0 degrees by Celsius in December. The Tuya LCD sensors started to show values like 6,551.5 C as a negative temperature. The displays themself on the devices show good values like -1.5. Those ones without LCD (Sonoff) show good values in HA even with -20.

I have installed the latest core and OS of HA but nothing changed. I have no access to the house, it stays alone and waits the spring, so no way to repair any device. But my heating (waterpipes!) is depending on the right temperature values, particularly from the negative ones!

Looking on the graphs, it seems the 6,55x represents the negative value. As on the 1st image, 6,553.4 should be -3.4 C. I followed the graph with the rising values 6,553.4 -> 6,552.1 → 6,551.7 → 6,551.5. These are certainly the negative temperature values: -3.4, -2.1, -1.7, -1.5. So, 6,55 is the "minus" degrees. On the second graph, the value +0.2 degrees C is positive and there is no 6,55x.xx before this value.

[image]

[image]

### Steps to reproduce

1. Add a Tuya TS0601 sensor to the ZHA integration.
2. Put the device in the refrigerator and wait when it becomes below the 0 degrees by Celsius.
3. Look the debug log or download the device's diagnostics how the temperature value for attribute "measured_value" changes and takes and puts before the real temperature the code 655xxx.

### Expected behavior

I checked the attributes “measured_value” and it is 655150 for one of the sensors and 655340 for another as the negative values (-1.5C and -3.4C). But it was 880 for a device in warmer room showing +8.8 C on the graph. So, "655" is definitely a negative temperature code followed by real value (-1.5) and the following "0" as a hundredth, [655]15[0]: -1.5C.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0302",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0402",
        "0x0405",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_qoy0ekbd",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_sensor.TuyaTempHumiditySensor_Square"
}
```

</details>

### Diagnostic information

<details><This is for the negative temperature -2.7 C.>Diagnostic information</summary>

```json
          "0x0402": {
            "endpoint_attribute": "temperature",
            "attributes": {
              "0x0000": {
                "attribute_name": "measured_value",
                "value": 655270
```

</details>

<details>< This is for the positive temperature +8.8 C>Diagnostic information</summary>

```json
          "0x0402": {
            "endpoint_attribute": "temperature",
            "attributes": {
              "0x0000": {
                "attribute_name": "measured_value",
                "value": 880
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Additional information

As it seems to be a Tuya's issue (but there are thousands of Tuyas' users out there), the value '655' should be inerpreted as '-' (minus or negative value).

And it should be nice to have some script to rewrite the historical erroneous values in the sql db.

almostlunatic, 2024-01-07:
I was totally wrong! The value 655 is a part of a signed int. I had today someone to go to the house. Two sensors in the cold rooms showed -4,6 and -4,5 C. These corresponded to the values of 654900 and 654910. If the ending zero would be omitted, it's unsigned int divided by 10. 65490 is -46.

It could be converted in a template but I don't know how to do it. The right thing to do would be to add the conversion to the Tuya Integration.

thsilva85, 2024-05-04:
Hi! I have the same problem. Any suggestion for fix this ?

closed 2024-11-07 by github-actions[bot]