zigpy/zha-device-handlers #2834: [Device Support Request]  Qirks for _TZE204_mhxn2jso

maxxxe73, 2023-12-12:
### Problem description

Hallo, this should be a Tuya TS0601 compatible device but in ZHA finds no entities
Did someone used this presence sensor?

Thanks in advance, Max

### Solution description

Need quirk for device

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
"data": {
    "ieee": "**REDACTED**",
    "nwk": 8645,
    "manufacturer": "_TZE204_mhxn2jso",
    "model": "TS0601",
    "name": "_TZE204_mhxn2jso TS0601",
    "quirk_applied": false,
    "quirk_class": "zigpy.device.Device",
    "quirk_id": null,
    "manufacturer_code": 4417,
    "power_source": "Mains",
    "lqi": 36,
    "rssi": null,
    "last_seen": "2023-12-12T10:01:41",
    "available": false,
    "device_type": "Router",
    "signature": {
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
      "endpoints": {
        "1": {
          "profile_id": "0x0104",
          "device_type": "0x0051",
          "input_clusters": [
            "0x0000",
            "0x0004",
            "0x0005",
            "0xef00"
          ],
          "output_clusters": [
            "0x000a",
            "0x0019"
          ]
        },
        "242": {
          "profile_id": "0xa1e0",
          "device_type": "0x0061",
          "input_clusters": [],
          "output_clusters": [
            "0x0021"
          ]
        }
      },
      "manufacturer": "_TZE204_mhxn2jso",
      "model": "TS0601"
    },
    "active_coordinator": false,
    "entities": [],
    "neighbors": [
      {
        "device_type": "Coordinator",
        "rx_on_when_idle": "On",
        "relationship": "Parent",
        "extended_pan_id": "**REDACTED**",
        "ieee": "**REDACTED**",
        "nwk": "0x0000",
        "permit_joining": "Unknown",
        "depth": "0",
        "lqi": "103"
      }
    ],
    "routes": [],
    "endpoint_names": [
      {
        "name": "SMART_PLUG"
      },
      {
        "name": "PROXY_BASIC"
      }
    ],
    "user_given_name": null,
    "device_reg_id": "4cd737c952175d894998f9f0d26ed655",
    "area_id": null,
    "cluster_details": {
      "1": {
        "device_type": {
          "name": "SMART_PLUG",
          "id": 81
        },
        "profile_id": 260,
        "in_clusters": {
          "0x0004": {
            "endpoint_attribute": "groups",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0005": {
            "endpoint_attribute": "scenes",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0xef00": {
            "endpoint_attribute": null,
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0000": {
            "endpoint_attribute": "basic",
            "attributes": {
              "0x0001": {
                "attribute_name": "app_version",
                "value": 74
              },
              "0x0004": {
                "attribute_name": "manufacturer",
                "value": "_TZE204_mhxn2jso"
              },
              "0x0005": {
                "attribute_name": "model",
                "value": "TS0601"
              }
            },
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      },
      "242": {
        "device_type": {
          "name": "PROXY_BASIC",
          "id": 97
        },
        "profile_id": 41440,
        "in_clusters": {},
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

nazar-pc, 2024-05-26:
Does https://github.com/Koenkk/zigbee-herdsman-converters/pull/6186 have sufficient details on how to implement support for this?
Also there is a duplicate report at https://github.com/zigpy/zha-device-handlers/issues/2894 and a forum thread at https://community.home-assistant.io/t/another-tuya-mmwave-human-presence-ts0601-tze204-mhxn2jso/681965?u=nazar-pc

nazar-pc, 2024-11-22:
Unfortunately device still isn't usable

nazar-pc, 2025-10-21:
Device is still unusable in 2025.10.3

nazar-pc, 2025-12-01:
Here is a working quirk on the forum, just needs to be upstreamed: https://community.home-assistant.io/t/another-tuya-mmwave-human-presence-ts0601-tze204-mhxn2jso/681965/11?u=nazar-pc