zigpy/zha-device-handlers #2835: New Ikea Somrig button, not able to connect zha

NicerDicer111, 2023-12-12:
### Problem description

Hello :slight_smile:
I bought the new Ikea Somrig buttons and would like to connect them to my HomeAssistant with Zigbee Dongle.
During the search of a new device, HA identifies the button but is not able to configure it.
Although the configuration process does not finish, the device is part of the device list. Sometimes there is a battery and identify entity and sometimes not. I can not control any entity.

### Solution description

 Is anybody aware if there is a way to connect a device which is currently not fully supported or do you know if there will be an integration?

### Screenshots/Video

[image]
[image]

### Device signature

IEEE: 5c:c7:c1:ff:fe:87:35:e2
Nwk: 0x046e
Device Type: EndDevice
LQI: 108
RSSI: Unbekannt
Zuletzt gesehen: 2023-12-12T20:13:52
Energiequelle: Battery or Unknown

### Diagnostic information

[zha-e958911dd7b4b0c8793c91dc810d16d3-IKEA of Sweden SOMRIG shortcut button-7dfa14459bf34ecb1709bcf1e8686b31.json.txt](https://github.com/zigpy/zha-device-handlers/files/13652515/zha-e958911dd7b4b0c8793c91dc810d16d3-IKEA.of.Sweden.SOMRIG.shortcut.button-7dfa14459bf34ecb1709bcf1e8686b31.json.txt)

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

MattWestb, 2023-12-12:
Its looks like your device is paired OK but its sending manufacture specific commands and need one quirk for getting  the events from it OK.
SYMFONISK II is using the same command so its shall not being so large problem getting  it done but i cant then im not at home for the moment.

NicerDicer111, 2023-12-13:
Thanks for your response! 
I will try to find a solution but to be honest I‘m a total newbie. So maybe it will take some time or i need your help. 😃

QuantumCorral, 2023-12-13:
is there any news about this ? would also be very interested.
Thanks for the effort you put into it.

NicerDicer111, 2023-12-14:
Unfortunatly not from my side. So I still need help to integrate it..

ender-null, 2023-12-16:
How do you pair it? I didn’t find any info about how to do it

Funder75, 2023-12-17:
I have a similar problem with the Somrig button.
I had to reset the button (4 clicks on the link button) for it to enter pairing mode. 
Then it seemed to pair successfully, but now I have only entities for Identity and battery.

[image]

HawkanForce, 2023-12-17:
Thank you Funder75, was loosing my mind. Tried long pressing, and pressing together with button 1 and even reading the manual. Thought the device was broken :D 
Now paired but only the same entities battery and identify

diogomendes-cmyk, 2023-12-17:
Indeed, IKEA lists no procedure to get into pairing mode, discovered yesterday by chance.. :(

Same issue in current HA, Bat + Identify but nothing else.

casparz, 2023-12-18:
I am working on a quirk for the remote, I have some issues contributing (its my first time).

It works as a custom quirk in my setup, so if you know how to add custom quirks you can copy my code for now.

NicerDicer111, 2023-12-18:
Thank you! Is it possible to use z2mqtt and zha simultaneously?
Currently I use zha and for the somrig button and your code I need to install z2mqtt right?
Or do I need to change my whole set up to z2mqtt?

casparz, 2023-12-18:
No, the point is to get it to work with ZHA (and it does :) )

joshuapfl, 2023-12-18:
WOULD love this to work soon toooo. No Idea how to accomplisch :)

QuantumCorral, 2023-12-19:
you have to create a new folder e.g. 

`/config/custom_zha_quirks`

in which you save the "somrigsmartbtn.py".
Then add the following to "configuration.yaml"
```
zha:
  enable_quirks: true
  custom_quirks_path: /config/custom_zha_quirks/
```

Restart HA and add the button again.

closed 2023-12-23 by TheJulianJES

Bluefarmer, 2024-01-03:
Hi, I'm trying to get this to work, but the button keeps on configuring after connecting. 

I made the file with the name mentioned, put in all the code, added the text to the configuration.yaml but it keeps on configuring

BenJamesAndo, 2024-01-05:
@Bluefarmer you don't need to add a custom quirk for this anymore as it's been merged. I set mine up today. I pressed the pairing button 4 times and it then showed up in ZHA on the add device page.

HawkanForce, 2024-01-06:
Did you get it to work @Bluefarmer ?

I removed Somrig from my devices, updated HA to the latest version, added zigbee devices and it just keeps configuring
[image]
it does show up in devices but no button presses are shown in the logbook or battery information etc.
[image]

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0006",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x0020",
        "0x1000",
        "0xfc7c",
        "0xfc80"
      ],
      "output_clusters": [
        "0x0003",
        "0x0004",
        "0x0006",
        "0x0008",
        "0x0019",
        "0x1000",
        "0xfc80"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x0006",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0xfc80"
      ],
      "output_clusters": [
        "0x0003",
        "0x0004",
        "0x0006",
        "0x0008",
        "0xfc80"
      ]
    }
  },
  "manufacturer": "IKEA of Sweden",
  "model": "SOMRIG shortcut button",
  "class": "zhaquirks.ikea.somrigsmartbtn.IkeaSomrigSmartButton"
}

MattWestb, 2024-01-06:
If the device is getting problem then being paired its not getting  OK configured..
Try reconfigure from the device card and waking it up then sending commands to it.
If not working restart HA and try agen. Dome new devices need potting the battery out and doing the reconfigure then its being putted back in for getting it OK and other times deleting it and wait one minute / or restart HA and adding it new.

Bluefarmer, 2024-01-06:
I still have the same issue if I connect the device the normal way, without quirks. After the interview is complete, it stays on configuring for a very long time. After this, the buttons are not available. This happens with two separate devices...

Bluefarmer, 2024-01-06:
Okay, never mind. Updated to latest version (wasn't correctly triggered...) and now everything works

HawkanForce, 2024-01-06:
Thank you, now I also get something in the log after reconfiguring.

[image]

deephack1982, 2024-01-06:
This worked perfectly for me. Will somrigsmartbtn.py be included in a future release?

casparz, 2024-01-07:
It is included in the most recent release of HA (2024.1). For those that have added it as custom quirk, it is best to:
- remove the custom quirk
- remove the somrig device
- install 2024.1
- reboot
- add the device again.

MattWestb, 2024-01-07:
> This worked perfectly for me. Will somrigsmartbtn.py be included in a future release?

Its in HA 2024.1

BatteryDie, 2024-01-21:
Seem this issue is still persist after I updated and restarted HA 2024.1.5. I tried remove and pair it. I do not have custom quirk on HA.

[image]

Here are log:
```
Received a packet: ZigbeePacket(timestamp=datetime.datetime(2024, 1, 21, 21, 25, 12, 328276, tzinfo=datetime.timezone.utc), src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=0, source_route=None, extended_timeout=False, tsn=7, profile_id=0, cluster_id=32822, data=Serialized[b'\x07\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
[0x0000:zdo] ZDO request ZDOCmd.Mgmt_Permit_Joining_rsp: [<Status.SUCCESS: 0>]
[0x0000:zdo] No handler for ZDO request:ZDOCmd.Mgmt_Permit_Joining_rsp([<Status.SUCCESS: 0>])
Received a packet: ZigbeePacket(timestamp=datetime.datetime(2024, 1, 21, 21, 25, 13, 808150, tzinfo=datetime.timezone.utc), src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x451F), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=1794, data=Serialized[b'\x08$\n\x00\x00%\xb3\xab\x00\x00\x00\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=28, non_member_radius=0, lqi=165, rssi=None)
[0x451F:1:0x0702] Received ZCL frame: b'\x08$\n\x00\x00%\xb3\xab\x00\x00\x00\x00'
[0x451F:1:0x0702] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=36, command_id=10, *direction=<Direction.Client_to_Server: 1>)
[0x451F:1:0x0702] Decoded ZCL frame: Metering:Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=uint48_t, value=43955))])
[0x451F:1:0x0702] Received command 0x0A (TSN 36): Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=uint48_t, value=43955))])
[0x451F:1:0x0702] Attribute report received: current_summ_delivered=43955
[0x451F:1:0x0702] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=36, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>)
[0x451F:1:0x0702] Sending reply: Default_Response(command_id=10, status=<Status.SUCCESS: 0>)
[0x35BF](E11-N1G): Device seen - marking the device available and resetting counter
[0x35BF](E11-N1G): Update device availability -  device available: True - new availability: True - changed: False
[0xDDE3](E11-N1G): last_seen is 19457349.267830133 seconds ago and ping attempts have been exhausted, marking the device unavailable
[0xDDE3](E11-N1G): Update device availability -  device available: False - new availability: False - changed: False
[0x37C9](TH01): Device seen - marking the device available and resetting counter
[0x37C9](TH01): Update device availability -  device available: True - new availability: True - changed: False
[0x9F1D](TS011F): Device seen - marking the device available and resetting counter
[0x9F1D](TS011F): Update device availability -  device available: True - new availability: True - changed: False
[0xF6F6](TS0202): Device seen - marking the device available and resetting counter
[0xF6F6](TS0202): Update device availability -  device available: True - new availability: True - changed: False
sensor.plug_02_active_power: polling for updated state
[0xCC8E:1:0x0b04]: async_update
[0xCC8E:1:0x0b04]: Reading attributes in chunks: ['active_power', 'rms_current', 'rms_voltage']
[0xCC8E:1:0x0b04] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=False, *is_general=True), tsn=19, command_id=<GeneralCommand.Read_Attributes: 0>, *direction=<Direction.Server_to_Client: 0>)
[0xCC8E:1:0x0b04] Sending request: Read_Attributes(attribute_ids=[1291, 1288, 1285])
Received a packet: ZigbeePacket(timestamp=datetime.datetime(2024, 1, 21, 21, 25, 14, 138297, tzinfo=datetime.timezone.utc), src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xCC8E), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=2820, data=Serialized[b'\x18\x13\x01\x0b\x05\x00!\x10\x00\x08\x05\x00!\xe5\x00\x05\x05\x00!}\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=29, non_member_radius=0, lqi=84, rssi=None)
[0xCC8E:1:0x0b04] Received ZCL frame: b'\x18\x13\x01\x0b\x05\x00!\x10\x00\x08\x05\x00!\xe5\x00\x05\x05\x00!}\x00'
[0xCC8E:1:0x0b04] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=19, command_id=1, *direction=<Direction.Client_to_Server: 1>)
[0xCC8E:1:0x0b04] Decoded ZCL frame: TuyaZBElectricalMeasurement:Read_Attributes_rsp(status_records=[ReadAttributeRecord(attrid=0x050B, status=<Status.SUCCESS: 0>, value=TypeValue(type=uint16_t, value=16)), ReadAttributeRecord(at [... cut ...] \xc9', *payload=201))]))
[0xE215:1:0xef00] Received command 0x01 (TSN 174): get_data(data=TuyaCommand(status=0, tsn=191, datapoints=[TuyaDatapointData(dp=1, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00\xc9', *payload=201))]))
[0x14D1](STARKVIND Air purifier table): Device seen - marking the device available and resetting counter
[0x14D1](STARKVIND Air purifier table): Update device availability -  device available: True - new availability: True - changed: False
```

dhzl84, 2024-01-22:
> Seem this issue is still persist after I updated and restarted HA 2024.1.5. I tried remove and pair it. I do not have custom quirk on HA.
> 
> ![image](https://private-user-images.githubusercontent.com/13942195/298416203-46d74d66-05e3-4fbf-a56e-c5c2c30a72f2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU5NTA3MTgsIm5iZiI6MTcwNTk1MDQxOCwicGF0aCI6Ii8xMzk0MjE5NS8yOTg0MTYyMDMtNDZkNzRkNjYtMDVlMy00ZmJmLWE1NmUtYzVjMmMzMGE3MmYyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMjIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTIyVDE5MDY1OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTVkYTdmOTllN2EzMzMwZDk3MzFkYjlhMzkzZjk3MmFmYjU2MGU4NTZkMmU5YTMxYzc3MjM0NTg0NTRhZTUzNTQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.n_gq3oD9ngdMEOapayMHOUmPmj6zbLej9XthcmqBbZ4)
> 
> Here are highlighted log
> Edit: I was able to pair between Somrig device and HA by spamming pair buttons (i don't know why red LED was off while pair mode it have new brand battery too) but pressing both two buttons do not works when it became unknown. I already replaced the battery two times.
> 
> ![image](https://private-user-images.githubusercontent.com/13942195/298419134-2069e066-6fb3-4ac6-aeab-f8e7a287be69.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU5NTA3MTgsIm5iZiI6MTcwNTk1MDQxOCwicGF0aCI6Ii8xMzk0MjE5NS8yOTg0MTkxMzQtMjA2OWUwNjYtNmZiMy00YWM2LWFlYWItZjhlN2EyODdiZTY5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMjIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTIyVDE5MDY1OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTUyM2JmZDViODUyNjhjZGY4Y2Y3YmMxN2ZmYjI1ZjlhMTFmNzVjMTQ0ZjlmNzIwZWI3NWY3MjBkNjNlMjE3OTAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.chiT_NRzhhrRHAVB-eFLpmqynK_cghsmSnfqrQhHFmU)

Exact same issue here, also running HA 2024.1.5.

RASTRILL0, 2024-01-22:
Same problem here. Device pairs and shows battery and Identify but is unable to bind the buttons.

MattWestb, 2024-01-22:
@RASTRILL0 This device can only being bond if its using in ZLL mode (direct paring with light without coordinator) the its sending On/Off light commands.
Then being paired with one coordinator its only sending short double and long press / release in matter standard commands for switches and lights is not understanding this commands.
You shall using automatons triggered from the events the remote is sending and defining actions they shall doing in the automations.

gimik91, 2024-01-29:
Same problem here. Device pairs and shows battery and Identify but is unable to bind the buttons.

RASTRILL0, 2024-01-29:
> Same problem here. Device pairs and shows battery and Identify but is unable to bind the buttons.

Tap the pair button 4 times to initiate the pairing process (red blinking light) then keep tapping the pair button until it’s complete. I think the remote is going to sleep before the process finishes. I got mine working.

MattWestb, 2024-01-30:
> Same problem here. Device pairs and shows battery and Identify but is unable to bind the buttons.

Pleas explaining wot you meaning with  `binding` then its only used with light controllers and not "other" controllers (and this is NOT one light controller like Rodret is).
And if being paired OK it shall being  OK sending the commands t the system but can need being reconfigured for doing it OK.

gimik91, 2024-01-31:
> Tap the pair button 4 times to initiate the pairing process (red blinking light) then keep tapping the pair button until it’s complete. I think the remote is going to sleep before the process [...]

I linked succesfully after 4 times click. I also added quirk and tried to run OTA (device is already latest version 1.0.20.  
battery and identify works, but i just cannot see events from clicking the buttons. Still not solved 
[image]

reinder83, 2024-01-31:
I also have trouble pairing these, first few times I did not get any entities at all, now I get these 
[image]
When I subscribe to zha_events in the developer tools I don't see anything

dhzl84, 2024-02-02:
As mentioned [here](https://github.com/zigpy/zha-device-handlers/issues/2835#issuecomment-1904734979) you have to enforce ZLL. For me this just worked by pairing Somrig via a Light device.
Not sure if one can enforce ZLL otherwise.
[image]

After pairing that way I receive the events
[image]

MattWestb, 2024-02-02:
The comment of ZLL  is ti getting it sending Light Command like On/Off and dimming up and down.
4 press and the LED is start blinking its going in normal paring mode and if being OK its sending commands with press from both button.
If not getting events try reconfigure and waking it up then sending commands. If not working restart HA and next is taking the battery out and putting it in and fast starting reconfigure.

BatteryDie, 2024-02-02:
> As mentioned [here](https://github.com/zigpy/zha-device-handlers/issues/2835#issuecomment-1904734979) you have to enforce ZLL. For me this just worked by pairing Somrig via a Light device.
> Not sure if one can enforce ZLL otherwise.

I removed Somrig from HA and I tried pair Somrig again via devices I have:
1. Ikea TRADFRI bulb E26 WW 806lm
2. Jasco Enbrighten Zigbee Switch (it is acting as light device)
3. Ikea STARKVIND Air purifier table
4. TuYa TS011F smart plugs

Seem HA is not detecting Somrig with all of the device on the list. Weird thing is, it was able to detect and configurate Somrig with direct zigbee hub (Sonoff Zigbee 3.0 USB Dongle Plus with Texas Instruments CC2652).

P.S. What do 'enforce ZLL' mean? Google is not helping me.

reinder83, 2024-02-02:
I've managed to pair it succesfully now through an zigbee lightbulb

MattWestb, 2024-02-02:
> P.S. What do 'enforce ZLL' mean? Google is not helping me.´

It was the first Zigbee standard for lights (Zigbee Light Link) but its now one part or Zigbee 3 and have some extra functionality like touch link paring and the device is making its own network (IKEA standalone device / direct paring with lights and Trädfri hub is using it) and Dirigera is using "normal" paring with coordinator that is doing all security.

Your IKEA light shall not have any problem if its running current production firmware and not one old ZLL version.

HawkanForce, 2024-02-04:
Is it just me or is only button1 working as of now? Nothing happens in the log when pressing button 2

[image]

reinder83, 2024-02-04:
> Is it just me or is only button1 working as of now? Nothing happens in the log when pressing button 2 [...]

I have the same with 2 of my 3 somrig's, with the first one I managed to pair, everything seems to be working, I thought probably something went wrong while paring, so maybe i'll try to re-pair them later

MattWestb, 2024-02-04:
If getting anything from the controller more then the checking every 50 minutes  do one reconfigure of the device from the device card and waking the device then sending somthing to it then its sleeping nearly all the time.
No events from button 2 and also being one hardware fault not likely one software problem.

HawkanForce, 2024-02-04:
These Ikea Zigbee devices Somrig and E2002 seem to be a pain. Never had any issues with Zigbee devices before. Tried to reconfigure but now back to no buttons working and when trying to pair its stuck on interview and if trying to reconfigure it just failes despite trying to wake the device in different ways. Seems to be up to luck if you get the bindings to work. I'll leave this as is for now. Have to try to pair it again some other time and just hope for the best.

drkx, 2024-02-20:
> Is it just me or is only button1 working as of now? Nothing happens in the log when pressing button 2 [...]

I have the same problem. Button 1 works perfectly, but the second one seems to do nothing. Is there a possibility of debugging? To see if anything is sent by button 2?

labmonkey, 2024-02-28:
> As mentioned [here](https://github.com/zigpy/zha-device-handlers/issues/2835#issuecomment-1904734979) you have to enforce ZLL. For me this just worked by pairing Somrig via a Light device. Not sure if one can enforce ZLL otherwise. ![Screenshot_20240202-082634](https://private-user-images.githubusercontent.com/5675570/301763073-8dd3c35c-c553-4e9f-a12f-f13220e59e41.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkxMTQyODYsIm5iZiI6MTcwOTExMzk4NiwicGF0aCI6Ii81Njc1NTcwLzMwMTc2MzA3My04ZGQzYzM1Yy1jNTUzLTRlOWYtYTEyZi1mMTMyMjBlNTllNDEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIyOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMjhUMDk1MzA2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZjAxNDBiZDNiZWI1NzgxMzk1NWVmODhiMWY2NWVmMTg0OGIwYjY4ZDliMDIxNmYxMzNjNDY1OTUwMDkwZmJkMCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.Hd6aMFwuMjLfTLg9wPMP3jjIfKfFYsGbyCtR4-nFa_c)
> 
> After pairing that way I receive the events ![Screenshot_20240202-082852](https://private-user-images.githubusercontent.com/5675570/301763488-acee7fa0-dce3-489c-a578-24be462fafca.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkxMTQyODYsIm5iZiI6MTcwOTExMzk4NiwicGF0aCI6Ii81Njc1NTcwLzMwMTc2MzQ4OC1hY2VlN2ZhMC1kY2UzLTQ4OWMtYTU3OC0yNGJlNDYyZmFmY2EucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIyOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMjhUMDk1MzA2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ODYwYmNjYmZjMjUwMjY0M2Y0MDg4MjAzMjM0NTY3YWMzZmUwMmI5ZWU0NTVmNzhmOWZmMzA1N2ZmNzYxNDczOCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.oo1cEJJbA-NOMBLQDUHEUua5sVbNfmNrdEwxg18yGK4)

I can confirm that. I was trying everything to connect SOMRIG via my Sonoff Zigbee dongle and even if I went through the configuration step somehow the buttons never worked. I've connected Ikea TRADFRI lighbulb first and then used it to connect the remote and suddenly everything works perfectly.

MattWestb, 2024-02-28:
Always have 3 or more good routers for end devices then the router part of all coordinators is very bad and being offline then restarting the  system.
If not you have not one mesh network but one star network with one point of (complete) failure and you is misusing the systems architecture / functionality.

petervk, 2024-02-28:
I just bought one of these and once I figured out how to put it in pairing mode it seems to pair ok but I don't get any buttons showing up for the device in Home Assistant. What I find odd is that when I click either button events show up in the logbook but I don't know how to trigger automations based on that. I'm running Home Assistant 2024.2.4 and an ezsp coordinator connected via usb.
[image]

petervk, 2024-02-28:
I have been able to get this working using this blueprint: https://community.home-assistant.io/t/ikea-somrig-remote-e2213-zha/668671 but I feel like it would be better if this functionality was directly exposed through home assistant.

petervk, 2024-02-28:
Oh, there they are. I thought I tried that before and didn't see anything. Thanks!

MattWestb, 2024-02-28:
For information is one firmware update in the pipe for Somrig its just being certificated by CSA so very likely its being released in some weeks.
https://csa-iot.org/csa_product/somrig-shortcut-button-2/
Its only bug fix then its jumping from 1.0.20 to 1.0.21.

BatteryDie, 2024-03-02:
> I removed Somrig from HA and I tried pair Somrig again via devices I have:
>
> 1. Ikea TRADFRI bulb E26 WW 806lm
> 2. Jasco Enbrighten Zigbee Switch (it is acting as light device)
> 3. Ikea STARKVIND Air [...]

I updated HA today then I was setting up STARKVIND Air purifier table and SOMRIG. I left them for over 10 minutes and they quietly completed pairing setup.

[image]

thurcombe, 2024-05-17:
I'm a little confused here, some are saying it works with the latest HA, some are saying you need the blueprint, some are saying spam the pairing button, some are saying the thing just pairs after an age, some are saying it needs to be forced into ZLL whereas other comments indicate a successful pairing so does it need ZLL? I am confused :)

With the exception of ZLL I have tried everything repeatedly with no luck. Occasionally I can get it to configure far enough that it at least reports battery status, occasionally I can see a button press event that fires whilst pairing but beyond that point.... nothing.

reinder83, 2024-05-17:
> I'm a little confused here, some are saying it works with the latest HA, some are saying you need the blueprint, some are saying spam the pairing button, some are saying the thing just pairs after [...]

You need to pair it through ZLL (through a light bulb) for the buttons to work, sometimes you'll need to try a few times for all the button events to work

MattWestb, 2024-05-17:
ZLL pairing (10 sec pushing paring button) is only used for TF and direct paring to lights.
For Dirigeraa and all open Zigbee system use normal Zigbee 3 paring (4 click paring button) and both DG and ZHA is working well but can getting time out and need reoving the battery for getting it working.

hugomn, 2024-09-23:
Did anyone manage to add any SOMRIG buttons via ZHA? Currently running Core 2024.9.2 and none of my 3 buttons are able to finish pairing: 

[image]

Sometimes they are added to the list of devices, but not functional at all (not triggering any events in the log, etc). I've tried all the tips on this thread (adding via TRADFRI lamp, adding the custom quirk, clicking pair button continuously, etc) and no luck.

[image]

What is weird is that one of my buttons was working fine (probably on a previous version), and now I've removed it to test and it also cannot be added anymore. Maybe something is broken in the newly added quirk? Thanks for any tips!

zapster, 2024-09-24:
Same issue as the previous poster. Maybe this issue should be reopened?

drkx, 2024-09-24:
I eventually gave up and bought a Rodret (https://www.ikea.com/nl/en/p/rodret-wireless-dimmer-power-switch-smart-white-20528128/), which looks exactly the same but does work in ZHA.

zapster, 2024-09-24:
Update: I manage to pair it via IKEA Tradfri bulb, 10 cm distance, after the 3rd or 4th try. Still, I think there is some room for improvement.

hugomn, 2024-09-25:
@TheJulianJES @casparz seems like something broke pairing with Ikea Somrig buttons. Do you mind reopening this issue?

MattWestb, 2024-09-26:
Was deleting one and restarting HA and adding it new without problems and its sending all commands and doing checking every 55 minutes.

What coordinator is you using in ZHA ?
TI or deCONZ ?

Also good Zigbee3 routers or CE ones ?
(CE = China  Export not EU cert)

zapster, 2024-09-26:
@MattWestb maybe there are different revisions of the Somrig devices? I bought mine a few weeks ago in Austria.

> What coordinator is you using in ZHA ?

Dresden Elektronik ConBee II on USB 2 port with extension cable.

> Also good Zigbee3 routers or CE ones ?

One Ikea repeater and a hand full of Bulbs (Osram, Ikea, Philips, ...). No "no name" or CE as far as I can tell.

Maybe also worth mentioning, I never had pairing issues with any another Zigbee devices.

MattWestb, 2024-09-26:
W1110 here and IKEA have only one hardware version if the Sommrig if looking on FCC and CSA.-OT docks.
And more then likely we have the same batch of device that IKEA AT is have getting.

IKEA reaprter is old (first gen) but is updated to true Zogbee 3 and is working OK for me.
Osram if older version is normally very bad routers and runs some old Zigbee 1/2.X. stack

Conbee II shall working OK but must running on the latest firmware for not having problems.

If you can only the Conbee and updated IKEA routers and trying pairing the remote with one router and not direct to the Conbee.

For trying paring:
1 delete the device in ZHA.
2 restart HA.
3 start add Zigbee device (bed from one good router and add device thru this. . .) and press the pairing button on Sommring 4 times (in 5 seconds).

PS: Conbee is having very strong power output so making its dominant / like have all devices direct to it self = not good in one mesh network.

[15 more replies]

PR #2844 (2023-12-23): Add support for IKEA Somrig shortcut button
## Proposed change
Quirk for new SOMRIG smart button from IKEA. I based everything off of the Symfonysk remote v2.

## Additional information
Fixes #2835

## Checklist

- [x] The changes are tested and work correctly
- [x] `pre-commit` checks pass / the code has been formatted using Black
- [ ] Tests have been added to verify that the new code works