zigpy/zha-device-handlers #2840: [Q]: how to provide different names for a repeated entity/metric

jmuf, 2023-12-17:
In [#2650](https://github.com/zigpy/zha-device-handlers/issues/2650) we are trying to develop a "quirk" for a power meter that will report three times the same metric: (e.g 3x "Active Power"):
* first clamp
* second clamp
* totals (as calculated by the device)

Done by having the two clamps report as new classes  ( `EnergyA`, `PowerA` and similar for `..B` and `..Total`) that listen for updates from the TUYA cluster, then using separate endpoints in the `replacement`:

```
            10: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.METER_INTERFACE,
                INPUT_CLUSTERS: [
                    EnergyA,
                    PowerA,
                ],
            },
```

This all seems to work, but the three metric have the same name in the GUI:

[image]
_Originally posted by @robertstumberger in https://github.com/zigpy/zha-device-handlers/issues/2650#issuecomment-1855224449_

Is there a way to "tag" these in the code so that they show up under different names ("Clamp A Active Power", "Clamp B Active Power", "Total Active Power") - perhaps some sort of sub-devicename that these classes or `endpoints` could set?

Alternatively, is there a better approach than to use separate endpoints here?

TheJulianJES, 2023-12-24:
> Alternatively, is there a better approach than to use separate endpoints here?

The sensors could be created directly in ZHA. That would also solve the name issues:
https://github.com/home-assistant/core/blob/fc349b08755d115356700cb5e71f0792d0a7fa42/homeassistant/components/zha/sensor.py#L318-L327

jmuf, 2023-12-28:
> TheJulianJES

Thanks for the update, although I am not sure whether I understand your advice - is this even something I could do via my "quirk"? 
`ElectricalMeasurementApparentPower` is contained in `ElectricalMeasurement`, i.e the class I am deriving "my" endpoints from.
Adding several new sensors names directly in ZHA (i.e something like `ElectricalMeasurementApparentPower2`, analog to the approach for `Tier{1..9}2SmartEnergySummation` ?) looks somewhat ugly - how many of these to create?
I see that we already have something like `phase_B`, `phase_C` in tuya. Although technically wrong (my quirk is for a single-phase meter, not tri-phase), I've tried to use these to report the additional data, but these metrics are not picked up by default in ZHA.

closed 2024-07-02 by github-actions[bot]