zigpy/zha-device-handlers #2841: [Device Support Request] IKEA of Sweden Remote Control N2

HawkanForce, 2023-12-17:
### Problem description

I just bought a Ikea Styrbar E2002. It paris but shows no entities or only battery and identify. Tried to reconfigure, remove, restart an readd.
It does not show firmware version, and shows to use Quirkv1 (dont know how to remove or change quirks)

### Solution description

just to get it to work :)

### Screenshots/Video

[image]

### Device signature

```json
<details><summary>Device signature</summary>

1
2
3
4
5
6
7
8
9
10
11
12
13
14
15
16
17
18
19
20
21
22
23
24
25
26
27
28
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0820",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x1000",
        "0xfc57"
      ],
      "output_clusters": [
        "0x0003",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0019",
        "0x1000"
      ]
    }
  },
[... 438 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

MattWestb, 2023-12-17:
The device looks OK and the system is only getting the version then the device is requesting OTA but is coming in some hours but you can also reading it on the basic cluster (SW version).
If you have doing reconfigure and was waking up the device and was getting green checks on  all it shall working OK.
But i think the best is letting it doing OTA update then its on shipping version and not the last production ones (after OTA you need deleting it and waiting one minute and then adding it new for working OK).

HawkanForce, 2023-12-17:
Thanks for the quick response.
The reconfig shows green for binding and reporting (and got green for battery once too) but now all orange. I'll let it sit for a couple of hours if it sorts itself out

[image]

MattWestb, 2023-12-17:
They can being tricky getting reconfigured but normally its being OK or one restart of HA.
But better getting OTA update done so not need doing all more times.

HawkanForce, 2023-12-18:
It shows button presses in the log now, but the ota doesn't seem to install. At one time it did show the firmware version as 0x00010024 and others seemed to have problems with the OTA for that one aswell so I'll try to poke a bit with that.
[image]

TheJulianJES, 2023-12-20:
>  It paris but shows no entities or only battery and identify.

That's expected. You won't get an entity for the button presses. You should get a `zha_event` on a button press though. Like mentioned, reconfiguring (whilst pressing buttons to wake the device) should help with this.

> It shows button presses in the log now

That's the expected behavior then. You can use device automation triggers (trigger: device) in automations with the different button presses then.

HawkanForce, 2023-12-21:
Today I somehow got the 2.4.5 firmware installed, but it no longer shows button presses in the logbook, it has switched to zhaquirks.ikea.fourbtnremote.IkeaTradfriRemoteV2
[image]

I really do not know what I am doing :) All other Zigbee devices have worked right out of the box but these Ikea ones seem to be giving at least me a bit of a workout

MattWestb, 2023-12-21:
Deleting it from the device card and waiting one minute or restarting HA and adding it new.
Then its OK paired do one reconfigure from the device card and waking up the device so i can reviving the commands.

HawkanForce, 2023-12-21:
deleted - waited and added new. Got the interview
[image]
the I went to the device, removed the battery for a while and took reconfig
[image]
tried a restart afterwards also
[image]

AllardKatan, 2023-12-21:
This sort of corresponds to my experience with the Ikea remotes. Getting them to work requires a lot waiting plus of trial and error (or more information than I could find). Once they work though, they stay good. If someone has more in-depth knowledge and/or knows a link to a good how-to guide for pairing Ikea remotes to ZHA, that would be great. Because they are cheap, easy to get and reliable once they start working.

HawkanForce, 2023-12-21:
Ok thanks, I'll try again tomorrow see if I get lucky :)

MattWestb, 2023-12-21:
If its in the network but not waking up OK the best is removing the battery and putting it back and reseeding the reconfigure. If getting all green it shall being OK and also they is doing checkins every 55 minutes. Also restart HA can helping but takes times and is little tricky.

HawkanForce, 2023-12-22:
For a while I got left and right button presses to work
[image]
and now I got the device added. But no button presses work. I'll leave it overnight and restart HA in the morning and continue. Maybe these are some babysteps atleast :)
[image]

henri98, 2023-12-29:
Im facing issues with binding groups in ZHA since firmware version 0x02040005. Does this work for you?

HawkanForce, 2023-12-29:
I Henri, I have not gotten it to work :/

MattWestb, 2023-12-29:
Group binding is not supported after firmware 1.0.024 (over one year old "news") and if running on that or older its working.
https://github.com/zigpy/zigpy/discussions/660#discussioncomment-4497890

HawkanForce, 2023-12-29:
I'm a noob and don't understand much like group bindings etc but I'll try to read the thread you posted maybe I learn something :) I'll try to downgrade the firmware also that was in the post. Let's see what happens.

MattWestb, 2023-12-29:
All IKEA controller that i have trying is blocked for firmware downgrade with OTA (also written in the Zigbee certificate) so must do it with SWD-Probe and flashing the old firmware with it.

HawkanForce, 2023-12-29:
Is it a brick otherwise? Maybe just better to give it to someone using ikea hub and buying another button. All other Zigbee devices have worked so easily so I thought all zigbee would be easy. So a lesson learned atleast :) Thanks for all the help

MattWestb, 2023-12-29:
Its depends what you like using it for.
Controlling lights with direct binding / device is working OK and also using the events for doing automatons only binding the remote to Zigbee groups is not working on the production firmware (that is bad for all light systems).

Then you can getting commands from the remote is not one brick only little tricky getting OK configurated.

PS: Binding is making the commands is working also then the host system is not online / not working OK and is great for light systems.

HawkanForce, 2023-12-29:
So I can check for button presses despite them not showing in the log for the device? I will the button presses just to run different automations so I guess the group bindings does not affect me

closed 2024-07-03 by github-actions[bot]

abacao, 2024-07-17:
I'm facing the more or less the same issues.

I have a Ikea Styrbar remote. It was not easily pair-able and only the arrow key register.
[image]

I have tried the update without success and the device is using a "quirk" different from everyone else.

Any suggestions?

MattWestb, 2024-07-17:
First update the firmware its out of date.
Second do one reconfigure and waking up the remote then sending things to it.
If it dont like communicating take the battery out and punting it back.

Sakura-sx, 2024-09-14:
> deleted - waited and added new. Got the interview
> 
> [image]
> 
> the I went to the device, removed the battery for a while and took reconfig
> 
> [image]
> 
> tried a restart afterwards also
> 
> [image]
> 
> 
> 
> 

It looks exactly the same for me, had the same issue reconfiguring, the only difference is that for me it shows zigpy.quirks.v2.CustomDeviceV2 as the quirk. Any ideas on how to get it working? I also had an issue with IKEA lights but it turned out I was resetting them the wrong way, I was turning them on and off 6 times from off state, but it turned out I had to do it from an on state, after doing it the correct way the lights flashed different and the device has worked perfectly ever since. I wonder if it's a similar issue on the remote and I am just resetting it wrong, the way I have been doing it is pressing the sync button 4 times in a 5 second time. Then the light on the front blinks orange.

MattWestb, 2024-09-14:
Take out the battery and putting it back and start the reconfigure then the device is rejoining the network.
Also possible try restarting HA but the battery trick looks working better.

Then the device is working OK you is getting checkins from it ever 55 minutes in the log.

The `zigpy.quirks.v2.CustomDeviceV2` is the current version for remotes on the latest production device firmware. If the system is loading one older version its needs deleting the device and wait one minute (or restart HA) and adding it new so the system is getting the new endpoint / cluster settings for using the right quirk.

Sakura-sx, 2024-09-14:
> Take out the battery and putting it back and start the reconfigure then the device is rejoining the network.
>
> Also possible try restarting HA but the battery trick looks working better.
>
>
>
> Then the [...]

I have tried the battery trick, it didn't work. I will try removing the battery, deleting the button from home assistant, restarting home assistant, and then adding it again. The problem is that I think I need to put it in sync mode or it will not add. How did you do this?

MattWestb, 2024-09-14:
`sync mode` dos not exist in Zigbee.
I think you meaning paring more (4 press pairing button in 5 sec).

Deleting and restarting HA and then paring it and then try reconfigure it and dont forgetting waking it up. If not working OK do the battery trick and clicking some times on one button for keep it awake the doing the reconfigure.