zigpy/zha-device-handlers #2843: Add support for EARU Breaker TS011F TZ3000_zrm3oxsh

PhilippeP62, 2023-12-17:
Problem : when added as is , consumption and amperage is way over normal.
I found similar issue for the Tongou breaker so

I copied the PR2252 code and made a custom quirck for my model by adjusting the model info and consumption and amperage are now normal.

Just adding the model info to the tsO11f plug quirck should solved this.

cordvision, 2024-02-28:
Any chance you could post the custom quirk? I'm having the dame issue. 
Also, do you get accurate wattage readings? I compared mine to a calibrated meter and they are off by 7-8 percent.

PhilippeP62, 2024-02-28:
Based on TS011F Plug Quirk
so i named the file 
ts011F_tz3000_zrm3oxsh.py

```"""TS011F plug."""

from zigpy.profiles import zgp, zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    OnOff,
    Ota,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.homeautomation import ElectricalMeasurement
from zigpy.zcl.clusters.lightlink import LightLink
from zigpy.zcl.clusters.measurement import TemperatureMeasurement
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODEL,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.quirk_ids import TUYA_PLUG_ONOFF
from zhaquirks.tuya import (
    TuyaNewManufCluster,
    TuyaZB1888Cluster,
    TuyaZBE000Cluster,
    TuyaZBElectricalMeasurement,
    TuyaZBExternalSwitchTypeCluster,
    TuyaZBMeteringCluster,
    TuyaZBMeteringClusterWithUnit,
    TuyaZBOnOffAttributeCluster,
)
from zhaquirks.tuya.mcu import EnchantedDevice

class Plug_CB_Metering(EnchantedDevice):
    """Circuit breaker with monitoring, e.g. Tongou TO-Q-SY1-JZT. First one using this definition was _TZ3000_qeuvnohg."""

    quirk_id = TUYA_PLUG_ONOFF

    signature = {
        MODEL: "TS011F",
        MODELS_INFO: [("_TZ3000_zrm3oxsh", "TS011F")],
        ENDPOINTS: {
        # <SimpleDescriptor endpoint=1 profile=260 device_type=266
        # device_version=1
        # input_clusters=[0, 3, 4, 5, 6, 1026, 1794, 2820, 57344, 57345]
        # output_clusters=[10, 25]>
        1: {
            PROFILE_ID: zha.PROFILE_ID,
            DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
            INPUT_CLUSTERS: [
                Basic.cluster_id,                                                    
                Identify.cluster_id,
                Groups.cluster_id,
                Scenes.cluster_id,
                OnOff.cluster_id,
                TemperatureMeasurement.cluster_id,  # Note: no temperature sensor claimed in manual and it's constant -6.3C
                Metering.cluster_id,
                ElectricalMeasurement.cluster_id,
                TuyaZBE000Cluster.cluster_id,
                TuyaZBExternalSwitchTypeCluster.cluster_id,
            ],
            OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
        },
        # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
        # device_version=1
        # input_clusters=[]
        # output_clusters=[33]>
        242: {
            PROFILE_ID: zgp.PROFILE_ID,
            DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
            INPUT_CLUSTERS: [],
            OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
        },
    },
}
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_PLUG_IN_UNIT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    TuyaZBMeteringCluster,
                    TuyaZBElectricalMeasurement,
                    TuyaZBE000Cluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }```

cordvision, 2024-02-28:
Thanks a lot for posting. For some reason it doesn't work on my end. 
Could it have something do to with the device type (I think mine is different but I don't really understand how it works)?

This is what I find in the signature tab (Zigbee ZHA) of my smart braker
```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x010a",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0402",
        "0x0702",
        "0x0b04",
        "0xe000",
        "0xe001"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_zrm3oxsh",
  "model": "TS011F",
  "class": "zigpy.device.Device"
}
```

PhilippeP62, 2024-02-28:
[image]

That's what I have so your quirk is not loading , there is some setup to do in configuration.yaml also ...

piavia, 2024-03-28:
@Koenkk
I think I found out the data of this device _TZ3000_zrm3oxsh,
could you look at the link
https://github.com/sprut/Hub/issues/2977

PhilippeP62, 2024-09-24:
Oh I see that there is a message in the logs 

Enregistreur: zhaquirks
Source: /usr/local/lib/python3.12/site-packages/zhaquirks/__init__.py:469
S'est produit pour la première fois: 21:57:30 (1 occurrences)
Dernier enregistrement: 21:57:30

Loaded custom quirks. Please contribute them to https://github.com/zigpy/zha-device-handlers

That's what this issue is about ...

closed 2025-08-15 by github-actions[bot]