zigpy/zha-device-handlers #2846: [Device Support Request] smarthjemme.dk QUAD-ZIG-SW

MTrab, 2023-12-18:
### Problem description

Tried generating a quirk using https://github.com/zigpy/quirk-generator but the device is still matched to `zigpy.device.Device` after being added again.

### Solution description

Correct matching of the device

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=64447, maximum_buffer_size=80, maximum_incoming_transfer_size=160, server_mask=0, maximum_outgoing_transfer_size=160, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0xfffe",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0007"
      ],
      "output_clusters": [
        "0x0000",
        "0x0001",
        "0x0012"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0xfffe",
      "input_clusters": [
        "0x0007"
      ],
      "output_clusters": [
        "0x0012"
      ]
    },
    "3": {
      "profile_id": "0x0104",
      "device_type": "0xfffe",
      "input_clusters": [
        "0x0007"
      ],
      "output_clusters": [
        "0x0012"
      ]
    },
    "4": {
      "profile_id": "0x0104",
      "device_type": "0xfffe",
      "input_clusters": [
        "0x0007"
      ],
      "output_clusters": [
        "0x0012"
      ]
    },
    "5": {
      "profile_id": "0x0104",
[... 789 lines cut ...]
                ],
                OUTPUT_CLUSTERS: [
                    MultistateInput.cluster_id,
                ],
            },
            5: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    OnOffConfiguration.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    MultistateInput.cluster_id,
                ],
            },
        },
    }

```

</details>

### Additional information

@TheJulianJES as requested :)

TheJulianJES, 2023-12-18:
How did you "install" the custom quirk?

MTrab, 2023-12-18:
Same as the other few quirks I have installed - put the file in `/config/custom_zha_quirks/` and have the following in my configuration file

```
zha:
  custom_quirks_path: /config/custom_zha_quirks/
```

Works like a charm for the other custom_quirks, just not this one.

TheJulianJES, 2023-12-18:
Hmm, try enabling [ZHA debug logs](https://www.home-assistant.io/integrations/zha/#debug-logging), then restart HA. The logs should contain more information on why zigpy doesn't match that quirk.
Also, check that `Loaded custom quirks. Please contribute them to` [...] is present in the logs.

MTrab, 2023-12-19:
I have a pretty huge setup, so enabling logs as stated in the docs is just absolutely FLOODING my logs, making it impossible to differentiate this single device.

Anything in particular I can use as search phrase?

TheJulianJES, 2023-12-24:
Try just searching for `QUAD-ZIG-SW`

bangert, 2024-10-26:
i am new to zha but would like to see support for this device. @MTrab did you get it to work at some point?

could you share the quirks file that didnt work?

Thanks

TheJulianJES, 2024-10-26:
Check:
- https://github.com/zigpy/zha-device-handlers/pull/3400

closed 2025-02-06 by TheJulianJES