zigpy/zha-device-handlers #2847: [Device Support Request] TS0021  _TZ3210_3ulg9kpo Tuya 2 gang switch

BenRoe, 2023-12-19:
### Problem description

Tuya TS0021 is found in ZHA, but offers no functionality. Shows a switch control that does nothing.

### Solution description

Add two way switch function.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>
[image]

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0402",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0500",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3210_3ulg9kpo",
  "model": "TS0021",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2023.12.3",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.11.6",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "6.1.21-v8+",
    "run_as_root": true
  },
  "custom_components": {
    "powercalc": {
[... 353 lines cut ...]

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

### Additional information

At zigbee2mqtt is already a open PR for that device.
https://github.com/Koenkk/zigbee2mqtt/discussions/19765

The device can be found here https://www.aliexpress.com/item/1005006126584104.html

accik, 2023-12-21:
[image]
I also have TS0021 but it's showing as "smoke". I can provide more info if needed, hope to see this included in ZHA.

CoarsePiglet, 2024-01-02:
I also have this device but for me it shows Opening as the Sensor
[image]

jaimezapa, 2024-02-03:
Hi, I had the same issue here. Didn't manage to write a quirk but instead wrote an automation that triggers when the error shows up in the logs. Works perfectly fine. From the "unknow cluster command " message you can determine both the button and the type of tap. See automation below. Hope it helps ;) 

alias: Boton setup jaime
description: ""
trigger:
  - platform: event
    event_type: system_log_event
    event_data:
      name: zigpy.zcl
    id: liso
condition:
  - condition: template
    value_template: "{{ \"0xCECD:1:0xef00\" in trigger.event.data.message[0] }}"
    enabled: true
action:
  - choose:
      - conditions:
          - condition: template
            value_template: "{{ \"\\\\x02\\\\x04\" in trigger.event.data.message[0] }}"
        sequence:
          - choose:
              - conditions:
                  - condition: template
                    value_template: "{{ \"\\\\x01\\\\x00\" in trigger.event.data.message[0] }}"
                sequence:
                  - service: light.toggle
                    target:
                      entity_id: light.suite_all_studio_all
                    data: {}
                alias: single
              - conditions:
                  - condition: template
                    value_template: "{{ \"\\\\x01\\\\x01\" in trigger.event.data.message[0] }}"
                sequence:
                  - service: switch.toggle
                    target:
                      entity_id: switch.adaptive_lighting_sleep_mode_al_studio
                    data: {}
                alias: double
              - conditions:
                  - condition: template
                    value_template: "{{ \"\\\\x01\\\\x02\" in trigger.event.data.message[0] }}"
                sequence: []
                alias: long
        alias: boton liso
      - conditions:
          - condition: template
            value_template: "{{ \"\\\\x01\\\\x04\" in trigger.event.data.message[0] }}"
        sequence:
          - choose:
              - conditions:
                  - condition: template
                    value_template: "{{ \"\\\\x01\\\\x00\" in trigger.event.data.message[0] }}"
                sequence:
                  - if:
                      - condition: device
                        type: is_off
                        device_id: 1220c3eafc6d7f8b55d55f95e5ebbfc4
                        entity_id: 50111a494b1a6048f7a9cbd971090e28
                        domain: light
                    then:
                      - service: light.turn_on
                        target:
                          device_id: []
                          area_id: []
                          entity_id: light.ewelink_ck_bl702_al_01_7009_z102lg03_1_light_8
                        data: {}
                      - service: light.turn_on
                        target:
                          device_id: []
                          area_id: []
                          entity_id: light.ewelink_ck_bl702_al_01_7009_z102lg03_1_light_8
                        data:
                          rgb_color:
                            - 255
                            - 56
                            - 0
                      - service: adaptive_lighting.set_manual_control
                        data:
                          manual_control: true
                          entity_id: switch.adaptive_lighting_adaptative_lights_1
                    else:
                      - service: adaptive_lighting.set_manual_control
                        data:
                          manual_control: false
                          entity_id: switch.adaptive_lighting_adaptative_lights_1
                      - delay:
                          hours: 0
                          minutes: 0
                          seconds: 5
                          milliseconds: 0
                      - service: light.turn_off
                        target:
                          device_id: []
                          area_id: []
                          entity_id: light.all_lights_hall_lights
                        data: {}
                alias: single
              - conditions:
                  - condition: template
                    value_template: "{{ \"\\\\x01\\\\x01\" in trigger.event.data.message[0] }}"
                sequence: []
                alias: double
              - conditions:
                  - condition: template
                    value_template: "{{ \"\\\\x01\\\\x02\" in trigger.event.data.message[0] }}"
                sequence:
                  - service: switch.toggle
                    target:
                      entity_id: switch.setup_jaime_switch
                    data: {}
                alias: long
        alias: boton rugoso
    default: []
  - delay:
      hours: 0
      minutes: 0
      seconds: 3
      milliseconds: 0
    enabled: true
mode: single

hexalm, 2024-02-12:
Well I've been making my first foray into zigpy for these switches. I've got the signature piece down, but could probably do with  little hint on what I've got wrong below. Presumably it's just the clusters I'm mapping don't have the needed functionality. I've tried a number of combinations without full success on using button presses to do anything.

Currently the devce is recognized, but looks like I have not properly assigned datapoint handlers (it's clear the TuyaNewManufCluster is falling through all of the things it can handle). I'm a bit hazy on how to determine if an existing class works or what. (The zigbee2mqtt implementation didn't look complicated, but I stll haven't grasped how it works enough to translate to zigpy.)

Including log results when I press the buttons now (I could have sworn dp 10 for batteries hadn't done this before when I understood ess of whqat I was doing, but oh well). It pretty straightforward, though

```
2024-02-11 20:23:20.264 DEBUG (MainThread) [zigpy.zcl] [0x3721:1:0xef00] No datapoint handler for TuyaDatapointData(dp=2, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>))
2024-02-11 20:29:15.481 DEBUG (MainThread) [zigpy.zcl] [0x3721:1:0xef00] No datapoint handler for TuyaDatapointData(dp=1, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x00', *payload=<enum8.undefined_0x00: 0>))
2024-02-11 21:19:40.839 DEBUG (MainThread) [zigpy.zcl] [0x3721:1:0xef00] No datapoint handler for TuyaDatapointData(dp=10, data=TuyaData(dp_type=<TuyaDPType.VALUE: 2>, function=0, raw=b'\x00\x00\x00\x00', *payload=0))
```

My quirk:
```
"""Tuya 2 Button Remote."""

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import Basic, OnOff, Ota, PowerConfiguration, Time
from zigpy.zcl.clusters.security import IasZone

from zhaquirks.const import (
    BUTTON_1,
    BUTTON_2,
    COMMAND,
    DEVICE_TYPE,
    DOUBLE_PRESS,
    ENDPOINT_ID,
    ENDPOINTS,
    INPUT_CLUSTERS,
    LONG_PRESS,
    MODEL,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    SHORT_PRESS,
)
from zhaquirks.tuya import TuyaNewManufCluster, TuyaNoBindPowerConfigurationCluster, TuyaSmartRemoteOnOffCluster, TuyaSwitch

class TuyaSmartRemote0021(TuyaSwitch):
    """Tuya 2-button remote device."""

    signature = {
        MODEL: "TS0021",
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    IasZone.cluster_id,
                    TuyaNewManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    IasZone.cluster_id,
                    TuyaNewManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.REMOTE_CONTROL,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaNoBindPowerConfigurationCluster,
                    TuyaSmartRemoteOnOffCluster,
                    TuyaNewManufCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.REMOTE_CONTROL,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaNoBindPowerConfigurationCluster,
                    TuyaSmartRemoteOnOffCluster,
                    TuyaNewManufCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
        },
    }

    device_automation_triggers = {
        (SHORT_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: SHORT_PRESS},
        (LONG_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: LONG_PRESS},
        (DOUBLE_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: DOUBLE_PRESS},
        (SHORT_PRESS, BUTTON_2): {ENDPOINT_ID: 2, COMMAND: SHORT_PRESS},
        (LONG_PRESS, BUTTON_2): {ENDPOINT_ID: 2, COMMAND: LONG_PRESS},
        (DOUBLE_PRESS, BUTTON_2): {ENDPOINT_ID: 2, COMMAND: DOUBLE_PRESS},
    }
```

I'm moderately experienced with python and handy with grep, so I can easily troubleshoot more or provide more info, but I'm hoping someone witth more knowledge of this software can easily spot the issue!

CoarsePiglet, 2024-02-17:
@jaimezapa Do you mind posting your reply as code block because I cannot get it to work. Thanks

CoarsePiglet, 2024-02-18:
@jaimezapa I have just tried an even simpler version and it doesn't work. I managed to troubleshoot it and the automation is not triggering because ZHA is firing "DEBUG" events and the system_log_event type only accepts ERROR and WARNING...

When I manually create a notification using system_log.write, with default level, it does trigger the automation. But not when using DEBUG.

Is ZHA sending ERROR/WARNING level into your log on your end? If so, how did you manage it to do that?

jaimezapa, 2024-02-18:
I see. Yesterday I upgraded HA to 2024.2 and looks like they've changed the log type. In the 2023.10 version zigpy would throw this as a warning instead of debug. Guess we'll have to come up with a workaround...

 [zigpy.zcl] [0xB9C5:1:0xef00] No explicit handler for cluster command 0x06: b'\x02r\x02\x04\x00\x01\x00'

jaimezapa, 2024-02-18:
Maybe monitor the stdout logs from outside the container and throw MQTT messages?

jaimezapa, 2024-02-18:
Feel free to try this code, I run it as a service and works like a charm. Note that debug should stay enabled for zigpy (tomorrow I´ll see if I can filter the with logger first)

'''
import os
import sys
import time
import re
import subprocess 
# Path to the Home Assistant log file
log_file_path = "volumes/home_assistant/home-assistant.log"

# Regular expression pattern_lucia to match the specified entry
pattern_lucia = re.compile(r"\[0xB9C5:1:0xef00\]")
pattern_jaime = re.compile(r"\[0xCECD:1:0xef00\] Unknown cluster command")
# Webhook URL for sending the curl command
webhook_url_lucia = "http://your-HA:8123/api/webhook/Setup-lucia"
webhook_url_jaime = "http://your-HA:8123/api/webhook/Setup-jaime"
# Function to send the curl command

def send_curl_command(code,webhook):
    curl_command = ["curl", "-X", "POST", "-d", f"code={code}", webhook]
    subprocess.run(curl_command)

# Function to check for the pattern_lucia in the log file
def check_log():
    with open(log_file_path, "r") as f:
        # Move the file pointer to the end of the file
        f.seek(0, os.SEEK_END)
        
        # Continuously monitor new lines added to the log file
        while True:
            line = f.readline()
            if line:
                # Check if the line contains the specified pattern_lucia
                if pattern_lucia.search(line):
                    # Extract the code from the log entry
                    match = re.search(r"b'([^']+)'", line)
                    if match:
                        code = match.group(1)
                        # Send the curl command with the code
                        send_curl_command(code,webhook_url_lucia)
                        
                # Check if the line contains the specified pattern_jaime
                if pattern_jaime.search(line):
                    # Extract the code from the log entry
                    match = re.search(r"b'([^']+)'", line)
                    if match:
                        code = match.group(1)
                        # Send the curl command with the code
                        send_curl_command(code,webhook_url_jaime)
            else:
                # Sleep for a short interval before checking again
                time.sleep(0.1)

try:
    # Start monitoring the log file
    check_log()
except KeyboardInterrupt:
    # Handle keyboard interrupt
    sys.exit(0)
except Exception as e:
    # Handle other exceptions
    print("Error:", e)

'''

NdK73, 2024-03-16:
I have the same switch. The first time I added it, it got recognized as smoke detector (like @accik 's one) then I removed it and added a second time and it got recognized as an opening detector (like @CoarsePiglet 's one). Removing and adding it again seems to retain the opening configuration.
I also tried initiating the pairing process from both buttons and tried to repeatly press 'em during pairing. No change. :(

NdK73, 2024-03-17:
I'm trying to write a quirk based on @hexalm 's code. But it seems ZHA does not use my quirk, just its default code:
- if I insert a syntax error, I find the error in the logs => the quirk gets read
- I tried with 'MODEL: "TS0021"' and with 'MODELS_INFO: [("_TZ3210_3ulg9kpo", "TS0021")]' (after importing MODELS_INFO, obv) => no difference
- I change the quirk, save it, remove the device from ZHA, restart HA, re-add the device

Enabling debug log, I can see:
```2024-03-17 09:40:55.501 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xAF62](TS0021): starting availability checks - interval: 87
2024-03-17 09:40:55.501 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] Discovering entities for device: a4:c1:38:a1:0b:25:6a:cf-_TZ3210_3ulg9kpo TS0021
2024-03-17 09:40:55.501 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] Attempting to discover quirks v2 entities for device: a4:c1:38:a1:0b:25:6a:cf-_TZ3210_3ulg9kpo TS0021
2024-03-17 09:40:55.501 DEBUG (MainThread) [homeassistant.components.zha.core.discovery] Device: a4:c1:38:a1:0b:25:6a:cf-_TZ3210_3ulg9kpo TS0021 is not a quirks v2 device - skipping discover_quirks_v2_entities
2024-03-17 09:40:55.501 DEBUG (MainThread) [homeassistant.components.zha.core.gateway] device - 0xAF62:a4:c1:38:a1:0b:25:6a:cf entering async_device_initialized - is_new_join: True
2024-03-17 09:40:55.501 DEBUG (MainThread) [homeassistant.components.zha.core.gateway] device - 0xAF62:a4:c1:38:a1:0b:25:6a:cf has joined the ZHA zigbee network
2024-03-17 09:40:55.502 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xAF62](TS0021): started configuration
2024-03-17 09:40:55.502 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0xAF62:ZDO](TS0021): 'async_configure' stage succeeded
2024-03-17 09:40:55.505 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0xAF62:1:0x0001]: Performing cluster binding
2024-03-17 09:40:55.507 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0xAF62:1:0x0000]: Configuring cluster attribute reporting
2024-03-17 09:40:55.507 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0xAF62:1:0x0000]: finished cluster handler configuration
2024-03-17 09:40:55.507 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0xAF62:1:0xef00]: Performing cluster binding
2024-03-17 09:40:55.508 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0xAF62:1:0x0019]: finished cluster handler configuration
2024-03-17 09:40:55.508 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0xAF62:1:0x0019]: finished cluster handler configuration
2024-03-17 09:40:55.707 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0xAF62:1:0x0500]: started IASZoneClusterHandler configuration
2024-03-17 09:40:55.784 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0xAF62:1:0x0001]: bound 'power' cluster: Status.SUCCESS
2024-03-17 09:40:55.784 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0xAF62:1:0x0001]: Configuring cluster attribute reporting
2024-03-17 09:40:56.206 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0xAF62:1:0x0500]: bound 'ias_zone' cluster: Status.SUCCESS
2024-03-17 09:40:56.284 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0xAF62:1:0x0001]: Failed to configure reporting for '[]' on 'power' cluster: [ConfigureReportingResponseRecord(status=<Status.SUCCESS: 0>), ConfigureReportingResponseRecord(status=<Status.SUCCESS: 0>)]
2024-03-17 09:40:56.284 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0xAF62:1:0x0001]: Successfully configured reporting for '{'battery_percentage_remaining', 'battery_voltage'}' on 'power' cluster
2024-03-17 09:40:56.284 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0xAF62:1:0x0001]: finished cluster handler configuration
2024-03-17 09:40:58.342 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0xAF62:1:0x0500]: Enroll requested
2024-03-17 09:40:58.342 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0xAF62:1:0x0500]: wrote cie_addr: 00:12:4b:00:25:90:6b:f1 to 'ias_zone' cluster
2024-03-17 09:40:58.342 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0xAF62:1:0x0500]: Sending pro-active IAS enroll response
2024-03-17 09:40:58.343 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0xAF62:1:0x0500]: finished IASZoneClusterHandler configuration```

Any hints? Tks.

hexalm, 2024-03-17:
@NdK73 apologies-I found out a bit ago that the quirk signature was not being recognized but, haven't gotten back to working on this yet.  I must have had a cached version or something when I thought it was working.

If you go to the service and pick the option to "configure zigbee device" (three dots next to the 'reconfigure' option) you can see the signature. I know I had it being recognized before I added the second endpoint (but still I never got the thing to give me any button entities). Tuya's stuff is just weird and I could stand help from someone who has worked on the ZHA handlers for them. 

Maybe I'll look at this again today, but I'm planning to redo my architecture for HA and might be switching to zigbee2mqtt.

XBigTK13X, 2024-03-19:
I am poking around at the quirks a bit to understand how they are written. Is it common to port zigbee2mqtt functionality as a quirk? I think I found this device's configs in that system.

https://github.com/Koenkk/zigbee2mqtt/discussions/19765

MikeInMaine, 2024-03-26:
> I'm moderately experienced with python and handy with grep, so I can easily troubleshoot more or provide more info, but I'm hoping someone witth more knowledge of this software can easily spot the issue!

Did you make any progress with this quirk?  I wasn't able to get yours to load for the TS0021 switch without making some changes, such as removing the second endpoints

Drethak, 2024-04-03:
So I somehow managed to get it so that switches were showing up in the entities instead of an IASZone sensor, however, neither responded to button presses from the actual device. Has anyone else made any progress with this? I was using a slightly modified version of the TS0042 code that already exists as a working quirk.

NdK73, 2024-04-19:
Probably the users saying it's working are using it via a Tuya gateway.
Too bad I've had no time yet to look at the issue :(

MikeInMaine, 2024-04-21:
I ended up ordering the following and it works great out of the box with zha.  I forget to check the compatibility list before ordering items and one really should.  This button has at least two functions for 1. click and 2. double-click and maybe a 3. long press.

https://www.aliexpress.us/item/3256805811361539.html?spm=a2g0o.order_list.order_list_main.29.3a951802SJXgMV&gatewayAdapt=glo2usa

> It's really shame, as I wanted to get into zigbee ecosystem and use a switch instead of my phone for two devices, I ordered this switch as it has 2 gangs, waited for a month and looked forward, just to find out about this issue... now I have a coordinator without any zigbee devices.

MikeInMaine, 2024-04-21:
> Believe it or not, the same device is in my cart, and I hesitated between
> that one and the one from OP. Unfortunately, I made a mistake and ordered
> this one just because it has 2 gangs.

lol, I did the same!

ariboehme, 2024-04-25:
I have the same issue here: I use HA with Tuya integration using a Tuya ZigBee gateway. In the first addition he appeared as Smoke, in the second as Contact. Pressing the second button shows it as Open in HA. But the rest doesn't work.

[image]

dtaylor, 2024-04-29:
I bought this same switch and went down the rabbit hole as well. The signature part was pretty easy to get right, but I didn't get nearly as close as @netizeni did.

MikeInMaine, 2024-05-06:
> This really bothers me, so I spend additional time to try to make it work.
> I've managed to add a correct signature combination, so now the device will
> show a switch toggle, and even tho it doesn't work, at least it's not
> "Opening" sensor anymore. Also, "Zigbee info" will now show that custom
> quirk is being used and under automation, you can see "first button
> pressed", double pressed and long pressed.
>
> Unfortunately, single, double and long press don't work, as it seems the
> device doesn't fire zha_event. I will share the custom quirk below, so
> hopefully someone will join in fixing it. It would be also helpful if you
> guys, who have this device, try different combination of clusters (from
> different switch classes of this project) inside INPUT_CLUSTERS and
> OUTPUT_CLUSTERS arrays of replacement objects and report back if you find
> something works. My assumption is once replacement objects gets filled
> with correct clusters, the device will start firing events and will work as
> it should.
> Custom quirk
>
> """Tuya 2 Button Remote."""
> from zigpy.profiles import zhafrom zigpy.quirks import CustomDevicefrom zigpy.zcl.clusters.general import Basic, OnOff, Ota, PowerConfiguration, Timefrom zigpy.zcl.clusters.security import IasZone
> from zhaquirks.const import (
>     BUTTON_1,
>     BUTTON_2,
>     COMMAND,
>     DEVICE_TYPE,
> [... 103 lines cut ...]
>     device_automation_triggers = {
>         (SHORT_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: SHORT_PRESS},
>         (LONG_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: LONG_PRESS},
>         (DOUBLE_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: DOUBLE_PRESS},
>     }

I bought a button that works OOtB from aliexpress for something like $5
delivered to the USA.  At my hourly rate, that's about  1/10th of an hour.
I threw this one away. I use the zigbee compatibility list before buying
anything these days.

MikeInMaine, 2024-05-07:
> I understand your point of view, Mike, it's valid, but at least in my
> case, this is not about the price itself. It's about helping the community
> and all the people who have or will have this switch. If there weren't all
> these people selflessly contributing before us, we wouldn't be able to use
> zha integration.
>
> All in all, I hope someone will join in to help and make this work. I
> assume Matt West would be able to solve this easily, but I don't want to
> tag him again.

I definitely agree. On the other hand, manufacturers who build nonstandard
stuff shouldn't be supported.

rbaron, 2024-07-05:
I finally came up with a hacky quirk that seem to work okay on my system (core `2024.7.0`):

```python
"""Device handler the TS0021."""
from zhaquirks import EventableCluster
from zigpy.quirks import CustomDevice
from zhaquirks.const import (
    ARGS,
    ATTRIBUTE_ID,
    CLUSTER_ID,
    COMMAND_ATTRIBUTE_UPDATED,
    COMMAND,
    SKIP_CONFIGURATION,
    DEVICE_TYPE,
    ENDPOINT_ID,
    ENDPOINTS,
    INPUT_CLUSTERS,
    LONG_PRESS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PARAMS,
    PROFILE_ID,
    SHORT_PRESS,
    DOUBLE_PRESS,
    VALUE,
)
from zhaquirks.tuya import (
    TUYA_CLUSTER_ID,
    TuyaNewManufCluster,
    DPToAttributeMapping,
)

BTN_1 = "Button 1"
BTN_2 = "Button 2"

COMMAND_1_PRESSED = 'btn_1_pressed'
COMMAND_2_PRESSED = 'btn_2_pressed'

class TuyaCustomCluster(TuyaNewManufCluster, EventableCluster):
    dp_to_attribute = {
        1: DPToAttributeMapping(
            TuyaNewManufCluster.ep_attribute,
            COMMAND_1_PRESSED,
        ),
        2: DPToAttributeMapping(
            TuyaNewManufCluster.ep_attribute,
            COMMAND_2_PRESSED,
        ),
    }

    data_point_handlers = {
        1: "_dp_2_attr_update",
        2: "_dp_2_attr_update",
    }

class TS0021(CustomDevice):
    signature = {
        MODELS_INFO: [("_TZ3210_3ulg9kpo", "TS0021")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: 0x0104,
                DEVICE_TYPE: 0x0402,
                INPUT_CLUSTERS: [
                    0x0000,
                    0x0001,
                    0x0500,
                    0xef00,
                ],
                OUTPUT_CLUSTERS: [
                    0x000a,
                    0x0019,
                ],
            },
        },
    }

    replacement = {
        SKIP_CONFIGURATION: False,
        ENDPOINTS: {
            1: {
                PROFILE_ID: 0x0104,
                DEVICE_TYPE: 0x0402,
                INPUT_CLUSTERS: [
                    0x0000,
                    0x0001,
                    0x0500,
                    TuyaCustomCluster,
                ],
                OUTPUT_CLUSTERS: [
                    0x000a,
                    0x0019,
                ],
            },
        },
    }

    device_automation_triggers = {
        (SHORT_PRESS, BTN_1): {
            COMMAND: COMMAND_ATTRIBUTE_UPDATED,
            CLUSTER_ID: TUYA_CLUSTER_ID,
            ENDPOINT_ID: 1,
            ARGS: {ATTRIBUTE_ID: COMMAND_1_PRESSED, VALUE: 0},
        },
        (DOUBLE_PRESS, BTN_1): {
            COMMAND: COMMAND_ATTRIBUTE_UPDATED,
            CLUSTER_ID: TUYA_CLUSTER_ID,
            ENDPOINT_ID: 1,
            ARGS: {ATTRIBUTE_ID: COMMAND_1_PRESSED, VALUE: 1},
        },
        (LONG_PRESS, BTN_1): {
            COMMAND: COMMAND_ATTRIBUTE_UPDATED,
            CLUSTER_ID: TUYA_CLUSTER_ID,
            ENDPOINT_ID: 1,
            ARGS: {ATTRIBUTE_ID: COMMAND_1_PRESSED, VALUE: 2},
        },
        (SHORT_PRESS, BTN_2): {
            COMMAND: COMMAND_ATTRIBUTE_UPDATED,
            CLUSTER_ID: TUYA_CLUSTER_ID,
            ENDPOINT_ID: 1,
            ARGS: {ATTRIBUTE_ID: COMMAND_2_PRESSED, VALUE: 0},
        },
        (DOUBLE_PRESS, BTN_2): {
            COMMAND: COMMAND_ATTRIBUTE_UPDATED,
            CLUSTER_ID: TUYA_CLUSTER_ID,
            ENDPOINT_ID: 1,
            ARGS: {ATTRIBUTE_ID: COMMAND_2_PRESSED, VALUE: 1},
        },
        (LONG_PRESS, BTN_2): {
            COMMAND: COMMAND_ATTRIBUTE_UPDATED,
            CLUSTER_ID: TUYA_CLUSTER_ID,
            ENDPOINT_ID: 1,
            ARGS: {ATTRIBUTE_ID: COMMAND_2_PRESSED, VALUE: 2},
        },
    }
    
```

Drethak, 2024-07-16:
> I finally came up with a hacky quirk that seems to work okay on my system (core `2024.7.0`):
> 
> ```python
> """Device handler the TS0021."""
> from zhaquirks import EventableCluster
> from zigpy.quirks import CustomDevice
> from zhaquirks.const import (
>     ARGS,
>     ATTRIBUTE_ID,
>     CLUSTER_ID,
>     COMMAND_ATTRIBUTE_UPDATED,
>     COMMAND,
>     SKIP_CONFIGURATION,
>     DEVICE_TYPE,
>     ENDPOINT_ID,
>     ENDPOINTS,
>     INPUT_CLUSTERS,
>     LONG_PRESS,
>     MODELS_INFO,
>     OUTPUT_CLUSTERS,
>     PARAMS,
>     PROFILE_ID,
>     SHORT_PRESS,
>     DOUBLE_PRESS,
>     VALUE,
> )
> from zhaquirks.tuya import (
>     TUYA_CLUSTER_ID,
>     TuyaNewManufCluster,
>     DPToAttributeMapping,
> )
> 
> 
> BTN_1 = "Button 1"
> BTN_2 = "Button 2"
> 
> COMMAND_1_PRESSED = 'btn_1_pressed'
> COMMAND_2_PRESSED = 'btn_2_pressed'
> 
> 
> class TuyaCustomCluster(TuyaNewManufCluster, EventableCluster):
>     dp_to_attribute = {
>         1: DPToAttributeMapping(
>             TuyaNewManufCluster.ep_attribute,
>             COMMAND_1_PRESSED,
>         ),
>         2: DPToAttributeMapping(
>             TuyaNewManufCluster.ep_attribute,
>             COMMAND_2_PRESSED,
>         ),
>     }
> 
>     data_point_handlers = {
>         1: "_dp_2_attr_update",
>         2: "_dp_2_attr_update",
>     }
> 
> 
> class TS0021(CustomDevice):
>     signature = {
>         MODELS_INFO: [("_TZ3210_3ulg9kpo", "TS0021")],
>         ENDPOINTS: {
>             1: {
>                 PROFILE_ID: 0x0104,
>                 DEVICE_TYPE: 0x0402,
>                 INPUT_CLUSTERS: [
>                     0x0000,
>                     0x0001,
>                     0x0500,
>                     0xef00,
>                 ],
>                 OUTPUT_CLUSTERS: [
>                     0x000a,
>                     0x0019,
>                 ],
>             },
>         },
>     }
> 
>     replacement = {
>         SKIP_CONFIGURATION: False,
>         ENDPOINTS: {
>             1: {
>                 PROFILE_ID: 0x0104,
>                 DEVICE_TYPE: 0x0402,
>                 INPUT_CLUSTERS: [
>                     0x0000,
>                     0x0001,
>                     0x0500,
>                     TuyaCustomCluster,
>                 ],
>                 OUTPUT_CLUSTERS: [
>                     0x000a,
>                     0x0019,
>                 ],
>             },
>         },
>     }
> 
>     device_automation_triggers = {
>         (SHORT_PRESS, BTN_1): {
>             COMMAND: COMMAND_ATTRIBUTE_UPDATED,
>             CLUSTER_ID: TUYA_CLUSTER_ID,
>             ENDPOINT_ID: 1,
>             ARGS: {ATTRIBUTE_ID: COMMAND_1_PRESSED, VALUE: 0},
>         },
>         (DOUBLE_PRESS, BTN_1): {
>             COMMAND: COMMAND_ATTRIBUTE_UPDATED,
>             CLUSTER_ID: TUYA_CLUSTER_ID,
>             ENDPOINT_ID: 1,
>             ARGS: {ATTRIBUTE_ID: COMMAND_1_PRESSED, VALUE: 1},
>         },
>         (LONG_PRESS, BTN_1): {
>             COMMAND: COMMAND_ATTRIBUTE_UPDATED,
>             CLUSTER_ID: TUYA_CLUSTER_ID,
>             ENDPOINT_ID: 1,
>             ARGS: {ATTRIBUTE_ID: COMMAND_1_PRESSED, VALUE: 2},
>         },
>         (SHORT_PRESS, BTN_2): {
>             COMMAND: COMMAND_ATTRIBUTE_UPDATED,
>             CLUSTER_ID: TUYA_CLUSTER_ID,
>             ENDPOINT_ID: 1,
>             ARGS: {ATTRIBUTE_ID: COMMAND_2_PRESSED, VALUE: 0},
>         },
>         (DOUBLE_PRESS, BTN_2): {
>             COMMAND: COMMAND_ATTRIBUTE_UPDATED,
>             CLUSTER_ID: TUYA_CLUSTER_ID,
>             ENDPOINT_ID: 1,
>             ARGS: {ATTRIBUTE_ID: COMMAND_2_PRESSED, VALUE: 1},
>         },
>         (LONG_PRESS, BTN_2): {
>             COMMAND: COMMAND_ATTRIBUTE_UPDATED,
>             CLUSTER_ID: TUYA_CLUSTER_ID,
>             ENDPOINT_ID: 1,
>             ARGS: {ATTRIBUTE_ID: COMMAND_2_PRESSED, VALUE: 2},
>         },
>     }
>     
> ```

Unfortunately this hasn't seemed to fix it for me, it's still either coming up as an IasZone sensor or a door opening sensor. Am I missing something else perhaps? (I just made a new custom quirk using this code)

rbaron, 2024-07-16:
@Drethak do you see the quirk active like this:

[image]

It will still show the "Opening" sensor. What the quirk enables are the automations for the buttons:

[image]

Could you check if those are available for you? Otherwise we'd need to take a look at the logs to figure out why the quirk is not kicking in.

Drethak, 2024-07-17:
@rbaron Thank you! I wasn't even aware this was a thing, many thanks for pointing this out.

NdK73, 2024-08-01:
Can confirm it "mostly" works. Buttons are detected, but it still gets reported as a smoke (or door) sensor and battery is fixed at 0%. But at least it can be used in automations! Great!
Hope it gets "backported" in ZHA soon.

closed 2024-08-11 by TheJulianJES

justinreeves00, 2024-08-17:
this is in production now? I am still showing these TS0021 switches as openers, it's recognizing button presses but doesn't seem to show anything but single button press actions. am I just doing something wrong? I have reset them multiple times. 
[image]

rbaron, 2024-08-18:
@justinreeves00 indeed even with quirk it still shows up as opener. The events you see on the logbook are not button presses, but "attribute updates". This is the custom way this button implements button presses. The updates are different for each event (single press A, double press B etc), but the UI does not show that.

The only thing the quirk currently does is enabling automations [like in this comment](https://github.com/zigpy/zha-device-handlers/issues/2847#issuecomment-2230199113). If you can set up different automations for single, double, long press and they are triggered, I'd say it's working. Otherwise let me know and I can try helping debug.

GarckaMan, 2026-02-07:
> Hi Since 2026.2 version the quirk doens't work anymore. Anyone else has the same issue ?

Yes. My 2 gang switches stopped working after updating to 2026.2. I'm not using the quirk, I was just using default ZHA.

PR #3247 (2024-08-11): Add support for the Tuya TS0021 2-gang switch
Fixes #2847, fixes #3007

## Proposed change
This PR adds support for the Tuya TS0021 2-button switch. Here's an [example listing](https://www.aliexpress.com/item/1005004644217026.html) for this device.

## Additional information
The implementation was tested on Home Assistant core `2024.7.0`.

## Checklist
- [x] The changes are tested and work correctly
- [x] `pre-commit` checks pass / the code has been formatted using Black
- [x] Tests have been added to verify that the new code works