zigpy/zha-device-handlers #2851: [Device Support Request]  TY0201 - Temperature + Humdity Sensor

metlinux, 2023-12-20:
### Problem description

This device is recognized and temperature is working, but the device also supports humidity which is unfortunately not working, It is missing.

### Solution description

Support for retrieving and displaying the humidity from this device

### Screenshots/Video

<details><summary>Screenshots/Video</summary>
[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0302",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0402",
        "0x0405"
      ],
      "output_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_zl1kmjqx",
  "model": "TY0201",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2023.12.0",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.11.6",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Amsterdam",
    "os_name": "Linux",
    "os_version": "6.6.7-arch1-1",
    "run_as_root": true
  },
  "custom_components": {
    "hacs": {
      "version": "1.33.0",
      "requirements": [
[... 322 lines cut ...]

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

### Additional information

The device seems (from physical aspect and the id code) as the 
Model IH-K009 manufactured by Tuya
Zigbee ID: TS0201 | _TZ3000_dowj6gyi | _TZ3000_8ybe88nf
Maybe it is the newer version of this and the solution can be re-used

CoderMal, 2024-01-12:
I have the same device with the same issue. However, if I run the Sonoff Bridge Pro (Tasmota) in non TCP mode (zha), Tasmota sees temperature, signal, battery and humidity properly.

chrizzo84, 2024-05-18:
This is working for me :) Needed some reloading, restarting and reconnecting of the device, but finally it seems to be working! Thx  @llucax

infinitytec, 2024-06-08:
@llucax that seems to be working for one of mine. I'm not sure what started it working for it, but I have several others that are not working still. Same manufacturer as yours. 

I also have some units that identify as TS0201 (instead of TY) and have the manufacturer "_TZ3000_dowj6gyi" and they just work without needing this quirk added. They look identical to these.

infinitytec, 2024-06-13:
> @llucax that seems to be working for one of mine. I'm not sure what started it working for it, but I have several others that are not working still. Same manufacturer as yours. However, it seems that if this quirk applies, I get "Firmware: 0x10013001".
> 
> I also have some units that identify as TS0201 (instead of TY) and have the manufacturer "_TZ3000_dowj6gyi" and they just work without needing this quirk added. They look identical to these.

OK, I actually got it to work? Kinda?

The key for me was to have it exactly like this, like you said (but I misunderstood and kept adding "TY0201" to the second model information:

`signature = {                                          
    MODELS_INFO: [
        ("_TZ3000_bjawzodf", "TY0201"),
        ("_TZ3000_zl1kmjqx", ""),
    ],`

Also, I was wrong about the firmware thing I think. I needed to go to Zigbee Info and identify the `Quirk: ty0201.TemperatureHumiditySensorWithScreen` line instead.

I have five of these sensors. I thought I got them all running great but then discovered last night that only one of them is reporting. They are all reporting that they are using the quirk.

PR #3201 (2024-06-26): Add Tuya TY0201 temperature/humidity sensor
This seems to be a pretty common sensor, for which several models seem to exist : _TZ3000_bjawzodf and _TZ3000_zl1kmjqx

## Proposed change

Add support for  another humidity/temperature sensor

## Additional information
First time contributing here. And I am not sure it is completely correct, but 
this quirk is confirmed to work for a wide number of people. I welcome any comments
to improve this.

Fixes #2862, fixes #2851, fixes #2701

 

## Checklist

- [x] The changes are tested and work correctly
- [ ] `pre-commit` checks pass / the code has been [... cut]
closed 2024-06-26 by TheJulianJES