zigpy/zha-device-handlers #2852: [Device Support Request] ZY-M100 Series Presence Sensor (TS0601 by _TZE204_ijxvkhd0)

logan893, 2023-12-20:
### Problem description

Pairing with Home Assistant but no sensors or controls available. Advertised as Presence sensor for use with Tuya zigbee hub/gateway.

Model: ZY-M100-1 (Side wall version); there is also the ZY-M100-2 Ceiling version, which I do not have myself.

Shows up in ZHA as TS0601 by _TZE204_ijxvkhd0

This is a mains powered 24 GHz presence sensor, zigbee and side wall edition.

https://www.aliexpress.com/item/1005006128737558.html

### Solution description

Presence Sensor functionality

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

logan893, 2023-12-20:
If there's an easy way to convert the implementation into a quirk or something, zigbee2mqtt has support, even though it doesn't seem to work flawlessly yet.

https://github.com/Koenkk/zigbee2mqtt/issues/18237

logan893, 2023-12-21:
Here is the data from Manage Zigbee Device -> Signature.

It looks identical to the motion sensor already included in the latest ZHA code with class MmwRadarMotionGPP in ts0601_motion.py. Same input clusters and output clusters for endpoints 1 and 242.

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE204_ijxvkhd0",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

logan893, 2023-12-21:
I added _TZE204_ijxvkhd0 to the MmwRadarMotionGPP, same as used by _TZE204_qasjif9e ( https://github.com/zigpy/zha-device-handlers/issues/2510 ).

I get a sensor for Occupancy but it is always set to Clear and doesn't change. No other sensors or controls that I can see.

I can get light sensor data from TuyaIlluminanceMeasurement (Endpoint id: 1, Id: 0x0400, Type: in), attribute measured_value (id: 0x0000). It changes the value read when I cover the sensor.

All the occupancy sensing output is None (from TuyaOccupancySensing (Endpoint id: 1, Id: 0x0406, Type: in)), except for occupancy (0x0000) which is always zero (0).

MmwRadarManufCluster, mcu_version returns 1.1.8.
Firmware is 0x4a (74)

I'm out of ideas for the moment.

logan893, 2023-12-21:
Occupancy seems to work (can show either Clear and Detected) if I switch occupancy from using dp_1 attribute, to using dp_112. 

It's used in the zigbee2mqtt implementation for "presence" true/false.
            [112, 'presence',  tuya.valueConverter.trueFalse1],

I don't know how to create a custom sensor for the presence_state type attribute.

visata, 2023-12-21:
I ordered several of these sensors for testing purposes and discovered they didn't work out of the box since I use ZHA. I don't want to migrate to Zigbee2MQTT just because of this issue.

@logan893, can you share how you added '_TZE204_ijxvkhd0' to the MmwRadarMotionGPP? I would like to test this.

Thank you

logan893, 2023-12-21:
@visata I don't know if it's all proper, but this is what I have changed and added so far.

```
--- ts0601_motion.py.latest
+++ ts0601_motion.py
@@ -147,11 +147,16 @@
             0xEF6A: ("dp_106", t.enum8, True),
             0xEF6B: ("dp_107", t.enum8, True),
             0xEF6C: ("dp_108", t.uint32_t, True),
+            0xEF6D: ("dp_109", t.uint32_t, True),
+            0xEF6E: ("dp_110", t.uint32_t, True),
+            0xEF6F: ("dp_111", t.uint32_t, True),
+            #0xEF70: ("occupancy", t.uint32_t, True),
         }
     )

     dp_to_attribute: Dict[int, DPToAttributeMapping] = {
-        1: DPToAttributeMapping(
+        # was: 1: DPToAttributeMapping(
+        112: DPToAttributeMapping(
             TuyaOccupancySensing.ep_attribute,
             "occupancy",
         ),
@@ -209,6 +214,18 @@
             TuyaMCUCluster.ep_attribute,
             "dp_108",
         ),
+        109: DPToAttributeMapping(
+            TuyaMCUCluster.ep_attribute,
+            "dp_109",
+        ),
+        110: DPToAttributeMapping(
+            TuyaMCUCluster.ep_attribute,
+            "dp_110",
+        ),
+        111: DPToAttributeMapping(
+            TuyaMCUCluster.ep_attribute,
+            "dp_111",
+        ),
     }

     data_point_handlers = {
@@ -226,6 +243,10 @@
         106: "_dp_2_attr_update",
         107: "_dp_2_attr_update",
         108: "_dp_2_attr_update",
+        109: "_dp_2_attr_update",
+        110: "_dp_2_attr_update",
+        111: "_dp_2_attr_update",
+        112: "_dp_2_attr_update",
     }

@@ -399,6 +420,7 @@
             ("_TZE200_sfiy5tfs", "TS0601"),
             ("_TZE200_mrf6vtua", "TS0601"),
             ("_TZE204_qasjif9e", "TS0601"),
+            ("_TZE204_ijxvkhd0", "TS0601"),
         ],
         ENDPOINTS: {
             1: {

```

visata, 2023-12-23:
@logan893 I'm adding this quirk from here https://fixtse.com/blog/zy-m100-full-zha-support

Will let you know if it works.

logan893, 2023-12-24:
@visata Looks good. Have you gotten the "target distance" to display as a sensor?

logan893, 2024-01-03:
I cobbled together a custom quirk based on fixtse's quirk ( https://fixtse.com/blog/zy-m100-full-zha-support ; https://gist.githubusercontent.com/fixtse/b95753b84c34b45f49b3116d23b66342/raw/0f84bd6e9b6c174970c7b5fc21078d4b4da06a15/TZE204_ijxvkhd0_e5m9c5hl.py ) with some influence from https://github.com/zigpy/zha-device-handlers/pull/2525#issuecomment-1826881992 and my own additions.

I haven't seen that there would be any "fade time remaining" parameter for _TZE204_ijxvkhd0, but I have no clue where to begin digging.

I added an Iaszone motion sensor to separately track the motion detected (seems to be a fixed fade time of roughly 5-6 seconds). Distance to target is via a pressure sensor, showing cm to target with unit hPa.

[image]

```
## originally based on
### https://fixtse.com/blog/zy-m100-full-zha-support
### ( https://gist.githubusercontent.com/fixtse/b95753b84c34b45f49b3116d23b66342/raw/0f84bd6e9b6c174970c7b5fc21078d4b4da06a15/TZE204_ijxvkhd0_e5m9c5hl.py )
## with inspiration from
### https://github.com/zigpy/zha-device-handlers/pull/2525#issuecomment-1826881992
## and my own changes and additions

import math
from typing import Dict
from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import (
    AnalogInput,
    AnalogOutput,
    Basic,
    GreenPowerProxy,
    Groups,
    Ota,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.measurement import (
    IlluminanceMeasurement,
    OccupancySensing,
    PressureMeasurement,
)

from zigpy.zcl.clusters.security import IasZone, ZoneType

from zhaquirks import Bus, LocalDataCluster

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    
    CLUSTER_COMMAND,
    ZONE_STATUS_CHANGE_COMMAND,
    ON,
    OFF,
)

from zhaquirks.tuya import (
    NoManufacturerCluster,
    TuyaLocalCluster,
    TuyaNewManufCluster,
)
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaAttributesCluster,
    TuyaMCUCluster,
)

class TuyaMmwRadarSelfTest(t.enum8):
    """Mmw radar self test values."""
    TESTING = 0
[... 252 lines cut ...]
            6: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaMmwRadarMotionSensitivity,
                ],
                OUTPUT_CLUSTERS: [],
            },
            7: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaMmwRadarPresenceSensitivity,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }
```

mariomor, 2024-02-02:
Hi @logan893 ,

it works like a charm. Thanks for sharing such a pearl :)
One question: how to correlate distance to target in cm's and hPA ? At first glance such units  don't correlate.
Can you shed some light here ?

Thank you.

ghaisasadvait, 2024-02-07:
> I cobbled together a custom quirk based on fixtse's quirk ( https://fixtse.com/blog/zy-m100-full-zha-support ; https://gist.githubusercontent.com/fixtse/b95753b84c34b45f49b3116d23b66342/raw/0f84bd6e9b6c174970c7b5fc21078d4b4da06a15/TZE204_ijxvkhd0_e5m9c5hl.py ) with some influence from [#2525 (comment)](https://github.com/zigpy/zha-device-handlers/pull/2525#issuecomment-1826881992) and my own additions.
> 
> I haven't seen that there would be any "fade time remaining" parameter for _TZE204_ijxvkhd0, but I have no clue where to begin digging.
> 
> I added an Iaszone motion sensor to separately track the motion detected (seems to be a fixed fade time of roughly 5-6 seconds). Distance to target is via a pressure sensor, showing cm to target with unit hPa.
> 
> ![image](https://private-user-images.githubusercontent.com/8500470/294000671-44014d4d-f7ec-48fe-8ef6-708a4f249161.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDczMzAxODEsIm5iZiI6MTcwNzMyOTg4MSwicGF0aCI6Ii84NTAwNDcwLzI5NDAwMDY3MS00NDAxNGQ0ZC1mN2VjLTQ4ZmUtOGVmNi03MDhhNGYyNDkxNjEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIwNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMDdUMTgxODAxWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZTlmNjllYTRjZTRiMzk2MjU3MWU4MzFlOWJjMjMzMDgwNmIwNWFjMjQzNGJhMTMyNGZkNWFiODJhZmEyZDcwZCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.kXnxHwxyoIHKWbpxhvcP_8D-9ADtmDVG6DVkCRpJ-Fs)
> 
> (this custom quirk also pasted in initial problem description)
> 
> ```python
> ## originally based on
> ### https://fixtse.com/blog/zy-m100-full-zha-support
> ### ( https://gist.githubusercontent.com/fixtse/b95753b84c34b45f49b3116d23b66342/raw/0f84bd6e9b6c174970c7b5fc21078d4b4da06a15/TZE204_ijxvkhd0_e5m9c5hl.py )
> ## with inspiration from
> ### https://github.com/zigpy/zha-device-handlers/pull/2525#issuecomment-1826881992
> ## and my own changes and additions
> 
> import math
> from typing import Dict
> from zigpy.profiles import zha
> from zigpy.quirks import CustomDevice
> import zigpy.types as t
> from zigpy.zcl import foundation
> from zigpy.zcl.clusters.general import (
>     AnalogInput,
>     AnalogOutput,
>     Basic,
>     GreenPowerProxy,
>     Groups,
>     Ota,
>     Scenes,
>     Time,
> )
> from zigpy.zcl.clusters.measurement import (
>     IlluminanceMeasurement,
>     OccupancySensing,
>     PressureMeasurement,
> )
> 
> from zigpy.zcl.clusters.security import IasZone, ZoneType
> 
> from zhaquirks import Bus, LocalDataCluster
> 
> from zhaquirks.const import (
>     DEVICE_TYPE,
>     ENDPOINTS,
>     INPUT_CLUSTERS,
>     MODELS_INFO,
>     OUTPUT_CLUSTERS,
>     PROFILE_ID,
>     
>     CLUSTER_COMMAND,
>     ZONE_STATUS_CHANGE_COMMAND,
>     ON,
>     OFF,
> )
> 
> from zhaquirks.tuya import (
>     NoManufacturerCluster,
>     TuyaLocalCluster,
>     TuyaNewManufCluster,
> )
> from zhaquirks.tuya.mcu import (
>     DPToAttributeMapping,
>     TuyaAttributesCluster,
>     TuyaMCUCluster,
> )
> 
> class TuyaMmwRadarSelfTest(t.enum8):
[... 266 lines cut ...]
>                 DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
>                 INPUT_CLUSTERS: [
>                     TuyaMmwRadarMotionSensitivity,
>                 ],
>                 OUTPUT_CLUSTERS: [],
>             },
>             7: {
>                 PROFILE_ID: zha.PROFILE_ID,
>                 DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
>                 INPUT_CLUSTERS: [
>                     TuyaMmwRadarPresenceSensitivity,
>                 ],
>                 OUTPUT_CLUSTERS: [],
>             },
>             242: {
>                 PROFILE_ID: 41440,
>                 DEVICE_TYPE: 97,
>                 INPUT_CLUSTERS: [],
>                 OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
>             },
>         }
>     }
> ```

This worked wonders for me!

Tovrin, 2024-02-08:
Hmmm ....

I picked up one of these sensors and it arrived today (a TS0601 _TZE204_e5m9c5hl and USB-C version).   I followed the instructions at https://fixtse.com/blog/zy-m100-full-zha-support and identified the correct code to paste in, but I've had no luck.  The device has been discovered, but no entities at all.

I'm not entirely sure where to go and what to do.  I'm afraid all this is a bit beyond me, but if there's information I can provide to figure it out, I'll give it a damn good try.

thefunkygibbon, 2024-02-14:
struggling to get it to pick up my model (_TZE204_qasjif9e).  I've added that to the sig part of the py script and reloaded but it just keeps on picking up the same (what seems to be the default)
Quirk: zhaquirks.tuya.ts0601_motion.MmwRadarMotionGPP

edsmith323, 2024-02-18:
I have a similar issue where the presence remains on detected and never changes to clear. 
The device is _TZE204_ztc6ggyl TS0601, I tried using the quirk posted, adding the device details to the signature section but get none the adjustment sliders in HA zigbee page.
I'm not sure what I'm missing.

logan893, 2024-02-26:
> Hi @logan893 ,
> 
> it works like a charm for _TZE204_ijxvkhd0. Thanks for sharing such a pearl :) One question: how to correlate distance to target in cm's and hPA ? At first glance such units don't correlate. Can you shed some light here ?
> 
> Thank you.

The distance is exactly the number of cm reported from the sensor. Minimum distance seems to be 150 cm (i.e. 150 hPa with the sensor), maximum 550 cm.

logan893, 2024-02-26:
> anyone still caring about these devices?

The _TZE204_qasjif9e variant is the 5.8GHz version and it behaves quite differently. This quirk of mine is for the 24GHz model only.

I also have that 5.8GHz variant with the same model as yours, and I too had problems with the default implementation. I haven't yet posted my updated quirk file that covers both, as I'm not sure it truly covers everything.

With my variant of a quirk for _TZE204_qasjif9e you get access to a trigger-delay-timer and clear-timer, you get distance sensor and a min/max sense range, sensitivity, occupancy, and illumination. There is no extra active-motion sensor like there is for the 24GHz variant. Again, I don't know if this quirk covers it all.

I'll see if I can post it later today.

thefunkygibbon, 2024-02-28:
hi, ah thats great, thank you.  I've just tested it and it actually works with the pressure/distance part which i just couldn't get working from hacking around with other quirks.    it does seem to be all over the place though but that won't be anything to do with your code and is likely to do with the radar sensor itself.  so after all of this its probably going to be a bit tricky to make any automations with the numbers it gives lol

thanks again @logan893 , you're a hero

thefunkygibbon, 2024-03-26:
@logan893  sorry to be a PITA,  i've just got another (ceiling, 24ghz model) which has a new model number I can't find any references to.   I've tried to use your above 24ghz quirk , added the model number, it picks it up,  but it says illuminance and pressure are 'unknown' and motion/presence are always 'Clear' and don't seem to pick any movement up.   any ideas?  or do you need to have a physical device to be able to advise?

thanks

thefunkygibbon, 2024-03-27:
ok so i managed to get the occupancy being reported  by literally using the quirk from 
https://raw.githubusercontent.com/zigpy/zha-device-handlers/dev/zhaquirks/tuya/ts0601_motion.py
and adding the signature to it.    
Not sure how to go about getting the illuminance and the settings (max distance etc) to work at all.   any ideas?

thefunkygibbon, 2024-03-28:
ugh, ok so i REALLY need to be able to configure the settings on this since it seems to "clear" after no movement within like 10 seconds.   it really shouldnt do that should it? i thought the whole point in radar was that it detected presence despite no movement!! :-(

logan893, 2024-04-06:
@thefunkygibbon I believe you'll need to either be very good or lucky at guessing, or have a Tuya gateway, to figure out how these new devices work. I don't have a gateway myself, so most of the quirks I posted here are based on information others have already discovered, and all things considered just a tiny bit of guesswork and trial and error.

Zigbee2MQTT has a tutorial for how to extract the device data points from logs using a Tuya gateway and the Tuya app.

https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html

closed 2025-08-15 by github-actions[bot]