zigpy/zha-device-handlers #2854: [Device Support Request]

Chris202212, 2023-12-21:
### Problem description

This device is recognized and LQI and RSSI is working, but the temperature  is unfortunately not working, It is missing.

### Solution description

Support for retrieving and displaying the temperature and humidity and maybe battery from this device

### Screenshots/Video

[image]

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

MattWestb, 2023-12-21:
Nice its one tuya MCU device.

For doing somthing some diagnostic and logs is needed also some more information of the device so the devs is knowing hat is expected (more then tuya is doing).

Chris202212, 2023-12-23:
I'm quite new to Home Assistant so if you could tell me what I need to do to provide more information

zener82, 2023-12-30:
I'm interested in this fix too :)

Below some additional information:

Signature:
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE204_upagmta9",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

zener82, 2023-12-30:
This is the error message during the association.
I cannot get more information in home assistant logs.

[image]

zener82, 2023-12-30:
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.12.4",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.6",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Paris",
    "os_name": "Linux",
    "os_version": "6.1.63-haos-raspi",
    "supervisor": "2023.12.0",
    "host_os": "Home Assistant OS 11.3.rc1",
    "docker_version": "24.0.7",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "hacs": {
      "version": "1.33.0",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    },
    "localtuya": {
      "version": "5.2.1",
      "requirements": []
    },
    "alexa_media": {
      "version": "4.8.1",
      "requirements": [
        "alexapy==1.27.10",
        "packaging>=20.3",
        "wrapt>=1.14.0"
      ]
    },
    "pronote": {
      "version": "0.13.6",
      "requirements": [
        "pronotepy==2.13.0"
      ]
    },
    "lektrico_custom": {
      "version": "1.1.11",
      "requirements": [
        "lektricowifi==0.0.25"
      ]
    },
    "variable": {
      "version": "3.4.0",
      "requirements": [
        "iso4217==1.11.20220401"
      ]
    },
    "carbu_com": {
      "version": "9.0.3",
      "requirements": [
        "bs4",
        "requests",
        "ratelimit"
      ]
    }
  },
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "after_dependencies": [
      "onboarding",
[... 244 lines cut ...]
              },
              "0x0005": {
                "attribute_name": "model",
                "value": "TS0601"
              }
            },
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}

zener82, 2023-12-30:
I'm new to zha, but can help with python developping.

Please guide me for further information to provide or code to develop or test :) Thx !

roblamoreaux, 2023-12-30:
I saw more info on this device at https://github.com/dresden-elektronik/deconz-rest-plugin/issues/7322 there was something about it needing to be sent some TUYA commands or it will drain the battery constantly sending some message.

roblamoreaux, 2023-12-30:
I think it is the same things asnhttps://github.com/zigpy/zha-device-handlers/issues/1561

jacekk015, 2024-03-13:
Quirk code is already in ZHA. Only device ID was missing:
_TZE204_upagmta9
https://github.com/zigpy/zha-device-handlers/pull/3036

roblamoreaux, 2024-09-09:
Still doesn't work with ZHA. I think everyone gave up on it ever working with ZHA, or went to Z2MQTT

WeterPeter, 2024-10-14:
> Has there been any progress on this? I recently switched to ZHA (because Z2MQTT gave me a headache) and am pretty dissappointed, that my cheap sensor is now a cheap paperweight.

Yeah, I got four of these not realising they were different and now are FOUR paperweights...

LennardPlay, 2024-10-14:
You might wanna look into this: https://github.com/zigpy/zha-device-handlers/issues/2694#issuecomment-2394973206

This worked for me anyway

WeterPeter, 2024-10-15:
> You might wanna look into this: [#2694 (comment)](https://github.com/zigpy/zha-device-handlers/issues/2694#issuecomment-2394973206)
> 
> This worked for me anyway

Yess!! It now does work for me too on all four. Some paired directly with all entities, some I had to remove and pair again to get all entities to show. Thanks!! 🙏😀

PR #3817 (2025-01-31): Fix Tuya temp/humidity sensor quirks (`_TZE200_upagmta9`, ...)
## Proposed change

Add Tuya `tze200 cirvgep4` and combine quirks to match z2m.

See: https://github.com/Koenkk/zigbee-herdsman-converters/blob/09cfce5153d3931c7cc009aa3ae3107d9b394f5c/src/devices/tuya.ts#L1073

## Additional information

Closes: #2577
Closes: #2854 
Closes: #2304 
Closes: #3260 
Closes: #2725 
Closes: #2929 

## Checklist

- [ ] The changes are tested and work correctly
- [x] `pre-commit` checks pass / the code has been formatted using Black
- [x] Tests have been added to verify that the new code works
closed 2025-01-31 by TheJulianJES