zigpy/zha-device-handlers #2855: Add power monitoring support for _TZ3000_aghwaexm TS0001 (CR-MNZ1)

rod451, 2023-12-21:
### Problem description

Hopefully, this is an easy fix.  When I add this device to ZHA 12.3, it comes up as a simple switch but I have seen reviews online that indicate it supports power monitoring with the following quirk.  Quirk:zhaquirks.tuya.ts000x.Switch_1G_Metering.  

### Solution description

I would like to see it come up as a device that also includes power monitoring (power, current, voltage, summation delivered, etc) as per something like _TZ3000_tgddllx4 found in an online review of the device.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4660, maximum_buffer_size=108, maximum_incoming_transfer_size=0, server_mask=11264, maximum_outgoing_transfer_size=0, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe001"
      ],
      "output_clusters": [
        "0x000a"
      ]
    }
  },
  "manufacturer": "_TZ3000_aghwaexm",
  "model": "TS0001",
  "class": "zigpy.device.Device"
}```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.12.3",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.6",
    "docker": true,
    "arch": "aarch64",
    "timezone": "America/Toronto",
    "os_name": "Linux",
    "os_version": "6.1.58-haos-raspi",
    "supervisor": "2023.12.0",
    "host_os": "Home Assistant OS 11.2",
    "docker_version": "24.0.7",
    "chassis": "embedded",
    "run_as_root": true
[... 318 lines cut ...]
    }
  }
}```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

PJNorrisS1, 2023-12-22:
I have the same one, and although ZHA shows the current, power and voltage, it's not reporting. I have heard that there are two models - one that does power monitoring and one that doesn't... I fear we have the one that doesn't!

_TZ3000_tgddllx4 apparently does have power monitoring...

https://smarthomescene.com/reviews/zigbee-relay-with-energy-monitoring-for-high-power-appliances/

rod451, 2023-12-22:
> I have the same one, and although ZHA shows the current, power and voltage, it's not reporting. I have heard that there are two models - one that does power monitoring and one that doesn't... I fear [...]

Thanks for the reply.  I contacted the author of that review and just got the same response that there were two models, one with power monitoring and one without.  You would think they would have different model numbers.  That said, ZHA for mine does not show current, power and voltage.  I've ordered another that specifically mentions power monitoring.  Hopefully that one will work.    Cheers.

PJNorrisS1, 2023-12-23:
Thanks for the update. Mind sharing the link where you ordered the power one from? Thanks

rod451, 2023-12-23:
Sure thing.  I just used the [Aliexpress link](https://www.aliexpress.com/item/1005005037224711.html?aff_fcid=49e1a7645d6c46bdb9ffffc91a6945ff-1703339566159-06094-_Dlrd0FJ&tt=CPS_NORMAL&aff_fsk=_Dlrd0FJ&aff_platform=shareComponent-detail&sk=_Dlrd0FJ&aff_trace_key=49e1a7645d6c46bdb9ffffc91a6945ff-1703339566159-06094-_Dlrd0FJ&terminal_id=116e9e16bc264b8dbdf99a6571ee6093&afSmartRedirect=y&gatewayAdapt=4itemAdapt) in that SmartHomeScene review.  Fingers crossed...

closed 2024-06-28 by github-actions[bot]