zigpy/zha-device-handlers #2862: [Device Support Request] TY0201 by _TZ3000_bjawzodf

idumzaes, 2023-12-26:
### Problem description

Round white temperature and humidity sensor with display purchased from Temu.
Only temperature and battery shows.

I attempted using the [ts0201.py](https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0201.py) and replacing `("_TZ3000_qaaysllp", "TS0201")` with `("_TZ3000_bjawzodf", "TY0201")` and it seems to be working.

### Solution description

Temperature, Humidity and Battery level support.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

Before Quirk
[image]

After Quirk
[image]

</details>

### Device signature

<details><summary>Device signature</summary>

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0302",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0402",
        "0x0405"
      ],
      "output_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_bjawzodf",
  "model": "TY0201",
  "class": "zigpy.device.Device"
}

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.12.3",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.6",
    "docker": true,
    "arch": "x86_64",
    "timezone": "America/New_York",
    "os_name": "Linux",
    "os_version": "6.1.63-haos",
    "supervisor": "2023.12.0",
[... 530 lines cut ...]
    replacement = {
        ENDPOINTS: {
            1: {
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    TemperatureMeasurement.cluster_id,
                    RelativeHumidityX10,
                    TuyaTemperatureHumidityAlarmCluster,
                ],
                OUTPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            },
        },
    }

</details>

### Additional information

https://www.temu.com/tuya-smart-zigbee-temperature-and-humidity-sensor-smart-life-app-control-indoor-humidity-detector-works-with-zigbee-gateway-hub-g-601099531358811.html

LukaszJedynowicz, 2023-12-29:
Hi. My equipment works the same. Only temperature and battery shows.  TY0201, _TZ3000_bjawzodf

idumzaes, 2023-12-29:
Try the custom quirk I listed on the original post. It should work for you as it is working for me.

idumzaes, 2024-01-01:
I take it back. The humidity stopped working for whatever reason.

LukaszJedynowicz, 2024-01-03:
Hi. My custom quirk works.
[ty0201.zip](https://github.com/zigpy/zha-device-handlers/files/13822854/ty0201.zip)

tobibachert, 2024-01-04:
> Hi. My custom quirk works. [ty0201.zip](https://github.com/zigpy/zha-device-handlers/files/13822854/ty0201.zip)

Hello,

I just received one TY0201 by _TZ3000_bjawzodf device today and encountered the same situation as described in this issue. I could add the device to HA in ZHA, but I only got battery and temperature, but not humidity.
My HA is running on a raspberry PI 4 with ConBee II Zigbee stick.

After some search, I found this issue and tried the ty0201.zip custom quirk from @LukaszJedynowicz .
I think i managed to activate it ( I have no experience with custom quirks until now, I followed [this description](https://github.com/zigpy/zha-device-handlers/pull/2419)) correctly as I have an additional line in the Zigbee info of the device including the typo with the missing "i" from the ty0201.py (see screenshots).
But still the humidity is missing (also see screenshots).
I deleted and newly added the device to be sure.
Did you do anything additionally to get it running with humidity?

Thank you in advance!
[image]
[image]

idumzaes, 2024-01-04:
It didn’t work for me at first either. Try restarting HA after applying the quirk and then adding the device again.

tobibachert, 2024-01-04:
> It didn’t work for me at first either. Try restarting HA after applying the quirk and then adding the device again.

Thanks for your answer.
I'm sorry I wasn't precise enough what I did, unfortunately I already did these steps.
In detail:
- applied quirk
- checked configuration (just to be sure...)
- restarted HA
- deleted device
- newly added device
- restarted HA 
--> Device is running but still without humidity.

I have the latest HA core update and some other updates open, so I will take the time to do all updates now.
I will give an update tomorrow if that changes anything.

tobibachert, 2024-01-05:
Update: I did not succeed in getting the humidity in home assistant. I will leave it as it is for the moment, maybe there will be some activity here in the future that helps me getting it running.

Thank you @idumzaes for you support!

davejarrard, 2024-01-05:
I don't know if this helps but here are a couple of internal pictures and a link to all the endpoints on the Tuya website as I too am experiencing the same problem.

[image]
[image]

https://developer.tuya.com/en/docs/connect-subdevices-to-gateways/zigbee-sensor?id=K9ik6zvmhrfh6

LukaszJedynowicz, 2024-01-05:
Hi, I have two identical thermometers. The second thermometer only shows the temperature, but both provide humidity information in their diagnostics.
Diagnostics are identicaly.
Check your diagnostic information to see if there's humidity information available.
          "0x0405": {
            **"endpoint_attribute": "humidity",**
            "attributes": {
              "0x0000": {
                "attribute_name": "measured_value",
                **"value": 4140**
              }
            },
            "unsupported_attributes": {
              "0x0000": {
                "attribute_name": "measured_value"
              }
            }
          }
 The thermometers are of very poor quality.
  
[image]
[image]

leopold-code, 2024-01-10:
Hello,
I got the same device today TY0201  _TZ3000_bjawzodf firmware: 0x10013001
The problem with ZHA is a little bit different. I can see the humidity but not the temperature. Currently I have no quirks installed. I am working with the current HA Version
[image]
I am very interested in the temperature not so much in the humidity. Because of this I am looking forward for a fix. Below a screenshot from my device configuration.
[image]

idumzaes, 2024-01-16:
> Hi. My custom quirk works. [ty0201.zip](https://github.com/zigpy/zha-device-handlers/files/13822854/ty0201.zip)

I can confirm this custom quirk works. I have been using it for 2 weeks without any issues.

[image]

Davej817, 2024-01-16:
Unfortunately, this still didn't work for me.
[image]

dolorosus, 2024-01-16:
Same here. The custom quirk doesn't work for me either.
[image]

ThorstenDtk, 2024-01-27:
How did you manage to pair the device? I tried to pair with PhosCon several times, also tried different devices, it never succeeded. Usually I'm using ioBroker with the Deconz adapter and RaspBee, but for testing I tried to pair using the Zigbee adapter in ioBroker and the Sonoff USB stick - that worked immediately.

idumzaes, 2024-01-27:
I’ve been using ZHA with a Sonoff Zigbee 3.0 USB Dongle Plus.  I’ve never used PhisCon, Deconz, ioBroker or RaspBee.

dolorosus, 2024-01-27:
I paired this device using  ```ZHA``` and ```SkyConnect``` as dongle.

mpactdev, 2024-01-29:
The custom quirk mentioned does not work for me either. I'm not experienced with quirk development, but I also don't know why it works for anyone (please someone explain if they know, I'm legit curious, not criticizing) - since on the surface it appears to make the replacement template identical to the original.

gielie, 2024-02-03:
For me the quirk posted here didn't work either so I changed it a bit and now it works for me.
this Is my quirk.
`"""Tuya TY0201 temperature, humidity and optional illumination sensors."""

from zigpy.profiles import zha
from zigpy.profiles.zha import DeviceType
from zigpy.quirks import CustomCluster, CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, Identify, Ota, PowerConfiguration, Time
from zigpy.zcl.clusters.measurement import (
    IlluminanceMeasurement,
    RelativeHumidity,
    TemperatureMeasurement,
)
from zigpy.zdo.types import NodeDescriptor

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    NODE_DESCRIPTOR,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)

class TemperatureHumidtySensorWithScreen(CustomDevice):
    """Temu temperature and humidity sensor with screen."""

    signature = {
        #  <SimpleDescriptor endpoint=1, profile=260, device_type="0x0302"
        #  input_clusters=["0x0000", "0x0001", "0x0003", "0x0402", "0x0405", "0xe002]
        #  output_clusters=["0x0003", "0x000a", "0x0019"]>
        MODELS_INFO: [("_TZ3000_bjawzodf", "TY0201")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: DeviceType.TEMPERATURE_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    TemperatureMeasurement.cluster_id,
                    RelativeHumidity.cluster_id,
                  ],
                OUTPUT_CLUSTERS: [
                    Ota.cluster_id,
                ],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    TemperatureMeasurement.cluster_id,
                    RelativeHumidity.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Ota.cluster_id,
                ],
            },
        },
    }`

vojacekj, 2024-02-03:
@gielie could you post it as a file? I think formatting broke it.

TarekTaha, 2024-02-04:
@gielie  I have the same problem: "Only temperature shown". I had a look at the code posted above; it's the same as in the quirk files above, but I don't see any difference. What did you actually change?

koverg70, 2024-02-04:
I had have the same issue: had no humidity only temperature in HA
Today I have upgraded to latest (2024.1.6) HA and added the custom quirk.
After that no measurement at all. Even the temperature and battery disappeared.
Then I stopped HA (docker container), removed the quirk and the cache in the custom quirk folder.
Started HA Docker container (on Raspberry Pi 4).

And it wokrs:
[image]

gielie, 2024-02-04:
> @gielie I have the same problem: "Only temperature shown". I had a look at the code posted above; it's the same as in the quirk files above, but I don't see any difference. What did you actually change?

I changed this part

   #  input_clusters=["0x0000", "0x0001", "0x0003", "0x0402", "0x0405"]
   #  output_clusters=["0x0019"]>

into this

  #  input_clusters=["0x0000", "0x0001", "0x0003", "0x0402", "0x0405", "**0xe002**]
  #  output_clusters=["**0x0003", "0x000a**", "0x0019"]>

koverg70, 2024-02-04:
Hi @gielie !

Thanks for sharing your config.
Let me ask two questions!
1. These modified lines are only source comments. How do these effect the configuration. There is also a typo in the first comment line "0xe002 missing closing quoutes.
> #  input_clusters=["0x0000", "0x0001", "0x0003", "0x0402", "0x0405", "0xe002]
>  #  output_clusters=["0x0003", "0x000a", "0x0019"]>
2. Shouldn't the modification be made on the following lists two? How do you do it?
> INPUT_CLUSTERS: [
>                 Basic.cluster_id,
>                 PowerConfiguration.cluster_id,
>                 Identify.cluster_id, [...]

Regards,

Gábor

Davej817, 2024-02-05:
This is very strange. Following the change to the quirks file, one of the sensors has started showing temperature and humidity but another is only showing temperature still.

I have tried repeatedly restarting HA and removing the cache file but no further change.

koningdde, 2024-02-06:
Hi, Iam trying to get this working also, I got all the same results as above except a working device.

I cannot find how the clear the cache for the quirks, can anyone point me in the right direction?

My feeling is that any change I make to the quirk is not loaded when I pair the device.

Thanks in advance.

Danny

Davej817, 2024-02-07:
@koningdde your quirks cache is in config\zha_quirks\__pycache__. You can also see if your quirk is being used by looking at the device info 
[image]

koningdde, 2024-02-07:
> @koningdde your quirks cache is in `config\zha_quirks\__pycache__`. You can also see if your quirk is being used by looking at the device info [...]

Thank you, 
I should have been looking a little closer as it is states obvious in the folder. 
I will restart this adventure now and see if the quirks is used again, indeed the quirk was not loaded at my configuation.

Davej817, 2024-02-07:
To add to the confusion I unpaired the working device, re-paired and the sensor returned to only showing temperature. It seems there is something more at play. I now have no idea how I got it fully working.

koningdde, 2024-02-08:
Weird results here also, i removed the devices, after pairing one is showing temperature, the other one has no data.
When pairing the endpoints looks fine.

[image]

Data is available on one device, but only temperature:
[image]

Seems to me we are on the wrong track or the device doen's communicate as i should, maybe a timing problem in the RF system.

I think the devices are rubbish :)

gielie, 2024-02-08:
> Weird results here also, i removed the devices, after pairing one is showing temperature, the other one has no data. When pairing the endpoints looks fine. [...]

Don’t give up, with my Quirk I have 5 devices working, I can remove them and add them without any problem. 

But the main thing is that it is an ZHA problem in Z2MQTT they are all working out of the box.

koningdde, 2024-02-08:
Thanks, i don't feel like migrating to Z2MQTT yet, because i need to pair everything again, 
Although i like the idea that i can manipulate the communication more.

insane2subro, 2024-02-09:
After repeatedly deleting the device and re-adding worked using the custom quirk given on this thread.

idumzaes, 2024-02-09:
> After repeatedly deleting the device and re-adding worked using the custom quirk given on this thread.

Pretty much what I did to get mine working with ZHA and it’s been working ever since. Though I only have one of these sensors.

tomeq82, 2024-02-11:
I have four of these devices, the same, and they act randomly - some of them register humidity, some of them temperature. I never got two readings at once. Suprisingly, they are ultra reliable under zigbee2mqtt, in contrast to other meters...

thuesen, 2024-02-13:
Here's the [ty0201.py.zip](https://github.com/zigpy/zha-device-handlers/files/14259766/ty0201.py.zip) that I ended up using which seems to work for me. I don't know if it continues to work, as of now, it does.

This is the file mentioned earlier in the thread, and with the changes from @gielie.   I made no other chances to it, I'm just posting it here as a file to make it easier to download :)

gielie, 2024-02-13:
Mine stopt working after a couple of days, can’t get them to include anymore in zha, I switched to z2mqtt, no more troubles. 

Someone needs to pick this up why zha doesn’t work but z2mqtt has no problem with this device.

thuesen, 2024-02-13:
> Mine stopt working after a couple of days, can’t get them to include anymore in zha, I switched to z2mqtt, no more troubles.

Well that doesn't bode well.

Also, I can't get my 2nd of these devices to give me humidity, despite the same quirk is being applied as for the 1st one. 

Weird.

tomeq82, 2024-02-13:
This seems to be something very elementary at the protocol level query - as those particular meters doesn't work well here, while they work perfect in z2mqtt, but on the contrary - meters from Sonoff (eg. TH01) doesn't work well in z2mqtt while they are perfect here (ZHA) - i whish I have skills to compare differences, when the devices "talks" and how to debug that...

idumzaes, 2024-02-13:
On damn. My single one of these stopped reporting humidity so I’m back to square one.  
I’m thinking about switching to Z2MQTT but that’s a drastic move for just one device. I’ll probably just end up not using this sensor.

philippedeswert, 2024-03-01:
The quirk did not immediately work for me. But after some other tweaking and restarting home assistant it finally got the humidity. I did notice the same thing as @theusen, which is that for some reason the humidity gets marked as unsupported. Thus I thought to maybe map it to something Tuya specific... No idea if this is even sane, as it is entirely new to me.

`
"""Tuya TY0201 temperature, humidity and optional illumination sensors."""

from zigpy.profiles import zha
from zigpy.profiles.zha import DeviceType
from zigpy.quirks import CustomCluster, CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, Identify, Ota, PowerConfiguration, Time
from zigpy.zcl.clusters.measurement import (
    IlluminanceMeasurement,
    RelativeHumidity,
    TemperatureMeasurement,
)
from zigpy.zdo.types import NodeDescriptor

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    NODE_DESCRIPTOR,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)

from zhaquirks.tuya.air import (
    TuyaAirQualityTemperature,
    TuyaAirQualityHumidity,
)

class TemperatureHumiditySensorWithScreen(CustomDevice):                             
    """Temu temperature and humidity sensor with screen."""
                                                                                     
    signature = {                                          
        #  <SimpleDescriptor endpoint=1, profile=260, device_type="0x0302"
        #  input_clusters=["0x000", "0x0001", "0x0003", "0x0402", "0x0405"]
        #  output_clusters=["0x0019"]>                                     
        MODELS_INFO: [("_TZ3000_bjawzodf", "TY0201")],                     
        ENDPOINTS: {                                                       
            1: {                                                           
                PROFILE_ID: zha.PROFILE_ID,                                
                DEVICE_TYPE: DeviceType.TEMPERATURE_SENSOR,                
                INPUT_CLUSTERS: [                                          
                    Basic.cluster_id,                                      
                    PowerConfiguration.cluster_id,                         
                    Identify.cluster_id,                                   
                    TemperatureMeasurement.cluster_id,                     
                    RelativeHumidity.cluster_id,                           
                  ],                                                       
                OUTPUT_CLUSTERS: [                                         
                    Ota.cluster_id,                                       
                ],                                                         
            },                                                            
        },                                                                 
    }                                                                     
                                                                           
    replacement = {                                                        
        ENDPOINTS: {                                                      
            1: {                                                           
                INPUT_CLUSTERS: [                                         
                    Basic.cluster_id,                                      
                    PowerConfiguration.cluster_id,         
                    Identify.cluster_id,                   
                    TuyaAirQualityTemperature,    
                    TuyaAirQualityHumidity,                
                ],                                
                OUTPUT_CLUSTERS: [                         
                    Ota.cluster_id,                   
                ],                                
            },                                        
        },                                        
    }                        
`

llucax, 2024-03-01:
Hi @philippedeswert, I was just trying to make this work and saw your comment and tried it out, and I can confirm that it works for me too! I couldn't get the original quirk to work at all, even after several repairs and restarts.

I had to figure out the correct file format because your comment formatting is messed up, but after applying the changes, restarting and re-pairing I finally got the humidity to show! :tada:

gielie, 2024-03-01:
> Hi @philippedeswert, I was just trying to make this work and saw your comment and tried it out, and I can confirm that it works for me too! I couldn't get the original quirk to work at all, even [...]

Could you share the file u used? Many thanks.

philippedeswert, 2024-03-03:
> Could you share the file u used? Many thanks.

I apologize for the bad formatting, the quoting/code block just did not want to work correctly. Had to rename the file to .txt, but I guess it should be ok. 

Anyway I hope it helps. I suspect there is still some Tuya specific thing with the clusters/power as reconfiguring now works for the temperature and humidity but not for the battery,

[ty0201.txt](https://github.com/zigpy/zha-device-handlers/files/14471995/ty0201.txt)

tobibachert, 2024-03-03:
> [philippedeswert](/philippedeswert)

Great...with this querk, humidity is finally working for me after newly adding the device.
Thanks a lot!

tobibachert, 2024-03-03:
> I apologize for the bad formatting, the quoting/code block just did not want to work correctly. Had to rename the file to .txt, but I guess it should be ok.
>
> Anyway I hope it helps. I suspect there [...]

Great...with this querk, humidity is finally working for me after newly adding the device.
Thanks a lot!

gielie, 2024-03-03:
> I apologize for the bad formatting, the quoting/code block just did not want to work correctly. Had to rename the file to .txt, but I guess it should be ok.
>
> Anyway I hope it helps. I suspect there [...]

great work, this works for all my devices, many thanks.

thuesen, 2024-03-03:
How do you re-add it / factory reset it? 

I can't seem to figure out how to get it work. Whenever I remove it from ZHA, and I start to pair it again, it just re-appears in the list.

It seems if I hold down the top-button on the device for more than 10 seconds, it seems to forget its connected to my ZHA. But once I add it again, it gets the old quirk added (I believe), despite I removed it and only use the new one.

Davej817, 2024-03-03:
Absolutely excellent @gielie  that did it. Now showing humidity. I hadn't realised that I had to remove the device and unpair the sensor by holding the button in until the signal strength on the display disappeared. I then paired as usual and voila it works.

koningdde, 2024-03-03:
I changed the network to Z2MQTT an it is working without problems, so i cannot test this anymore, but i hope for you guys this is working now.

krecik, 2024-03-04:
> I apologize for the bad formatting, the quoting/code block just did not want to work correctly. Had to rename the file to .txt, but I guess it should be ok.
>
> Anyway I hope it helps. I suspect there [...]

After applying this quick, my ZHA shows nothing - no temp, no humidity.

philippedeswert, 2024-03-04:
> Someone wrote this morning that:
> 
> > After applying this quick, my ZHA shows nothing - no temp, no humidity.
> 
> But that message is now gone. If that someone managed to fix that issue, please enlighten us :D I'm stuck at the same, after finally getting the device removed from ZHA (I think).

Usually restarting ZHA helps with that, but only after you added the device back... I had some cases when trying things out that the device would not report temp and/or humidity but would be fine after a ZHA restart.

thuesen, 2024-03-04:
Thank you philippe and gielie for your suggestions. I was not aware of the __pycache__ folder, File Editor was hiding it. I tried to clear it, restart ZHA, HA, re-pair etc etc., but it keeps appearing with only "Firmware", no temperature or humidity.

I guess I'll let it be for now. Maybe it will change its mind eventually.

vojacekj, 2024-03-04:
> Thank you philippe and gielie for your suggestions. I was not aware of the __pycache__ folder, File Editor was hiding it. I tried to clear it, restart ZHA, HA, re-pair etc etc., but it keeps [...]

Same here with @philippedeswert quirk I now don't even see temperature, which was only thing working out of the box.

I used txt file provided, I see that this quirk was applied but temperature and humidity is missing. (Did restart, remove device and cache, etc )

philippedeswert, 2024-03-04:
> Same here with @philippedeswert quirk I now don't even see temperature, which was only thing working out of the box.
>
> I used txt file provided, I see that this quirk was applied but temperature and [...]

I hope you changed the txt extension back to .py?

vojacekj, 2024-03-04:
> I hope you changed the txt extension back to .py?

Yes I did as, I don't think it would be applied otherwise.

antonGritsenko, 2024-03-15:
> Thank you philippe and gielie for your suggestions. I was not aware of the **pycache** folder, File Editor was hiding it. I tried to clear it, restart ZHA, HA, re-pair etc etc., but it keeps [...]

You are not alone :) Same store, diagnostic shows that now temperature is not supported, but show nothing for both )))

tobibachert, 2024-03-15:
> > Thank you philippe and gielie for your suggestions. I was not aware of the **pycache** folder, File Editor was hiding it. I tried to clear it, restart ZHA, HA, re-pair etc etc., but it keeps appearing with only "Firmware", no temperature or humidity.
> > I guess I'll let it be for now. Maybe it will change its mind eventually.
> 
> You are not alone :) Same story, diagnostic shows that now temperature is not supported, but show nothing for both )))
> 
> UPD: after yet another restart the humidity back to live, but not temperature )) Magic... I can read both values using Manage Zigbee Device, btw...
> 
> UPD2: hm, now Download Diagnostic shows that all *value attributes supported, I did restart of the HA and guess what... it works... May be my manual execution of the Read Values in the Manage Zigbee Device did the trick, not sure..
> 
> UPD3: similar comment [here](https://github.com/zigpy/zha-device-handlers/issues/2851#issuecomment-1985536173), so may be really reading values manually do the magic...

UPD3 seems to do it...same procedure as described, manual reading, temperature and humidity could be read but were not available as entities in the device, then restarting HA and - MAGIC - they're here and updating!
Let's see if it stays this way, if you compare my comment some days ago I was somehow able to get both values directly after adding the device with the custom quirk, but then the values stopped both updating.

vojacekj, 2024-03-16:
> UPD3 seems to do it...same procedure as described, manual reading, temperature and humidity could be read but were not available as entities in the device, then restarting HA and - MAGIC - they're [...]

Do you use any quirk? If yes which one?

tobibachert, 2024-03-16:
> > > > Thank you philippe and gielie for your suggestions. I was not aware of the **pycache** folder, File Editor was hiding it. I tried to clear it, restart ZHA, HA, re-pair etc etc., but it keeps appearing with only "Firmware", no temperature or humidity.
> > > > I guess I'll let it be for now. Maybe it will change its mind eventually.
> > > 
> > > 
> > > You are not alone :) Same story, diagnostic shows that now temperature is not supported, but show nothing for both )))
> > > UPD: after yet another restart the humidity back to live, but not temperature )) Magic... I can read both values using Manage Zigbee Device, btw...
> > > UPD2: hm, now Download Diagnostic shows that all *value attributes supported, I did restart of the HA and guess what... it works... May be my manual execution of the Read Values in the Manage Zigbee Device did the trick, not sure..
> > > UPD3: similar comment [here](https://github.com/zigpy/zha-device-handlers/issues/2851#issuecomment-1985536173), so may be really reading values manually do the magic...
> > 
> > 
> > UPD3 seems to do it...same procedure as described, manual reading, temperature and humidity could be read but were not available as entities in the device, then restarting HA and - MAGIC - they're here and updating!
> > Let's see if it stays this way, if you compare my comment some days ago I was somehow able to get both values directly after adding the device with the custom quirk, but then the values stopped both updating.
> 
> Do you use any quirk? If yes which one?
> 
> Edit: yes it looks like it is finally working also for me without any quirk. Just read those values and restart the HA ( I recently also updated HA to newest version 2024.3.1, if anybody interested, but I am not sure if that has anything to do with this, so upgrade on your own risk)

I was using this querk from philipedeswert above https://github.com/zigpy/zha-device-handlers/issues/2862#issuecomment-1975062539. I am running HA 2024.2.5 at the moment. The values are still updating this morning :-)
I will try without querk after your comment and give an update later

[16 more replies]

PR #3201 (2024-06-26): Add Tuya TY0201 temperature/humidity sensor
This seems to be a pretty common sensor, for which several models seem to exist : _TZ3000_bjawzodf and _TZ3000_zl1kmjqx

## Proposed change

Add support for  another humidity/temperature sensor

## Additional information
First time contributing here. And I am not sure it is completely correct, but 
this quirk is confirmed to work for a wide number of people. I welcome any comments
to improve this.

Fixes #2862, fixes #2851, fixes #2701

 

## Checklist

- [x] The changes are tested and work correctly
- [ ] `pre-commit` checks pass / the code has been [... cut]
closed 2024-06-26 by TheJulianJES