zigpy/zha-device-handlers #2865: [Device Support Request] Ecodim 05

mano3m, 2023-12-28:
### Problem description

My Ecodim 05 two gang dimmer (see https://www.ecodim.nl/nl/eco-dim05.html) does not report its status back to ZHA (in a way it understands it anyways). This gives errors and results in out of sync issues, see for example: https://community.home-assistant.io/t/eco-dim-05-dual-dimmer-not-supported-in-zha/623843

### Solution description

In Z2M Ecodim devices have their own herdsman converter:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/ecodim.ts

It would be great if that can be converted to a quirk.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.FullFunctionDevice|MainsPowered|RxOnWhenIdle|AllocateAddress: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0101",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x1000"
      ],
      "output_clusters": [
        "0x0019"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x0101",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x1000"
      ],
      "output_clusters": [
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "EcoDim BV",
  "model": "EcoDim-Zigbee 3.0",
[... 773 lines cut ...]
  File "/usr/src/homeassistant/homeassistant/components/light/__init__.py", line 591, in async_handle_light_off_service
    await light.async_turn_off(**filter_turn_off_params(light, params))
  File "/usr/src/homeassistant/homeassistant/components/zha/light.py", line 471, in async_turn_off
    result = await self._on_off_cluster_handler.off()
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/zha/core/cluster_handlers/__init__.py", line 83, in wrapper
    with wrap_zigpy_exceptions():
  File "/usr/local/lib/python3.11/contextlib.py", line 155, in __exit__
    self.gen.throw(typ, value, traceback)
  File "/usr/src/homeassistant/homeassistant/components/zha/core/cluster_handlers/__init__.py", line 66, in wrap_zigpy_exceptions
    raise HomeAssistantError(
homeassistant.exceptions.HomeAssistantError: Failed to send request: device did not respond
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

alrassia, 2023-12-28:
I have exactly the same device and do not have the issues you describe. The device seems to follow the zigbee spec and should not really need a quirk. Looking at the z2m converter it does not seem to do anything out of the ordinary.

Duchadian, 2024-01-07:
I have exactly the same issue @mano3m is describing. Also have another Ecodim dimmer ( [a Dim.07](https://www.ecodim.nl/en/eco-dim07-zigbee-basic.html) ), which works perfectly fine. 

Don't have the knowledge of Zigbee to figure out what is going on, but I can do some checks or debug if anyone can tell me how to do so.

mano3m, 2024-01-07:
Hi Duchadian. Do you have both dimmer channels connected to a lamp or only one? I had contact with EcoDim support recently. They are going on a Home Assistant course soon, but currently dont support HA. They did tell me both channels need to be connected to a lamp/load (which is not the case for me). I wanted to use the dimmer to dim one dumb led lamp and use the other channel through HA to dim a Philips Hue lamp (so without using the actual dimmer, just the Zigbee signal).

alrassia, 2024-01-07:
The dimmer is not manufactured or developed by ecodim, its actually an oem branded device by HZC electric. So their support might also be limited in their capacity to really diagnose the issue.

the oem product you cannot easily find but here is a link: https://x.alibaba.com/AvRCTo?ck=pdp
I tried to order directly considering the price versus the ecodim price, but they claim not to ship to the netherlands.

Duchadian, 2024-01-07:
@mano3m , for me both channels are connected to lamps. 

Exactly the same issue regardless. Interestingly, sometimes when both channels are turned on, and I turn the second one off via HA, the first channel starts switching on/off a couple of times before going back to the right state. Haven't yet encountered this behaviour the other way around, but that might be a matter of time. 

@alrassia, interesting. I didn't expect Ecodim to make / design them in-house, so I tried finding it as well. Seems like you succeeded where I didn't :)

mano3m, 2024-01-07:
@Duchadian Hmm, I have only connected the first channel. I though Home Assistant retries a few times before giving up. If I turn it on, then off, it goes on and off a few times. I am beginning to suspect we have a faulty unit........?

JIK12345, 2024-01-15:
Exactly the same issue here with botch channels connected to a lamp.

Pantsema, 2024-02-24:
One additional vote for this device support. Both channels connected and dimming is working, but on/off commands lead into a device did not respond error in HA. Work around is using automations to set brightness to 100% and 0%. Triggering these automations do turn on and off the lights, but when dimming to 0% (thus turning off the lights) the status in HA is off sync (keeps reporting on on the last know brightness level).

odinson394, 2024-04-10:
Hi, I am experiencing exactly the same issue, also with two lamps connected. Would be great if this can be solved, couldn’t find any other duo dimmer solutions!

FireWhisperer, 2024-10-11:
Exactly the same issue here. 4 3.8W GU10 led-lamps in parallel per channel.

mano3m, 2024-10-12:
It seems they fixed it in #3362 for another brand of the same dimmer